strongtz/i915-sriov-dkms #117: Unable to compile on 6.2.16-18-pve

idontcare99999, 2023-10-14:
Was previously working just fine until most recent proxmox kernel upgrade. Now receive this message when attempting to compile:

make -j8 KERNELRELEASE=6.2.16-18-pve -j8 -C /lib/modules/6.2.16-18-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build KVER=6.2.16-18-pve..................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.16-18-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.

root@pm01:~/915# cat /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log
DKMS make.log for i915-sriov-dkms-6.1 for kernel 6.2.16-18-pve (x86_64)
Sat Oct 14 03:51:26 PM PDT 2023
make: Entering directory '/usr/src/linux-headers-6.2.16-18-pve'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/soc/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/soc/intel_gmch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/soc/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_memcpy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sw_fence.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sw_fence_work.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_syncmap.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_user_extensions.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioc32.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_debugfs_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_pipe_crc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_pmu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen2_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen6_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen6_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen7_renderclear.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_sseu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_heartbeat.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_pm.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.c:103:12: warning: ‘gen12_pte_encode_shim’ defined but not used [-Wunused-function]
  103 | static u64 gen12_pte_encode_shim(dma_addr_t addr,
      |            ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_user.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_execlists_submission.o
[... 223 lines cut ...]
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_hdmi.c: In function ‘intel_hdmi_get_modes’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_hdmi.c:2558:56: warning: passing argument 2 of ‘intel_connector_update_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2558 |         return intel_connector_update_modes(connector, edid);
      |                                                        ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_hdmi.c:49:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_connector.h:28:47: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
   28 |                                  struct edid *edid);
      |                                  ~~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_lvds.c: In function ‘intel_lvds_get_modes’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_lvds.c:485:37: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  485 |                 struct edid *edid = drm_edid_raw(fixed_edid);
      |                                     ^~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_lvds.c: In function ‘intel_lvds_init’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_lvds.c:975:58: warning: passing argument 2 of ‘drm_add_edid_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  975 |                 if (drm_add_edid_modes(&connector->base, edid)) {
      |                                                          ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_lvds.c:39:
./include/drm/drm_edid.h:572:70: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
  572 | int drm_add_edid_modes(struct drm_connector *connector, struct edid *edid);
      |                                                         ~~~~~~~~~~~~~^~~~
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.16-18-pve'

YTOAA, 2023-10-25:
I'm the same as above.

koda-git, 2023-10-26:
Same here! Worked around this for now by booting into 6.2.16-15-pve. Would love to get it fixed

spideyfusion, 2023-10-26:
https://github.com/strongtz/i915-sriov-dkms/pull/98 strikes again. This time it's caused by Canonical cherry-picking the [commit](https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/jammy/commit/?h=hwe-6.2-next&id=dc928990255aae26ecf84530d8a0a157a7b47684) into their recent 6.2 [release](https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/jammy/commit/?h=hwe-6.2-next&id=0b6a250fd5bb058a9965d904b3e6f83b87a0c3b7). Proxmox is unfortunately affected because they [derived](https://pve.proxmox.com/wiki/Proxmox_VE_Kernel) from the Ubuntu 23.04 base.

I made a quick and dirty PoC on my local repository to see if the module would compile, and sure enough, it does:

```diff
diff --git a/drivers/gpu/drm/i915/display/intel_dp.c b/drivers/gpu/drm/i915/display/intel_dp.c
index 587dd9a..21d5eea 100644
--- a/drivers/gpu/drm/i915/display/intel_dp.c
+++ b/drivers/gpu/drm/i915/display/intel_dp.c
@@ -3884,7 +3884,7 @@ intel_dp_mst_hpd_irq(struct intel_dp *intel_dp, u8 *esi, u8 *ack)
 {
        bool handled = false;

-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE > KERNEL_VERSION(6,4,5)
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE >= KERNEL_VERSION(6,2,0)
        drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled);
 #else
        drm_dp_mst_hpd_irq(&intel_dp->mst_mgr, esi, &handled);
@@ -3966,7 +3966,7 @@ intel_dp_check_mst_status(struct intel_dp *intel_dp)
                if (!intel_dp_ack_sink_irq_esi(intel_dp, ack))
                        drm_dbg_kms(&i915->drm, "Failed to ack ESI\n");

-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE > KERNEL_VERSION(6,4,5)
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE >= KERNEL_VERSION(6,2,0)
                if (ack[1] & (DP_DOWN_REP_MSG_RDY | DP_UP_REQ_MSG_RDY))
                         drm_dp_mst_hpd_irq_send_new_request(&intel_dp->mst_mgr);
 #endif
```

Does anyone have an idea how we could make the backwards compatibility checks without breaking vanilla kernels? The check is already messy as-is. 😭

koda-git, 2023-10-26:
> > #98 strikes again. This time it's caused by Canonical cherry-picking the [commit](https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/jammy/commit/?h=hwe-6.2-next&id=dc928990255aae26ecf84530d8a0a157a7b47684) into their recent 6.2 [release](https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/jammy/commit/?h=hwe-6.2-next&id=0b6a250fd5bb058a9965d904b3e6f83b87a0c3b7). Proxmox is unfortunately affected because they [derived](https://pve.proxmox.com/wiki/Proxmox_VE_Kernel) from the Ubuntu 23.04 base.
> > I made a quick and dirty PoC on my local repository to see if the module would compile, and sure enough, it does:
> > ```diff
> > diff --git a/drivers/gpu/drm/i915/display/intel_dp.c b/drivers/gpu/drm/i915/display/intel_dp.c
> > index 587dd9a..21d5eea 100644
> > --- a/drivers/gpu/drm/i915/display/intel_dp.c
> > +++ b/drivers/gpu/drm/i915/display/intel_dp.c
> > @@ -3884,7 +3884,7 @@ intel_dp_mst_hpd_irq(struct intel_dp *intel_dp, u8 *esi, u8 *ack)
> >  {
> >         bool handled = false;
> > 
> > -#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE > KERNEL_VERSION(6,4,5)
> > +#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE >= KERNEL_VERSION(6,2,0)
> >         drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled);
> >  #else
> >         drm_dp_mst_hpd_irq(&intel_dp->mst_mgr, esi, &handled);
> > @@ -3966,7 +3966,7 @@ intel_dp_check_mst_status(struct intel_dp *intel_dp)
> >                 if (!intel_dp_ack_sink_irq_esi(intel_dp, ack))
> >                         drm_dbg_kms(&i915->drm, "Failed to ack ESI\n");
> > 
> > -#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE > KERNEL_VERSION(6,4,5)
> > +#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6,1,40) && LINUX_VERSION_CODE <= KERNEL_VERSION(6,1,99)) || LINUX_VERSION_CODE >= KERNEL_VERSION(6,2,0)
> >                 if (ack[1] & (DP_DOWN_REP_MSG_RDY | DP_UP_REQ_MSG_RDY))
> >                          drm_dp_mst_hpd_irq_send_new_request(&intel_dp->mst_mgr);
> >  #endif
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Does anyone have an idea how we could make the backwards compatibility checks without breaking vanilla kernels? The check is already messy as-is. 😭
> 
> So is there any quick fix for mere mortals who have updated their Proxmox kernel?

I was able to compile on 6.2.16-18pve by editing /drivers/gpu/drm/i915/display/intel_dp.c as mentioned above!

makoONE, 2023-10-26:
thanks rjblake and koda-git for the workaround, it works fine.

rjblake, 2023-10-27:
Thanks @spideyfusion & @koda-git. Clearly wasn't thinking for myself earlier! Even worse, I kept editing the wrong file today and kept wondering why it kept failing and my edits kept getting reverted and then realised I wasn't editing the file in my /usr/src/i915-sriov-dkms-6.1/drivers/gpu/drm/i915/display/ folder. Duh!

YTOAA, 2023-10-27:
Unfortunately, the kernel built using the method presented above only reaches c3 pkg from Powertop.

salty72, 2023-11-04:
Works great. thanks.

Coder84619, 2023-11-04:
I don't understand...how does one apply the fix?

spideyfusion, 2023-11-06:
> I don't understand...how does one apply the fix?

You can manually edit the affected file as [mentioned](https://github.com/strongtz/i915-sriov-dkms/issues/117#issuecomment-1781367045) above.

rjblake, 2023-11-06:
@Coder84619 - the diff output shows the lines to remove (prefixed with a '-') and the ones to add (prefixed with a '+'). What it ultimately does is edit the following text in the file drivers/gpu/drm/i915/display/intel_dp.c (in my case file is in the folder /usr/src/i915-sriov-dkms-6.1/drivers/gpu/drm/i915/display/intel_dp.c

`> KERNEL_VERSION(6,4,5)`

and changes it to 

`>= KERNEL_VERSION(6,2,0)`

LazyBoBo, 2023-11-09:
> #98 strikes again. This time it's caused by Canonical cherry-picking the [...]
My pve version is 8.0 ， Kernel version 6.2.16-19

I tried this method ,  still failed.
The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
I changed  6,4,5 to 6,2,0 ，still failed
------------------------------------------------------------------------
#if LINUX_VERSION_CODE >= KERNEL_VERSION(6,2,0)
        drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled);
#else

#if LINUX_VERSION_CODE >= KERNEL_VERSION(6,2,0)
                if (ack[1] & (DP_DOWN_REP_MSG_RDY | DP_UP_REQ_MSG_RDY))
                        drm_dp_mst_hpd_irq_send_new_request(&intel_dp->mst_mgr);
#endif

Coder84619, 2023-11-09:
> @Coder84619 - no idea as to your skill level, but since you're here on github I assume you have some basic coding ability. The diff output shows the lines to remove (prefixed with a '-') and the ones to add (prefixed with a '+'). What it ultimately does is refer to the edit the following text in the file drivers/gpu/drm/i915/display/intel_dp.c (in my case file is in the folder /usr/src/i915-sriov-dkms-6.1/drivers/gpu/drm/i915/display/intel_dp.c
> 
> `> KERNEL_VERSION(6,4,5)`
> 
> and change it to
> 
> `>= KERNEL_VERSION(6,2,0)`
> 
> This can be done using the editor of your choice (e.g. nano, vi, vim, etc.). Make sure you have an updated pull from this Git repo, or you'll find yourself with other issues. As for suggesting that others should provide you with a CLI command to make the fix is probably not the right way about getting help. You can use a Stream Editor such as 'sed', but this carries its own set of risks if not done correctly.

Ok, after doing a fresh GIT pull, this worked as expected. Thanks!

Coder84619, 2023-11-09:
> ## I tried this method , still failed.
> The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
> I changed 6,4,5 to 6,2,0 ，still failed
> #if LINUX_VERSION_CODE >= [...]

I think you are using an older version of the dkms files. Do a fresh github pull. Worked for me.

ravnoor, 2023-11-10:
`sed -i 's/KERNEL_VERSION(6,4,5)/KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`

this did the trick for me

Hell0wor1d, 2023-11-11:
> `sed -i 's/KERNEL_VERSION(6,4,5)/KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`
> 
> this did the trick for me

Good job, did you try 6.2.16-19?

rjblake, 2023-11-11:
> `sed -i 's/KERNEL_VERSION(6,4,5)/KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`
> 
> this did the trick for me

Careful with this, as you’re missing the change of the ‘=‘ to a ‘>=‘ so may need to be changed later again

Coder84619, 2023-11-11:
> > `sed -i 's/KERNEL_VERSION(6,4,5)/KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`
> > this did the trick for me
> 
> Good job, did you try 6.2.16-19?

I did manual edits and yes it works with 6.2.16-19.

ravnoor, 2023-11-12:
> > `sed -i 's/KERNEL_VERSION(6,4,5)/KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`
> > this did the trick for me
> 
> Good job, did you try 6.2.16-19?

It did work with just the `sed` command on `6.2.16-19`

cdru, 2023-11-12:
> > `sed -i 's/KERNEL_VERSION(6,4,5)/KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`
> > this did the trick for me
> 
> Careful with this, as you’re missing the change of the ‘=‘ to a ‘>=‘ so may need to be changed later again

For anyone who doesn't speak regex or sed, here's an updated command to include the equals sign:
`sed -i 's/ KERNEL_VERSION(6,4,5)/= KERNEL_VERSION(6,2,0)/g' drivers/gpu/drm/i915/display/intel_dp.c`

samgprince, 2023-11-13:
Thank you! It worked on 6.2.16-19.

LazyBoBo, 2023-11-13:
> I think you are using an older version of the dkms files. Do a fresh github pull. Worked for me.

You are right...It was an older version. Thank you for your adivice.

Coder84619, 2023-11-23:
Proxmox 8.1 is now HA, with kernel 6.5. DKMS fails to install:

Building module:
Cleaning build area...
make -j16 KERNELRELEASE=6.5.11-4-pve -j16 -C /lib/modules/6.5.11-4-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build KVER=6.5.11-4-pve..........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.11-4-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.

Hell0wor1d, 2023-11-24:
> Proxmox 8.1 is now GA, with kernel 6.5. DKMS fails to install:
> 
> Building module: Cleaning build area... make -j16 KERNELRELEASE=6.5.11-4-pve -j16 -C /lib/modules/6.5.11-4-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build KVER=6.5.11-4-pve..........(bad exit status: 2) Error! Bad return status for module build on kernel: 6.5.11-4-pve (x86_64) Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.

Ugly quick fix, manually edit the **dkms.conf** file:

before
```
PACKAGE_NAME="@_PKGBASE@"
PACKAGE_VERSION="@PKGVER@"
```

after
```
PACKAGE_NAME="i915-sriov-dkms"
PACKAGE_VERSION="6.5"
```

Coder84619, 2023-11-24:
@Hell0wor1d Ya get past one error, but still get the boot time errors #122

lightmaster, 2023-12-05:
I was originally getting this when installing proxmox-kernel-6.2.16-19-pve. I ended using `cd /usr/src/i915-sriov-dkms-6.1 && git pull` and then it let me finish installing the new kernel with `apt install -f`

closed 2024-01-29 by strongtz