strongtz/i915-sriov-dkms #118: [BUG] Kernel 6.5.7-200.fc38.x86_64, invalidation response timeout- crash when using virtualized gpu

sihawken, 2023-10-24:
Does not work on kernel 6.5.7-200.fc38.x86_64

The desktop crashes when using 0000:00:02.1, set up using virt manager.

CPU: 12th Gen Intel® Core™ i7-1260P × 16

`[ 3004.780435] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[ 3015.412724] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 72
[ 3016.436819] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 73
[ 3017.460788] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 74
[ 3018.485044] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 75
[ 3019.509783] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 76
[ 3020.533098] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 77
[ 3021.557126] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 78
[ 3021.561824] i915 0000:00:02.1: [drm] *ERROR* GT0: rcs0 TLB invalidation did not complete in 4ms!
[ 3022.580909] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 79
[ 3022.585856] i915 0000:00:02.1: [drm] *ERROR* GT0: rcs0 TLB invalidation did not complete in 4ms!
[ 3023.604903] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 80
[ 3023.609872] i915 0000:00:02.1: [drm] *ERROR* GT0: rcs0 TLB invalidation did not complete in 4ms!
[ 3024.629169] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 81
[ 3024.633886] i915 0000:00:02.1: [drm] *ERROR* GT0: rcs0 TLB invalidation did not complete in 4ms!
[ 3025.653260] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 82
[ 3025.657902] i915 0000:00:02.1: [drm] *ERROR* GT0: rcs0 TLB invalidation did not complete in 4ms!
[ 3026.677200] i915 0000:00:02.1: [drm] *ERROR* tlb invalidation response timed out for seqno 83
[ 3026.681917] i915 0000:00:02.1: [drm] *ERROR* GT0: rcs0 TLB invalidation did not complete in 4ms!
`

resiliencer, 2023-11-02:
I get the same error on 6.5.7-zen and on 6.5.9-zen (but seqno is 23) but it works for hour or maybe two and then hosts runs out of memory. Memory leak occurs somewhere.

githubpanguan, 2023-12-12:
> 我在 6.5.7-zen 和 6.5.9-zen 上遇到相同的错误（但 seqno 是 23），但它可以工作一个小时或两个小时，然后主机内存不足。 某处发生内存泄漏。

Me too, the kernel memory overflowed, not the process

suki78, 2024-06-28:
ok so I figured it out, I found this commit that solved TLB invalidation kernel error
https://github.com/intel/linux-intel-lts/commit/c75552e4fc1b556c4d33ec20ec1e5c99f666068e

closed 2024-12-29 by strongtz