strongtz/i915-sriov-dkms #120: how to bypass virtual machine check

nhamhuynh79, 2023-10-29:
Hello everyone, how to bypass intel graphic driver check "hide kvm"
When hide vm state get error code 43 :(

michael-pptf, 2024-04-14:
set your vm cpu type to host.

liuran001, 2024-04-25:
I have the same problem. When I try to add the parameter `args: -cpu host,hypervisor=off` to pve's qemu config file, the device manager in my Windows VM says `Windows has stopped this device due to a problem with it (code 43)`. I've tried a number of things, and it's apparently impossible to hide the VM without adding `hypervisor=off`, and once this parameter is added, the device manager frustratingly displays error code 43. I haven't tried this in Linux for now (and I don't have a need to hide the VM in Linux), but in Windows, a lot of the software and games detect the presence of a VM and refuse to start it once it is detected (e.g. Genshin Impact), while it is impractical to run these games without a GPU. I assume that Intel's driver detects the VM state or has a special adaptation for SR-IOV, and once the VM state is removed, the driver will not run in the correct state (maybe?). My CPU is an i7-12700H and this problem has been bothering me for a while, hopefully someone can fix it.

michael-pptf, 2024-04-25:
> I have the same problem. When I try to add the parameter `args: -cpu host,hypervisor=off` to pve's qemu config file, the device manager in my Windows VM says `Windows has stopped this device due to [...]

Here's the thing - this is a kernel module developed by Intel, and its intention is to support SR-IOV. There should be no reason whatsoever to hide that the vGPU is used in a VM, because that's what it is for. This is the reason why you can't set the `hypervisor=off` and expect this vGPU to function correctly.

If you want to play games or what not, pass a real GPU to the VM might have better chance of success.

closed 2024-12-29 by strongtz