strongtz/i915-sriov-dkms #121: Unable to compile on 6.2.16-19-pve

er1czhang, 2023-11-07:
Was previously working just fine until most recent proxmox kernel upgrade. Now receive this message when attempting to compile:

DKMS make.log for i915-sriov-dkms-6.2 for kernel 6.2.16-19-pve (x86_64)
Tue Nov  7 11:33:04 PM CST 2023
make: Entering directory '/usr/src/linux-headers-6.2.16-19-pve'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/soc/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/soc/intel_gmch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/soc/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_memcpy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_mm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_sw_fence.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_sw_fence_work.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_syncmap.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_user_extensions.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_ioc32.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_debugfs_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_pipe_crc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/i915_pmu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen2_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen6_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen6_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen7_renderclear.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen8_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_context.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_context_sseu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_engine_heartbeat.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_engine_pm.o
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.c:103:12: warning: ‘gen12_pte_encode_shim’ defined but not used [-Wunused-function]
  103 | static u64 gen12_pte_encode_shim(dma_addr_t addr,
      |            ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_engine_user.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_execlists_submission.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_ggtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_ggtt_fencing.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_gt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_gt_buffer_pool.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/gt/intel_gt_clock_utils.o
[... 217 lines cut ...]
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_dp.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_hdmi.c: In function ‘intel_hdmi_get_modes’:
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_hdmi.c:2558:56: warning: passing argument 2 of ‘intel_connector_update_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2558 |         return intel_connector_update_modes(connector, edid);
      |                                                        ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_hdmi.c:49:
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_connector.h:28:47: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
   28 |                                  struct edid *edid);
      |                                  ~~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_lvds.c: In function ‘intel_lvds_get_modes’:
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_lvds.c:485:37: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  485 |                 struct edid *edid = drm_edid_raw(fixed_edid);
      |                                     ^~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_lvds.c: In function ‘intel_lvds_init’:
/var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_lvds.c:975:58: warning: passing argument 2 of ‘drm_add_edid_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  975 |                 if (drm_add_edid_modes(&connector->base, edid)) {
      |                                                          ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.2/build/drivers/gpu/drm/i915/display/intel_lvds.c:39:
./include/drm/drm_edid.h:572:70: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
  572 | int drm_add_edid_modes(struct drm_connector *connector, struct edid *edid);
      |                                                         ~~~~~~~~~~~~~^~~~
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.2/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.16-19-pve'

jobsidi, 2023-11-07:
the same error

LazyBoBo, 2023-11-09:
> the same error. Read [#117 (comment)](https://github.com/strongtz/i915-sriov-dkms/issues/117#issuecomment-1795047758) all installed fine

I tried the method mentioned in #117 ,  still failed.
The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
I changed  6,4,5 to 6,2,0 ，same error
------------------------------------------------------------------------
#if LINUX_VERSION_CODE >= KERNEL_VERSION(6,4,5)
        drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled);
#else

#if LINUX_VERSION_CODE >= KERNEL_VERSION(6,4,5)
                if (ack[1] & (DP_DOWN_REP_MSG_RDY | DP_UP_REQ_MSG_RDY))
                        drm_dp_mst_hpd_irq_send_new_request(&intel_dp->mst_mgr);
#endif

woodcoal, 2023-11-10:
> ## I tried the method mentioned in #117 , still failed.
> The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
> I changed 6,4,5 to 6,2,0 ，same error
> #if [...]

> ## I tried the method mentioned in #117 , still failed.
> The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
> I changed 6,4,5 to 6,2,0 ，same error
> #if [...]

> ## I tried the method mentioned in #117 , still failed.
> The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
> I changed 6,4,5 to 6,2,0 ，same error
> #if [...]

方案可以用，我使用此方案已经通过了。
CPU：13700
系统：PVE 8.0.4 
内核：6.2.16-19-pve

通过几天反复测试，我估计还是和 CPU 核心有关，尝试单线程编译通过。

具体方法：
1. 将库文件部署到 `/usr/src` `下，改名为：`i915-sriov-dkms-6.2`
2. 修改 `dkms.conf` 文件，将 `PACKAGE_NAME` 改为 `i915-sriov-dkms`；`PACKAGE_VERSION` 改为 `6.2`；删除 4、5 行中的 -`j$(nproc)`
3. 修改 `/usr/src/i915-sriov-dkms-6.1/drivers/gpu/drm/i915/display/intel_dp.c` 文件，将 `> KERNEL_VERSION(6,4,5)` 批量替换成 `>= KERNEL_VERSION(6,2,0)`
4. 更新源，下载库
```
apt install pve-headers-$(uname -r)
apt install dkms build-* -y
```
5. 编译，注意最后一定要加上参数 `-j 1` 即：`dkms install --force -m i915-sriov-dkms -v 6.1 -j 1`
6. 等待编译成功，祝你好运！

最重要的是**使用单线程编译**！

LazyBoBo, 2023-11-13:
> > > the same error. Read [#117 (comment)](https://github.com/strongtz/i915-sriov-dkms/issues/117#issuecomment-1795047758) all installed fine
> > 
> > 
> > ## I tried the method mentioned in #117 , still failed.
> > The sentence before "KERNEL_VERSION" in my file is some kind different from the instruction.
> > I changed 6,4,5 to 6,2,0 ，same error
> > #if LINUX_VERSION_CODE >= KERNEL_VERSION(6,4,5) drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled); #else
> > #if LINUX_VERSION_CODE >= KERNEL_VERSION(6,4,5) if (ack[1] & (DP_DOWN_REP_MSG_RDY | DP_UP_REQ_MSG_RDY)) drm_dp_mst_hpd_irq_send_new_request(&intel_dp->mst_mgr); #endif
> 
> 方案可以用，我使用此方案已经通过了。 CPU：13700 系统：PVE 8.0.4 内核：6.2.16-19-pve
> 
> 通过几天反复测试，我估计还是和 CPU 核心有关，尝试单线程编译通过。
> 
> 具体方法：
> 
> 1. 将库文件部署到 `/usr/src` 下，改名为：`i915-sriov-dkms-6.2`
> 2. 修改 `dkms.conf` 文件，将 `PACKAGE_NAME` 改为 `i915-sriov-dkms`；`PACKAGE_VERSION` 改为 `6.2`；删除 4、5 行中的 `-j $(nproc)`
> 3. 修改 `/usr/src/i915-sriov-dkms-6.1/drivers/gpu/drm/i915/display/intel_dp.c` 文件，将 `> KERNEL_VERSION(6,4,5)` 批量替换成 `>= KERNEL_VERSION(6,2,0)`
> 4. 更新源，下载库
> 
> ```
> apt install pve-headers-$(uname -r)
> apt install dkms build-* -y
> ```
> 
> 5. 编译，注意最后一定要加上参数 `-j 1` 即：`dkms install --force -m i915-sriov-dkms -v 6.2 -j 1`
> 6. 大概需要几分钟，等待结果，祝你好运！
> 
> 最重要的是**使用单线程编译**！
> 
> 附：dkms.conf 文件修改后内容
> 
> ```
> PACKAGE_NAME="i915-sriov-dkms"
> PACKAGE_VERSION="6.2"
> 
> MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build KVER=${kernelver}"
> CLEAN="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build KVER=${kernelver} clean"
> 
> BUILT_MODULE_NAME[0]="i915"
> DEST_MODULE_LOCATION[0]=/kernel/drivers/gpu/drm/i915
> 
> AUTOINSTALL=yes
> ```

感谢答复，我已编译成功。

我的代码不是最新的，虽然执行了pull，但是文件往/usr/src 复制时没有强制覆盖，导致编译的代码依然是老版本的。强制覆盖后编译成功。

我是多线程编译的。

感谢你的回答，给了我排查问题的启发。

9426224, 2023-12-06:
上面的方案这些我都看过了 说下我这边的情况
使用的CPU是12700H ES QY0Z 核显是UHD770 编译内核版本为6.2.16-3-pve
这个问题其实和dkms 单线程编译应该都没有直接的关系 查看make.log日志发现报错主要在于这一句

` implicit declaration of function ‘drm_dp_mst_hpd_irq_handle_event’ [-Werror=implicit-function-declaration] drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled);`

这句话的意思就是发现了一个函数的隐式声明，但是在代码的当前位置之前没有提供该函数的显式声明或定义
实际上就可以理解为dkms在编译这个新版本的仓库的时候发现了drm_dp_mst_hpd_irq_handle_event这个函数没有被定义，那么发现了问题我就去查询到了这个函数的定义来源于这个头文件 

`#include <drm/drm_dp_mst_helper.h>`

那么通过find命令可以查询到这个头文件在
**/usr/src/你编译仓库时的内核版本/include/drm/display/drm_dp_mst_helper.h**
这里，直接vi进去搜索是否存在drm_dp_mst_hpd_irq_handle_event的定义，发现我使用的这个内核版本 6.2.16-3-pve 里面没有定义这个函数 

同理我查询了我的debian上的其他几个内核中同样的文件内是否存在这个函数定义，这里包含5.15.104-1-pve/ 6.1.10-1-pve，这两个内核中同样没有定义这个函数，于是我直接安装了最新的6.5.11-6-pve内核，头文件中果然有了这个函数的定义

`int drm_dp_mst_hpd_irq_handle_event(struct drm_dp_mst_topology_mgr *mgr,
                                    const u8 *esi,
                                    u8 *ack,
                                    bool *handled);`

得出的结论就是其实还是和内核版本强依赖，如果你的内核版本报出了同样的错误，那么请安装比起你当前内核版本更新一点的内核头文件，然后去查看其中是否存在了这个函数的支持，如果没有发现，请继续安装更高版本的头文件，直到发现了合适的版本后

安装这个版本对应的内核，使用proxmox-boot-tool kernel pin工具固定新版本，同时编译命令设置为dkms install --force -m i915-sriov-dkms -v 6.2 -k 6.5.11-6-pve 使用-k来指定你要编译的dkms对应查找的内核版本，如果编译成功那么就没有问题了

lianghanqiang003, 2024-02-03:
> 上面的方案这些我都看过了 说下我这边的情况 使用的CPU是12700H ES QY0Z 核显是UHD770 编译内核版本为6.2.16-3-pve 这个问题其实和dkms 单线程编译应该都没有直接的关系 查看make.log日志发现报错主要在于这一句
>
> ` implicit declaration of function ‘drm_dp_mst_hpd_irq_handle_event’ [...]

6.2-16-3 安装6.5内核后成功了，感谢

ooninoo, 2024-06-16:
> 上面的方案这些我都看过了 说下我这边的情况 使用的CPU是12700H ES QY0Z 核显是UHD770 编译内核版本为6.2.16-3-pve 这个问题其实和dkms 单线程编译应该都没有直接的关系 查看make.log日志发现报错主要在于这一句
>
> ` implicit declaration of function ‘drm_dp_mst_hpd_irq_handle_event’ [...]

6.5.13-3 成功了