strongtz/i915-sriov-dkms #122: pve 6.5.3-1-pve error

skill7899, 2023-11-09:
[   50.620385] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[   50.620492] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   50.620534] i915 0000:00:02.1: Running in SR-IOV VF mode
[   50.621594] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.621842] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.623873] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   50.623918] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   50.624597] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.624808] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.625794] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   50.625803] i915 0000:00:02.1: HuC firmware PRELOADED
[   50.629147] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   50.629157] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   50.629383] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 1
[   50.629705] pci 0000:00:02.2: [8086:a780] type 00 class 0x030000
[   50.629734] pci 0000:00:02.2: DMAR: Skip IOMMU disabling for graphics
[   50.629829] pci 0000:00:02.2: Adding to iommu group 46
[   50.629839] pci 0000:00:02.2: vgaarb: bridge control possible
[   50.629841] pci 0000:00:02.2: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[   50.629846] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   50.629851] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   50.629941] i915 0000:00:02.2: enabling device (0000 -> 0002)
[   50.629970] i915 0000:00:02.2: Running in SR-IOV VF mode
[   50.630218] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.630280] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.630961] i915 0000:00:02.2: [drm] VT-d active for gfx access
[   50.631080] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[   50.631585] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.631626] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.631971] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   50.631975] i915 0000:00:02.2: HuC firmware PRELOADED
[   50.634331] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[   50.634342] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[   50.634600] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[   50.635065] pci 0000:00:02.3: [8086:a780] type 00 class 0x030000
[   50.635105] pci 0000:00:02.3: DMAR: Skip IOMMU disabling for graphics
[   50.635238] pci 0000:00:02.3: Adding to iommu group 47
[   50.635257] pci 0000:00:02.3: vgaarb: bridge control possible
[   50.635260] pci 0000:00:02.3: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[   50.635266] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   50.635273] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.635278] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   50.635391] i915 0000:00:02.3: enabling device (0000 -> 0002)
[   50.635431] i915 0000:00:02.3: Running in SR-IOV VF mode
[   50.635683] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.635748] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.636352] i915 0000:00:02.3: [drm] VT-d active for gfx access
[   50.636376] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[   50.636854] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.636926] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.637325] i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   50.637328] i915 0000:00:02.3: HuC firmware PRELOADED
[   50.639686] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[   50.639692] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[   50.639864] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[   50.640194] pci 0000:00:02.4: [8086:a780] type 00 class 0x030000
[   50.640218] pci 0000:00:02.4: DMAR: Skip IOMMU disabling for graphics
[   50.640340] pci 0000:00:02.4: Adding to iommu group 48
[   50.640348] pci 0000:00:02.4: vgaarb: bridge control possible
[   50.640349] pci 0000:00:02.4: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[   50.640354] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   50.640359] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.640362] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.640366] i915 0000:00:02.3: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   50.640427] i915 0000:00:02.4: enabling device (0000 -> 0002)
[   50.640456] i915 0000:00:02.4: Running in SR-IOV VF mode
[   50.640619] i915 0000:00:02.4: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.640685] i915 0000:00:02.4: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.641142] i915 0000:00:02.4: [drm] VT-d active for gfx access
[   50.641164] i915 0000:00:02.4: [drm] Using Transparent Hugepages
[... 54 lines cut ...]
[   50.653654] pci 0000:00:02.7: [8086:a780] type 00 class 0x030000
[   50.653676] pci 0000:00:02.7: DMAR: Skip IOMMU disabling for graphics
[   50.653761] pci 0000:00:02.7: Adding to iommu group 51
[   50.653770] pci 0000:00:02.7: vgaarb: bridge control possible
[   50.653772] pci 0000:00:02.7: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
[   50.653777] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   50.653782] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.653786] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.653790] i915 0000:00:02.3: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.653793] i915 0000:00:02.4: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.653797] i915 0000:00:02.5: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   50.653801] i915 0000:00:02.6: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   50.653875] i915 0000:00:02.7: enabling device (0000 -> 0002)
[   50.653901] i915 0000:00:02.7: Running in SR-IOV VF mode
[   50.654089] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.654158] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.654539] i915 0000:00:02.7: [drm] VT-d active for gfx access
[   50.654568] i915 0000:00:02.7: [drm] Using Transparent Hugepages
[   50.655031] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[   50.655078] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
[   50.655417] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   50.655420] i915 0000:00:02.7: HuC firmware PRELOADED
[   50.657384] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[   50.657393] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[   50.657616] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.7 on minor 7

skill7899, 2023-11-20:
内核退回到旧的也是报这个错，只能整个回滚pve。报这个错的时候，windows的vf不行了，dsm的也不行

zimmra, 2023-11-20:
I experience this as well.

The VFs are created, but when a VM started that use one it fails to start, and then all of the VFs dissappear.

zimmra, 2023-11-21:
@skill7899 

我想你能翻译我的英语。

I was able to make this work with some modifications

You can try my fork at [zimmra/i915-sriov-dkms](https://github.com/zimmra/i915-sriov-dkms) if you'd like

You can make the modification yourself too, see commit [777e10c ](https://github.com/zimmra/i915-sriov-dkms/commit/777e10cad67c873139f253194ec5a415133f6b66) for the file I changed to prevent this error

After doing this, I was successfully able to use the VFs. (You also need to make sure you do not select 'All Functions' when adding the PCI Device, as this will auto-select the 'master' PCIe device (e.g. 0000:00:02) instead of the 'vf' device (e.g. 0000:00:02.1). Even if you have a 'vf' device selected it will still revert to the 'master' device.)

skill7899, 2023-11-22:
> After doing this, I was successfully able to use the VFs. (You also need to make sure you do not select 'All Functions' when adding the PCI Device, as this will auto-select the 'master' PCIe device [...]

thanks you，i will test it

skill7899, 2023-11-22:
@zimmra  windows 10 and dsm sa6400 can‘not driver，windows driver code 43 

[image]

skill7899, 2023-11-22:
i guess GT0: IOV: Found interface version 0.1.4.1 is influenced by pve-firmware version，but i had rollback~

Coder84619, 2023-11-23:
@zimmra How do I rebuild dkms with the modified file? I still get the 0.1.4.1 error.
 
apt install git dkms build-* unzip -y
dkms install --force -m i915-sriov-dkms -v 6.1

skill7899, 2023-11-24:
> @zimmra How do I rebuild dkms with the modified file? I still get the 0.1.4.1 error.
> 
> apt install git dkms build-* unzip -y dkms install --force -m i915-sriov-dkms -v 6.1

dkms build --force -m i915-sriov-dkms -v 6.1
dkms install --force -m i915-sriov-dkms -v 6.1

Coder84619, 2023-11-24:
Proxmox 8.1 with Windows 11 Pro is working by following this post: https://www.derekseaman.com/2023/11/proxmox-ve-8-1-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

SGodV, 2024-01-21:
> 内核退回到旧的也是报这个错，只能整个回滚pve。报这个错的时候，windows的vf不行了，dsm的也不行

之前有版本可以吗，是哪个内核版本还是pve版本。我直接安装的pve8.1，内核6.5.11-7-pve。在DSM6400中直通整个核显成功，使用sriov的vGPU失败
[image]
在win10中无论是直通整个核显还是使用sriov都报错代码43。我确认我在使用sriov的时候没有勾选全部功能。
我使用的cpu是奔腾金牌7505

closed 2024-03-08 by skill7899

scyto, 2024-04-06:
> Proxmox 8.1 with Windows 11 Pro is working by following this post: https://www.derekseaman.com/2023/11/proxmox-ve-8-1-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

not for me it isn't, dreaded code 43, used to work on earlier 8.0 kernel versions, i am on proxmox 8.1 and 6.5.13-3-pve kernel

what intel driver version are you using?

skill7899, 2024-04-07:
> not for me it isn't, dreaded code 43, used to work on earlier 8.0 kernel versions, i am on proxmox 8.1 and 6.5.13-3-pve kernel
>
> what intel driver version are you using?

the laster

scyto, 2024-04-07:
> the laster

Thanks, appreciate it.  I think i just this second found my issue - i had forgotten there was a bunch of args in the vm10.conf file 
lesson learnt - always delete VMs rather than re-installing in them, just incase
i also think i am narrowing down yet one more cause of the dreaded win11 automatic repair boot loop....