strongtz/i915-sriov-dkms #123: Kernel 6.6 compile error

Bill5122, 2023-11-12:
The driver no longer compiles on kernel 6.6

[make.log](https://github.com/strongtz/i915-sriov-dkms/files/13328839/make.log)

Seems to be due to https://github.com/torvalds/linux/commit/71a7974ac7019afeec105a54447ae1dc7216cbb3

ich777, 2023-11-30:
@strongtz I'm able to reproduce the same on Kernel `6.6.3`, but it seems that more work needs to be done for 6.6.3 but I haven't got enough time yet to completely fix everything.

mio-19, 2024-01-25:
I am able to reproduce this on 6.6.13

a17t, 2024-01-27:
Fix:
The makro 
`FB_GEN_DEFAULT_DEFERRED_IO_OPS` 
used in file  `./drivers/gpu/drm/i915/display/intel_fbdev.c` 
has been renamed to
`FB_GEN_DEFAULT_DEFERRED_IOMEM_OPS`

This at least makes the driver compile on 6.6

ich777, 2024-01-27:
> This at least makes the driver compile on 6.6

Are you able to male a PR for that?

a17t, 2024-01-27:
@ich777 done

ich777, 2024-01-28:
@a17t can confirm that it is working fine now! Thanks. :)

I think this issue can be closed @Bill5122

mio-19, 2024-01-29:
What version is your 6.6? I am having other errors on 6.6.14 lts https://github.com/strongtz/i915-sriov-dkms/issues/135

ich777, 2024-01-29:
I have compiled a package for Kernel `6.6.13` [here](https://github.com/ich777/unraid-i915-sriov/releases/tag/6.6.13-Unraid).

ich777, 2024-02-12:
@strongtz I think this can be closed since it's working again. :)

closed 2024-02-12 by strongtz