strongtz/i915-sriov-dkms #125: Kernel 6.5.11-4-pve errors in dmesg (seem harmless)

Coder84619, 2023-11-25:
These errors seem to be harmless, as vGPU is working with Windows 11. But maybe some code needs a bit of cleanup? Proxmox 8.1.3 on an Intel i5-1240P. 

`32.076467] kvmgt: disagrees about version of symbol i915_gem_object_set_to_cpu_domain
[   32.076470] kvmgt: Unknown symbol i915_gem_object_set_to_cpu_domain (err -22)
[   32.076518] kvmgt: disagrees about version of symbol i915_gem_object_alloc
[   32.076519] kvmgt: Unknown symbol i915_gem_object_alloc (err -22)
[   32.076530] kvmgt: disagrees about version of symbol i915_gem_object_create_shmem
[   32.076531] kvmgt: Unknown symbol i915_gem_object_create_shmem (err -22)
[   32.076550] kvmgt: disagrees about version of symbol i915_gem_object_pin_map
[   32.076551] kvmgt: Unknown symbol i915_gem_object_pin_map (err -22)
[   32.076561] kvmgt: disagrees about version of symbol i915_gem_gtt_insert
[   32.076561] kvmgt: Unknown symbol i915_gem_gtt_insert (err -22)
[   32.076563] kvmgt: disagrees about version of symbol i915_request_add
[   32.076564] kvmgt: Unknown symbol i915_request_add (err -22)
[   32.076572] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_fini
[   32.076573] kvmgt: Unknown symbol i915_gem_ww_ctx_fini (err -22)
[   32.076607] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_init
[   32.076608] kvmgt: Unknown symbol i915_gem_ww_ctx_init (err -22)
[   32.076609] kvmgt: disagrees about version of symbol i915_reserve_fence
[   32.076610] kvmgt: Unknown symbol i915_reserve_fence (err -22)
[   32.076614] kvmgt: disagrees about version of symbol i915_gem_object_ggtt_pin_ww
[   32.076615] kvmgt: Unknown symbol i915_gem_object_ggtt_pin_ww (err -22)
[   32.076656] kvmgt: disagrees about version of symbol i915_gem_object_init
[   32.076657] kvmgt: Unknown symbol i915_gem_object_init (err -22)
[   32.076695] kvmgt: disagrees about version of symbol _i915_vma_move_to_active
[   32.076696] kvmgt: Unknown symbol _i915_vma_move_to_active (err -22)
[   32.076698] kvmgt: disagrees about version of symbol i915_request_create
[   32.076699] kvmgt: Unknown symbol i915_request_create (err -22)
[   32.076714] kvmgt: disagrees about version of symbol i915_ppgtt_create
[   32.076715] kvmgt: Unknown symbol i915_ppgtt_create (err -22)
[   32.076732] kvmgt: disagrees about version of symbol i915_gem_ww_ctx_backoff
[   32.076732] kvmgt: Unknown symbol i915_gem_ww_ctx_backoff (err -22)
[   32.076735] kvmgt: disagrees about version of symbol i915_unreserve_fence
[   32.076735] kvmgt: Unknown symbol i915_unreserve_fence (err -22)
[   32.076741] kvmgt: disagrees about version of symbol __i915_gem_object_set_pages
[   32.076742] kvmgt: Unknown symbol __i915_gem_object_set_pages (err -22)
[   32.076759] kvmgt: disagrees about version of symbol i915_request_wait
[   32.076759] kvmgt: Unknown symbol i915_request_wait (err -22)
[   32.076767] kvmgt: disagrees about version of symbol __i915_gem_object_flush_map
[   32.076767] kvmgt: Unknown symbol __i915_gem_object_flush_map (err -22)`

kinsong, 2023-11-26:
After multiple successful attempts, the operation is as follows:

nano /etc/default/grub
###GRUB_CMDLINE_LINUX_DEFAULT="quiet"
## GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on i915.enable_guc=3 i915.max_vfs=7"
update-grub
update-initramfs -u

apt install sysfsutils -y
echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7" >> /etc/sysfs.conf
reboot

chmod +x /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
echo 3 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
chmod -x /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
ll /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
## 3
# or
sudo nano /etc/rc.local
  #!/bin/bash
  echo 3 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
  exit 0
chmod +x /etc/rc.local
reboot

rm -rf /var/lib/dkms/i915-sriov-dkms*
rm -rf /usr/src/i915-sriov-dkms*
rm -rf ~/i915-sriov-dkms

cd ~
git clone https://github.com/strongtz/i915-sriov-dkms.git
cd ~/i915-sriov-dkms
cp -a ~/i915-sriov-dkms/dkms.conf{,.1.bak}
sed -i 's/"@_PKGBASE@"/"i915-sriov-dkms"/g' ~/i915-sriov-dkms/dkms.conf
sed -i 's/"@PKGVER@"/"6.5.11-4"/g' ~/i915-sriov-dkms/dkms.conf
sed -i 's/ -j$(nproc)//g' ~/i915-sriov-dkms/dkms.conf
cat ~/i915-sriov-dkms/dkms.conf
dkms add .

cd /usr/src/i915-sriov-dkms-6.5.11-4
sudo dkms status
## i915-sriov-dkms/6.5.11-4: added
sudo dkms install -m i915-sriov-dkms -v 6.5.11-4 -k 6.5.11-4-pve --force -j 1
## i915-sriov-dkms/6.5.11-4, 6.5.11-4-pve, x86_64: installed

lspci | grep VGA
## 00:02.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
## 00:02.1 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
## 00:02.2 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)
## 00:02.3 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 730] (rev 0c)

Coder84619, 2023-11-26:
@kinsong I got to the point: cd /usr/src/i915-sriov-dkms-6.5.11-4

and the directory is not found.

kinsong, 2023-11-26:
@Coder84619 You need to execute it first in the download directory：
dkms add .
By default, it will be generated in usr/src/

Coder84619, 2023-11-26:
I worked through it and now the dmesg logs are clean. Thanks! I did need to make one tweak:

sed -i 's/"@pkgver@"/"6.5.11-4"/g' ~/i915-sriov-dkms/dkms.conf

should be (PKGVER in all caps):

sed -i 's/"@PKGVER@"/"6.5.11-4"/g' ~/i915-sriov-dkms/dkms.conf

Otherwise I got:
root@proxmox1:~/i915-sriov-dkms# cat ~/i915-sriov-dkms/dkms.conf
PACKAGE_NAME="i915-sriov-dkms"
PACKAGE_VERSION="@PKGVER@"

kinsong, 2023-11-26:
The comment section will be automatically formatted, just set it to shell now

85550867, 2023-12-02:
6.5.11-6  
dkms install -m i915-sriov-dkms -v 6.5.11-6 -k 6.5.11-6-pve --force -j 1

error:
DKMS make.log for i915-sriov-dkms-6.5.11-6 for kernel 6.5.11-6-pve (x86_64)
Sat Dec  2 06:45:13 PM CST 2023
make: Entering directory '/usr/src/linux-headers-6.5.11-6-pve'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Debian 12.2.0-14) 12.2.0
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.11-6/build/drivers/gpu/drm/i915/i915_driver.o
gcc: error: unrecognized command-line option ‘-mharden-sls=all’
gcc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
gcc: error: unrecognized command-line option ‘-fzero-call-used-regs=used-gpr’
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5.11-6/build/drivers/gpu/drm/i915/i915_driver.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.5.11-6-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5.11-6/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.11-6-pve'

edgesider, 2024-01-14:
You may have added "kvmgt" in `/etc/modules` according to pve's [mdev documentation](https://pve.proxmox.com/wiki/PCI(e)_Passthrough#_mediad_devices_vgpu_gvt_g). If using sriov, the "kvmgt" module is unnecessary, just remove it.

mio-19, 2024-01-23:
@kinsong I used a similar setup but things were not working.

```bash
apt install git dkms build-essential pve-headers -y
git clone --depth 1 https://github.com/strongtz/i915-sriov-dkms.git /usr/src/i915-sriov-dkms-6.5.11-7
cd /usr/src/i915-sriov-dkms-6.5.11-7
sed -i 's|@_PKGBASE@|i915-sriov-dkms|g' dkms.conf
sed -i 's|@PKGVER@|6.5.11-7|g' dkms.conf
dkms install --force -m i915-sriov-dkms -v 6.5.11-7 -k 6.5.11-7-pve
nano /etc/default/grub # GRUB_CMDLINE_LINUX_DEFAULT  intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 quiet
update-grub && update-initramfs -u && pve-efiboot-tool refresh && echo Done~~
apt install -y sysfsutils
lspci | grep VGA # Should be: 00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3" > /etc/sysfs.conf

```

After Reboot

```

$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
$ echo 3 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
write: Numerical result out of range
```

closed 2024-12-29 by strongtz