strongtz/i915-sriov-dkms #129: Proxmox 8.1.3 Kernel 6.5.11-6 - i915 Setup Error not enough MMIO resources for SR-IOV after reboot

mofakhar, 2023-12-05:
Hi, I did try following the steps that you have shared but once the node is rebooted I'm getting the following errors:
`
root@pve:~# dmesg|grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.11-6-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.098964] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.11-6-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    3.279619] i915: module verification failed: signature and/or required key missing - tainting kernel
[    3.597246] i915 0000:00:02.0: Running in SR-IOV PF mode
[    3.597828] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.597864] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.597927] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.598315] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.599844] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[    3.601286] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    3.610521] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.13.1
[    3.610524] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    3.627559] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    3.628034] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    3.628036] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    3.628504] i915 0000:00:02.0: [drm] GuC RC: enabled
[    3.629556] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    3.629667] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.665846] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.667048] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.667172] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    3.670046] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.670544] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    3.670978] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.879979] i915 0000:00:02.0: not enough MMIO resources for SR-IOV
[    4.983212] i915 0000:00:02.0: [drm] *ERROR* Failed to enable 7 VFs (-ENOMEM)
[    5.087218] i915 0000:00:02.0: not enough MMIO resources for SR-IOV
[    5.189850] i915 0000:00:02.0: [drm] *ERROR* Failed to enable 7 VFs (-ENOMEM)
`

I'm running a Proxmox 8.1.3 with Kernel 6.5.11-6.

In addition, after the module compilatoon, I got the following file generated:  /var/lib/dkms/i915-sriov-dkms/6.5.11-6/6.5.11-6-pve/x86_64/module/i915.ko

However, in the log file i saw a couples of warnning as you can see below:
`
[make.log](https://github.com/strongtz/i915-sriov-dkms/files/13555203/make.log)
`

VictC79, 2023-12-05:
I also tried the PVE8.1.3 version, and it was exactly the same as your compilation log, but I checked that there was an error in dmesg.

root@pve:~# dmesg |grep "0000:00:02"
[    0.522729] pci 0000:00:02.0: [8086:a7a8] type 00 class 0x030000
[    0.522736] pci 0000:00:02.0: reg 0x10: [mem 0x6123000000-0x6123ffffff 64bit]
[    0.522741] pci 0000:00:02.0: reg 0x18: [mem 0x4000000000-0x400fffffff 64bit pref]
[    0.522744] pci 0000:00:02.0: reg 0x20: [io  0x3000-0x303f]
[    0.522756] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.522758] pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    0.522781] pci 0000:00:02.0: reg 0x344: [mem 0x611c000000-0x611cffffff 64bit]
[    0.522782] pci 0000:00:02.0: VF(n) BAR0 space: [mem 0x611c000000-0x6122ffffff 64bit] (contains BAR0 for 7 VFs)
[    0.522786] pci 0000:00:02.0: reg 0x34c: [mem 0x6000000000-0x601fffffff 64bit pref]
[    0.522786] pci 0000:00:02.0: VF(n) BAR2 space: [mem 0x6000000000-0x60dfffffff 64bit pref] (contains BAR2 for 7 VFs)
[    0.565485] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    0.565485] pci 0000:00:02.0: vgaarb: bridge control possible
[    0.565485] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.583200] pci 0000:00:02.0: Adding to iommu group 0
[    3.721711] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[    3.721731] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV

Diluka, 2024-01-06:
Proxmox 8.1.3 with Kernel 6.5.11-7

restart twice

gongdao123, 2024-02-28:
Did you enable SR-IOV in BIOS settings?

https://zhuanlan.zhihu.com/p/561002399

mofakhar, 2024-04-05:
Yes the Bios entry was already set to enable.

closed 2024-12-29 by strongtz