strongtz/i915-sriov-dkms #130: N100 Unable to virtualize device properly

Meilinhost, 2023-12-11:
I've tried to install it more than 20 times according to the tutorials on the web, but it fails.
I read that 6.2.16-9 has the best support for sriov, so I locked the kernel to 6.2.16-9, compiled, setup, rebooted, and no device.
I don't know what's wrong, I'm running out of ideas.
[image]
[image]
[image]

closed 2023-12-11 by Meilinhost

gavinsama, 2024-04-28:
me too,I'm so tired!!!

Meilinhost, 2024-04-28:
> 我也是，好累！！！

A while ago I finished sriov. configuration utilizing the pve_source script, you can try it out

Meilinhost, 2024-04-28:
> > > me too,I'm so tired!!!
> > 
> > 
> > A while ago I finished sriov. configuration utilizing the pve_source script, you can try it out
> 
> Thank you ！what's your PVE version ？Kernel version？

pve is using the latest version 8.x, the kernel is also updated to the latest, pve version and kernel is not the main problem, but the driver problem, you can go to see the article, the keyword is "pve_source".

gavinsama, 2024-04-29:
> pve is using the latest version 8.x, the kernel is also updated to the latest, pve version and kernel is not the main problem, but the driver problem, you can go to see the article, the keyword is [...]

thank you so much~