strongtz/i915-sriov-dkms #132: Document why this should be used instead of Intel's official DKMS

eero-t, 2023-12-22:
Intel's official DKMS sources: https://github.com/intel-gpu/intel-gpu-i915-backports

Available from Intel driver package repositories: https://dgpu-docs.intel.com/driver/installation.html

Already support SR-IOV: https://dgpu-docs.intel.com/driver/kernel-driver-types.html#differences-between-the-out-of-tree-driver-and-the-upstream-kernel

What would be the reasons to use this DKMS instead (please document them in your README)?

Diluka, 2024-01-06:
Have you successfully enabled SRIOV on PVE 8.1.x according to the official documentation? I tried it over a month ago with no success.

brndd, 2024-01-08:
The official backports seem to only support a small handful of LTS enterprise distros.

eero-t, 2024-01-08:
> Have you successfully enabled SRIOV on PVE 8.1.x according to the official documentation? I tried it over a month ago with no success.

What's "PVE"?

Diluka, 2024-01-08:
> > Have you successfully enabled SRIOV on PVE 8.1.x according to the official documentation? I tried it over a month ago with no success.
> 
> What's "PVE"?

https://www.proxmox.com based on Debian 12 bookworm

eero-t, 2024-01-08:
> The official backports seem to only support a small handful of LTS enterprise distros.

supported = validated on that set.  But because we're talking just DKMS, only the kernel is relevant.

Because Intel backports repo states same `backport/main` branch supporting different kernel versions in SLES, RHEL and Ubuntu, I would assume DKMS being fairly likely to work also in other distributions having same kernel versions, and the problem being more with the distro specific DKMS packaging?

(Ubuntu is basically stabilized Debian testing snapshot with Ubuntu additions, so DKMS package for Ubuntu could work on Debian too, for compatible kernel version.)

> https://www.proxmox.com/ based on Debian 12 bookworm

Bookwork seems to be running 6.1 kernel: https://packages.debian.org/bookworm-updates/linux-image-amd64

@Diluka What kernel you tried, and was the DKMS package failing to install on that kernel version [1], or were you "just" unable to get SRIOV working when DKMS otherwise worked OK?   If latter, what HW you tried this on, and did you see any kernel error messages for your kernel command line SR-IOV parameters?

----

[1] As to DKMS install, I recently tried Intel DKMS with Ubuntu 22.04 on Intel Arc HW (which is lacking SRIOV).  Latest 22.04 HWE kernel is currently 6.2.  While _earlier_ (Intel "client" repo) DKMS version did not support 6.2 or earlier 6.1 HWE, current one did install / work on 6.2:
* https://github.com/intel-gpu/documentation/issues/84
* https://github.com/intel-gpu/intel-gpu-i915-backports/blob/backport/main/docs/README_ubuntu.md

brndd, 2024-01-08:
> supported = validated on that set. But because we're talking just DKMS, only the kernel is relevant.

And the supported distros seem to only support fairly old kernels. I could not find a list of supported kernel versions anywhere on the linked Intel pages though, but I doubt 6.5 is supported.

eero-t, 2024-01-09:
> And the supported distros seem to only support fairly old kernels. I could not find a list of supported kernel versions anywhere on the linked Intel pages though

Supported kernels are listed in the backports DKMS repo main README: https://github.com/intel-gpu/intel-gpu-i915-backports/blob/backport/main/README.md

If one wants to know which particular kernel(s) given DKMS package version supports, I guess one can check the matching tag from backports repo, and docs belonging to it, although it's rather inconvenient: https://github.com/intel-gpu/intel-gpu-i915-backports/tags

> but I doubt 6.5 is supported.

Not by backport, right.

First `i915` KMD dGPU support bits were upstreamed to kernel 5.15 I think, and support was enabled in 6.2 i.e. it's first kernel version supporting Intel dGPUs without `i915.force_probe`.

So for "general GPU use", there may be no need for the backport on newer kernel versions. But backport DKMS indeed does support couple of extra features like SRIOV, that the upstream i915 does not: https://dgpu-docs.intel.com/driver/kernel-driver-types.html

=> maybe README in this repo could point to above 2 docs, as reason for its existence?

----

PS. Because this repo is based on the backport driver, I would guess this not being validated for the other kernel versions either, or is somebody running at least kernel i-g-t & Mesa piglit test-suites for the other kernel versions on some HW?
* https://gitlab.freedesktop.org/drm/igt-gpu-tools/
* https://gitlab.freedesktop.org/mesa/piglit

brndd, 2024-01-10:
> So for "general GPU use", there may be no need for the backport on newer kernel versions. But backport DKMS indeed does support couple of extra features like SRIOV, that the upstream i915 does not

I think SR-IOV is the only reason anyone would be desperate enough to use a random experimental kernel module from GitHub anyway ;-). Once SR-IOV is upstreamed, this dkms has no reason to exist.

Diluka, 2024-01-30:
tried [official installation](https://dgpu-docs.intel.com/driver/installation.html) in Ubuntu 22.04 HWE server VM. stuck [here](https://dgpu-docs.intel.com/driver/installation.html#ubuntu-package-installation)
```
Error! Bad return status for module build on kernel: 6.5.0-15-generic (x86_64)
Consult /var/lib/dkms/intel-i915-dkms/1.23.9.11.231003.15/build/make.log for more information.
dpkg: error processing package intel-i915-dkms (--configure):
 installed intel-i915-dkms package post-installation script subprocess returned error exit status 10
```

[make.log](https://github.com/strongtz/i915-sriov-dkms/files/14091560/make.log)

I don't know anything about Linux, hardware, or C language, so there is absolutely nothing I can do about this kind of mistake.
That's why I use this repo

michael-pptf, 2024-04-14:
Wow we've got someone working for Intel to comment on this hack!

The reason to use this is very simple. Intel Official DKMS doesn't work with Proxmox, and people want to use SRIOV with actual VIRTUAL MACHINES.

Does Intel really expects anyone who wants to use DKMS to install vanilla Ubuntu/Debian/RHEL with some very specific kernels and slap KVM/QEMU on it to run VMs? I guess the majority of people would like to use a hypervisor such as Proxmox to make life easier.

johntdavis84, 2024-04-25:
> Wow we've got someone working for Intel to comment on this hack!
> 
> The reason to use this is very simple. Intel Official DKMS doesn't work with Proxmox, and people want to use SRIOV with actual VIRTUAL MACHINES.
> 

This is what brought me here. And Proxmox is _very aggressive_ with their kernel versions. This repo is recommended by Wendell at Level1Techs for use with Proxmox, as well.

The 6.5 branch was standard on Proxmox 8.1, and Proxmox 8.2 came out this week on kernel 6.8. I had to install the last iteration of the 6.5 kernel to get this module installed.

brndd, 2024-06-03:
@eero-t Would you happen to know if it's planned to upstream SR-IOV support into mainline kernels? I believe I heard last year that it was planned for Q1 2024, but evidently that didn't end up happening.

It is a quite useful feature outside of the datacenter too, for running accelerated virtual machines using Looking Glass.