strongtz/i915-sriov-dkms #139: Introduced instability/panic on Kernel 6.5

a17t, 2024-02-05:
After fixing some minor compile bugs (see previous PRs) this module built for kernel 6.5.11 on Proxmox 8.1.
The idea being, to pass through the GPU of the intel n100 to VMs, enabling hardware encoding.

Unfortunately with the module loaded and enabled, the system panics every so often (4-12h). I tried getting a dmesg or any further log output, which was impossible up to now. (System runs headless) 

I will enrich this issue with the panic (I will connect a screen and try to get that panic on screen). In case it  turns out, it is not this driver, I will of course close this. 

Anyone else experiencing similar behaviour.

Hardware used:
Intel N100 (13 Gen)
32GB DDR5 Ram

(Random Aliexpress board...)

mayanez, 2024-03-02:
Can confirm I have the same issue. 

I tried various methods to try and get logging capturing any issue, but there was never any I could identify. It occurs as a hard shutdown from what I can tell. I'm running a i5-13400k. I've since disabled the DKMS and stability has been rock solid, so I can rule out bad HW.

a17t, 2024-03-03:
Unfortunately did not have the time to investigate further. But if disabling dkms solved the instability for you I will give it a try. I anyway want to run that all on a headless machine.

BenSchweikert, 2024-03-07:
Same here with a ASRock Z690 Phantom Gaming 4 and Intel Core i5-12500. I checked PowerSaving Options, Bios settings and Memtest. I am using now iGPU passthrough to a VM which is solid. So I currently think it is related to the dkms. I checked several Kernel Versions starting with 6.2 up to 6.5 with a Proxmox Host. The hard reboots occured with all kernels but I think the frequency was different.

azerty9971, 2024-05-19:
Could this be because of this : https://patchwork.kernel.org/project/intel-gfx/patch/BYAPR03MB4168C6D020B750EAF8021731ADE22@BYAPR03MB4168.namprd03.prod.outlook.com/ ?

closed 2024-12-29 by strongtz