strongtz/i915-sriov-dkms #140: Does not build on Kernel 6.7.8

ich777, 2024-03-06:
i915-sriov does not build on Kernel version 6.7.8 and reports that message:
```/
usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: implicit declaration of function 'INTEL_MTL_M_IDS'; did you mean 'INTEL_MTL_IDS'? [-Werror=implicit-function-declaration]
  220 |         INTEL_MTL_M_IDS(0),
      |         ^~~~~~~~~~~~~~~
      |         INTEL_MTL_IDS
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: initializer element is not constant
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/intel_device_info.c:220:9: note: (near initialization for 'subplatform_m_ids[0]')
  CC [M]  /usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_mm.o
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/intel_device_info.c:224:9: error: implicit declaration of function 'INTEL_MTL_P_IDS'; did you mean 'INTEL_MTL_IDS'? [-Werror=implicit-function-declaration]
  224 |         INTEL_MTL_P_IDS(0),
      |         ^~~~~~~~~~~~~~~
      |         INTEL_MTL_IDS
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/intel_device_info.c:224:9: error: initializer element is not constant
```

@a17t did you yet build against a newer Kernel, I think it has to do because they got rid of the subplatforms see [here](https://patchwork.kernel.org/project/intel-gfx/patch/20230718222753.1075713-16-matthew.d.roper@intel.com/#25432147).

YuriTukhachevsky, 2024-03-12:
Same issue here on Kernel version 6.7.7-1.1 with exactly the same output  
```
/var/lib/dkms/i915-sriov-dkms/6.7/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: implicit declaration of function ‘INTEL_MTL_M_IDS’; did you mean ‘INTEL_MTL_IDS’? [-Werror=implicit-function-declaration]
  220 |         INTEL_MTL_M_IDS(0),
      |         ^~~~~~~~~~~~~~~
      |         INTEL_MTL_IDS
/var/lib/dkms/i915-sriov-dkms/6.7/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: initializer element is not constant
/var/lib/dkms/i915-sriov-dkms/6.7/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: note: (near initialization for ‘subplatform_m_ids[0]’)
/var/lib/dkms/i915-sriov-dkms/6.7/build/drivers/gpu/drm/i915/intel_device_info.c:224:9: error: implicit declaration of function ‘INTEL_MTL_P_IDS’; did you mean ‘INTEL_MTL_IDS’? [-Werror=implicit-function-declaration]
  224 |         INTEL_MTL_P_IDS(0),
      |         ^~~~~~~~~~~~~~~
      |         INTEL_MTL_IDS
/var/lib/dkms/i915-sriov-dkms/6.7/build/drivers/gpu/drm/i915/intel_device_info.c:224:9: error: initializer element is not constant
```
Awaiting for solutions

ich777, 2024-03-12:
@strongtz do you plan to support Kernel 6.7+?
I just took a quick look at it but it seems that this change is a bit more involved since much changed from what I saw.

strongtz, 2024-03-26:
> @strongtz do you plan to support Kernel 6.7+? I just took a quick look at it but it seems that this change is a bit more involved since much changed from what I saw.

Sorry but I don't use SR-IOV on my machine anymore, so I'm not able to test it.  :(

ich777, 2024-03-26:
> Sorry but I don't use SR-IOV on my machine anymore, so I'm not able to test it. :(

Oh no too bad...
So the project is basically dead or better speaking relies on community support to keep it alive? :/

chchia, 2024-03-31:
this is the biggest bad news....

lintush, 2024-03-31:
it seems intel i915 sr-iov patch is mainlined in kernel 6.9.
https://www.phoronix.com/news/Intel-Xe-DRM-Linux-6.9-Pull
i'm planning to buy N100(alder) pc for vm host. could someone test new 6.9 mainline kernel to check sr-iov working on alder or later?

KaarelP2rtel, 2024-03-31:
@lintush you seem to be wrong. They started work on supporting sr-iov for the new xe driver. Nothing on i915 yet.

lintush, 2024-04-01:
> @lintush you seem to be wrong. They started work on supporting sr-iov for the new xe driver. Nothing on i915 yet.

oh I miswrote. you're correct currently i915 has no sriov support on mainline. the new "xe" driver for tiger lake and newer gpu (event it don't have "Xe" branding...) seem to have support on mainline 6.9. sorry for inconvenience I just want someone to test :)
I think the users of 6.7.8 can smoothly move to 6.9 if the xe driver works.

KaarelP2rtel, 2024-04-02:
Sriov does not work yet. All they have done is started working on sriov full support.
I tried 6.9 and after forcing xe driver usage I was not able to sriov working.

https://www.reddit.com/r/VFIO/comments/17sw0bt/intel_begins_sorting_out_sriov_support_for_the_xe/

NovaPixell, 2024-04-24:
Same issue while building dkms for 6.8.4-2 PVE, for Intel N95 GPU

`/var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: implicit declaration of function ‘INTEL_MTL_M_IDS’; did you mean ‘INTEL_MTL_IDS’? [-Werror=implicit-function-declaration]
  220 |         INTEL_MTL_M_IDS(0),
      |         ^~~~~~~~~~~~~~~
      |         INTEL_MTL_IDS
/var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: initializer element is not constant
/var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: note: (near initialization for ‘subplatform_m_ids[0]’)
/var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:224:9: error: implicit declaration of function ‘INTEL_MTL_P_IDS’; did you mean ‘INTEL_MTL_IDS’? [-Werror=implicit-function-declaration]
  224 |         INTEL_MTL_P_IDS(0),
      |         ^~~~~~~~~~~~~~~
      |         INTEL_MTL_IDS
/var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:224:9: error: initializer element is not constant
/var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:224:9: note: (near initialization for ‘subplatform_p_ids[0]’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.8.4-2-pve/Makefile:1926: /var/lib/dkms/i915-sriov-dkms/6.8.4-2/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.4-2-pve'`

Will the repo be updated to fix the bug? Or is it something that will take a long time?

ich777, 2024-04-25:
@NovaPixell 4th comment in this issue from @strongtz 

If someone has the time/hardware to create a PR and submit it then it would be totally possible.

Hell0wor1d, 2024-04-26:
> Same issue while building dkms for 6.8.4-2 PVE, for Intel N95 GPU
>
> ```
> /var/lib/dkms/i915-sriov-dkms/6.8.4-2/build/drivers/gpu/drm/i915/intel_device_info.c:220:9: error: implicit declaration of [...]

me too, could anyone help to fix the issue?

ich777, 2024-05-07:
@shenwii builds fine against 6.7.12, have to do some testing and report back. Great work so far! Thanks!

BTW the build fails against 6.8.9 which I already mentioned in your PR

ich777, 2024-05-07:
Fixed in #168

closed 2024-05-08 by ich777