strongtz/i915-sriov-dkms #141: Parsec shows that The host encoder failed [-15002].

withlqs, 2024-03-08:
I follow https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/  and everything is going well, except for Parsec.

When I connect to Parsec on the windows guest, Parsec shows that The host encoder failed [-15002].

withlqs, 2024-03-08:
Here is the log of Parsec host on the windwos guest VM.

> [I 2024-03-08 11:12:10] STUN reply from ::ffff:35.169.37.243:3478
[D 2024-03-08 11:12:10] net           = BUD|::ffff:192.168.50.56|31858
[D 2024-03-08 11:12:10] Virtual Display Driver State: OK
[I 2024-03-08 11:12:10] withlqs#9877093 connected.
[D 2024-03-08 11:12:10] dxgi          = 1.5
[I 2024-03-08 11:12:10] * aexclusive_start/acapture_start[459] = -9000
[D 2024-03-08 11:12:10] format        = BGRA
[I 2024-03-08 11:12:10] * encode_init[127] = -15101
[I 2024-03-08 11:12:10] * encode_init[127] = -15101
[I 2024-03-08 11:12:10] 'SetDisplayConfig' failed because no displays are connected.
[I 2024-03-08 11:12:10] * aexclusive_start/acapture_start[459] = -9000
[D 2024-03-08 11:12:10] display_x     = 3840
[D 2024-03-08 11:12:10] display_y     = 2160
[D 2024-03-08 11:12:10] display_hz    = 60
[I 2024-03-08 11:12:10] * aexclusive_start/acapture_start[459] = -9000
[I 2024-03-08 11:12:11] * aexclusive_start/acapture_start[459] = -9000
[D 2024-03-08 11:12:11] encoder       = intel
[D 2024-03-08 11:12:11] input         = rgba
[D 2024-03-08 11:12:11] codec         = h265
[D 2024-03-08 11:12:11] encode_x      = 1920
[D 2024-03-08 11:12:11] encode_y      = 1080
[I 2024-03-08 11:12:11] * aexclusive_start/acapture_start[459] = -9000
[I 2024-03-08 11:12:11] * mfxe_encode/MFXVideoCORE_SyncOperation[482] = -1
[I 2024-03-08 11:12:11] * aexclusive_start/acapture_start[459] = -9000
[D 2024-03-08 11:12:11] display_x     = 1920
[D 2024-03-08 11:12:11] display_y     = 1080
[D 2024-03-08 11:12:11] display_hz    = 60
[I 2024-03-08 11:12:13] Virtual tablet removed due to client disconnect
[I 2024-03-08 11:12:13] * host_msg_thread_cleanup[2433] = -15002
[I 2024-03-08 11:12:13] withlqs#9877093 disconnected.

michael-pptf, 2024-04-14:
Is your VF correctly identified by Windows? Go to Device Manager and check if it is code 43. 

As far as I know, the drivers used could be a problem. gfx_win_101.4146 is the one that works for me.

withlqs, 2024-04-14:
> Is your VF correctly identified by Windows? Go to Device Manager and check if it is code 43.
> 
> As far as I know, the drivers used could be a problem. gfx_win_101.4146 is the one that works for me.

My VF is correctly identified by Windows and the VF in Device Manager dose not show any error code.

michael-pptf, 2024-04-25:
> > Is your VF correctly identified by Windows? Go to Device Manager and check if it is code 43.
> > As far as I know, the drivers used could be a problem. gfx_win_101.4146 is the one that works for me.
> 
> My VF is correctly identified by Windows and the VF in Device Manager dose not show any error code.

Yes, just happened to me, too.

I was able to remove the driver and use this instead, version 4316.
https://www.intel.com/content/www/us/en/download/741626/780560/intel-arc-pro-graphics-windows.html

Proxmox 8.1 + 6.5-13 kernel

XuJiaobin, 2024-05-30:
> > > Windows 是否正确识别了您的 VF？转到设备管理器并检查其代码是否为 43。
> > > 据我所知，使用的驱动程序可能存在问题。gfx_win_101.4146 适合我的那个。
> > 
> > 
> > 我的 VF 被 Windows 正确识别，并且设备上的 VF 没有显示任何错误代码。
> 
> 是的，我也刚刚遇到了痛苦。
> 
> 删除该驱动程序并使用这个版本 4316。https [://www.intel.com/content/www/us/en/download/741626/780560/intel-arc-pro-graphics-windows.html](https://www.intel.com/content/www/us/en/download/741626/780560/intel-arc-pro-graphics-windows.html)
> 
> Proxmox 8.1 + 6.5-13 内核 ![CleanShot 2024-04-25 于 01 01 09](https://private-user-images.githubusercontent.com/59861321/325524230-e0d52266-733c-4096-a72b-386ee2319754.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwNDg0ODMsIm5iZiI6MTcxNzA0ODE4MywicGF0aCI6Ii81OTg2MTMyMS8zMjU1MjQyMzAtZTBkNTIyNjYtNzMzYy00MDk2LWE3MmItMzg2ZWUyMzE5NzU0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTMwVDA1NDk0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhYmRjZTcyM2ZjYTViZDAxMmU1ZDFlODJlMDQ3NjE1ZjM0YzY5ZGEzYmI2NDMyMjBkOGMzOGU0NTJiZDM4MDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.sxceprfrXHGBmM4y05OJdI92IFlacfIgc-BCCGZkUvM)

Oh! This problem has been a long-standing issue for me.The issue has been resolved, Thank you.

GuyPaddock, 2024-06-09:
I ran into a similar issue in a VM that's running on QEMU KVM with a Tesla M40. Parsec was working, then it stopped working around the time I installed the nVidia drivers, so I thought that's what the issue was. I was wrong.

Instead, I had forgotten that I had changed the startup resolution of QEMU to 1440x900 so that the machine was more usable in Virt Manager through HP iLO. When I changed the resolution to 1024x768 or 1920x1080, Parsec was able to stream without the blank screen and cryptic log errors. So, see if you get different results with different resolutions.

working-name, 2024-12-15:
Same/similar issue on a real host with 11th gen i7 and a 3060, but nothing plugged into the 3060, just the motherboard's HDMI (using integrated gpu).

```
[I 2024-12-14 21:46:08] STUN reply from ::ffff:52.86.26.213:3478
[I 2024-12-14 21:46:09] * nat_read_packet[469] = -700000
[D 2024-12-14 21:46:09] net = BUD|::ffff:14.11.11.16|30734
[D 2024-12-14 21:46:09] BUD AES_GCM = 256
[D 2024-12-14 21:46:09] VirtualUsbControllerIsCompatiblePresent failed with e0000001 for api version 1, no compatible driver found.
[I 2024-12-14 21:46:09] Parsec Virtual USB Adapter failed to initialize, no virtual device support
[I 2024-12-14 21:46:09] Failed to initialize host gamepad input via ViGEm
[I 2024-12-14 21:46:09] * vmouse_create/PCVUD_vmouse_device_create[51] = -536870911
[I 2024-12-14 21:46:09] wurstelkrak#12174180 connected.
[D 2024-12-14 21:46:09] dxgi = 1.5
[D 2024-12-14 21:46:09] format = BGRA
[D 2024-12-14 21:46:09] nvEncOpenEncodeSessionEx failed: 1
[I 2024-12-14 21:46:09] * encode_init[127] = -15007
[I 2024-12-14 21:46:09] * encode_init[127] = -15101
[D 2024-12-14 21:46:09] encoder = intel
[D 2024-12-14 21:46:09] input = rgba
[D 2024-12-14 21:46:09] codec = h265
[D 2024-12-14 21:46:09] encode_x = 1920
[D 2024-12-14 21:46:09] encode_y = 1080
[I 2024-12-14 21:46:10] * mfxe_encode/MFXVideoCORE_SyncOperation[482] = -17
[I 2024-12-14 21:46:10] * host_msg_thread_cleanup[2506] = -15002
[I 2024-12-14 21:46:10] wurstelkrak#12174180 disconnected.
```