strongtz/i915-sriov-dkms #142: ERROR (dkms apport): binary package for i915-sriov-dkms: x.x.x-x not found

farjar, 2024-03-08:
root@redroid:~# sudo dkms install -m i915-sriov-dkms -v 6.2.0-31 -k 6.2.0-31-generic --force -j 1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j1 KERNELRELEASE=6.2.0-31-generic -j2 -C /lib/modules/6.2.0-31-generic/build M=/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build KVER=6.2.0-31-generic......^C
root@redroid:~# sudo dkms install -m i915-sriov-dkms -v 6.2.0-31 -k 6.2.0-31-generic --force

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=6.2.0-31-generic -j2 -C /lib/modules/6.2.0-31-generic/build M=/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build KVER=6.2.0-31-generic.............................................(bad exit status: 2)
ERROR (dkms apport): binary package for i915-sriov-dkms: 6.2.0-31 not found
Error! Bad return status for module build on kernel: 6.2.0-31-generic (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/make.log for more information.

root@redroid:/usr/src/i915-sriov-dkms-6.2.0-31# cat /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/make.log
DKMS make.log for i915-sriov-dkms-6.2.0-31 for kernel 6.2.0-31-generic (x86_64)
Fri Mar  8 05:05:05 AM UTC 2024
make: Entering directory '/usr/src/linux-headers-6.2.0-31-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/i915_driver.o
 ......
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.c:103:12: warning: ‘gen12_pte_encode_shim’ defined but not used [-Wunused-function]
  103 | static u64 gen12_pte_encode_shim(dma_addr_t addr,
      |            ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/gt/intel_context.o
 ......
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp_aux.o
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c: In function ‘intel_dp_mst_hpd_irq’:
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c:3888:9: error: implicit declaration of function ‘drm_dp_mst_hpd_irq_handle_event’ [-Werror=implicit-function-declaration]
 3888 |         drm_dp_mst_hpd_irq_handle_event(&intel_dp->mst_mgr, esi, ack, &handled);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c: In function ‘intel_dp_check_mst_status’:
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c:3971:25: error: implicit declaration of function ‘drm_dp_mst_hpd_irq_send_new_request’ [-Werror=implicit-function-declaration]
 3971 |                         drm_dp_mst_hpd_irq_send_new_request(&intel_dp->mst_mgr);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c: In function ‘intel_dp_get_modes’:
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c:4847:69: warning: passing argument 2 of ‘intel_connector_update_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 4847 |                 num_modes = intel_connector_update_modes(connector, edid);
      |                                                                     ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c:55:
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_connector.h:28:47: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
   28 |                                  struct edid *edid);
      |                                  ~~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c: In function ‘intel_edp_init_connector’:
/var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c:5362:51: warning: passing argument 2 of ‘drm_add_edid_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 5362 |                 if (drm_add_edid_modes(connector, edid)) {
      |                                                   ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.c:44:
./include/drm/drm_edid.h:572:70: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
  572 | int drm_add_edid_modes(struct drm_connector *connector, struct edid *edid);
      |                                                         ~~~~~~~~~~~~~^~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp_aux_backlight.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp_hdcp.o
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build/drivers/gpu/drm/i915/display/intel_dp.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.2.0-31/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.0-31-generic'

Doug411, 2024-04-23:
Did you figure this out?   I have same issue with 6.5.0-28-generic

farjar, 2024-04-24:
it's solved，tks，pls close！ 

Farjar
***@***.***

Doug411, 2024-04-30:
If its solved what is the solution  I have tried to build the module on both ubuntu 22.04 and debian 12.5 and I encounter this same problem

pasbec, 2024-07-23:
Debian 12 and Ubuntu 22.04 shoud be working now.

closed 2024-12-29 by strongtz