strongtz/i915-sriov-dkms #143: sriov_numvfs  can not write

ganyao114, 2024-03-09:
root@pve:~# echo 6(1~7) > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory

root@pve:~# echo 8(>7) > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: Numerical result out of range

echo 0 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs is ok

I7 13620H
SIROV enabled
Linux pve 6.5.13-1-pve #1 SMP PREEMPT_DYNAMIC PMX 6.5.13-1

ganyao114, 2024-03-09:
Capabilities: [320 v1] Single Root I/O Virtualization (SR-IOV)
                IOVCap: Migration- 10BitTagReq- Interrupt Message Number: 000
                IOVCtl: Enable- Migration- Interrupt- MSE- ARIHierarchy- 10BitTagReq-
                IOVSta: Migration-
                Initial VFs: 7, Total VFs: 7, Number of VFs: 0, Function Dependency Link: 00
                VF offset: 1, stride: 1, Device ID: a7a8
                Supported Page Size: 00000553, System Page Size: 00000001
                Region 0: Memory at 000000611d000000 (64-bit, non-prefetchable)
                Region 2: Memory at 0000006000000000 (64-bit, prefetchable)
                VF Migration: offset: 00000000, BIR: 0

ieiian, 2024-03-12:
I'm sure I encountered the same problem. It seems to be a problem with pve8.1, because I used pve8.0 (upgraded to pve8.1) and it was ok. I also used the 6.5.13-1-pve kernel. I hope anyone can find the problem. Please share the solution, thank you.

N100
BIOS-SIROV enabled
Linux pve 6.5.13-1-pve

Akouniza, 2024-03-12:
If you find the solution for the iGPU passthrough, please reply :) .

Intel VT-d enabled
Linux 6.5.13-1-pve

ieiian, 2024-03-22:
> I'm sure I encountered the same problem. It seems to be a problem with pve8.1, because I used pve8.0 (upgraded to pve8.1) and it was ok. I also used the 6.5.13-1-pve kernel. I hope anyone can find the problem. Please share the solution, thank you.
> 
> N100 BIOS-SIROV enabled Linux pve 6.5.13-1-pve

Good news, I have solved this problem. My problem occurred with secure boot. If yours is the same as mine, you can try to turn off secure boot or load the DKMS key. The general process is as follows:
mokutil --import /var/lib/dkms/mok.pub (set a password)
reboot
<Startup screen operation>:
<Enroll MOK>, <Continue>, <Yes>, <*password*>, <Reboot>
Hope it can help friends who encounter the same problem

iiilin, 2024-03-22:
> Good news, I have solved this problem. My problem occurred with secure boot. If yours is the same as mine, you can try to turn off secure boot or load the DKMS key. The general process is as [...]

i have the same problem. but this don't work for me. i don't enable secure boot.

chchia, 2024-03-31:
for me it was secure boot. proxmox enabled secure boot by default now.

chilltrout, 2024-09-05:
did anyone resolve this? I'm also secure boot turned off and can only see 00:02.0 after following the docs on 6.8.12-1. I do however have the whole device passed to my HAOS VM running frigate inside it and it sees the GPU and tries to do HW decoding but wont have a bar of my reolink cameras or openvino

polyfloyd, 2024-12-11:
I encountered the same issue and was able to resolve it by updating the linux-firmware files.

Debian Bookworm specifically ships with firmware blobs that are too old for SR-IOV support on my Alder Lake hardware. So I installed the firmware-intel-graphics from backports (make sure to add `non-free-firmware` to your sources!) and now writing the sysfs works :)

closed 2024-12-29 by strongtz