strongtz/i915-sriov-dkms #148: SR-IOV Support for Intel Arc A380 on AMD Systems with i915-sriov-dkms?

10bn, 2024-04-07:
I'm interested in using the i915-sriov-dkms module for SR-IOV with an Intel Arc A380 GPU on an AMD system. 

Does anyone know if this is generally possible at this stage? 

Thanks for your work on this project!

michael-pptf, 2024-04-14:
A series does not support SRIOV.

nikolaisen-hhpv, 2024-09-19:
There're rumours you could unlock it for the A770. Any updates on that matter?

michael-pptf, 2025-01-25:
> There're rumours you could unlock it for the A770. Any updates on that matter? Reference: LVL1 Tech Forum / Youtube Videos

Nope. Wendell only stated he had one A770 working by cross-flashing to DG2 datacenter CPU, and only that particular model of A770 can be flashed. He did not mention which model it is and the process to perform the cross-flash. 

You can safely treat this as a deader-than-dead-end.

closed 2025-04-04 by 10bn