strongtz/i915-sriov-dkms #150: Version 1.9

mio-19, 2024-04-08:
Do we need to update the https://github.com/strongtz/i915-sriov-dkms/blob/master/drivers/gpu/drm/i915/gt/uc/abi/guc_version_abi.h file?

```
dmesg | grep i915
[    0.000000] Command line: initrd=\EFI\proxmox\6.5.13-3-pve\initrd.img-6.5.13-3-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs     intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt
[    0.053624] Kernel command line: initrd=\EFI\proxmox\6.5.13-3-pve\initrd.img-6.5.13-3-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs     intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt
[    7.801936] i915: module verification failed: signature and/or required key missing - tainting kernel
[    8.162514] i915 0000:00:02.0: Running in SR-IOV PF mode
[    8.163033] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    8.163062] i915 0000:00:02.0: vgaarb: deactivate vga console
[    8.163119] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    8.163534] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    8.166825] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    8.171993] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[    8.172006] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    8.176865] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    8.177398] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    8.177401] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    8.177743] i915 0000:00:02.0: [drm] GuC RC: enabled
[    8.178570] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    8.179248] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    8.180957] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.181191] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    8.181218] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    8.181504] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   12.091631] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[   12.091729] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   12.091758] i915 0000:00:02.1: Running in SR-IOV VF mode
[   12.092224] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   12.092266] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   12.092793] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   12.092824] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   12.093428] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   12.093468] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   12.093785] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   12.093789] i915 0000:00:02.1: HuC firmware PRELOADED
[   12.096357] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   12.096370] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   12.096477] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 1
[   12.096851] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   12.096856] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   12.096943] i915 0000:00:02.2: enabling device (0000 -> 0002)
[   12.096973] i915 0000:00:02.2: Running in SR-IOV VF mode
[   12.097254] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   12.097296] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   12.098290] i915 0000:00:02.2: [drm] VT-d active for gfx access
[   12.098322] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[   12.098846] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   12.098864] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   12.099133] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   12.099138] i915 0000:00:02.2: HuC firmware PRELOADED
[   12.101299] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[   12.101310] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[   12.101418] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[   12.102207] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[   12.102213] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[   12.102217] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   12.102302] i915 0000:00:02.3: enabling device (0000 -> 0002)
[   12.102329] i915 0000:00:02.3: Running in SR-IOV VF mode
[   12.102964] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   12.103004] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   12.103391] i915 0000:00:02.3: [drm] VT-d active for gfx access
[   12.103420] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[   12.103970] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   12.103988] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   12.104286] i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   12.104291] i915 0000:00:02.3: HuC firmware PRELOADED
[   12.106051] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[   12.106062] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[   12.106259] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[   12.106472] i915 0000:00:02.0: Enabled 3 VFs
[   18.172347] i915 0000:00:02.0: VF1 FLR
[   20.985903] i915 0000:00:02.0: VF1 FLR
[   21.158086] i915 0000:00:02.0: VF1 FLR
```

scyto, 2024-04-10:
where did you get the 7.20.0 version guc from? 
as well as the version 2.20 adlp__dmc?

mio-19, 2024-04-10:
> where did you get the 7.20.0 version guc from? as well as the version 2.20 adlp__dmc?

Maybe they are from the intel-microcode debian package? I haven't installed other firmware files.

rjblake, 2024-04-10:
> where did you get the 7.20.0 version guc from? as well as the version 2.20 adlp__dmc?

I believe they are from the pve-firmware:amd64 (3.9-2, 3.10-1) update from a few days back

scyto, 2024-04-10:
thanks, i will give it a few days and upgrade my kernel and firmware's to the latest

what was interesting to me is those version are not in the intel repo 
https://github.com/intel-gpu/intel-gpu-firmware 

i wonder if they will help with my weird issues:

1. cant run a VF in a VM with a CPU set to anything but host
2. the intel official drivers cause me a spinlock crash (compared to this that doesn't)

mio-19, 2024-04-19:
After upgrading pve kernel today:

```
dmesg | grep i915
[    0.000000] Command line: initrd=\EFI\proxmox\6.5.13-5-pve\initrd.img-6.5.13-5-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs     intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt
[    0.053350] Kernel command line: initrd=\EFI\proxmox\6.5.13-5-pve\initrd.img-6.5.13-5-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs     intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt
[    9.768193] i915: unknown parameter 'max_vfs' ignored
[    9.769206] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    9.769239] i915 0000:00:02.0: vgaarb: deactivate vga console
[    9.769293] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    9.769774] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    9.772848] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    9.777892] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[    9.777904] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    9.782642] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads
[    9.783121] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    9.783124] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    9.783418] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    9.784545] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    9.785454] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    9.787077] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    9.787365] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    9.787659] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   13.936644] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
[   13.936680] i915 0000:00:02.0: driver does not support SR-IOV configuration via sysfs
```

rjblake, 2024-04-19:
> After upgrading pve kernel today:
>
> ```
> dmesg | grep i915
> [    0.000000] Command line: initrd=\EFI\proxmox\6.5.13-5-pve\initrd.img-6.5.13-5-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs     intel_iommu=on [...]

That seems very strange. I upgraded my PVE Kernel a couple of weeks back and have none of those issues. Log as below:
```root@hp-em800g9-i7-1:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    0.039634] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
[    4.171048] i915: module verification failed: signature and/or required key missing - tainting kernel
[    4.395396] i915 0000:00:02.0: Running in SR-IOV PF mode
[    4.396169] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.396294] i915 0000:00:02.0: vgaarb: deactivate vga console
[    4.396336] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    4.396989] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.398667] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    4.402133] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[    4.402136] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    4.404545] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    4.404872] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    4.404874] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    4.405316] i915 0000:00:02.0: [drm] GuC RC: enabled
[    4.405859] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.460461] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    4.462343] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.462467] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    4.489237] fbcon: i915drmfb (fb0) is primary device
[    4.572026] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    5.202438] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    5.202488] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    5.202500] i915 0000:00:02.1: Running in SR-IOV VF mode
[    5.203642] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[    5.205088] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    5.205103] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    5.205444] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[    5.206128] i915 0000:00:02.1: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    5.206130] i915 0000:00:02.1: HuC firmware PRELOADED
[    5.208685] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.208688] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    5.208852] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 1```

ER-EPR, 2024-04-21:
```
[    9.787365] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    9.787659] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
```
I try to install vGPU and encounter this as well, what is the last known version that will work with zfs systemd uefi boot? @mio-19

michael-pptf, 2024-04-21:
> After upgrading pve kernel today:
>
> ```
> dmesg | grep i915
> [    0.000000] Command line: initrd=\EFI\proxmox\6.5.13-5-pve\initrd.img-6.5.13-5-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs     intel_iommu=on [...]

After doing an upgrade for my Proxmox node, I have to reinstall the dkms and clean up the kernel commandline in grub. Here are the steps:
1. Install Proxmox headers. You can't build the dkms for the current kernel without it. `apt install pve-headers-$(uname -r)`
2. Clone the repo again, I think there's a PR merge for a fix in 6.5
3. Do the usual steps for dkms.conf (change the package name and version to match 6.5)
4. Run the dkms install 
5. Change kernel commandline in grub, namely the guc and vfs_enabled parts.
6. Reboot

Here's what I got:
`
...
[    7.802015] i915 0000:00:02.7: Running in SR-IOV VF mode
[    7.802212] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.0.0
[    7.802491] i915 0000:00:02.7: [drm] VT-d active for gfx access
[    7.802513] i915 0000:00:02.7: [drm] Using Transparent Hugepages
[    7.803054] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.0.0
[    7.803327] i915 0000:00:02.7: GuC firmware PRELOADED version 1.0 submission:SR-IOV VF
[    7.803330] i915 0000:00:02.7: HuC firmware PRELOADED
[    7.804864] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.804872] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[    7.805053] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.7 on minor 7
[    7.805327] i915 0000:00:02.0: Enabled 7 VFs
`

DarkPhyber-hg, 2024-04-24:
i'm on 6.5.13-5 and it seems to be working?  I haven't tested it fully, but i'm getting the 1.9 errors as well.

```

[    6.518617] i915 0000:00:02.1: Running in SR-IOV VF mode
[    6.519129] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[    6.519190] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[    6.521051] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    6.521064] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    6.521533] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[    6.521583] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[    6.521957] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    6.521959] i915 0000:00:02.1: HuC firmware PRELOADED
```

is the solution still to do what @rjblake suggested earlier on and change GUC_VF_VERSION_ALTERNATE_MINOR?

rjblake, 2024-04-24:
> i'm on 6.5.13-5 and it seems to be working? I haven't tested it fully, but i'm getting the 1.9 errors as well.
>
> ```
>
> [    6.518617] i915 0000:00:02.1: Running in SR-IOV VF mode
> [    6.519129] i915 [...]

I haven't checked lately, but seems that the guc_version_abi.h file has not been updated in the Master, so I'd update it

michael-pptf, 2024-04-25:
> I haven't checked lately, but seems that the guc_version_abi.h file has not been updated in the Master, so I'd update it

Could you show the content of your `guc_verison_abi.h`? I have both the minor and minor_alternate = 4 and still got the error.

rjblake, 2024-04-26:
> Could you show the content of your `guc_verison_abi.h`? I have both the minor and minor_alternate = 4 and still got the error.

Contents of my file located here: `/usr/src/i915-sriov-dkms-6.5/drivers/gpu/drm/i915/gt/uc/abi/guc_verison_abi.h`
```
/* SPDX-License-Identifier: MIT */
/*
 * Copyright © 2022 Intel Corporation
 */

#ifndef _ABI_GUC_VERSION_ABI_H
#define _ABI_GUC_VERSION_ABI_H

#define GUC_VF_VERSION_LATEST_MAJOR	1
#define GUC_VF_VERSION_LATEST_MINOR	9

#endif /* _ABI_GUC_VERSION_ABI_H */
```

michael-pptf, 2024-04-26:
> > > > i'm on 6.5.13-5 and it seems to be working? I haven't tested it fully, but i'm getting the 1.9 errors as well.
> > > > ```
> > > > 
> > > > [    6.518617] i915 0000:00:02.1: Running in SR-IOV VF mode
> > > > [    6.519129] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
> > > > [    6.519190] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
> > > > [    6.521051] i915 0000:00:02.1: [drm] VT-d active for gfx access
> > > > [    6.521064] i915 0000:00:02.1: [drm] Using Transparent Hugepages
> > > > [    6.521533] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
> > > > [    6.521583] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
> > > > [    6.521957] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
> > > > [    6.521959] i915 0000:00:02.1: HuC firmware PRELOADED
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > is the solution still to do what @rjblake suggested earlier on and change GUC_VF_VERSION_ALTERNATE_MINOR?
> > > 
> > > 
> > > I haven't checked lately, but seems that the guc_version_abi.h file has not been updated in the Master, so I'd update it
> > 
> > 
> > Could you show the content of your `guc_verison_abi.h`? I have both the minor and minor_alternate = 4 and still got the error.
> 
> Contents of my file located here: `/usr/src/i915-sriov-dkms-6.5/drivers/gpu/drm/i915/gt/uc/abi/guc_version_abi.h`. My bad, was not the ALTERNATE_MINOR, but the LATEST_MINOR
> 
> ```
> /* SPDX-License-Identifier: MIT */
> /*
>  * Copyright © 2022 Intel Corporation
>  */
> 
> #ifndef _ABI_GUC_VERSION_ABI_H
> #define _ABI_GUC_VERSION_ABI_H
> 
> #define GUC_VF_VERSION_LATEST_MAJOR	1
> #define GUC_VF_VERSION_LATEST_MINOR	9
> 
> #endif /* _ABI_GUC_VERSION_ABI_H */
> ```

Huh, that's weird. How were you able to build it? If I changed the file to match yours, it returns error:
```/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_query.c:106:103: error: ‘GUC_VF_VERSION_ALTERNATE_MINOR’ undeclared (first use in this function); did you mean ‘GUC_VF_VERSION_LATEST_MINOR’?
  106 |         if (major != GUC_VF_VERSION_LATEST_MAJOR || (minor != GUC_VF_VERSION_LATEST_MINOR && minor != GUC_VF_VERSION_ALTERNATE_MINOR))
      |                                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                                       GUC_VF_VERSION_LATEST_MINOR
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_query.c:106:103: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_bios.o
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_query.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve'```

michael-pptf, 2024-04-27:
> > > > > i'm on 6.5.13-5 and it seems to be working? I haven't tested it fully, but i'm getting the 1.9 errors as well.
> > > > > ```
> > > > > 
> > > > > [    6.518617] i915 0000:00:02.1: Running in SR-IOV VF mode
> > > > > [    6.519129] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
> > > > > [    6.519190] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
> > > > > [    6.521051] i915 0000:00:02.1: [drm] VT-d active for gfx access
> > > > > [    6.521064] i915 0000:00:02.1: [drm] Using Transparent Hugepages
> > > > > [    6.521533] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
> > > > > [    6.521583] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
> > > > > [    6.521957] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
> > > > > [    6.521959] i915 0000:00:02.1: HuC firmware PRELOADED
> > > > > ```
> > > > > 
> > > > > 
> > > > >     
> > > > >       
> > > > >     
> > > > > 
> > > > >       
> > > > >     
> > > > > 
> > > > >     
> > > > >   
> > > > > is the solution still to do what @rjblake suggested earlier on and change GUC_VF_VERSION_ALTERNATE_MINOR?
> > > > 
> > > > 
> > > > I haven't checked lately, but seems that the guc_version_abi.h file has not been updated in the Master, so I'd update it
> > > 
> > > 
> > > Could you show the content of your `guc_verison_abi.h`? I have both the minor and minor_alternate = 4 and still got the error.
> > 
> > 
> > Contents of my file located here: `/usr/src/i915-sriov-dkms-6.5/drivers/gpu/drm/i915/gt/uc/abi/guc_version_abi.h`. My bad, was not the ALTERNATE_MINOR, but the LATEST_MINOR
> > ```
> > /* SPDX-License-Identifier: MIT */
> > /*
> >  * Copyright © 2022 Intel Corporation
> >  */
> > 
> > #ifndef _ABI_GUC_VERSION_ABI_H
> > #define _ABI_GUC_VERSION_ABI_H
> > 
> > #define GUC_VF_VERSION_LATEST_MAJOR	1
> > #define GUC_VF_VERSION_LATEST_MINOR	9
> > 
> > #endif /* _ABI_GUC_VERSION_ABI_H */
> > ```
> 
> Huh, that's weird. How were you able to build it? If I changed the file to match yours, it returns error:
> 
> ```
>   106 |         if (major != GUC_VF_VERSION_LATEST_MAJOR || (minor != GUC_VF_VERSION_LATEST_MINOR && minor != GUC_VF_VERSION_ALTERNATE_MINOR))
>       |                                                                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>       |                                                                                                       GUC_VF_VERSION_LATEST_MINOR
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_query.c:106:103: note: each undeclared identifier is reported only once for each function it appears in
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_bios.o
> make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_query.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
> make: *** [Makefile:234: __sub-make] Error 2
> make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve'```
> ```

Solved with #159.

pasbec, 2024-07-19:
I don't know why PR #159 was never merged, but the same change was part of #178 (I'd propose to just close #159).
With PR #179 there is now even an option to use an older firmware if necessary for whatever reason:
```
GUCFIRMWARE_MINOR=4 dkms install -m dkms install -m i915-sriov-dkms -v 2024.07.17 --force
```

johntdavis84, 2024-07-19:
Don’t quote me on this, but my understanding was that #159 was not merged because the contributor couldn’t test all the older versions of the Linux kernel that they wanted to test.

It makes sense to me to close it and open a fresh issue if needed—the codebase has changed quite a bit since then, and we’ve got new options to control the firmware, as you mentioned.

-- -- --
John T Davis
***@***.***

closed 2024-07-22 by strongtz

scyto, 2024-09-23:
The assumption these are older firmware's i think is wrong - they are different firmware.  For example after copying all the latest i915 from the intel github repo, the kernel repo my interface is firmly at 1.4.1 still.

The issue I am convinced is the comparison of 1.4 vs 0.1.4.1 - not the overall version of the bin / guc.  Unless someone can point me to a yet another location of later i.915 drivers.... i think the version 1.4 vs 1.9 is more about what hardware one has...