strongtz/i915-sriov-dkms #151: install error on pve 8.1 kernel version : 6.5.13-5

keivenliao, 2024-04-09:
CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sriov_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_event.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_memirq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_provisioning.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_query.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_relay.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_service.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_state.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/hsw_ips.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_atomic.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_atomic_plane.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_audio.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_bios.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_bw.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_cdclk.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_color.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_combo_phy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_connector.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_crtc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_crtc_state_dump.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_cursor.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_power.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_power_map.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_power_well.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dmc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dpio_phy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dpll.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dpll_mgr.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dpt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_drrs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dsb.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_fb.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_fb_pin.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_fbc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_fdi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_fifo_underrun.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_frontbuffer.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_global_state.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_hdcp.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_hotplug.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_hti.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_lpe_audio.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_modeset_verify.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_modeset_setup.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_overlay.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_pch_display.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_pch_refclk.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_plane_initial.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_psr.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_quirks.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_sprite.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_tc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_vblank.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_vga.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/i9xx_plane.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/skl_scaler.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/skl_universal_plane.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/skl_watermark.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_acpi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_opregion.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_fbdev.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/dvo_ch7017.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/dvo_ch7xxx.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/dvo_ivch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/dvo_ns2501.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/dvo_sil164.o
[... 16 lines cut ...]
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dsi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dsi_dcs_backlight.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dsi_vbt.o
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:86:35: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
   86 |                 crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                                   ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:906:17: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
  906 |                 drm_dp_calc_pbn_mode(mode->clock, min_bpp, false) > port->full_pbn) {
      |                 ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve'

jeanpaulrh, 2024-04-09:
I have this error too. Following

Fallesen82, 2024-04-09:
follow

makazeu, 2024-04-09:
Applying this change can fix this issue.

ovizii, 2024-04-11:
I'm having a problem following this guide. I then tried uninstalling the module like this:

```
dkms remove -m i915-sriov-dkms -v 6.5
Module i915-sriov-dkms-6.5 for kernel 6.5.13-3-pve (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

i915.ko:
 - Uninstallation
   - Deleting from: /lib/modules/6.5.13-3-pve/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...

```

This seems to have worked but when I try a apt-get dist-upgrade I get errors as for some reason it is still trying to rebuild with the i915 dkms module which I just uninstalled.

Any further tips? As you can see, it was working with 6.5.13-3-pve but now the upgrade to 6.5.13-5-pve fails.

```
Setting up proxmox-kernel-6.5.13-5-pve-signed (6.5.13-5) ...
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/dkms 6.5.13-5-pve /boot/vmlinuz-6.5.13-5-pve
dkms: running auto installation service for kernel 6.5.13-5-pve.
Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.5.13-5-pve -j8 -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.13-5-pve..................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.5.13-5-pve failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/proxmox-kernel-6.5.13-5-pve-signed.postinst line 20.
dpkg: error processing package proxmox-kernel-6.5.13-5-pve-signed (--configure):
 installed proxmox-kernel-6.5.13-5-pve-signed package post-installation script subprocess returned error exit status 2
dpkg: dependency problems prevent configuration of proxmox-kernel-6.5:
 proxmox-kernel-6.5 depends on proxmox-kernel-6.5.13-5-pve-signed | proxmox-kernel-6.5.13-5-pve; however:
  Package proxmox-kernel-6.5.13-5-pve-signed is not configured yet.
  Package proxmox-kernel-6.5.13-5-pve is not installed.
  Package proxmox-kernel-6.5.13-5-pve-signed which provides proxmox-kernel-6.5.13-5-pve is not configured yet.

dpkg: error processing package proxmox-kernel-6.5 (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 proxmox-kernel-6.5.13-5-pve-signed
 proxmox-kernel-6.5
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

lalalamiheart, 2024-04-13:
> I'm having a problem following this guide. I then tried uninstalling the module like this:
> 
> ```
> dkms remove -m i915-sriov-dkms -v 6.5
> Module i915-sriov-dkms-6.5 for kernel 6.5.13-3-pve (x86_64).
> Before uninstall, this module version was ACTIVE on this kernel.
> 
> i915.ko:
>  - Uninstallation
>    - Deleting from: /lib/modules/6.5.13-3-pve/updates/dkms/
>  - Original module
>    - No original module was found for this module on this kernel.
>    - Use the dkms install command to reinstall any previous module version.
> depmod...
> ```
> 
> This seems to have worked but when I try a apt-get dist-upgrade I get errors as for some reason it is still trying to rebuild with the i915 dkms module which I just uninstalled.
> 
> Any further tips? As you can see, it was working with 6.5.13-3-pve but now the upgrade to 6.5.13-5-pve fails.
> 
> I tried fixing it with:
> 
> ` apt-get install proxmox-kernel-6.5.13-5-pve proxmox-headers-6.5.13-5-pve -u`
> 
> but then it still fails with:
> 
> ```
> dpkg: dependency problems prevent configuration of proxmox-kernel-6.5:
>  proxmox-kernel-6.5 depends on proxmox-kernel-6.5.13-5-pve-signed | proxmox-kernel-6.5.13-5-pve; however:
>   Package proxmox-kernel-6.5.13-5-pve-signed is not installed.
>   Package proxmox-kernel-6.5.13-5-pve is not configured yet.
>   Package proxmox-kernel-6.5.13-5-pve-signed which provides proxmox-kernel-6.5.13-5-pve is not installed.
> ```
> 
> ```
> Setting up proxmox-kernel-6.5.13-5-pve-signed (6.5.13-5) ...
> Examining /etc/kernel/postinst.d.
> run-parts: executing /etc/kernel/postinst.d/dkms 6.5.13-5-pve /boot/vmlinuz-6.5.13-5-pve
> dkms: running auto installation service for kernel 6.5.13-5-pve.
> Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
> Signing key: /var/lib/dkms/mok.key
> Public certificate (MOK): /var/lib/dkms/mok.pub
> 
> Building module:
> Cleaning build area...
> make -j8 KERNELRELEASE=6.5.13-5-pve -j8 -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.13-5-pve..................(bad exit status: 2)
> Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
> Consult /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log for more information.
> Error! One or more modules failed to install during autoinstall.
> Refer to previous errors for more information.
> dkms: autoinstall for kernel: 6.5.13-5-pve failed!
> run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
> Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/proxmox-kernel-6.5.13-5-pve-signed.postinst line 20.
> dpkg: error processing package proxmox-kernel-6.5.13-5-pve-signed (--configure):
>  installed proxmox-kernel-6.5.13-5-pve-signed package post-installation script subprocess returned error exit status 2
> dpkg: dependency problems prevent configuration of proxmox-kernel-6.5:
>  proxmox-kernel-6.5 depends on proxmox-kernel-6.5.13-5-pve-signed | proxmox-kernel-6.5.13-5-pve; however:
>   Package proxmox-kernel-6.5.13-5-pve-signed is not configured yet.
>   Package proxmox-kernel-6.5.13-5-pve is not installed.
>   Package proxmox-kernel-6.5.13-5-pve-signed which provides proxmox-kernel-6.5.13-5-pve is not configured yet.
> 
> dpkg: error processing package proxmox-kernel-6.5 (--configure):
>  dependency problems - leaving unconfigured
> Errors were encountered while processing:
>  proxmox-kernel-6.5.13-5-pve-signed
>  proxmox-kernel-6.5
> E: Sub-process /usr/bin/dpkg returned an error code (1)
> ```
> 
> **Turns out I had to manually remove /var/lib/dkms/i915-sriov-dkms/**

@ovizii so,you fixed it? after i remove it, i still yeilded the same result... could you post detailedly?

lalalamiheart, 2024-04-14:
#

> @ovizii so,you fixed it? after i remove it, i still yeilded the same result... could you post detailedly?

actually, I succeeded~ it need to uninstall the module

ovizii, 2024-04-14:
> actually, I succeeded~ it need to uninstall the module

Sorry I should have given more info. Basically this was what I did:

1. uninstall dmks i915 module
2. Remove the folder I mentioned,
3. update the kernel
4. reboot
5. rebuild dkms module

xbb, 2024-04-15:
The patch provided in the pull request can be applied conditionally with a regex:

In the module source, create the directory `patches` and download the patch under it (to get a patch add `.patch` at the end of the [pull request URL](https://patch-diff.githubusercontent.com/raw/strongtz/i915-sriov-dkms/pull/152.patch), or the [commit URL](https://github.com/strongtz/i915-sriov-dkms/commit/ffc23727f106995d89bc7ad32df4f1a3809ee737.patch))

Edit `dkms.conf` and add `PATCH[0]=<patch_file_name>` and `PATCH_MATCH[0]=<regex>` to match the kernel version:

The example below, if I got the regex right, matches `6.5.13-5-pve` (.13-5) and above.

```
PATCH[0]=ffc23727f106995d89bc7ad32df4f1a3809ee737.patch
PATCH_MATCH[0]="6\.5\.((\d{3,}|[2-9]\d|1[4-9])-\d+|13-(\d{2,}|[5-9]))-pve"
```

When the module is building you should see the patch being applied if the regex matches the kernel version:

```
applying patch ffc23727f106995d89bc7ad32df4f1a3809ee737.patch...patching file drivers/gpu/drm/i915/display/intel_dp_mst.c
```

Also check `dkms status` to see what's currently installed

rdosrun, 2024-04-16:
> The patch provided in the pull request can be applied conditionally with a regex:
>
> In the module source, create the directory `patches` and download the patch under it (to get a patch add `.patch` [...]

where in the dkms.conf would I put this line?

xbb, 2024-04-16:
> where in the dkms.conf would I put this line?

It shouldn't matter, I put the two lines at the end after the `AUTOINSTALL=yes` line

keqingf, 2024-04-19:
> 应用此更改可以解决此问题：https://github.com/strongtz/i915-sriov-dkms/pull/152/files
> 
> 指导：
> 
> 1. 重命名`./drivers/gpu/drm/i915/display/intel_dp_mst.c`为`intel_dp_mst.c.bak`
> 2. 下载`https://raw.githubusercontent.com/makazeu/i915-sriov-dkms/ffc23727f106995d89bc7ad32df4f1a3809ee737/drivers/gpu/drm/i915/display/intel_dp_mst.c`到`./drivers/gpu/drm/i915/display/`
> 3. 重建
> 
> ```
> Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
> Signing key: /var/lib/dkms/mok.key
> Public certificate (MOK): /var/lib/dkms/mok.pub
> 
> Building module:
> Cleaning build area...
> make -j4 KERNELRELEASE=6.5.13-5-pve -j4 -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.13-5-pve...................................
> Signing module /var/lib/dkms/i915-sriov-dkms/6.5/build/i915.ko
> Cleaning build area...
> 
> i915.ko:
> Running module version sanity check.
>  - Original module
>  - Installation
>    - Installing to /lib/modules/6.5.13-5-pve/updates/dkms/
> depmod...
> ```
我这么做后还是出现

PR #152 (2024-04-20): Changed the signature of `drm_dp_calc_pbn_mode`
**Prcoceeding with this merge may break builds in PVE with kernel < 6.5.13-5-pve**

Changed the signature of `drm_dp_calc_pbn_mode` to fix the install error on the PVE kernel `6.5.13-5-pve`

Applying this change can fix #151
closed 2024-04-20 by strongtz