strongtz/i915-sriov-dkms #153: Status of this Patch in Kernel 6.8?

johntdavis84, 2024-04-12:
Is this patch still needed to enable SR-IOV on, e.g., an Alder Lake 12-500T iGPU, on Kernel 6.8?

Proxmox is moving to 6.8 in a couple of months and I'm wondering if this will still be necessary.

ich777, 2024-04-12:
Duplicate of #140 
Please read the above issue, there are all answeres you need.

Yes, still needed and no, not working.

johntdavis84, 2024-04-12:
Thanks! Sorry for my search fail.

Sounds like the new Intel iGPU driver won't implement SR-IOV support until _sometime_ in the kernel 6.9 development cycle.

I'm not familiar with Linux kernel development timelines. We're just now seeing 6.8 roll out into products like Proxmox. When are we likely to see it replaced with 6.9 (or whatever the next LTS is)?

ich777, 2024-04-12:
> Sounds like the new Intel iGPU driver won't implement SR-IOV support until _sometime_ in the kernel 6.9 development cycle.

I think that would even take longer than that.

> We're just now seeing 6.8 roll out into products like Proxmox. When are we likely to see it replaced with 6.9 (or whatever the next LTS is)?

If distributions switch to a newer Kernel >= 6.6 then this SRIOV driver won't work anymore, at least not until someone patches it but this time it could be a bit more complicated because things in the Kernel changed quiet a lot.

johntdavis84, 2024-04-12:
> If distributions switch to a newer Kernel >= 6.6 then this SRIOV driver won't work anymore, at least not until someone patches it but this time it could be a bit more complicated because things in [...]

That makes sense, but it's certainly unfortunate. One reason I bought a mini-PC with an i5-12500T was so I could run a bunch of otherwise low-power resource-light VMs that needed a GPU (Plex, GPU-accelerated remote desktop for graphics tasks, etc.). I chose the hardware because my understanding was that SR-IOV would let me do what I wanted. I wasn't clever enough to realize Intel didn't have the drivers together on Linux.

Pure PCIe passthrough won't work because of the IOMMU groups: the iGPU is in group 9 but insists on pulling in the Intel sound hardware with it, and that's in group 9 with a bunch of other stuff.

I think I'm stuck until the standard intel driver, without any DKMS mods, actually implements the SR-IOV features in their own GPU. 

So, I'm back to trying to get an all-software solution working with SPICE, where I still haven't solved the audio lag.

From that point of view, this has been an expensive experiment that didn't go how I wanted because I didn't know I couldn't assume Intel would have SR-IOV working in Linux (since I've never had problems using an iGPU in Linux for actual display out). Sure, that's on me, but it's got me to the point where this is probably the last mini PC I'm ever putting Proxmox on. I can still get great use out of this one, but I'm tired of spending so much time trying and failing to work around not having PCIe slots. :P

ich777, 2024-04-12:
Can't you still stay on the old version from Proxmox where it is working currently? Just saying.
I personally don't use Proxmox and instead I'm using Unraid because Proxmox is a bit too blown up for my likings for a Home Lab.

Sorry I can't help here.

Maybe someone will fix/fork the repo and produce a working build. ;)

johntdavis84, 2024-04-12:
Proxmox is an odd duck. It depends on the Debian repos (in addition to its own repos), but uses a modified Ubuntu kernel. And that kernel seems to get minor updates at least once every 1-2 months, even on the stable Enterprise repo that I use. And the underlying KVM and ZFS systems are getting updated all the time in lockstep with the kernel. Bug fixes, stability enhancements, etc. (Think about how often the Ubuntu kernel itself gets updated, and then PVE has its own tweaks.)

Pinning the kernel and trying to update the rest of the system seems like a recipe for trouble. :P 

No need to apologize. It's not on you at all. You've made a great DKMS here, and brought joy to a lot of people with it.

My aggravation at this point is entirely at Intel. They haven't gotten SR-IOV working on their iGPUs in the mainline kernel by now, especially after taking away GVT-g and telling the community at the time that it would be replaced with the better SR-IOV. And they absolutely advertise their 12th + gen stuff as having SR-IOV features. They just don't tell you it doesn't work in Linux.

In my setup, that means I can't use a mini PC that consumes much less than a 100 watts to do GPU acceleration, and instead have to depend on second, bigger server that consumes ~100 watts ... before activating the Tesla P4 that wants another 60 watts or so.

ich777, 2024-04-12:
Sorry but I‘m not the maintainer from that repo, I just answered because I saw your request. 😉

I would never buy bleeding edge hardware for a Linux server because manufacturers or better speaking the community are always a bit behind with driver support for Linux.

However I think you missunderstood the comment above, isn‘t there a way to stay on the specific version that are you on on Proxmox?

However even Proxmox can‘t go higher than Kernel 6.7 because they are also limited by ZFS since OpenZFS only supports up to Kernel 6.7 of course if a new version from ZFS is released then nothing stands in the way.

pcmike, 2024-04-24:
> That makes sense, but it's certainly unfortunate. One reason I bought a mini-PC with an i5-12500T was so I could run a bunch of otherwise low-power resource-light VMs that needed a GPU (Plex, [...]

Can you switch to LXCs instead of full blown VMs as a work around for now?

johntdavis84, 2024-04-24:
> 
> Can you switch to LXCs instead of full blown VMs as a work around for now?
>

Proxmox just released a version of the 6.5 kernel today that's compatible with the DKMS module in this repo. I need to get it working before I can do any VM vs. LXC experiments.

ich777, 2024-04-25:
@johntdavis84 can you please close at least one issue here?
There is also issue #140 which is basically the same as this here.

johntdavis84, 2024-04-25:
Sorry about that. Thought I closed it with my last comment.

closed 2024-04-25 by johntdavis84