strongtz/i915-sriov-dkms #155: Setup Issues in Proxmox 8.1, Kernel 6.5.13-1-pve?

johntdavis84, 2024-04-21:
I got as far as trying to install and build the new kernel, and ended up failing with a build error. Unfortunately, I accidentally cleared it before I could copy it.

However, I wanted to check and confirm whether anyone actually has this working with my configuration before I try again.

The hardware is an Intel i5-12500T (UHD 770), but as I didn't make it past installing the DKMS module, I don't think that's relevatn to the issue. I hope.

Here's my config:

```
pveversion -v
proxmox-ve: 8.1.0 (running kernel: 6.5.13-1-pve)
pve-manager: 8.1.4 (running version: 8.1.4/ec5affc9e41f1d79)
proxmox-kernel-helper: 8.1.0
proxmox-kernel-6.5.13-1-pve-signed: 6.5.13-1
proxmox-kernel-6.5: 6.5.13-1
proxmox-kernel-6.5.11-8-pve-signed: 6.5.11-8
ceph-fuse: 17.2.7-pve2
corosync: 3.1.7-pve3
criu: 3.17.1-2
glusterfs-client: 10.3-5
ifupdown2: 3.2.0-1+pmx8
ksm-control-daemon: 1.4-1
libjs-extjs: 7.0.0-4
libknet1: 1.28-pve1
libproxmox-acme-perl: 1.5.0
libproxmox-backup-qemu0: 1.4.1
libproxmox-rs-perl: 0.3.3
libpve-access-control: 8.1.2
libpve-apiclient-perl: 3.3.1
libpve-common-perl: 8.1.1
libpve-guest-common-perl: 5.0.6
libpve-http-server-perl: 5.0.5
libpve-network-perl: 0.9.5
libpve-rs-perl: 0.8.8
libpve-storage-perl: 8.1.0
libspice-server1: 0.15.1-1
lvm2: 2.03.16-2
lxc-pve: 5.0.2-4
lxcfs: 5.0.3-pve4
novnc-pve: 1.4.0-3
proxmox-backup-client: 3.1.4-1
proxmox-backup-file-restore: 3.1.4-1
proxmox-kernel-helper: 8.1.0
proxmox-mail-forward: 0.2.3
proxmox-mini-journalreader: 1.4.0
proxmox-offline-mirror-helper: 0.6.5
proxmox-widget-toolkit: 4.1.4
pve-cluster: 8.0.5
pve-container: 5.0.8
pve-docs: 8.1.4
pve-edk2-firmware: 4.2023.08-4
pve-firewall: 5.0.3
pve-firmware: 3.9-2
pve-ha-manager: 4.0.3
pve-i18n: 3.2.1
pve-qemu-kvm: 8.1.5-3
pve-xtermjs: 5.3.0-3
qemu-server: 8.0.10
smartmontools: 7.3-pve1
spiceterm: 3.3.0
swtpm: 0.8.0+pve1
vncterm: 1.8.0
zfsutils-linux: 2.2.2-pve2
```

ich777, 2024-04-22:
Maybe related to #151 and should be fixed with #152 as long as you are on Kernel 6.5.13-5-pve which you are not from what I see from the description.

johntdavis84, 2024-04-23:
> Maybe related to #151 and should be fixed with #152 as long as you are on Kernel 6.5.13-5-pve which you are not from what I see from the description.

Thanks!

That definitely did something. I'm getting a different (and hopefully easier to fix) error about a function having too few arguments now.

I've updated my original comment with the details.

ich777, 2024-04-23:
> proxmox-ve: 8.1.0 (running kernel: 6.5.13-1-pve)

Sorry for the maybe dumb question but can't you just update to a version which includes Kernel version Kernel 6.5.13-**5**-pve instead of 6.5.13-**1**-pve to solve your issue.
I linked the PR above because it states:
`Prcoceeding with this merge may break builds in PVE with kernel < 6.5.13-5-pve`

johntdavis84, 2024-04-23:
> > proxmox-ve: 8.1.0 (running kernel: 6.5.13-1-pve)
> 
> Sorry for the maybe dumb question because I don't use Proxmox but can't you just update to a Proxmox version which includes Kernel version Kernel 6.5.13-**5**-pve instead of 6.5.13-**1**-pve to solve your issue. I linked the PR above because it states: `Prcoceeding with this merge may break builds in PVE with kernel < 6.5.13-5-pve`

...
That is not a dumb question. I'm a bit of an idiot, apparently.

My version is the latest version I can get when i do a `dist-upgrade`, but I'm also not on the test (beta) repo. 
Let me look and see what I can find out. I'll report back.

johntdavis84, 2024-04-25:
Thanks everyone. I'm going to close this as solved shortly. :)

I was running on the latest enterprise version of the 6.5.13 PVE kernel, which wasn't new enough. (I think it was the "_1" variant.) On Proxmox 8.1, the newer ones were in the test repo, which I keep disabled.

The day Proxmox 8.2 was released, 6.5.13-5-pve became available in the non-test repos (or at least, the enterprise repo). 

So I was able to get it added and installed and the system boots.

I'm going to suggest to the guide author I used to add a section about requiring 6.5.13-5-pve, which is currently not mentioned, with instructions on how to install it if you're on a fresh install of Proxmox 8.2, which doesn't include the 6.5.13-5-pve by default. It can be installed with a single command, though. :)

closed 2024-04-25 by johntdavis84