strongtz/i915-sriov-dkms #156: Proxmox 8.2 with Kernel 6.5.13-5-pve (Secure Boot Disabled, SystemD): DKMS module not loading due to tainted kernel

johntdavis84, 2024-04-25:
Hello,

I've read the github instructions, and because I'm unfamiliar with installing DKMS modules, followed the install guide here: https://www.derekseaman.com/2023/11/proxmox-ve-8-1-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

I got the module added and installed, but it's refusing to load on boot because it's unable to verify the module signature.
`[Thu Apr 25 17:34:56 2024] i915: module verification failed: signature and/or required key missing - tainting kernel`

Which makes sense as Secure Boot is off and this install of Proxmox was done with it off. Turning it on and reinstalling Proxmox is not an option on this machine.
**I have Secure Boot disabled because my BIOS implements it in a horribly confusing way. I've already bricked my entire Proxmox install once trying to make it work, so I'm sticking with SystemD with no Secure Boot.**

I'm not sure what to do at this point. I saw the MOK keys auto-generated when the module was installed via the "Sign Command." I need that not to happen, I think? Or at the very least, I need do something now to get it to get past the module verification error.

I'd really appreciate some help. Thanks!

Let me grab some system info:
```
pveversion 
pve-manager/8.2.2/9355359cd7afbae4 (running kernel: 6.5.13-5-pve)

uname -a
Linux andromeda2 6.5.13-5-pve #1 SMP PREEMPT_DYNAMIC PMX 6.5.13-5 (2024-04-05T11:03Z) x86_64 GNU/Linux

dmesg | grep -i "command line"
[    0.000000] Command line: initrd=\EFI\proxmox\6.5.13-5-pve\initrd.img-6.5.13-5-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=3 console=ttyS0,115200n8 console=tty0
[    0.042660] Kernel command line: initrd=\EFI\proxmox\6.5.13-5-pve\initrd.img-6.5.13-5-pve root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=3 console=ttyS0,115200n8 console=tty0
```

gfgjs, 2024-04-27:
If PVE uses [Systemd-boot] to start, please change [Systemd-boot] to [Grub], which may be helpful.

reference：
https://juejin.cn/post/7330920549771837481
https://zoe.red/2024/186.html

johntdavis84, 2024-04-27:
Thanks for your reply. As I noted above,

> I have Secure Boot disabled because my BIOS implements it in a horribly confusing way. I've already bricked my entire Proxmox install once trying to make it work, so I'm sticking with SystemD with no Secure Boot.

My particular machine makes using Secure Boot with GRUB too much of an issue. I actually broke my Proxmox install trying to set it up correctly and had to completely wipe the boot disk and start over.

Hopefully there's a way to do this without GRUB, otherwise I'm going to have to give up on it until official support makes it into mainline.

michael-pptf, 2024-04-27:
I have the same "module verification failed" message and everything still works fine.

johntdavis84, 2024-04-27:
> I have the same "module verification failed" message and everything still works fine.

Interesting. Are you in Non-Secure Boot/SystemD mode? Sounds like that's a red herring. Good, since I don't know how to fix it. :P 

My issue is that I don't actually have any VFs to use. I've set it up for 3 to start with, but I don't have any.

```
# lspci -nn | grep -i vga
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-S GT1 [UHD Graphics 770] [8086:4690] (rev 0c)

root@andromeda2:~# batcat /etc/kernel/cmdline
───────┬────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: /etc/kernel/cmdline
───────┼────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 console=ttyS0,115200n8 console=tty0
───────┴──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
```

michael-pptf, 2024-04-27:
> Interesting. Are you in Non-Secure Boot/SystemD mode? Sounds like that's a red herring. Good, since I don't know how to fix it. :P
>
> My issue is that I don't actually have any VFs to use. I've set it [...]

Secure boot disabled and using GRUB. 

```
root@minilab2:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet i915.force_probe=46d2
[    0.050050] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet i915.force_probe=46d2
[    3.497337] i915: module verification failed: signature and/or required key missing - tainting kernel
[    4.021793] i915 0000:00:02.0: Running in SR-IOV PF mode
[    4.022160] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.022190] i915 0000:00:02.0: vgaarb: deactivate vga console
[    4.022254] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    4.022625] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    4.023536] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[    4.040241] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    4.876808] i915 0000:00:02.0: [drm] [ENCODER:184:DDI A/PHY A] failed to retrieve link info, disabling eDP
[    4.896263] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[    4.896270] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    4.901784] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    4.902425] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    4.902428] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    4.902801] i915 0000:00:02.0: [drm] GuC RC: enabled
[    4.903743] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    4.903916] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.920162] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    4.921867] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.922089] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    4.922504] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.922931] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.923040] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    5.469004] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    5.469102] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    5.469128] i915 0000:00:02.1: Running in SR-IOV VF mode
[    5.469470] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[    5.469968] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    5.469995] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    5.470565] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[    5.471236] i915 0000:00:02.1: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    5.471240] i915 0000:00:02.1: HuC firmware PRELOADED
[    5.473674] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.473683] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    5.473797] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 1
[    5.474691] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    5.474695] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    5.474778] i915 0000:00:02.2: enabling device (0000 -> 0002)
[    5.474805] i915 0000:00:02.2: Running in SR-IOV VF mode
[    5.476273] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[    5.476783] i915 0000:00:02.2: [drm] VT-d active for gfx access
[    5.476826] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[    5.477464] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[    5.478174] i915 0000:00:02.2: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    5.478178] i915 0000:00:02.2: HuC firmware PRELOADED
[    5.480703] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.480711] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[    5.481097] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[    5.481458] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=io+mem
[    5.481462] i915 0000:00:02.1: vgaarb: changed VGA decodes: olddecodes=none,decodes=none:owns=none
[    5.481465] i915 0000:00:02.2: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[    5.481543] i915 0000:00:02.3: enabling device (0000 -> 0002)
[    5.481567] i915 0000:00:02.3: Running in SR-IOV VF mode
[    5.482708] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.9.0
[    5.483232] i915 0000:00:02.3: [drm] VT-d active for gfx access
[    5.483261] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[    5.483866] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.9.0
[    5.484508] i915 0000:00:02.3: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    5.484511] i915 0000:00:02.3: HuC firmware PRELOADED
[... cut ...]
[    5.505105] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.9.0
[    5.505570] i915 0000:00:02.7: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    5.505572] i915 0000:00:02.7: HuC firmware PRELOADED
[    5.507128] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.507136] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[    5.507245] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.7 on minor 7
[    5.507480] i915 0000:00:02.0: Enabled 7 VFs
```

johntdavis84, 2024-04-27:
Thanks! I found my issue ... though as I suspected, it's exposed another error I need to deal with. This was indeed a red herring.

I have this set up now:
```
cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3
```

However, I've realized that on boot, ` /sys/bus/pci/devices/0000:00:02.0/sriov_numvfs = 0` .
Echoing 3 into it got me three SR-IOV GPU devices.

So, I guess I need the full path in the conf file? Let me check that.

michael-pptf, 2024-04-27:
> Thanks! I found my issue ... though as I suspected, it's exposed another error I need to deal with. This was indeed a red herring.
>
> I have this set up now:
>
> ```
> cat /etc/sysfs.conf [...]

This is what mine looks like:

```
root@minilab2:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
```

johntdavis84, 2024-04-27:
> > Thanks! I found my issue ... though as I suspected, it's exposed another error I need to deal with. This was indeed a red herring.
> > I have this set up now:
> > ```
> > cat /etc/sysfs.conf
> > devices/pci0000:00/0000:00:02.0/sriov_numvfs = 3
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > However, I've realized that on boot, ` /sys/bus/pci/devices/0000:00:02.0/sriov_numvfs = 0` . Echoing 3 into it got me three SR-IOV GPU devices.
> > So, I guess I need the full path in the conf file? Let me check that.
> 
> This is what mine looks like:
> 
> ```
> root@minilab2:~# cat /etc/sysfs.conf
> devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
> ```
> 
> I'm curious what kind of platform are you using?

Thanks for all the diagnostic text, and your patience. I got it to work with the same sysfs conf file you're using. My entire issue was the lack of sysfsutils being installed. I can't believe I missed that to the point that I actually tried switching to Secure Boot with GRUB and blew up my system.

Sysfsutils is not installed on a default PVE 8.2 clean install.

closed 2024-04-27 by johntdavis84

michael-pptf, 2024-04-27:
> Thanks for all the diagnostic text, and your patience. I got it to work with the same sysfs conf file you're using. My entire issue was the lack of sysfsutils being installed. I can't believe I [...]

Yes, `sysfsutils` is not a standard package and it needs to be installed manually. I think this is in the README.md.

johntdavis84, 2024-04-28:
So, I will admit that I was originally using a tutorial from another site based on this repo (that also covered how to install the Windows VM and do some other things), and did not read the directions here first. 

I goofed up there. 

(I also didn't realize sysfsutils wasn't standard, at least for Ubuntu/Debian-based distros. For some reason, it was just there the first time I needed it on another distribution, but now I suspect it got installed by something else, then, too. I'm guessing it's not standard because being able to easily modify those settings with a plain text file is a bit too much of a security risk?)