strongtz/i915-sriov-dkms #161: Error! Bad return status for module build on kernel: 6.2.6-1-pve (x86_64)

gavinsama, 2024-04-28:
I tried countless times, #151 method also tried, still can't, every time when compiling will appear the following error.

.................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.2.6-1-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.

michael-pptf, 2024-04-28:
> I tried countless times, #151 method also tried, still can't, every time when compiling will appear the following error.
>
> .................(bad exit status: 2) Error! Bad return status for module [...]

```
cat /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log
```

and post the output

gavinsama, 2024-04-28:
DKMS make.log for i915-sriov-dkms-6.1 for kernel 6.2.6-1-pve (x86_64)
Sun 28 Apr 2024 03:24:42 PM CST
make: Entering directory '/usr/src/linux-headers-6.2.6-1-pve'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/soc/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/soc/intel_gmch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/soc/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_memcpy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_mm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sw_fence.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_sw_fence_work.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_syncmap.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_user_extensions.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_ioc32.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_debugfs_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_pipe_crc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/i915_pmu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen2_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen6_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen6_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen7_renderclear.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_breadcrumbs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_context_sseu.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/gen8_ppgtt.c:103:12: warning: ‘gen12_pte_encode_shim’ defined but not used [-Wunused-function]
  103 | static u64 gen12_pte_encode_shim(dma_addr_t addr,
      |            ^~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_heartbeat.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_engine_user.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/gt/intel_execlists_submission.o
[... cut ...]
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:832:5: note: declared here
  832 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
make[1]: *** [scripts/Makefile.build:260: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make: *** [Makefile:2026: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.2.6-1-pve'
root@pve:/usr/src/i915-sriov-dkms-6.1#

michael-pptf, 2024-04-28:
> DKMS make.log for i915-sriov-dkms-6.1 for kernel 6.2.6-1-pve (x86_64) Sun 28 Apr 2024 03:24:42 PM CST make: Entering directory '/usr/src/linux-headers-6.2.6-1-pve' CC [M] [...]

...and the output of ```dkms status```.

I suggest removing the dkms, the existing repo, then clone it again. You can also upgrade to 6.5 kernel since the fix is for 6.5 and you are on 6.2.

tomcatzh, 2024-04-29:
I have try 6.2.16 on PVE 8.2.

6.1.15 and 6.2.9-1 on PVE 7.4

both got this error
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:85:24: error: too few arguments to function ‘drm_dp_calc_pbn_mode’

I have success built and module in 6.5.13 on PVE 8.2. But i can not start vgpu when
echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs

returned
echo: write error: No such file or directory

michael-pptf, 2024-04-29:
> I have try 6.2.16 on PVE 8.2.
>
> 6.1.15 and 6.2.9-1 on PVE 7.4
>
> both got this error /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:85:24: error: too few arguments [...]

Did you install ```sysfsutils```? It is listed in the README.md under step 9.

ahit976, 2024-05-07:
> I tried countless times, #151 method also tried, still can't, every time when compiling will appear the following error.
>
> .................(bad exit status: 2) Error! Bad return status for module [...]

The same problem. 
Tried several times with kernel 6.2.9.1 and 6.2.x, and the making failed!

michael-pptf, 2024-05-07:
> > I tried countless times, #151 method also tried, still can't, every time when compiling will appear the following error.
> > .................(bad exit status: 2) Error! Bad return status for module build on kernel: 6.2.6-1-pve (x86_64) Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.
> 
> The same problem. Tried several times with kernel 6.2.9.1 and 6.2.x, and the making failed at Step 5.
> 
> Execute command dkms install -m i915-sriov-dkms -v 6.1 --force. -m

A series of recent PRs broke compatibility with older kernels. If you want to stay with older ones (likely anything before 6.5), try reset the ```HEAD``` of the repo to before those commits were made.

pasbec, 2024-07-19:
The errors from wrong `drm_dp_calc_pbn_mode` interfaces should be gone now for Proxmox VE with the latest PRs #178 and #179. However, I have not tested the 6.2 kernel due to lack of SecureBoot support. Would be nice if someone could re-check. 🌞

closed 2024-07-22 by strongtz