strongtz/i915-sriov-dkms #162: Build SRIOV fails for Debian 12.5 and Ubuntu 22.04 guest

Doug411, 2024-04-30:
I'm successfully able to enable SRIOV on my proxmox host.  I am able to pass a VF into a jellyfin LXC.   However, I have not been able to build the module on a Ubuntu 22.04 or Debian 12.5 VM.   On Ubuntu 22.04 kernel 6.5 I get an error ERROR (dkms apport): binary package for i915-sriov-dkms: 6.1 not found.  I've tried setting the version to 6.5 etc, but still get the error.

oug@appserver:/usr/src$ sudo git clone https://github.com/strongtz/i915-sriov-dkms.git i915-sriov-dkms-6.1
Cloning into 'i915-sriov-dkms-6.1'...
remote: Enumerating objects: 2499, done.
remote: Counting objects: 100% (587/587), done.
remote: Compressing objects: 100% (281/281), done.
remote: Total 2499 (delta 374), reused 310 (delta 298), pack-reused 1912
Receiving objects: 100% (2499/2499), 3.93 MiB | 9.83 MiB/s, done.
Resolving deltas: 100% (1395/1395), done.
doug@appserver:/usr/src$ ls
i915-sriov-dkms-6.1             linux-headers-6.5.0-28-generic  python3.10
i915-sriov-dkms-6.5             linux-hwe-6.5-headers-6.5.0-27
linux-headers-6.5.0-27-generic  linux-hwe-6.5-headers-6.5.0-28
doug@appserver:/usr/src$ cd 915-sriov-dkms-6.1
bash: cd: 915-sriov-dkms-6.1: No such file or directory
doug@appserver:/usr/src$ cd i915-sriov-dkms-6.1
doug@appserver:/usr/src/i915-sriov-dkms-6.1$ ls
dkms.conf  i915-sriov-dkms-git.install  Makefile  README.md
drivers    include                      PKGBUILD
doug@appserver:/usr/src/i915-sriov-dkms-6.1$ sudo nano dkms.conf
doug@appserver:/usr/src/i915-sriov-dkms-6.1$ dkms install --force -m i915-sriov-dkms -v 6.1
Error! You must be root to use this command.
doug@appserver:/usr/src/i915-sriov-dkms-6.1$ sudo dkms install --force -m i915-sriov-dkms -v 6.1
Creating symlink /var/lib/dkms/i915-sriov-dkms/6.1/source -> /usr/src/i915-sriov-dkms-6.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.5.0-28-generic -j4 -C /lib/modules/6.5.0-28-generic/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build KVER=6.5.0-28-generic................(bad exit status: 2)
ERROR (dkms apport): binary package for i915-sriov-dkms: 6.1 not found
Error! Bad return status for module build on kernel: 6.5.0-28-generic (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.
doug@appserver:/usr/src/i915-sriov-dkms-6.1$

michael-pptf, 2024-04-30:
This is where ```cat /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log``` comes in handy. Please include this so we know what exactly is going on during the build process.

dariuszsekula, 2024-04-30:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:85:31: error: too few arguments to function ‘drm_dp_calc_pbn_mode’
   85 |             crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                               ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:898:13: error: too few arguments to function ‘drm_dp_calc_pbn_mode’
  898 |             drm_dp_calc_pbn_mode(mode->clock, min_bpp) > port->full_pbn) {
      |             ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.0-28-generic/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.0-28-generic'

Doug411, 2024-05-02:
I tried the three steps described in that link
1.  Rename ./drivers/gpu/drm/i915/display/intel_dp_mst.c to intel_dp_mst.c.bak
2.  Download https://raw.githubusercontent.com/makazeu/i915-sriov-dkms/ffc23727f106995d89bc7ad32df4f1a3809ee737/drivers/gpu/drm/i915/display/intel_dp_mst.c to ./drivers/gpu/drm/i915/display/
3.  Re-build

Similar errors.   Help Please :)

 CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dsi_vbt.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dvo.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:85:31: error: too few arguments to function ‘drm_dp_calc_pbn_mode’
   85 |             crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                               ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:898:13: error: too few arguments to function ‘drm_dp_calc_pbn_mode’
  898 |             drm_dp_calc_pbn_mode(mode->clock, min_bpp) > port->full_pbn) {
      |             ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.0-28-generic/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.0-28-generic'

Doug411, 2024-05-02:
FYI.  Error log on debian 12.5
 CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_aux.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.c: In function ‘intel_dp_get_modes’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.c:4847:69: warning: passing argument 2 of ‘intel_connector_update_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 4847 |                 num_modes = intel_connector_update_modes(connector, edid);
      |                                                                     ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.c:55:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_connector.h:28:47: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
   28 |                                  struct edid *edid);
      |                                  ~~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.c: In function ‘intel_edp_init_connector’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.c:5362:51: warning: passing argument 2 of ‘drm_add_edid_modes’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 5362 |                 if (drm_add_edid_modes(connector, edid)) {
      |                                                   ^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp.c:44:
/usr/src/linux-headers-6.1.0-20-common/include/drm/drm_edid.h:575:70: note: expected ‘struct edid *’ but argument is of type ‘const struct edid *’
  575 | int drm_add_edid_modes(struct drm_connector *connector, struct edid *edid);
      |                                                         ~~~~~~~~~~~~~^~~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_aux_backlight.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_hdcp.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_link_training.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dsi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dsi_dcs_backlight.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dsi_vbt.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:85:31: error: too few arguments to function ‘drm_dp_calc_pbn_mode’
   85 |             crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                               ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
/usr/src/linux-headers-6.1.0-20-common/include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:898:13: error: too few arguments to function ‘drm_dp_calc_pbn_mode’
  898 |             drm_dp_calc_pbn_mode(mode->clock, min_bpp) > port->full_pbn) {
      |             ^~~~~~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.1.0-20-common/include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
make[1]: *** [/usr/src/linux-headers-6.1.0-20-common/scripts/Makefile.build:255: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [/usr/src/linux-headers-6.1.0-20-common/Makefile:2035: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.0-20-amd64'

michael-pptf, 2024-05-02:
Try this in your **Linux Guest**:

Git clone this repo:
```git clone https://github.com/strongtz/i915-sriov-dkms```

List past commits:
```git log -n 10 --oneline```

```1e1daf4`` is what we looking for, use it as ```HEAD```:
```git reset --hard 1e1daf4```

Then proceed to the normal installation in Linux guest.

Doug411, 2024-05-06:
Hey Michael.   Its working now!   I built the module yesterday after setting the commit head as you described.   It finished but it had some warnings/errors regarding secure boot (even though my secure boot was off).   However, I was unable to start the VM.  Today, I removed the module, recloned the repository and rebuilt (without applying the HEAD commit fix.   This time it ran through successfully without warnings  (not sure why).   I encountered the same issue when starting the VM.  After a little googling, I discovered that when adding the VF to the  VM I needed to leave "all functions" unchecked, but the "primary GPU" should be checked.  I had done the opposite.   Once I made that change, it worked perfectly.   Thanks for all your help!   I got my mini PC in March.   SRIOV/GPU passthrough was the one thing I could never get working.   I started using an unpriviledged LXC with hardware acceleration.   Then I was able to get SRIOV enabled on the host and passed it into the LXC.   Thanks to your help, I have it working in a VM.   I'm running ZFS, and when I first started I I had some issues with docker in the LXC, so I was running Jellyfin native in the LXC.   I'm happy that I can have jellyfin running in docker in a VM alongside my other services.   My non-scientific observations are that my CPU load was lower before SRIOV was enabled (when I was just passing my render to the LXC).   When I enabled SRIOV on the host and passed in a VF to the LXC, 4K transcoding consumed a little more CPU and was a little flakier when multiple 4k streams were running, and fast forwarding, etc (would crash the LXC at times).    I've been playing with Jellyfin in docker on Ubuntu 22.04.   CPU consumption is slightly higher (I believe).  However it is just rock solid.  I've got 6 4k transcodes running and have moved around in them, fast forwarding, rewinding, etc, and haven't been able to make it crash.  I still have lots of CPU headroom.   Thanks again for your help

Doug411, 2024-05-06:
I take it back.   It's not using more CPU  (It was still scanning my collection before).    Now that the initial scan is done, I'm able to transcodes 6 simultaneous 4K streams all the way down to 720p 6mbps while displaying embedded subtitles, and my CPU stays 5-10%.   All while running 30+ docker containers, and 4 VMs.   Very Stable and efficient.  No crashes. Thanks.   FYI this is also a i9 12900h MS-01.  LOL while I was type that last sentence, it finally crashed.  After a short investigation, no worries.  I ran out of disk space on the VM because the Jellyfin transcode directory was so full from all the testing.  :)

ToniCipriani, 2024-07-01:
I ran into this issue with kernel `6.5.0-41` but it works without modification with `6.5.0-35`.

sheetsg77, 2024-07-10:
I can confirm @ToniCipriani comment, Ubuntu 23.10 VM (first installed Nov 2023) has Kernel 6.5.0-35, and dkms built and installed without issue.  However, Ubuntu 23.10 (downloaded July 2024) has Kernel 6.5.0-41 and dkms has build issue (Error 2 - similar as reported above).

ToniCipriani, 2024-07-16:
Still doesn't work with 6.5.0-44. I guess I'll wait till I upgrade to Ubuntu 24.04 LTS with a different kernel and try again.

sheetsg77, 2024-07-16:
I looked at 24.04 LTS, and I think (??) kernel is 6.8, which will not work with dkms per readme

Greg

pasbec, 2024-07-19:
I've created a PR #181 to fix issues with all 6.2 and 6.5 kernels in Ubuntu 22.04 related to drm_dp_calc_pbn_mode.