strongtz/i915-sriov-dkms #163: Installing into Lubuntu/Ubuntu LTS 22.04 Jammy (Lubuntu) VM: Won't Build on Kernel 6.5.0-28-generic -- Sanity Check on Kernel Versions?

johntdavis84, 2024-05-01:
Hello,

I apologize if this is an obvious question, but I've been learning a lot about different kernel versions and how they all fit together (and in particular impact this module) over the last week or so. I think I know the answer to this, but I wanted to double check.

I have the Proxmox 6.5.13_5 kernel pinned on Proxmox 8.2, and got the DKMS built and installed after some trial and error. It's working great and I have a GPU-accelerated Windows VM whose only problem is the fact that it is, unfortunately, Windows.

**tl;dr I know the Proxmox host now requires kernel 6.5.13_5, but what's the minimum requirement in Ubuntu-based guests? And if we need to move Ubuntu to a newer kernel than what's in the LTS, what's the safest way to do that?**

I'm trying to get Lubuntu 22.04 (Ubuntu 22.04 Jammy LTS) running as a guest, and running into an issue with the kernel version.

Here's what the VM reports after it's been updated.
```
$ uname -r
6.5.0-28-generic

$ hwe-support-status --verbose
Your Hardware Enablement Stack (HWE) is supported until April 2027.

 cat /proc/version_signature 
Ubuntu 6.5.0-28.29~22.04.1-generic 6.5.13
```

Apt indicates headers are installed:
```
linux-headers-generic-hwe-22.04/jammy-security,jammy-updates,now 6.5.0.28.29~22.04.1 amd64 [installed]
  Generic Linux kernel headers
```

The kernel fails to build, with the same error I saw when my version of the Proxmox kernel was too old.

Except ... there's no newer kernel in the official Ubuntu repo to upgrade to. There are a bunch of random older 6.0-6.2 kernels, but I'm not sure which of them I'm supposed to use if I want to go back, and given that this is a customized distro, I suspect some of the customization will break if I go back to 6.2.

These are the repos I'm hitting (distro specific repos that don't have kernels in them removed):
```
Hit:2 http://archive.ubuntu.com/ubuntu jammy InRelease                                                                                                                                                                                                                       
Hit:4 http://archive.ubuntu.com/ubuntu jammy-backports InRelease                                                             
Hit:5 http://archive.ubuntu.com/ubuntu jammy-security InRelease                                                       
Hit:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease                                                        
Hit:7 http://archive.canonical.com/ubuntu jammy InRelease                                               
```

Enough of the extra tools (whose repos I've omitted) have been patched to need/to assume 6.5 is present that I'm not sure what will happen if I try to downgrade to 6.2.

michael-pptf, 2024-05-02:
I think you are having the same issue with https://github.com/strongtz/i915-sriov-dkms/issues/162

I would say that the kernel and firmware shipped with PVE 6.5.13-5 is probably VERY edge. Looking at issue 162's build log I can see that one file coming up over and over again, and it was fixed by a PR just weeks ago.

Unfortunately I don't have a better option. Your first option is to try to downgrade Linux guest VM to something 6.2 and try to build it with the repo before the PR is made, and this is EXACTLY WHY I put a "DO NOT PR" tag in my PR for the firmware fix, because it may potentially break kernel compatibility.

michael-pptf, 2024-05-02:
See https://github.com/strongtz/i915-sriov-dkms/issues/162#issuecomment-2090567795

johntdavis84, 2024-05-02:
> I think you are having the same issue with #162
>
> I would say that the kernel and firmware shipped with PVE 6.5.13-5 is probably VERY edge. Looking at issue 162's build log I can see that one file [...]

Thanks! I'll take a look at that thread more closely later today. I don't mind trying to downgrade to 6.2. I even suspected that might be the answer, but I wanted to check before I started messing with that. Until I started working with SR-IOV iGPU stuff, I'd never actually done any sort of kernel tweaking. 

I've been wanting to try OpenSUSE Tumbleweed, as well as Debian 12-based Crunchbang++. It'll be interesting to see if I can make those work.

>  I would say that the kernel and firmware shipped with PVE 6.5.13-5 is probably VERY edge. 

For reference, I enabled the mainline kernel repo in my Lubuntu VM, and the newest kernel it offered me was 6.5.11.

I can't find it now, but I'm pretty sure that PR actually made the build require 6.5.13-3 (though I don't know if the "_3" is an actual Linux thing, or a Proxmox-specific thing). It was on 6.5.13-1 and it wouldn't work until I upgraded.

It's ironic that it was easier to get it working in a Windows guest. ;)

johntdavis84, 2024-05-03:
I'm trying the solution in that thread now. I need to get a Remote Desktop app up to test, but it seems like it worked.

I got the module built and was able to add one of my VFs in Primary GPU mode.

Of course, then apt wanted to run an update, and I noticed this.

Do I need to rebuild the module, or is this warning just going to happen?
[image]

johntdavis84, 2024-05-03:
Installed and built per the instructions in that thread, but alas, not working.

The device shows up inside the VM, but I get an error I've never seen before, and lspci gives me a lot of data about the device but doesn't know what it is compared to the host.

```
johntdavis84@andromeda2-dragonOS:~$ sudo lspci -nn -s 01:00 -vvv
01:00.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4690] (rev 0c) (prog-if 00 [VGA controller])
        Subsystem: Hewlett-Packard Company Device [103c:8952]
        Physical Slot: 0
        Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Region 0: Memory at 80000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at 380000000000 (64-bit, prefetchable) [size=512M]
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM not supported
                        ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed unknown (downgraded), Width x0 (downgraded)
                        TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Not Supported, TimeoutDis- NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit-
                Address: 00000000  Data: 0000
                Masking: 00000000  Pending: 00000000
        Kernel modules: i915

johntdavis84@andromeda2-dragonOS:~$ sudo dmesg | grep -i i915
[    3.469259] i915 0000:01:00.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
[    3.469953] i915 0000:01:00.0: Device initialization failed (-5)
[    3.469956] i915 0000:01:00.0: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[    3.469961] i915: probe of 0000:01:00.0 failed with error -5
```

I'm still on kernel 6.5.0. That one might just be too old.
Upgrading to 6.5.11 failed since I couldn't download the headers.

johntdavis84, 2024-05-04:
Re-did everything to be safe.

I still got those warnings pictured above, but everything worked and I have a GPU accelerated Remote Desktop session.

Closing this as solved, though I don't quite understand what changed by rolling back to the earlier repo commit?

closed 2024-05-04 by johntdavis84

michael-pptf, 2024-05-04:
I am no expert of Git, but you can see what each PR/Commit/Merge did to the repo.

If you want to find out, you can read it here: https://github.com/strongtz/i915-sriov-dkms/pull/152/commits/ffc23727f106995d89bc7ad32df4f1a3809ee737

This commit changed how a build function parses the parameter, which are only found in 6.5.13-5 and onwards. If you are building the module in a Linux guest that's lower than that, it would not have worked since it does not have the parameters needed to do so.

See those "too few arguments" or "too many arguments" messages? That is the problem.

By rebasing your repo to a point before the PR merge was made, you are back to the state where it can be built with older kernels - not just PVE, but Linux kernels in general.

johntdavis84, 2024-05-05:
> I am no expert of Git, but you can see what each PR/Commit/Merge did to the repo.
> 
> If you want to find out, you can read it here: [ffc2372](https://github.com/strongtz/i915-sriov-dkms/commit/ffc23727f106995d89bc7ad32df4f1a3809ee737)
> 
> This commit changed how a build function parses the parameter, which are only found in 6.5.13-5 and onwards. If you are building the module in a Linux guest that's lower than that, it would not have worked since it does not have the parameters needed to do so.
> 
> See those "too few arguments" or "too many arguments" messages? That is the problem.
> 
> By rebasing your repo to a point before the PR merge was made, you are back to the state where it can be built with older kernels - not just PVE, but Linux kernels in general.

Thanks!

So, at this point, the only way to use the current commit of this repo in a guest is for that guest to be running 6.5.13-3 or later?

That seems like it's probably a very small pool of non-rolling release distros, unless you manually install an upstream mainline kernel (easy in theory, unpredictable and not much fun in practice). I've yet to find an Ubuntu-based repo that uses it, aside from Proxmox itself.