strongtz/i915-sriov-dkms #165: New PVE install on a Minisfroum MS1 I9-12900h cpu 64gigs ram kernel 6.5.13-5-pve

patjsimpson, 2024-05-04:
I performed a new proxmox install from the ISO 8.1 I believe.
Pinned the kernel - proxmox-boot-tool kernel pin 6.5.11-8-pve
Used the post install helper script to config and upgrade PVE. Once it was upgraded I pinned the kernel from the one above to this kernel.
proxmox-boot-tool kernel pin 6.5.13-5-pve
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
I previously tried this on a mini computer with a J5005 I believe with out success because after the reboot my system would always loose its network when it came back up.  However the dkms install always worked on that system while following the same steps.
Got my memory for the MS1 and decided to try it against the newer I9-12900h to see if I had better results but no luck.

Proceeded to follow the steps here to install all required and the headers.
Tried to install 6.5 version of the dkms and I get this error.

dkms install --force -m i915-sriov-dkms -v 6.5
Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.5.13-5-pve -j20 -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.13-5-pve.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log for more information.
root@pve-i9:~/i915-sriov-dkms# cat /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log
DKMS make.log for i915-sriov-dkms-6.5 for kernel 6.5.13-5-pve (x86_64)
Sat May  4 08:02:37 AM PDT 2024
make: Entering directory '/usr/src/linux-headers-6.5.13-5-pve'
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/soc/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/soc/intel_gmch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/soc/intel_pch.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_memcpy.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_mm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sw_fence.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sw_fence_work.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_syncmap.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_user_extensions.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_ioc32.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_debugfs_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_pipe_crc.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_pmu.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/gen2_engine_cs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/gen6_engine_cs.o
[... 239 lines cut ...]
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:86:43: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
   86 |                         crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                                           ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:906:17: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
  906 |                 drm_dp_calc_pbn_mode(mode->clock, min_bpp, false) > port->full_pbn) {
      |                 ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_hdmi.c: In function ‘intel_hdmi_handle_sink_scrambling’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_hdmi.c:2682:29: warning: unused variable ‘adapter’ [-Wunused-variable]
 2682 |         struct i2c_adapter *adapter =
      |                             ^~~~~~~
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve'

patjsimpson, 2024-05-04:
I

> After changing the kernel, did you update the GRUB and LINUX_CMDLINE_DEFAULT?

I had not yet done that as it was further down in the instructions... Previously I had not modified that on the earlier generation and did not get the error above.

just did that 
now contains "intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet"

root@pve-i9:~# nano /etc/default/grub
root@pve-i9:~# update-grub
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.8.4-2-pve
Found initrd image: /boot/initrd.img-6.8.4-2-pve
Found linux image: /boot/vmlinuz-6.5.13-5-pve
Found initrd image: /boot/initrd.img-6.5.13-5-pve
Found linux image: /boot/vmlinuz-6.5.11-8-pve
Found initrd image: /boot/initrd.img-6.5.11-8-pve
Found memtest86+ 64bit EFI image: /boot/memtest86+x64.efi
Adding boot menu entry for UEFI Firmware Settings ...
done
root@pve-i9:~# update-initramfs -u
update-initramfs: Generating /boot/initrd.img-6.8.4-2-pve **_(Is this a concern?  I did see that on previous gen also)_**
Running hook script 'zz-proxmox-boot'..
Re-executing '/etc/kernel/postinst.d/zz-proxmox-boot' in new private mount namespace..
No /etc/kernel/proxmox-boot-uuids found, skipping ESP sync.

currently rebooting the host again and will retry it.

patjsimpson, 2024-05-04:
no change after reboot and retry
root@pve-i9:~# dkms install --force -m i915-sriov-dkms -v 6.5
Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.5.13-5-pve -j20 -C /lib/modules/6.5.13-5-pve/build M=/                                                                                                 var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.13-5-pve.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log for more information.

michael-pptf, 2024-05-04:
> update-initramfs: Generating /boot/initrd.img-6.8.4-2-pve (Is this a concern? I did see that on previous gen also)
No, when updating ```initramfs``` it generates config for all kernels currently present on the system.

Can you confirm IOMMU is actually enabled in BIOS and working? 
```dmesg | grep -e DMAR -e IOMMU```

patjsimpson, 2024-05-04:
sure , It appears to be but I am not an expert on this.

root@pve-i9:~# dmesg | grep -e DMAR -e IOMMU
[    0.019758] ACPI: DMAR 0x0000000042767000 000088 (v02 INTEL  EDK2     0000000                                                                                                 2      01000013)
[    0.019786] ACPI: Reserving DMAR table memory at [mem 0x42767000-0x42767087]
[    0.084499] DMAR: IOMMU enabled
[    0.176020] DMAR: Host address width 39
[    0.176021] DMAR: DRHD base: 0x000000fed90000 flags: 0x0
[    0.176024] DMAR: dmar0: reg_base_addr fed90000 ver 4:0 cap 1c0000c40660462 e                                                                                                 cap 29a00f0505e
[    0.176026] DMAR: DRHD base: 0x000000fed91000 flags: 0x1
[    0.176030] DMAR: dmar1: reg_base_addr fed91000 ver 5:0 cap d2008c40660462 ec                                                                                                 ap f050da
[    0.176031] DMAR: RMRR base: 0x0000004c000000 end: 0x000000503fffff
[    0.176033] DMAR-IR: IOAPIC id 2 under DRHD base  0xfed91000 IOMMU 1
[    0.176034] DMAR-IR: HPET id 0 under DRHD base 0xfed91000
[    0.176034] DMAR-IR: Queued invalidation will be enabled to support x2apic an                                                                                                 d Intr-remapping.
[    0.177559] DMAR-IR: Enabled IRQ remapping in x2apic mode
[    0.596428] pci 0000:00:02.0: DMAR: Skip IOMMU disabling for graphics
[    0.662992] DMAR: No ATSR found
[    0.662992] DMAR: No SATC found
[    0.662993] DMAR: IOMMU feature fl1gp_support inconsistent
[    0.662993] DMAR: IOMMU feature pgsel_inv inconsistent
[    0.662994] DMAR: IOMMU feature nwfs inconsistent
[    0.662994] DMAR: IOMMU feature dit inconsistent
[    0.662995] DMAR: IOMMU feature sc_support inconsistent
[    0.662995] DMAR: IOMMU feature dev_iotlb_support inconsistent
[    0.662996] DMAR: dmar0: Using Queued invalidation
[    0.662997] DMAR: dmar1: Using Queued invalidation
[    0.663491] DMAR: Intel(R) Virtualization Technology for Directed I/O

patjsimpson, 2024-05-04:
Before installing I confirmed that the two settings I know of were enabled in the uefi... Is there a possible problem with secure boot being enabled?  I did notice that and don't remember if that is ok but the install went fine so assumed proxmox must support it.

michael-pptf, 2024-05-04:
It is my opinion to have it disabled for better compatibility with stuff. If this is a new install and you are comfortable with the risks involved of having it turned off like me, I would say just nuke the installation and start over after disabling secure boot.

michael-pptf, 2024-05-04:
See PR #164. This is potentially what's preventing the build. I can't say for sure. 

After cloning the repo, try reset the ```HEAD``` to ```d2b3b6e``` see if it works. But even if it does, if you want to build the module in a Linux guest, some more tweaking might be needed.

patjsimpson, 2024-05-04:
I won't deny that most of that is over my head.  While I have compiled stuff like this before it is usually following a tutorial like this and doing a ton of do overs after google searches on errors.  I am not positive but I think I pulled this last night after those changes were posted.  I feel like you are hinting that I should do something different based on that post above but it is beyond my comprehension as to if it is to pull the repo and try again or something more detailed that I can't build in my head with my lack of overall understanding.  I do appreciate the effort to help!

patjsimpson, 2024-05-04:
sorry I missed this keep part... will see if I can figure out that part "After cloning the repo, try reset the HEAD to d2b3b6e see if it works"

michael-pptf, 2024-05-04:
> sorry I missed this keep part... will see if I can figure out that part "After cloning the repo, try reset the HEAD to [d2b3b6e](https://github.com/strongtz/i915-sriov-dkms/commit/d2b3b6e374c2f1799fb5bc07aad1f4ed5d058919) see if it works"

https://github.com/strongtz/i915-sriov-dkms/issues/162#issuecomment-2090567795
Follow the steps and change the commit number to ```d2b3b6e```

patjsimpson, 2024-05-04:
Was about to fix the head and noticed the one just before it....
19a8abe (HEAD -> master, origin/master, origin/HEAD) Merge pull request #164 from WR-Maox/fix_6.5.x
**e418a0f fix bug onkernel-6.5.x**
d2b3b6e Merge pull request #160 from michael-pptf/README-update

Should I still fix head to d2bb36e or go with e418a0f?

michael-pptf, 2024-05-04:
use #159 which I have not commited. You can also just clone my fork at https://github.com/michael-pptf/i915-sriov-dkms

patjsimpson, 2024-05-04:
got it.

patjsimpson, 2024-05-04:
this looks positive 

root@pve-i9:~/intelforked# dkms install --force -m i915-sriov-dkms -v 6.5
Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.5.13-5-pve -j20 -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.13-5-pve........
Signing module /var/lib/dkms/i915-sriov-dkms/6.5/build/i915.ko
Cleaning build area...

i915.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.5.13-5-pve/updates/dkms/
depmod...

patjsimpson, 2024-05-04:
So after continuing on and rebooting it does not appear to be showing virtuals

 lspci  | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
^C
root@pve-i9:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet
[    0.084096] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet
root@pve-i9:~#

I checked for errors in dmesg but not seeing any.

patjsimpson, 2024-05-04:
not sure why the line strikes happened there.

patjsimpson, 2024-05-04:
I believe I should see something like this

00:02.0 VGA compatible controller: Intel Corporation Device 4692 (rev 0c)
00:02.1 VGA compatible controller: Intel Corporation Device 4692 (rev 0c)
00:02.2 VGA compatible controller: Intel Corporation Device 4692 (rev 0c)
00:02.3 VGA compatible controller: Intel Corporation Device 4692 (rev 0c)

michael-pptf, 2024-05-04:
output of ```dkms status```?

patjsimpson, 2024-05-04:
similar but only one line

root@pve-i9:~# dkms status
i915-sriov-dkms/6.5, 6.5.13-5-pve, x86_64: installed

michael-pptf, 2024-05-04:
have you gone through the README.md from Step 9?

patjsimpson, 2024-05-04:
I will check again.

patjsimpson, 2024-05-04:
yes that is what I did

root@pve-i9:~# cat /etc/sysfs.conf
devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7
root@pve-i9:~#

patjsimpson, 2024-05-04:
I rebooted as requested after that.... I half expected the network to not come back up after that reboot as that is what happened on the J5005 cpu based system I was playing with before at that point but it did come back up so that is positive. Just need to figure out the virtuals issue.

michael-pptf, 2024-05-04:
Are you using the system with a screen attached? or just using it remotely?

patjsimpson, 2024-05-04:
SSH I  pulled the hdmi before I started this.  should I maybe power it off and them back on?  Not sure I have done a full power off since pulling the hdmi.

michael-pptf, 2024-05-04:
Shouldn't need to be.
So when you do ```dmesg | grep i915``` it basically came up empty?

patjsimpson, 2024-05-04:
I get this... not sure if you would call this empty

root@pve-i9:~# dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet
[    0.084096] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 iommu=pt quiet

patjsimpson, 2024-05-04:
should I have rebuilt it with out secure boot enabled?  sorry i thought maybe that was negated by you second post above.

michael-pptf, 2024-05-04:
> I get this... not sure if you would call this empty
>
> root@pve-i9:~# dmesg | grep i915 [ 0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.13-5-pve root=/dev/mapper/pve-root ro intel_iommu=on [...]

Yea that's empty. You are not even seeing the i915 DKMS being loaded. Well, it could be that SB is preventing it from being loaded, I can't say for sure. 

Like I said, nuke the installation and start over with SB turned off. I thought you were doing it this way already. My bad.

patjsimpson, 2024-05-04:
No problem  this is a new build and so no problem nuking it.

patjsimpson, 2024-05-04:
That must have been it.  So you mentioned above   "But even if it does, if you want to build the module in a Linux guest, some more tweaking might be needed."  can you point me to somewhere that details what you are hinting at?

root@pve-i9:~# lspci  | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.1 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.2 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.3 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.4 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.5 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.6 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.7 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)

patjsimpson, 2024-05-04:
I think I want to do a mix of lxc and possibly VM.s

michael-pptf, 2024-05-04:
https://github.com/strongtz/i915-sriov-dkms/issues/162#issuecomment-2090567795

Try this in Linux guest VM. You don't need this for LXC.

patjsimpson, 2024-05-04:
going back to start reading the readme again like I should first ;-)

michael-pptf, 2024-05-04:
Here's a quick run down:
LXC - uses the host capabilities, no drivers needed, just passthrough the VF and set it correctly in the configuration file.

Windows VM - needs Windows drivers, only a bunch of specific version would work.

Linux VM - you need to build the same module as this one. Depending on the kernel used by the VM, recent PRs are potentially problematic so you'd need to reset the ```HEAD``` to whenever before those commits are merged.

More often than not, you can take a look into the closed issues for solution.

patjsimpson, 2024-05-05:
so quick clarification...
I assume I would pass the individual card and render  and not the /dev/dri /dev/dri like I would when doing a container. When I do the dev/dri all the virtuals show up.... Also assuming I do not start with the card0 as I am assuming it is the same as doing the zero virtual below... Hash tag in front of both that I believe I am not supposed to use.

#00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.1 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.2 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.3 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.4 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.5 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.6 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:02.7 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)

lxc.cgroup2.devices.allow: c 226:1 rwm
lxc.cgroup2.devices.allow: c 226:129 rwm
lxc.cgroup2.devices.allow: c 29:0 rwm
lxc.mount.entry: /dev/fb0 dev/fb0 none bind,optional,create=file
#lxc.mount.entry: /dev/dri dev/dri none bind,optional,create=dir
lxc.mount.entry: /dev/dri/renderD129 dev/dri/renderD129 none bind,optional,create=file
lxc.mount.entry: /dev/card1 dev/card1 none bind,optional,create=file

thanks again for all the help here.

Coder84619, 2024-05-05:
I'm confused...what's the solution here? I have secure boot, re-installed my system from scratch, and get the same errors in the logs.

patjsimpson, 2024-05-05:
> I'm confused...what's the solution here? I have secure boot, re-installed my system from scratch, and get the same errors in the logs.

to get past the compile / install issues I used  Michael's repo   You can also just clone my fork at https://github.com/michael-pptf/i915-sriov-dkms

But after install secure boot was preventing the modules from loading so I rebuilt the system after turning off secure boot and did it all over again with success finally.

michael-pptf, 2024-05-05:
> I'm confused...what's the solution here? I have secure boot, re-installed my system from scratch, and get the same errors in the logs.

For a guaranteed successful install, you would have to turn off Secure Boot and use a commit that is not affected by recent changes. Luckily, my fork ```https://github.com/michael-pptf/i915-sriov-dkms``` is such one. 

You can, of course, leave Secure Boot on, but you will be looking at a completely different process that it may or may not work (it certainly did not work for me): ```https://www.derekseaman.com/2023/11/proxmox-ve-8-1-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html```

patjsimpson, 2024-05-05:
I finally got my tdarr lxc working but it is pitifully slow transcode speeds...  Not sure why that would be as it is supposed to be pretty fast. I am getting like 29 fps.  I have been following this tuturial which is from 2022 so maybe I should be using something more updated.  https://geekistheway.com/2022/12/23/setting-up-intel-gpu-passthrough-on-proxmox-lxc-containers/

patjsimpson, 2024-05-05:
when I try to run intel_gpu_top I get this error.

root@tdarr-i9:~# intel_gpu_top
Failed to detect engines! (No such file or directory)
(Kernel 4.16 or newer is required for i915 PMU support.)

both on proxmox and the LXC.

michael-pptf, 2024-05-05:
> I finally got my tdarr lxc working but it is pitifully slow transcode speeds... Not sure why that would be as it is supposed to be pretty fast. I am getting like 29 fps. I have been following this [...]

https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/
I haven't got time to update it to the current repo but for the LXC part it is largely the same process.

> when I try to run intel_gpu_top I get this error.
> 
> root@tdarr-i9:~# intel_gpu_top Failed to detect engines! (No such file or directory) (Kernel 4.16 or newer is required for i915 PMU support.)
> 
> both on proxmox and the LXC.

try this:
```intel_gpu_top -d sriov```

Coder84619, 2024-05-05:
> to get past the compile / install issues I used Michael's repo You can also just clone my fork at https://github.com/michael-pptf/i915-sriov-dkms
>
> But after install secure boot was preventing the [...]

I found that 6.5.13-3 (not -5) worked with secure boot on my second N305 server.

patjsimpson, 2024-05-06:
> https://www.michaelstinkerings.org/gpu-virtualization-with-intel-12th-gen-igpu-uhd-730/ I haven't got time to update it to the current repo but for the LXC part it is largely the same process.
>
>
> try [...]

Thanks that link  helped greatly!  Got it all working now in Tdarr lxc

patjsimpson, 2024-05-06:
> > scratch that it is working but every video is getting corrupted with green screen with artifacts. Could this be driver related? transcode was successful and it was hevc after complete with no errors.
> 
> Are you doing HDR tonemapping? That would not work and the symptom is the same as you described.

Not sure I am just using a community plugin in Tdarr.  Not sure if this would be in plugin setting or somewhere else in tdarr.

specifically this one  Tdarr_Plugin_drdd_standardise_all_in_one
DrDD H265 MKV AC3 audio subtitles [VAAPI & NVENC]

patjsimpson, 2024-05-06:
I wanted to use this one but had no luck..

Boosh-Transcode using QSV GPU & FFMPEG

==DETAILS== This is a QSV plugin. 8th+ gen INTEL QSV enabled CPUs are recommended. VAAPI is NOT used. ==OS SUPPORT== This plugin supports Linux & Windows using QSV. Mac is supported though cannot use QSV and relies on 'VideoToolBox' - Expect to see different encode speed & quality on Mac compared to other platforms. Ensure you set your node settings accordingly! ==LOGIC== Files will be transcoded into H265/HEVC using Quick Sync Video (QSV) via Intel GPU using ffmpeg. Settings are dependant on file bitrate working by the logic that H265 can support the same amount of data at half the bitrate of H264. This plugin will skip files already in HEVC, AV1 & VP9 unless "reconvert_hevc" is marked as true. If it is then these will be reconverted again if they exceed the bitrate specified in "hevc_max_bitrate". This plugin relies on understanding the accurate video bitrate of your files. It's highly recommended to remux into MKV & enable "Run mkvpropedit on files before running plugins" under Tdarr>Options.

michael-pptf, 2024-05-06:
What is the source file? The video that you are trying to transcode.

patjsimpson, 2024-05-06:
Yes plex media

Coder84619, 2024-05-06:
> > scratch that it is working but every video is getting corrupted with green screen with artifacts. Could this be driver related? transcode was successful and it was hevc after complete with no errors.
> 
> Are you doing HDR tonemapping? That would not work and the symptom is the same as you described.

I can confirm Plex STILL does not support vGPU via LXC on Proxmox. HDR tone mapping produces a corrupted video stream with torn, green video blocks. I raised this issue with Plex 6 months, and they said it would be put on their engineering roadmap. That was about Nov'23...nothing yet.

patjsimpson, 2024-05-06:
Is this something that would need to be turned off in the tdarr plugin or at a system level?

patjsimpson, 2024-05-06:
also I am using tdarr to convert all plex media to h265.

patjsimpson, 2024-05-08:
Seems like we just need to wait for them to fix dkms on build 8.2 so that eventually when they release cernel 6.9 .x.x on pve we can take advantage of it... maybe I am not reading that thread correct. I have limited understanding of this all.

thx for the updates.

lyj0309, 2024-05-08:
> > sorry I missed this keep part... will see if I can figure out that part "After cloning the repo, try reset the HEAD to [d2b3b6e](https://github.com/strongtz/i915-sriov-dkms/commit/d2b3b6e374c2f1799fb5bc07aad1f4ed5d058919) see if it works"
> 
> [#162 (comment)](https://github.com/strongtz/i915-sriov-dkms/issues/162#issuecomment-2090567795) Follow the steps (in your PVE main, not Linux guest - that's for the guy I was replying to) and change the commit number to `d2b3b6e`

@michael-pptf  in my case, it's not working

```sh
root@pve:~/i915-sriov-dkms# git log -n 1 HEAD
commit d2b3b6e374c2f1799fb5bc07aad1f4ed5d058919 (HEAD -> master)
Merge: 8a868c9 24586a4
Author: Sophon <wuxilin123@gmail.com>
Date:   Sun Apr 28 10:27:25 2024 +0800

    Merge pull request #160 from michael-pptf/README-update

root@pve:~/i915-sriov-dkms# dkms install -m i915-sriov-dkms -v 6.1 --force
Sign command: /lib/modules/6.5.13-5-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.5.13-5-pve -j12 -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.1/build KVER=6.5.13-5-pve...........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.1/build/make.log for more information.

 CC [M]  /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_gmbus.o
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:86:35: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
   86 |                 crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                                   ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:906:17: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
  906 |                 drm_dp_calc_pbn_mode(mode->clock, min_bpp, false) > port->full_pbn) {
      |                 ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.1/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.1/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.13-5-pve
```

is there any solution

michael-pptf, 2024-05-08:
just remove everything, clone my repo, and start over

ksqeib, 2024-05-23:
> just remove everything, clone my repo, and start over

It did work,thanks guy.

pbvdven, 2024-05-26:
I have  a issue.  

i changed machines from i7 1270P  to i9 13900H the minisforum ms01. but i just swapped the ssd's changed the network config and voila everything seems working just fine windows vm, lxc docker with plex and jellyfin and a other lxc docker with frigate for cctv. 

But now i rebuild the jellyfin container and i cant get hw working everything seems fine no error messages but it wont work but if i restore the backup its working again. also if i switch back to the same jellyfin version its not working anymore. but backup with running conainer from the 1270p is working fine. same for frigate after i recreate the container on the ms01 hw stops working but the backup with the running container from 1270p still works on the ms01. the windows VM doesnt have any issues at all with the gpu. plex i didn't try jet because i didn't want to break everything at once.

like a said i just swapped the ssd i thought its basically the same machine why reinstall everything. but now I'm not sure how i should fix this. should i just remove the dkms and reinstall it? like i said it looks like its working just fine i have 7vf also plex that i didnt touch also does hw transcoding on the VF. I really don't get why this isn't working but clean install is not really a option.

i do get some weird message from the i915 or gpu at boot just after grub something with cant confirm version 1.9?.0 for every VF its show the adress 02.0.0.0 ,1,2 and so on intel gpu with that message and then a other line with same but it was something with ignored. sorry i cant reboot at the moment i will check later and add the correct error message.

also i can to intel_gpu_top ? sriov and get output

but dmesg | grep i915 is giving me nothing but lspci | grep VGA is giving back the 7VF and the status of the dkms is installed i am pinned down to kernel 6.5.11-7-pve but that also didn't change. i do have a HDMI attached 

maybe someone has a solution or have done something similar before.

sheetsg77, 2024-07-12:
I'm having issues with installation on a guest VM (Ubuntu Mantic 23.10 Kernal 6.5.0-44).  I'm running PXE 8.2.4 with kernel pinned at 6.5.13-5-pve.  I have 7 VF's enabled on host.

On guest VM
```
greg@dkms-test:~/i915-sriov-dkms$ sudo dkms install --force -m i915-sriov-dkms -v 6.5
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.5.0-35-generic -j4 -C /lib/modules/6.5.0-35-generic/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.0-35-generic............(bad exit status: 2)
ERROR (dkms apport): binary package for i915-sriov-dkms: 6.5 not found
Error! Bad return status for module build on kernel: 6.5.0-35-generic (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log for more information.
```

greg@dkms-test:~/i915-sriov-dkms$ more /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log
DKMS make.log for i915-sriov-dkms-6.5 for kernel 6.5.0-35-generic (x86_64)
Fri Jul 12 16:15:56 UTC 2024
make: Entering directory '/usr/src/linux-headers-6.5.0-35-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_ioctl.o
....
....
...
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sriov.o
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c: In function ‘gsc_destroy_one’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:150:71: warning: passing argument 2 of ‘intel_huc_unregister_gsc_notifier’ dis
cards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  150 |                                                           aux_dev->dev.bus);
      |                                                           ~~~~~~~~~~~~^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_uc.h:14,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gt_types.h:19,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_engine.h:18,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_drv.h:46,
                 from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:8:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:70:80: note: expected ‘struct bus_type *’ but argument is of type ‘const st
ruct bus_type *’
   70 | void intel_huc_unregister_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
      |                                                               ~~~~~~~~~~~~~~~~~^~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c: In function ‘gsc_init_one’:
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:264:61: warning: passing argument 2 of ‘intel_huc_register_gsc_notifier’ disca
rds ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  264 |                                                 aux_dev->dev.bus);
      |                                                 ~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:69:78: note: expected ‘struct bus_type *’ but argument is of type ‘const st
ruct bus_type *’
   69 | void intel_huc_register_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
      |                                                             ~~~~~~~~~~~~~~~~~^~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:271:71: warning: passing argument 2 of ‘intel_huc_unregister_gsc_notifier’ dis
cards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  271 |                                                           aux_dev->dev.bus);
      |                                                           ~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:70:80: note: expected ‘struct bus_type *’ but argument is of type ‘const st
ruct bus_type *’
   70 | void intel_huc_unregister_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
      |                                                               ~~~~~~~~~~~~~~~~~^~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sriov_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_event.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/iov/intel_iov_memirq.o
[... 5 lines cut ...]
[... cut ...]
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp, bool dsc);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.5.0-35-generic/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.5.0-35-generic'

ToniCipriani, 2024-07-16:
> I'm having issues with installation on a guest VM (Ubuntu Mantic 23.10 Kernal 6.5.0-44). I'm running PXE 8.2.4 with kernel pinned at 6.5.13-5-pve. I have 7 VF's enabled on host.
> 
> On guest VM
> 
> ```
> greg@dkms-test:~/i915-sriov-dkms$ sudo dkms install --force -m i915-sriov-dkms -v 6.5
> Sign command: /usr/bin/kmodsign
> Signing key: /var/lib/shim-signed/mok/MOK.priv
> Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
> 
> Building module:
> Cleaning build area...
> make -j4 KERNELRELEASE=6.5.0-35-generic -j4 -C /lib/modules/6.5.0-35-generic/build M=/var/lib/dkms/i915-sriov-dkms/6.5/build KVER=6.5.0-35-generic............(bad exit status: 2)
> ERROR (dkms apport): binary package for i915-sriov-dkms: 6.5 not found
> Error! Bad return status for module build on kernel: 6.5.0-35-generic (x86_64)
> Consult /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log for more information.
> ```
> 
> Make.log
> 
> ```
> greg@dkms-test:~/i915-sriov-dkms$ more /var/lib/dkms/i915-sriov-dkms/6.5/build/make.log
> DKMS make.log for i915-sriov-dkms-6.5 for kernel 6.5.0-35-generic (x86_64)
> Fri Jul 12 16:15:56 UTC 2024
> make: Entering directory '/usr/src/linux-headers-6.5.0-35-generic'
> warning: the compiler differs from the one used to build the kernel
>   The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
>   You are using:           gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_driver.o
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_drm_client.o
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_config.o
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_getparam.o
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_hwmon.o
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_ioctl.o
> ....
> ....
> ...
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sriov.o
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c: In function ‘gsc_destroy_one’:
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:150:71: warning: passing argument 2 of ‘intel_huc_unregister_gsc_notifier’ dis
> cards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
>   150 |                                                           aux_dev->dev.bus);
>       |                                                           ~~~~~~~~~~~~^~~~
> In file included from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_uc.h:14,
>                  from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gt_types.h:19,
>                  from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_engine.h:18,
>                  from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_drv.h:46,
>                  from /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:8:
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:70:80: note: expected ‘struct bus_type *’ but argument is of type ‘const st
> ruct bus_type *’
>    70 | void intel_huc_unregister_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
>       |                                                               ~~~~~~~~~~~~~~~~~^~~
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c: In function ‘gsc_init_one’:
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:264:61: warning: passing argument 2 of ‘intel_huc_register_gsc_notifier’ disca
> rds ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
>   264 |                                                 aux_dev->dev.bus);
>       |                                                 ~~~~~~~~~~~~^~~~
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:69:78: note: expected ‘struct bus_type *’ but argument is of type ‘const st
> ruct bus_type *’
>    69 | void intel_huc_register_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
>       |                                                             ~~~~~~~~~~~~~~~~~^~~
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:271:71: warning: passing argument 2 of ‘intel_huc_unregister_gsc_notifier’ dis
> cards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
>   271 |                                                           aux_dev->dev.bus);
>       |                                                           ~~~~~~~~~~~~^~~~
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:70:80: note: expected ‘struct bus_type *’ but argument is of type ‘const st
> ruct bus_type *’
>    70 | void intel_huc_unregister_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
>       |                                                               ~~~~~~~~~~~~~~~~~^~~
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/i915_sriov_sysfs.o
[... 12 lines cut ...]
>   CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dsi_vbt.o
> /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
[... cut ...]
> make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> make[1]: *** [/usr/src/linux-headers-6.5.0-35-generic/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5/build] Error 2
> make: *** [Makefile:234: __sub-make] Error 2
> make: Leaving directory '/usr/src/linux-headers-6.5.0-35-generic'
> ```

Are you sure you're on 6.5.0-44? The logs seem to indicate you're compiling against 6.5.0-35-generic.

sheetsg77, 2024-07-16:
I tried numerous versions all the same results in 6.5.0-(35 or 41 or 44).  Back in Oct 2023, it did work with Ubuntu 23.10 (kernel 6.5.0-13 at that time) but trying to recreate that version did not work either.

I did switch to Debian, kernel 6.7.12, and it worked.

Greg

pasbec, 2024-07-19:
All the errors related to `drm_dp_calc_pbn_mode` should be gone now in Proxmox VE with the latest PRs #178 and #179.

closed 2024-07-22 by strongtz