strongtz/i915-sriov-dkms #166: Proxmox 8.2 Kernel 6.5.13-5 Error 2

Coder84619, 2024-05-05:
I have two 12th Gen Intel mini-PCs. On one I can get Kernel 6.5.13-5 compiled and it works. However, on the other I get the following two blocks of warnings/errors. I tried to clean up the system and re-try, but I get the same build failures.

Ideas?

`  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/uc/intel_uc_debugfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/uc/intel_uc_fw.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.o
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.c: In function ‘gsc_destroy_one’:
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:150:71: warning: passing argument 2 of ‘intel_huc_unregister_gsc_notifier’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  150 |                                                           aux_dev->dev.bus);
      |                                                           ~~~~~~~~~~~~^~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/uc/intel_uc.h:14,
                 from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gt_types.h:19,
                 from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_engine.h:18,
                 from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/i915_drv.h:46,
                 from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:8:
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:70:80: note: expected ‘struct bus_type *’ but argument is of type ‘const struct bus_type *’
   70 | void intel_huc_unregister_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
      |                                                               ~~~~~~~~~~~~~~~~~^~~
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.c: In function ‘gsc_init_one’:
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:264:61: warning: passing argument 2 of ‘intel_huc_register_gsc_notifier’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  264 |                                                 aux_dev->dev.bus);
      |                                                 ~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:69:78: note: expected ‘struct bus_type *’ but argument is of type ‘const struct bus_type *’
   69 | void intel_huc_register_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
      |                                                             ~~~~~~~~~~~~~~~~~^~~
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/intel_gsc.c:271:71: warning: passing argument 2 of ‘intel_huc_unregister_gsc_notifier’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  271 |                                                           aux_dev->dev.bus);
      |                                                           ~~~~~~~~~~~~^~~~
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/gt/uc/intel_huc.h:70:80: note: expected ‘struct bus_type *’ but argument is of type ‘const struct bus_type *’
   70 | void intel_huc_unregister_gsc_notifier(struct intel_huc *huc, struct bus_type *bus);
      |                                                               ~~~~~~~~~~~~~~~~~^~~
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/i915_sriov.o`

More:

`  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_link_training.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_find_vcpi_slots_for_bpp’:
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:86:43: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
   86 |                         crtc_state->pbn = drm_dp_calc_pbn_mode(adjusted_mode->crtc_clock,
      |                                           ^~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_display_types.h:36,
                 from /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:40:
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c: In function ‘intel_dp_mst_mode_valid_ctx’:
/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:906:17: error: too many arguments to function ‘drm_dp_calc_pbn_mode’
  906 |                 drm_dp_calc_pbn_mode(mode->clock, min_bpp, false) > port->full_pbn) {
      |                 ^~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:835:5: note: declared here
  835 | int drm_dp_calc_pbn_mode(int clock, int bpp);
      |     ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:251: /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.5.13-5-pve/Makefile:2039: /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build] Error 2
make: *** [Makefile:234: __sub-make] Error 2`

Coder84619, 2024-05-05:
Building module:
Cleaning build area...
make -j1 KERNELRELEASE=6.5.13-5-pve -C /lib/modules/6.5.13-5-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build KVER=6.5.13-5-pve...................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.13-5-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/make.log for more information.
i915-sriov-dkms/6.5.13-5: added
root@proxmox3:/usr/src/i915-sriov-dkms-6.5.13-5# more /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/make.log
DKMS make.log for i915-sriov-dkms-6.5.13-5 for kernel 6.5.13-5-pve (x86_64)
Sat May  4 07:35:22 PM PDT 2024

michael-pptf, 2024-05-05:
https://github.com/strongtz/i915-sriov-dkms/issues/165#issuecomment-2094350380

pasbec, 2024-07-19:
This should be working now with the latest PRs #178 and #179.

closed 2024-07-22 by strongtz