strongtz/i915-sriov-dkms #167: AUX USBC1/DDI TC1/PHY TC1: not started

bolzerrr, 2024-05-06:
I successfully configured the module, thanks for that in the first place!

Over time i get following errors. I dont have a display attached at all.
```
[ 9150.453575] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4002ff)
[ 9150.465574] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4002ff)
[ 9150.477573] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4002ff)
[ 9150.489366] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4002ff)
[ 9150.501572] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: did not complete or timeout within 10ms (status 0xad4002ff)
[ 9150.501580] i915 0000:00:02.0: [drm] *ERROR* AUX USBC1/DDI TC1/PHY TC1: not done (status 0xad4002ff)
[ 9150.537571] ------------[ cut here ]------------
[ 9150.537576] i915 0000:00:02.0: AUX USBC1/DDI TC1/PHY TC1: not started (status 0xad4002ff)
[ 9150.537599] WARNING: CPU: 1 PID: 163 at /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/drivers/gpu/drm/i915/display/intel_dp_aux.c:246 intel_dp_aux_xfer+0x60b/0x680 [i915]
[ 9150.537744] Modules linked in: rpcsec_gss_krb5 veth ebtable_filter ebtables ip_set ip6table_raw iptable_raw ip6table_filter ip6_tables iptable_filter bpfilter msr nf_tables bonding tls softdog nfnetlink_log nfnetlink binfmt_misc intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp kvm_intel i915(OE) mei_pxp mei_hdcp kvm crct10dif_pclmul polyval_clmulni polyval_generic ghash_clmulni_intel sha256_ssse3 sha1_ssse3 aesni_intel drm_buddy crypto_simd ttm cryptd cmdlinepart rapl drm_display_helper spi_nor intel_cstate pcspkr ee1004 wmi_bmof mtd cec mei_me rc_core ftdi_sio cp210x drm_kms_helper mei usbserial i2c_algo_bit acpi_pad acpi_tad mac_hid nfsd vhost_net vhost vhost_iotlb tap nct6775 nct6775_core hwmon_vid coretemp vfio_pci vfio_pci_core auth_rpcgss irqbypass nfs_acl vfio_iommu_type1 lockd vfio grace iommufd drm efi_pstore sunrpc dmi_sysfs ip_tables x_tables autofs4 zfs(PO) spl(O) btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c
[ 9150.537800]  raid1 raid0 multipath linear simplefb ax88179_178a usbnet mii hid_generic uas usbhid usb_storage hid nvme nvme_core r8169 crc32_pclmul ahci xhci_pci i2c_i801 xhci_pci_renesas spi_intel_pci nvme_common spi_intel i2c_smbus realtek libahci xhci_hcd video wmi
[ 9150.537836] CPU: 1 PID: 163 Comm: kworker/1:3 Tainted: P     U     OE      6.5.13-5-pve #1
[ 9150.537840] Hardware name: Default string N100M/N100M, BIOS 1.10 04/21/2023
[ 9150.537844] Workqueue: events i915_hotplug_work_func [i915]
[ 9150.537972] RIP: 0010:intel_dp_aux_xfer+0x60b/0x680 [i915]
[ 9150.538086] Code: 00 4c 8b 77 50 4d 85 f6 75 03 4c 8b 37 e8 3d 25 1c c7 41 89 d8 4c 89 e9 4c 89 f2 48 89 c6 48 c7 c7 c0 87 a7 c1 e8 e5 df 81 c6 <0f> 0b 48 8b 45 a8 89 98 10 06 00 00 bb f0 ff ff ff e9 ce fd ff ff
[ 9150.538091] RSP: 0018:ffffafdbc04af990 EFLAGS: 00010246
[ 9150.538094] RAX: 0000000000000000 RBX: 00000000ad4002ff RCX: 0000000000000000
[ 9150.538096] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[ 9150.538099] RBP: ffffafdbc04afa28 R08: 0000000000000000 R09: 0000000000000000
[ 9150.538101] R10: 0000000000000000 R11: 0000000000000000 R12: ffff90868cf18000
[ 9150.538103] R13: ffff90868a39b640 R14: ffff908681a8d290 R15: 0000000000000002
[ 9150.538106] FS:  0000000000000000(0000) GS:ffff908ddf880000(0000) knlGS:0000000000000000
[ 9150.538109] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 9150.538111] CR2: 00007ffcb85c0000 CR3: 000000016773e000 CR4: 0000000000752ee0
[ 9150.538114] PKRU: 55555554
[ 9150.538116] Call Trace:
[ 9150.538118]  <TASK>
[ 9150.538121]  ? show_regs+0x6d/0x80
[ 9150.538127]  ? __warn+0x89/0x160
[ 9150.538131]  ? intel_dp_aux_xfer+0x60b/0x680 [i915]
[ 9150.538244]  ? report_bug+0x17e/0x1b0
[ 9150.538248]  ? handle_bug+0x46/0x90
[ 9150.538252]  ? exc_invalid_op+0x18/0x80
[ 9150.538255]  ? asm_exc_invalid_op+0x1b/0x20
[ 9150.538259]  ? intel_dp_aux_xfer+0x60b/0x680 [i915]
[ 9150.538370]  intel_dp_aux_transfer+0x12f/0x330 [i915]
[ 9150.538480]  drm_dp_dpcd_access+0xa5/0x140 [drm_display_helper]
[ 9150.538499]  drm_dp_dpcd_probe+0x41/0x100 [drm_display_helper]
[ 9150.538516]  intel_dp_init_lttpr_and_dprx_caps+0xbe/0x2c0 [i915]
[ 9150.538626]  intel_dp_get_dpcd+0x16/0xd0 [i915]
[ 9150.538737]  intel_dp_detect+0x45b/0x750 [i915]
[ 9150.538846]  ? ww_mutex_lock+0x19/0xa0
[ 9150.538850]  drm_helper_probe_detect_ctx+0x57/0x120 [drm_kms_helper]
[ 9150.538874]  drm_helper_probe_detect+0xa0/0xc0 [drm_kms_helper]
[ 9150.538894]  intel_encoder_hotplug+0x4b/0x130 [i915]
[ 9150.539013]  intel_ddi_hotplug+0xa2/0x5a0 [i915]
[ 9150.539125]  ? __perf_event_task_sched_in+0x88/0x200
[ 9150.539130]  i915_hotplug_work_func+0x21e/0x330 [i915]
[ 9150.539248]  process_one_work+0x23b/0x450
[ 9150.539252]  worker_thread+0x50/0x3f0
[ 9150.539255]  ? __pfx_worker_thread+0x10/0x10
[ 9150.539258]  kthread+0xef/0x120
[ 9150.539261]  ? __pfx_kthread+0x10/0x10
[ 9150.539263]  ret_from_fork+0x44/0x70
[ 9150.539266]  ? __pfx_kthread+0x10/0x10
[ 9150.539268]  ret_from_fork_asm+0x1b/0x30
[ 9150.539271]  </TASK>
[ 9150.539273] ---[ end trace 0000000000000000 ]---
```

michael-pptf, 2024-05-07:
Supposedly PR #168 would fix it. 

If it is not merged into the main branch of this repo by the time you are reading this, you can go over to the committer's repo to clone it. Here's the link for your convenience.
https://github.com/shenwii/i915-sriov-dkms

bolzerrr, 2024-05-13:
Just updated and looks good so far! Thank you for the hint

closed 2024-05-13 by bolzerrr