strongtz/i915-sriov-dkms #169: Does not compile against Kernel 6.8+

ich777, 2024-05-08:
As a follow up to #168, does not compile against Kernel 6.8+

```
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:36:8: error: type defaults to 'int' in declaration of 'DEFINE_STATIC_KEY_FALSE' [-Werror=implicit-int]
   36 | static DEFINE_STATIC_KEY_FALSE(has_movntdqa);
      |        ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:36:1: warning: parameter names (without types) in function declaration
   36 | static DEFINE_STATIC_KEY_FALSE(has_movntdqa);
      | ^~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c: In function 'i915_memcpy_from_wc':
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:118:13: error: implicit declaration of function 'static_branch_likely' [-Werror=implicit-function-declaration]
  118 |         if (static_branch_likely(&has_movntdqa)) {
      |             ^~~~~~~~~~~~~~~~~~~~
  CC [M]  /usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_sw_fence_work.o
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:118:35: error: 'has_movntdqa' undeclared (first use in this function)
  118 |         if (static_branch_likely(&has_movntdqa)) {
      |                                   ^~~~~~~~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:118:35: note: each undeclared identifier is reported only once for each function it appears in
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c: In function 'i915_unaligned_memcpy_from_wc':
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:148:17: error: implicit declaration of function 'memcpy' [-Werror=implicit-function-declaration]
  148 |                 memcpy(dst, src, x);
      |                 ^~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:29:1: note: include '<string.h>' or provide a declaration of 'memcpy'
   28 | #include "i915_memcpy.h"
  +++ |+#include <string.h>
   29 | 
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:148:17: warning: incompatible implicit declaration of built-in function 'memcpy' [-Wbuiltin-declaration-mismatch]
  148 |                 memcpy(dst, src, x);
      |                 ^~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:148:17: note: include '<string.h>' or provide a declaration of 'memcpy'
  CC [M]  /usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_syncmap.o
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c: In function 'i915_memcpy_init_early':
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:165:13: error: implicit declaration of function 'static_cpu_has' [-Werror=implicit-function-declaration]
  165 |         if (static_cpu_has(X86_FEATURE_XMM4_1) &&
      |             ^~~~~~~~~~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:166:14: error: implicit declaration of function 'boot_cpu_has' [-Werror=implicit-function-declaration]
  166 |             !boot_cpu_has(X86_FEATURE_HYPERVISOR))
      |              ^~~~~~~~~~~~
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:167:17: error: implicit declaration of function 'static_branch_enable' [-Werror=implicit-function-declaration]
  167 |                 static_branch_enable(&has_movntdqa);
      |                 ^~~~~~~~~~~~~~~~~~~~
  CC [M]  /usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_user_extensions.o
/usr/src/i915-sriov-dkms/drivers/gpu/drm/i915/i915_memcpy.c:167:39: error: 'has_movntdqa' undeclared (first use in this function)
  167 |                 static_branch_enable(&has_movntdqa);
      |                                       ^~~~~~~~~~~~
```

BenSchweikert, 2024-06-14:
Hello guys, what's your strategy at the moment because with Kernels > 6.7 it is not building and strengtz does not need/maintain it any more. Do you put your "apps" from VM to Host or do you stay forever at an old kernel version or what do plan to do?

Best regards
Ben

ich777, 2024-06-14:
@BenSchweikert this comment maybe won't help you much but I personally also don't use this driver but I compile the driver packages for Unraid so that Unraid users can easily make use of it.

My current recommendation is that users stay on the latest stable Unraid version which currently users Kernel 6.1.79 and don't upgrade to the upcoming Unraid version (which will user Kernel 6.8+) if they want to still be able to use this driver.

Hopefully someone familiar with this repository will take a look at it and issue a PR.

However Intel is already working on a implementation upstream to the Kernel which should be very similar to this repository but they delay it from Kernel release to Kernel release, you can read more about that [here](https://github.com/intel/linux-intel-lts/issues/33).
I hope the official support for i915 SRIOV will be released soon...

sshaikh, 2024-06-24:
Perhaps naïve, but at this point might it not be easier to DKMS-fying the relevant i915 codebase from Intel? Similar to the work that was carried out at the start of this repo?

fanshu93, 2024-06-28:
+1 Expect 6.8 kernel support！

Daniel15, 2024-06-28:
6.8 is EOL so this should target 6.9 now.

Has anyone tried reaching out to the original author of the code at Intel to see if they can help get this running on 6.9?

Daniel15, 2024-06-28:
So I was looking into this, and it turns out Intel are adding a new graphics driver that has native SR-IOV support:

* https://www.phoronix.com/news/Intel-Xe-DRM-Linux-6.9-Pull
* https://lore.kernel.org/dri-devel/CAPj87rO4K6QS8hVn-d6N8CEi+Uibmgo6mZ5bNGz2rZDUMshvxA@mail.gmail.com/T/

This driver is for Intel Xe graphics, which includes the integrated graphics in Tiger Lake (11th gen) and newer, as well as their discrete GPUs.

What's unclear to me is:
1. Has this actually been merged yet? Is this driver ready to test in 6.9, or are more improvements needed?
2. Will it support SR-IOV in integrated GPUs or only discrete GPUs?
3. When will the driver be considered production-ready? The pull request says it's experimental for Tiger Lake (11th gen) to Meteor Lake (1st gen of Core Ultra mobile CPUs, released December 2023), and will be used as the primary driver for the next generation onwards

Daniel15, 2024-06-28:
Also, to add to the above, it looks like Intel's fork of Linux **does** have support for SR-IOV in the i915 driver in Linux v6.9 (https://github.com/intel/mainline-tracking/commit/ec2aa96813012243cf62765d42da51d880eb4b3c). Could that be merged into this repo?

ich777, 2024-06-28:
@Daniel15 they delayed it multiple times already, please follow this issue: https://github.com/intel/linux-intel-lts/issues/33

It's simply not done yet and not fully supported.

Daniel15, 2024-06-28:
> @Daniel15 they delayed it multiple times already, please follow this issue: [intel/linux-intel-lts#33](https://github.com/intel/linux-intel-lts/issues/33)
> 
> It's simply not done yet and not fully supported.

Makes sense. What about the SR-IOV code in the i915 driver in 6.9? https://github.com/intel/mainline-tracking/commit/ec2aa96813012243cf62765d42da51d880eb4b3c

> BTW, please don't do multiple posts (exactly the same) in different places, you already got a answer on the Unraid forums.

This Github repo is separate to Unraid. It doesn't make sense for me to just post in the Unraid forum as this repo is broader than that. I'm using it on a non-Unraid server too.

ich777, 2024-06-28:
> Makes sense. What about the SR-IOV code in the i915 driver in 6.9? [intel/mainline-tracking@ec2aa96](https://github.com/intel/mainline-tracking/commit/ec2aa96813012243cf62765d42da51d880eb4b3c)

IIRC it's not fully functional in Kernel 6.9, I compiled a custom Unraid version with that Kernel for a user to test, because I got no compatible hardware and ultimately it was not working.

Daniel15, 2024-06-28:
> IIRC it's not fully functional in Kernel 6.9, I compiled a custom Unraid version with that Kernel for a user to test, because I got no compatible hardware and ultimately it was not working.

@ich777 You compiled the Intel kernel fork specifically? Would you be interested in testing if someone were to send you compatible hardware?

> This is a confusing process to follow because Intel is slowly staging out this driver in phases. Kernel 6.8, 6.9, and 6.10 have all added various bits of software infrastructure that needs to be in place before the Xe driver is even in shape to be used by regular users for testing.
> 
> Compare that to other peripherals where support usually just shows up all at once in a kernel update or official/from the manufacturer DKMS, etc.

@johntdavis84 Writing a driver for something as complex as a graphics card is *hard*,  it sometimes takes a long time to get code merged into the kernel, and nobody wants to review giant pull requests, so it makes sense that they're breaking up the work into smaller, more manageable chunks.

ich777, 2024-06-28:
> @ich777 You compiled the Intel kernel fork specifically?

I compiled the Kernel which includes that changes and it doesn't work.

> Would you be interested in testing if someone were to send you compatible hardware?

Sorry, but I don't have any capacity to do more as I'm doing already.
I have about 180 Applications/Plugins on the CA App and if something is not working or broken I fix it or in the case of this plugin (which I only compile the driver for) I create Issues or sometimes PRs if it's a minor fix and have time to look at it.
Hope you understand that.

Daniel15, 2024-06-28:
Thank you @ich777. Makes sense. The whole Unraid community appreciates your work :)

pasbec, 2024-07-17:
Hi, I have pushed some modifications for kernel 6.8 into the [pve branch](https://github.com/pasbec/i915-sriov-dkms/tree/pve) in my own fork of this repo. It should work with all recent Proxmox 6.5/6.8 kernels (sorry for the partly German screenshots):

[image]
Feel free to test this but be aware that I have changed a few other things related to the config (fixed module version and Makefile changes).

Please also note that I'm no kernel developer and I really have no idea how the driver works internally. I just checked for differences and "fixed" build errors based on kernel source diffs until it compiled. The changes were tested with Proxmox kernels 6.5.13-3-pve, 6.5.13-5-pve, 6.8.4-3-pve and 6.8.8-2-pve with two Windows 11 VMs on Raptor Lake Refresh ("14th" gen) and Alder Lake (12th gen) hardware. I have not tested any 6.2 kernel due to lacking SecureBoot support. I also have no idea about any Linux guests so far - will probably start testing some some Debian guest with backport kernel soon.

ich777, 2024-07-17:
> Feel free to test it but be aware that I have [changed](https://github.com/pasbec/i915-sriov-dkms/pull/1/files) a few other things related to the config (fixed module version and Makefile changes). Any feedback is very welcome.

Create a PR, I'll try to compile it later today and try to find some testers in the Unraid community.

@Daniel15 are you interested in testing?

johntdavis84, 2024-07-17:
Thanks for your efforts on this. I’m really hopeful this makes it into the main branch.

I haven’t really run into problems on the host (Proxmox) side of things with my kernel pinned to 6.5.13-5, but the kernel restriction inside the VM is starting to cause me some issues with newer ISOs of various things.

-- -- --
John T Davis
***@***.***

Daniel15, 2024-07-17:
> Create a PR, I'll try to compile it later today and try to find some testers in the Unraid community.
>
> @Daniel15 are you interested in testing?

Sure! I don't have a test system and don't want to upgrade Unraid on my live system yet, so I'll have to see if I can run the new Unraid beta with a different USB stick.

ich777, 2024-07-17:
Working fine on Unraid 7.0.0-beta.2:
[image]

This is a version compiled from #179

closed 2024-07-17 by ich777

trustno1foxm, 2024-09-07:
I have also errors like that: Purging GPU memory, 0 pages freed, 0 pages still pinned, 227 pages left available.
Every week the Proxmox Server crashes always at night. Any ideas? Since Kernel updates these things happen.
There is a Thread about it too: https://forum.proxmox.com/threads/proxmox-reboots-after-purging-gpu-memory.142142/

thx!

so long

ich777, 2024-09-07:
> so long

I think it would be better to open up your own Issue, this Issue is already closed.