strongtz/i915-sriov-dkms #173: How to remove i915-sriov-dkms

klcuser, 2024-05-25:
Help, how to remove i915-sriov-dkms, I need restore to display output from HDMI.

pasbec, 2024-07-23:
The i915-sriov-dkms module is usually installed with two commands from the root of the cloned repository (where the `dkms.conf` file is). For example with module `version=2024.07.19` and `kernel=6.5.13-5-pve` on Proxmox VE 8.2, the commands would be: 
1. `dkms .`
2. `dkms install -m i915-sriov-dkms -v $version -k $kernel --force`

According to the [dkms manpage](https://manpages.debian.org/testing/dkms/dkms.8.en.html),
- the first command (`dkms add`) will **copy** the source from the current working directory (.) to `/usr/src/i915-sriov-dkms-$version`. It further creates an empty directory `/var/lib/dkms/i915-sriov-dkms/$version/build` and adds a symbolic link `/var/lib/dkms/i915-sriov-dkms/$version/source -> /usr/src/i915-sriov-dkms-$version`.
- The second command (`dkms install`) will try to install the module for the given kernel:
  - If the module is not already built, it will first **build the module** which we could have done seperately via `dkms build -m i915-sriov-dkms -v $version -k $kernel`. The build process will compile (and sign) the module (with the dkms machine owner key) for the selected kernel temporarily in `/var/lib/dkms/i915-sriov-dkms/$version/build` and once finished store the module in `/var/lib/dkms/i915-sriov-dkms/$version/$kernel/$(uname -m)/module/i915.ko`, where `uname -m` is the architecture like `x86_64`.
  - To **install the module**, it will then be copied from `/var/lib/dkms/i915-sriov-dkms/$version/$kernel/$(uname -m)/module/i915.ko` to **`/lib/modules/$kernel/updates/dkms/i915.ko`**. Though **the original i915 module is kept** and still exists in `/lib/modules/$kernel/kernel/drivers/gpu/drm/i915/i915.ko` However, the dkms module from `/lib/modules/$kernel/updates/dkms/i915.ko` will be *preferred* whenever the module gets reloaded and overrides the original one. [This page](https://www.suse.com/de-de/support/kb/doc/?id=000018455) nicely explains how kernel modules can be "overridden" (e.g. for testing) by storing them in the `/lib/modules/$kernel/updates` folder. The location of the currently loaded i915 module can easily be found via `modinfo -n i915`.

**Removing the i915-sriov-dkms module is therefore simple (reboot or reload the module afterwards):**
```
kernel=6.5.13-5-pve
rm /lib/modules/$kernel/updates/dkms/i915.ko
```
If access to the machine is lost due to missing graphics output, mount the root file system from initramfs (press "e" key in grub or systemd-boot and add `break=premount` to the kernel command line in order to get console access) or use some live system and remove the module file from there. If the module is already loaded in initramfs and causes issues, it can be blacklisted (press "e" key in grub or systemd-boot and add `modprobe.blacklist=i915` to the kernel command line).

**Removing any traces from the i915-sriov-dkms module is also simple:**
```
rm -rf /usr/src/i915-sriov-dkms-*
rm -rf /var/lib/dkms/i915-sriov-dkms
find /lib/modules -regex ".*/updates/dkms/i915.ko" -delete
```

**Be aware that all the paths might be different for other Linux distributions and be careful with copy-and-pasting the `rm -rf` commands!**

closed 2025-01-03 by strongtz

zmh1985, 2025-03-28:
dpkg --remove --force-remove-reinstreq intel-i915-dkms

PR #239 (2025-01-03): ci: auto build package and create release
May simplify the installation and update process.
GitHub Action will automatically create a Release when the version is bumped.

Closes #173 
Closes #200 
Closes #213 
Closes #222