strongtz/i915-sriov-dkms #174: [Report] build error on pve 6.5.13-5-pve

flyinskyin2013, 2024-05-28:
After 8a868c90d920c9ac8870bc92f1a62bba8d33e215  (Apr 26, 2024), the rep can't build on Linux pve 6.5.13-5-pve

It should need more branchs/tags for compatibility.

johntdavis84, 2024-05-28:
Hello! 

Looking at that merge request, all this should have done was keeping the build from running on an unsupported version of the Linux kernel. Supported kernels should be anything in the range of 6.1.x to 6.5.x. including 6.5.13-5-pve.

There might be an issue with the regex that checks the version number (I don't yet speak regular expressions, unfortunately).

What do you get on your Proxmox host when you run `uname -a`?

flyinskyin2013, 2024-05-30:
`uname -a` : Linux pve 6.5.13-5-pve #1 SMP PREEMPT_DYNAMIC PMX 6.5.13-5 x86_64 GNU/Linux

flyinskyin2013, 2024-05-30:
Besides, After reference https://github.com/strongtz/i915-sriov-dkms/pull/159, modify file (`drivers/gpu/drm/i915/gt/uc/abi/guc_version_abi.h`)  from `#define GUC_VF_VERSION_LATEST_MINOR	0` to `#define GUC_VF_VERSION_LATEST_MINOR	9`, then it will work well.

m0eak, 2024-06-09:
same problem

ddscentral, 2024-06-17:
There's also a bug in drivers/gpu/drm/i915/display/intel_dp_mst.c
drm_dp_calc_pbn_mode has extra parameter added for kernels under 6.6, which breaks build on 6.5.13-5-pve. Removing this extra parameter seems to fix the build. Not sure if anything else ain't broken though.

pasbec, 2024-07-16:
Changes of the kernel interfaces are not limited to different kernel versions. There may also be differences depending on distro and kernel type. E.g., for Proxmox VE (pve) the signature of drm_dp_calc_pbn_mode changed from 6.5.13-3-pve to 6.5.13-4-pve. This cannot be tested solely based on the kernel version 6.5.13. The code in this repo currently any kernel version 6.6.14 - I don't understand who pushed that and why.

[image]

I've pushed some modifications into the [pve branch](https://github.com/pasbec/i915-sriov-dkms/tree/pve)) of my own fork that seem to work on all? of the current proxmox kernel versions - including the 6.8 ones. So far I've positively tested the following kernels using two Win11 VMs in parallel on some Alder Lake N100 mini PC:
- 6.5.13-3-pve
- 6.5.13-5-pve
- 6.8.4-3-pve
- 6.8.8-2-pve
Haven't looked into Linux clients, yet.

pasbec, 2024-07-19:
The build errors should be gone now with the latest PRs #178 and #179.

closed 2024-07-22 by strongtz