strongtz/i915-sriov-dkms #175: Larger memory usage with SRIOV

sshaikh, 2024-06-05:
Arch linux VM, using latest commit (r113.3d7a1b3) and abi version fix:
```
❯ uname -a
Linux pc-office 6.6.32-1-lts #1 SMP PREEMPT_DYNAMIC Sat, 25 May 2024 20:20:51 +0000 x86_64 GNU/Linux
```
Without the module on boot:
```
❯ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.7Gi       358Mi       7.4Gi       796Ki       163Mi       7.4Gi
Swap:             0B          0B          0B
```
After starting firefox (4 tabs):
```
❯ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.7Gi       1.7Gi       5.5Gi       132Mi       893Mi       6.0Gi
Swap:             0B          0B          0B
```
With the module on boot:
```
❯ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.7Gi       369Mi       7.4Gi       1.2Mi       164Mi       7.4Gi
Swap:             0B          0B          0B
```
After starting firefox (same tabs):
```
❯ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.7Gi       2.3Gi       4.9Gi       626Mi       1.4Gi       5.4Gi
Swap:             0B          0B          0B
```
I don't believe this is an issue with FF, it just seems to exacerbate the issue. I am getting OOM issues with the module enabled with modest use.

I am accessing the headless VM using xrdp (with glamor extensions - xorg-xrdp-glamor). vGPU is detected fine and acceleration is there (for as long as apps are usable before being killed!).

Any tips on how to diagnose?

sshaikh, 2024-06-06:
The idle VM is continuing to use more memory. After 24h:

```
❯ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.7Gi       4.2Gi       2.6Gi       565Mi       1.8Gi       3.6Gi
Swap:             0B          0B          0B
```

xp499, 2024-06-15:
Hello, suffering from the same issue. Clearly it's a memory leak. Same distro, same kernel, hardware is a 14900K. Didn't tested if 6.1 lts is affected bc for now i'm sticking with 6.9 without this driver settled up.

It gets to a point that system force closes the rest of "critical" processes like xfce and lightdm for me. It resets the entire session and sometimes leaving the computer on the CLI. I've plenty more ram than you (128GiB) and over time it overrruns entire available amount.

xp499, 2024-06-15:
Didn't setup VF nor using them at all. Is just starting and using the system with this installed.

sshaikh, 2024-06-17:
Interesting - I only see it when I "install" the virtual card in the VM. Without that the system runs okay.

EricCorleone, 2024-08-05:
Same issue, the memory consumption is just keep increasing and never released.

EricCorleone, 2024-08-05:
Same issue, the memory consumption just keeps increasing and never releases.

lucker999, 2024-09-03:
Can confirm on arch lts 6.6.48-1 the ram usage steadily increases with each minute even if no apps launched. in about 5 minutes 22GB in use with only FF and konsole running. No VM or virt-manager launched as well.

bbaa-bbaa, 2024-10-02:
It would be great if anyone can test the fix #204, thanks.

vit0sd, 2024-10-03:
No leak in arch 6.11.1 after replacing drivers/gpu/drm/i915/gem/i915_gem_shmem.c with the original 6.11.1-arch1-1 version.

closed 2024-10-04 by strongtz

lucker999, 2024-10-07:
No leaks in arch 6.6.53-1 lts with new commit

PR #204 (2024-10-04): fix possible memory leak at 6.5+
Porting the missing section on [0b62af2](https://github.com/gregkh/linux/commit/0b62af28f249b9c4036a05acfb053058dc02e2e2)

May fix: #137
May fix: #175