strongtz/i915-sriov-dkms #177: Fail to load in debain12 kernel version 6.5.0 and 6.1（pve）

1402366912, 2024-07-16:
i am using q35 debain 12 6.5.0-0.deb12.4-amd64 

[    0.000000] Command line: initrd=\529e6f9543d943f9ac13a7647402bb07\6.5.0-0.deb12.4-amd64\initrd.img-6.5.0-0.deb12.4-amd64 root=UUID=110984f8-050b-4901-8423-e91d1ab28452 ro rootflags=subvol=rootfs quiet systemd.machine_id=529e6f9543d943f9ac13a7647402bb07 i915.enable_guc=3
[    0.018983] Kernel command line: initrd=\529e6f9543d943f9ac13a7647402bb07\6.5.0-0.deb12.4-amd64\initrd.img-6.5.0-0.deb12.4-amd64 root=UUID=110984f8-050b-4901-8423-e91d1ab28452 ro rootflags=subvol=rootfs quiet systemd.machine_id=529e6f9543d943f9ac13a7647402bb07 i915.enable_guc=3
[    3.646912] i915 0000:01:00.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
[    3.650742] i915 0000:01:00.0: Device initialization failed (-5)
[    3.650790] i915: probe of 0000:01:00.0 failed with error -5

 when i change the model to i440fx it ,it shows
sudo dmesg|grep i915

[ 0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.5.0-0.deb12.4-amd64 root=/dev/mapper/debian--vg-root ro quiet splash i915.enable_guc=3
[ 0.022749] Kernel command line: BOOT_IMAGE=/vmlinuz-6.5.0-0.deb12.4-amd64 root=/dev/mapper/debian--vg-root ro quiet splash i915.enable_guc=3
however The GPU was not loaded correctly
cpu：intel N100
pve kernel 6.5.13-3-pve
I have also test in 6.1

q35 :
[ 0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.1.0-22-amd64 root=/dev/mapper/debian--vg-root ro quiet splash i915.enable_guc=3
[ 0.018968] Kernel command line: BOOT_IMAGE=/vmlinuz-6.1.0-22-amd64 root=/dev/mapper/debian--vg-root ro quiet splash i915.enable_guc=3
[ 2.740477] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 2.740594] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 3.341323] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 3.341427] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 3.341583] i915 0000:01:00.0: drm_WARN_ON(s_en != 0x1)
[ 3.341610] WARNING: CPU: 1 PID: 138 at drivers/gpu/drm/i915/gt/intel_sseu.c:278 intel_sseu_info_init+0xc17/0xda0 [i915]
[ 3.341698] Modules linked in: i915(+) hid_generic usbhid hid drm_buddy video wmi i2c_algo_bit sr_mod(+) cdrom drm_display_helper sd_mod t10_pi cec ahci xhci_pci crc64_rocksoft crc64 rc_core libahci crc_t10dif crct10dif_generic ttm xhci_hcd virtio_gpu virtio_dma_buf drm_shmem_helper virtio_scsi drm_kms_helper ehci_pci virtio_net uhci_hcd net_failover failover ehci_hcd crct10dif_pclmul crct10dif_common psmouse crc32_pclmul libata drm scsi_mod virtio_pci virtio_pci_legacy_dev crc32c_intel virtio_pci_modern_dev i2c_i801 i2c_smbus scsi_common usbcore lpc_ich virtio usb_common virtio_ring floppy(+) button
[ 3.341739] RIP: 0010:intel_sseu_info_init+0xc17/0xda0 [i915]
[ 3.341845] ? intel_sseu_info_init+0xc17/0xda0 [i915]
[ 3.341927] ? intel_sseu_info_init+0xc17/0xda0 [i915]
[ 3.341995] ? intel_sseu_info_init+0xc17/0xda0 [i915]
[ 3.342064] ? fwtable_read32+0x96/0x220 [i915]
[ 3.342129] ? intel_uncore_forcewake_for_reg+0x45/0xf0 [i915]
[ 3.342191] intel_gt_init_mmio+0x1f/0x30 [i915]
[ 3.342264] i915_driver_probe+0x46b/0xe20 [i915]
[ 3.342371] i915_init+0x1f/0x7f [i915]
[ 3.943074] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 3.943180] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 4.543876] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 4.543968] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 5.144657] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 5.144751] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 5.144884] i915 0000:01:00.0: [drm] L3 bank mask is all zero!
[ 5.745450] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 5.745541] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 6.346297] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 6.346417] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 6.947378] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 6.947489] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 7.548212] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 7.548327] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 8.149056] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 8.149178] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 8.750033] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 8.750146] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 9.350856] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 9.350972] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 9.951680] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 9.951777] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 10.552488] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 10.552591] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 11.153300] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 11.153395] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 11.754085] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 11.754214] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 12.354960] [drm:fw_domains_get_with_fallback [i915]] ERROR render: timed out waiting for forcewake ack to clear.
[ 12.355069] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x20c/0x230 [i915]
[ 12.955769] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 12.955868] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[... 13 lines cut ...]
[ 15.967482] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 15.967591] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 16.568307] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 16.568407] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 17.169120] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 17.169212] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 17.769900] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 17.770005] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 18.370694] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[ 18.370787] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[ 18.971482] [drm:fw_domains_get_with_fallback [i915]] ERROR gt: timed out waiting for forcewake ack request.
[... cut ...]
[ 20.173293] i915 0000:01:00.0: [drm] ERROR Failed to map the ggtt page table
[ 20.223208] i915 0000:01:00.0: Device initialization failed (-12)
[ 20.223238] i915: probe of 0000:01:00.0 failed with error -12
and i440fx
sudo dmesg|grep i915
[ 0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.5.0-0.deb12.4-amd64 root=/dev/mapper/debian--vg-root ro quiet splash i915.enable_guc=3
[ 0.022749] Kernel command line: BOOT_IMAGE=/vmlinuz-6.5.0-0.deb12.4-amd64 root=/dev/mapper/debian--vg-root ro quiet splash i915.enable_guc=3

Does anyone have any ideas？

1402366912, 2024-07-16:
i am using sriov

pasbec, 2024-07-19:
I have tested it yesterday with a Debain 12 VM and was able to get it loaded with kernels 6.5.0-0.deb12.4, 6.6.13+bpo and also 6.7.12+bpo.
My guess is that you are missing the `firmware-linux` (`firmware-linux-free`, `firmware-linux-nonfree`) package which requires `firm`.

```bash
cat /etc/apt/sources.list.d/debian.sources
```
```txt
URIs: mirror+file:///etc/apt/mirrors/debian.list
Suites: bookworm bookworm-updates bookworm-backports
Components: main contrib non-free non-free-firmware

Types: deb deb-src
URIs: mirror+file:///etc/apt/mirrors/debian-security.list
Suites: bookworm-security
Components: main contrib non-free non-free-firmware
```

This is how I installed it (**careful with copy/paste, there are some "rm -rf"s**):
```bash
# Set desired kernel
kernel=6.5.0-0.deb12.4-amd64

# Install packages
apt -y install dkms vainfo intel-media-va-driver firmware-linux linux-headers-$kernel linux-image-$kernel

# Clone i915-sriov-dkms repo
rm -rf /root/i915-sriov-dkms
git clone https://github.com/strongtz/i915-sriov-dkms.git /root/i915-sriov-dkms

# Install i915-sriov-dkms module
cd /root/i915-sriov-dkms
rm -rf /usr/src/i915-sriov-dkms-* /var/lib/dkms/i915-sriov-dkms
dkms add .
dkms install -m i915-sriov-dkms -v 2024.07.17 -k $kernel --force

# Set i915-sriov-dkms module parameters and enable grub kernel selection
echo 'GRUB_TIMEOUT=3' > /etc/default/grub.d/15_timeout.cfg
echo 'GRUB_CMDLINE_LINUX_DEFAULT="i915.enable_guc=3"' > /etc/default/grub.d/i915.cfg

# Update grub and initramfs
update-grub
update-initramfs -u -k all

# Reboot and select the kernel from above
```

`dmesg | grep i915`:
```txt
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.0-0.deb12.4-amd64 root=UUID=8f994d1d-8ae8-4dff-aaa5-9aef4574329b ro console=tty0 console=ttyS0,115200 earlyprintk=ttyS0,115200 consoleblank=0 i915.enable_guc=3
[    0.041078] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.5.0-0.deb12.4-amd64 root=UUID=8f994d1d-8ae8-4dff-aaa5-9aef4574329b ro console=tty0 console=ttyS0,115200 earlyprintk=ttyS0,115200 consoleblank=0 i915.enable_guc=3
[    2.372766] i915: loading out-of-tree module taints kernel.
[    2.373208] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.621702] i915 0000:01:00.0: Running in SR-IOV VF mode
[    2.623003] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.9.0
[    2.624539] i915 0000:01:00.0: vgaarb: deactivate vga console
[    2.624950] i915 0000:01:00.0: [drm] Using Transparent Hugepages
[    2.626786] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.9.0
[    2.627649] i915 0000:01:00.0: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    2.628195] i915 0000:01:00.0: HuC firmware PRELOADED
[    2.631343] i915 0000:01:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.631951] i915 0000:01:00.0: [drm] PMU not supported for this GPU.
[    2.632483] [drm] Initialized i915 1.6.0 20201103 for 0000:01:00.0 on minor 0
```

`vainfo --display drm --device /dev/dri/renderD128`:
```txt
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 23.1.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP
```

azsde, 2024-10-08:
Thank you @pasbec I had issues making things work but it turns out it was a kernel issue, following your instructions worked !

woolmonkey, 2024-10-11:
@azsde any idea what you did?  I get the same error.  Do we still need to use the 6.5 kernel?

azsde, 2024-10-11:
@woolmonkey I wrote down the steps I followed here : https://gist.github.com/azsde/e92b1f219712045636cc3f8581c4542b#file-readme-md

closed 2024-12-29 by strongtz