strongtz/i915-sriov-dkms #180: [issue] pve 8.2.4 and linux 6.8.8-2-pve compiling issue

ErikBcd, 2024-07-17:
Following the pve host installation steps from the README. 
When compiling the module in step 5, I get the following errors:

```
root@proxmox1:~/Sources/i915-sriov-dkms# dkms install -m i915-sriov-dkms -v 2024.07.17 --force
Sign command: /lib/modules/6.8.8-2-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.8.8-2-pve -C /lib/modules/6.8.8-2-pve/build M=/var/lib/dkms/i915-sriov-dkms/2024.07.17/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.8-2-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/make.log for more information.
```

/var/lib/dkms/i915-sriov-dkms/2024.07.17/build/make.log: 
```
root@proxmox1:~/Sources/i915-sriov-dkms# cat /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/make.log 
DKMS make.log for i915-sriov-dkms-2024.07.17 for kernel 6.8.8-2-pve (x86_64)
Wed Jul 17 08:13:18 PM CEST 2024
make: Entering directory '/usr/src/linux-headers-6.8.8-2-pve'
KERNELRELEASE=6.8.8-2-pve
KERNELVERSION=6.8.8
EXTRAVERSION_MAJOR=2
EXTRAVERSION_MINOR=0
EXTRAVERSION_NAME=pve
EXTRAVERSION_DEFINE=EXTRAVERSION_PVE
LSBRELEASE=12
LSBRELEASE_MAJOR=12
LSBRELEASE_MINOR=0
LSBRELEASE_NAME=Debian
LSBRELEASE_DEFINE=RELEASE_DEBIAN
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_getparam.o
<command-line>: error: macro names must be identifiers
<command-line>: error: macro names must be identifiers
<command-line>: error: macro names must be identifiers
<command-line>: error: macro names must be identifiers
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_config.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_getparam.o] Error 1
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_drm_client.o] Error 1
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/i915_driver.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.8.8-2-pve/Makefile:1926: /var/lib/dkms/i915-sriov-dkms/2024.07.17/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.8-2-pve'
```

### System: 
Fresh install of Proxmox 8.2.4/Linux 6.8.8-2-pve 
Intel N100

I also tried to compile using #179 , but the error is the same. Not sure if this is related.

Any help in identifying the issue is much appreciated!

pasbec, 2024-07-17:
Hi, it's me who is to blame since I have not properly checked the Makefile before opening PR #178.
But the error `<command-line>: error: macro names must be identifiers` clearly shows me that there is an empty define.

Are you sure you are using the correction from PR #179? Also be aware that `dkms add .` will actully *copy* the repo content to `/usr/src/i915-sriov-dkms-2024.07.17/`. If you change the repo later, it won't update the latter automatically.

So you could try the following:
```bash
# Change into the root of the cloned repository
cd i915-sriov-dkms

# Only needed with #178 (without #179)
sed -i 's/DEFINE_EXTRAVERSION/EXTRAVERSION_DEFINE/' Makefile

# Remove the old copy of the repo
rm -rf /usr/src/i915-sriov-dkms-2024.07.17

# Remove the build stuff (just to be sure)
rm -rf /var/lib/dkms/i915-sriov-dkms

# Add the dkms repo again (copy to /usr/src/i915-sriov-dkms-2024.07.17)
dkms add .

# Install the dkms module again
dkms install -m i915-sriov-dkms -v 2024.07.17 --force
```

ErikBcd, 2024-07-17:
Thank you a lot for your help, I didn't properly clone your PR so it was my fault. Should've seen that when I looked into the Makefile..  
Can confirm now that #179 fixes everything!

closed 2024-07-17 by ErikBcd