strongtz/i915-sriov-dkms #182: Kernel NULL pointer dereference with Kernel 6.8

makoONE, 2024-07-20:
I am delighted that kernel 6.8 is now supported.
Unfortunately, however, freezes occur after a short time under Proxmox with kernel 6.8.8.3-pve. The error entry with dmesg is:

[ 800.513559] BUG: kernel NULL pointer dereference, address: 000000000000057d
[800.513567] #PF: supervisor read access in kernel mode
[ 800.513569] #PF: error_code(0x0000) - not-present page

It would be great if the error could be fixed, thanks.

johntdavis84, 2024-07-21:
Is anyone else experiencing this problem?

Is it a reason not to move to 6.8 yet?

-- -- --
John T Davis
***@***.***

pasbec, 2024-07-21:
I'm running the module with the 6.8.8-2-pve kernel without problems so far on Raptorlake Refresh hardware (ASUS W680 + i9-14900K) and also on some Beelink Mini EQ 12 (Alderlake N100). In the meantime, even all seven VFs were in use at the same time on the large machine for testing multiple Windows RDP sessions with 4K video decoding and 3D, Debian 12 and Ubuntu 22.04 with VA API and 3D acceleration in parallel. But maybe I was just lucky. 🫤 

This repo is currently based on [lts-v6.1.26-linux-230504T201607Z](https://github.com/intel/linux-intel-lts/tree/lts-v6.1.26-linux-230504T201607Z/drivers/gpu/drm/i915) from [linux-intel-lts](https://github.com/intel/linux-intel-lts) which [was merged](https://github.com/strongtz/i915-sriov-dkms/commit/16f413b45941e9ac5cf662537e1f598f6690d161) by @zhtengw on May 8, 2023. Afterwards there seem to be mostly build fixes. Looking at the diff (`git diff --stat lts-v6.1.26-linux-230504T201607Z lts-v6.1.95-linux-240708T112901Z drivers/gpu/drm/i915`) between [lts-v6.1.26-linux-230504T201607Z](https://github.com/intel/linux-intel-lts/tree/lts-v6.1.26-linux-230504T201607Z/drivers/gpu/drm/i915) and the latest [lts-v6.1.95-linux-240708T112901Z](https://github.com/intel/linux-intel-lts/tree/lts-v6.1.95-linux-240708T112901Z/drivers/gpu/drm/i915), there is a whopping number of 440 files changed, 32882 insertions(+), 20631 deletions(-). The [commit history](https://github.com/intel/linux-intel-lts/commits/lts-v6.1.95-linux-240708T112901Z/drivers/gpu/drm/i915?after=8d688d675795b3c6f21dec799f8042c8b8ed9c79+69) also contains a noticable amount of "fixes". If someone would have the time, and would be both *able* and *willing* to merge those changes into this repo, chances would likely be high that this addresses your issue, @makeONE. I don't know how much effort it would require to do this.

I've really just recently decied to give Intel SR-IOV a try (it was on the todo list for at least a yer 😄) and did the attempt to fix the build for 6.8 kernels only since I feared that this repo might not get updated anymore ([@strongtz seems to have stopped using it](https://github.com/strongtz/i915-sriov-dkms/issues/140#issuecomment-2019183047)). As mentioned in PR #178, I'm also not a kernel/drm developer and lack any experience in debugging kernel modules efficiently, sorry. But even some senior veteran drm developer would have to reproduce/understand such issues a bit more in detail (more logs, stacktrace, ...).

Just FYI (see below), there is already another issue on my hardware with some out of bounds (OOB) error on **Ubuntu 24.04 guest VM** and kernel 6.8 if the module is initialized as *minor 0* instead of *minor 1*. It is also worth to mention that some memory leaks have been reported in GH-175.

Given that Intel is already implementing [SR-IOV support into its new xe driver](https://github.com/torvalds/linux/commits/master/drivers/gpu/drm/xe) maybe in kernel 6.12 (or later)
- https://www.phoronix.com/news/Intel-Xe-DRM-Linux-6.9-Pull
- https://www.phoronix.com/news/Linux-6.11-DRM-Intel-Xe-Next

this repo might need to survive only for a few more months, *I guess* (Intel delayed the release for quite some time now but Arrow Lake will be released in Q4 2024).

So is it really worth merging [lts-v6.1.95-linux-240708T112901Z](https://github.com/intel/linux-intel-lts/tree/lts-v6.1.95-linux-240708T112901Z/drivers/gpu/drm/i915)? 

<details>
<summary>OOB error on Ubuntu 24.04 (click to expand)</summary>

```
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-35-generic root=UUID=9fd8e454-829a-44fd-9646-0d3ff78d763f ro i915.enable_guc=3
[    0.011671] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0-35-generic root=UUID=9fd8e454-829a-44fd-9646-0d3ff78d763f ro i915.enable_guc=3
[    0.374380] integrity: Loaded X.509 cert 'i915-sriov-07 Secure Boot Module Signature key: fc587d97a7953cb6be3d0df197c7049c60131607'
[    1.055948] systemd[1]: Hostname set to <i915-sriov-07>.
               use xe.force_probe='a780' and i915.force_probe='!a780'
[    1.695047] i915: loading out-of-tree module taints kernel.
[    1.917402] i915 0000:01:00.0: Running in SR-IOV VF mode
[    1.918023] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.9.0
[    1.918390] i915 0000:01:00.0: [drm] Using Transparent Hugepages
[    1.920489] i915 0000:01:00.0: [drm] GT0: GUC: interface version 0.1.9.0
[    1.921855] i915 0000:01:00.0: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    1.921859] i915 0000:01:00.0: HuC firmware PRELOADED
[    1.925762] i915 0000:01:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    1.925771] i915 0000:01:00.0: [drm] PMU not supported for this GPU.
[    1.925941] [drm] Initialized i915 1.6.0 20201103 for 0000:01:00.0 on minor 0
[    3.791062] UBSAN: array-index-out-of-bounds in /var/lib/dkms/i915-sriov-dkms/2024.07.17/build/drivers/gpu/drm/i915/gem/i915_gem_context.c:769:41
[    3.791066] index 1 is out of range for type 'i915_engine_class_instance [*]'
[    3.791089]  set_proto_ctx_param+0x741/0x760 [i915]
[    3.791195]  create_setparam+0x5e/0xa0 [i915]
[    3.791275]  i915_user_extensions+0xbb/0x170 [i915]
[    3.791345]  i915_gem_context_create_ioctl+0x1d0/0x300 [i915]
[    3.791420]  ? __pfx_i915_gem_context_create_ioctl+0x10/0x10 [i915]
[    3.791501]  ? __pfx_i915_gem_context_create_ioctl+0x10/0x10 [i915]
```
</details>

makoONE, 2024-07-21:
I have the said freezes by kernel null pointer dereference also with the Proxmox kernel 6.8.8-2-pve and 6.8.4-3-pve in connection with a Win11 VM and the vGPU usage.
The same VM runs otherwise with the Proxmox 6.5 kernel without any problems.

pasbec, 2024-07-21:
I sadly won't be able to help you with debugging/updating the module. @JTR-Tech and I are using the same [UHD Graphics](https://en.wikipedia.org/wiki/List_of_Intel_graphics_processing_units#Gen12) with device ID A780. The issue might be related to your UHD Graphics with device ID 46A3. But it could also be other effects like memory limits or the VM driver. I'm using the latest driver [32.0.101.5762](https://www.intel.com/content/www/us/en/download/785597/intel-arc-iris-xe-graphics-windows.html) for the Windows VMs with hardware settings as following:
[image]

makoONE, 2024-07-21:
My VM configuration is largely the same and I am also using the latest Intel Graphics Driver 32.0.101.5762 in the Win11 VM.

I just found out that the current i915-sriov-dkms release 2024.07.19 does not work with a 6.5 kernel (6.5.13-5-pve) anymore, same error kernel null pointer dereference.

Probably there is no other choice but to go back to i915-sriov-dkms release before 2024.07.17, i.e. without kernel 6.8 support.

pasbec, 2024-07-21:
Before 2024.07.17, kernel 6.5.13-5-pve was not working due to build issues. It was impossible to build and only 6.5.13-3-pve compiled. The build issues with 6.5.13-5-pve have been fixed with #178 / #179. I have tested and was running *all* kernels 6.5.13-3-pve, 6.5.13-5-pve and 6.8.4-3-pve and 6.8.8-2-pve without issues afterwards.

Also please note that there was *no functional change* for Proxmox with #178 and #179 except for the firmware version. The only thing that changed was header includes for kernel 6.8.* and a fix for 6.5.13-5-pve that would have prevented building the module anyway.

It is technically impossible that the module change from 6.1 to 2024.07.17 or 2024.07.19 has changed anything with the 6.5.13-3-pve kernel or the 6.5.13-5-pve kernel (which was broken anyway) *unless it is related to the firmware version*.

What is the output of `dmesg | grep i915` after boot?

makoONE, 2024-07-21:
Output of `dmesg | grep i915` after boot with current state of i915-sriov-dkms and kernel 6.8.8-3-pve:

[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.8.8-3-pve root=/dev/mapper/nab6--vg-root ro quiet mitigations=off intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 cpufreq.default_governor=powersave
[    0.101347] Kernel command line: BOOT_IMAGE=/vmlinuz-6.8.8-3-pve root=/dev/mapper/nab6--vg-root ro quiet mitigations=off intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7 cpufreq.default_governor=powersave
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[    1.681479] i915: loading out-of-tree module taints kernel.
[    1.681510] i915: module verification failed: signature and/or required key missing - tainting kernel
[    1.857258] i915 0000:00:02.0: Running in SR-IOV PF mode
[    1.857828] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.875918] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.875967] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.876334] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.877907] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    1.886258] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.20.0
[    1.886261] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    1.900528] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    1.901378] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    1.901379] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    1.901804] i915 0000:00:02.0: [drm] GuC RC: enabled
[    1.902284] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    1.932033] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[    1.933160] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    1.966144] fbcon: i915drmfb (fb0) is primary device
[    1.966148] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    3.999089] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    6.016961] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[    6.017020] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    6.017033] i915 0000:00:02.1: Running in SR-IOV VF mode
[    6.017463] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[    6.018848] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    6.018875] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    6.019215] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[    6.019992] i915 0000:00:02.1: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    6.019993] i915 0000:00:02.1: HuC firmware PRELOADED
[    6.022863] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.022866] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    6.023043] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[    6.023230] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[    6.023233] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[    6.023280] i915 0000:00:02.2: enabling device (0000 -> 0002)
[    6.023288] i915 0000:00:02.2: Running in SR-IOV VF mode
[    6.023761] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[    6.024884] i915 0000:00:02.2: [drm] VT-d active for gfx access
[    6.024895] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[    6.025180] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[    6.025752] i915 0000:00:02.2: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    6.025754] i915 0000:00:02.2: HuC firmware PRELOADED
[    6.027423] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.027426] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[    6.027574] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[    6.027801] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[    6.027804] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[    6.027806] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46a3' and i915.force_probe='!46a3'
[    6.027852] i915 0000:00:02.3: enabling device (0000 -> 0002)
[    6.027864] i915 0000:00:02.3: Running in SR-IOV VF mode
[    6.027976] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.9.0
[    6.028215] i915 0000:00:02.3: [drm] VT-d active for gfx access
[    6.028226] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[    6.028476] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.9.0
[    6.028844] i915 0000:00:02.3: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    6.028845] i915 0000:00:02.3: HuC firmware PRELOADED
[    6.030471] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.030475] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[    6.030630] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[... cut ...]
[    6.039221] i915 0000:00:02.7: [drm] GT0: GUC: interface version 0.1.9.0
[    6.039536] i915 0000:00:02.7: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[    6.039538] i915 0000:00:02.7: HuC firmware PRELOADED
[    6.040794] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.040798] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[    6.040853] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.7 on minor 7
[    6.041004] i915 0000:00:02.0: Enabled 7 VFs

pasbec, 2024-07-21:
If you want to test the previously allowed guc firmware minor version (0 or 4) instead of the corrected minor version 9. You could just run the follwing in the root of your the i915-sriov-dkms repo:

```
sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-0/' Makefile
rm -rf /usr/src/i915-sriov-dkms-* /var/lib/dkms/i915-sriov-dkms
dkms add .
dkms install -m i915-sriov-dkms -v $(cat VERSION) -k 6.8.8-2-pve --force
```

This will change `GUCFIRMWARE_MINOR=9` to `GUCFIRMWARE_MINOR=0` in the Makefile.

After successful compilation and reboot the `dmesg | grep i915` should then contain error messages like `*ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)` and `*ERROR* GT0: IOV: Found interface version 0.1.9.0` (see below) but interestingly the VFs still seem to work in Windows. It seems those errors could just be ignored and that was the reason why it worked before (confiremd by using `vainfo` and `intel_gpu_top -d sriov` on the host). 🤔 Could you kindly check if that fixes your freezes?

<details>
<summary>dmesg with errors due to firmware mismatch (click to expand)</summary>

```
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[    7.612985] i915 0000:00:02.0: Running in SR-IOV PF mode
[    7.615514] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    7.632422] i915 0000:00:02.0: vgaarb: deactivate vga console
[    7.632592] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    7.633162] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    7.641523] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    7.648210] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[    7.648229] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    7.676756] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    7.679104] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    7.679108] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    7.679683] i915 0000:00:02.0: [drm] GuC RC: enabled
[    7.681998] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.712463] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[    7.715226] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    7.743274] fbcon: i915drmfb (fb0) is primary device
[    7.819351] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   11.093865] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[   11.102209] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   11.103637] i915 0000:00:02.1: Running in SR-IOV VF mode
[   11.105737] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   11.107389] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   11.109574] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   11.111043] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   11.113438] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   11.115044] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   11.117154] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   11.118546] i915 0000:00:02.1: HuC firmware PRELOADED
[   11.123146] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.124415] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   11.126586] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[   11.134819] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   11.136053] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[   11.143573] i915 0000:00:02.2: enabling device (0000 -> 0002)
[   11.144852] i915 0000:00:02.2: Running in SR-IOV VF mode
[   11.146469] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   11.147856] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   11.150315] i915 0000:00:02.2: [drm] VT-d active for gfx access
[   11.151587] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[   11.153667] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[   11.155077] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   11.157371] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   11.158598] i915 0000:00:02.2: HuC firmware PRELOADED
[   11.162428] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.163650] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[   11.166127] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[   11.174452] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   11.175651] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[   11.176838] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[... 70 lines cut ...]
[... cut ...]
[   11.326937] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0
[   11.328272] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   11.329207] i915 0000:00:02.7: HuC firmware PRELOADED
[   11.332430] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.333311] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[   11.334300] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.7 on minor 7
[   11.336020] i915 0000:00:02.0: Enabled 7 VFs
```

<details>

makoONE, 2024-07-21:
I followed your advice with the modified minor version, dmesg says:

[ 5.863414] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.9 (0000000000000000)
[5.863572] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.9.0

Unfortunately the machine continues to freeze.

[86.393495] BUG: kernel NULL pointer dereference, address: 000000000000057d
[86.393502] #PF: supervisor read access in kernel mode
[ 86.393503] #PF: error_code(0x0000) - not-present page

MiKing233, 2024-07-21:
my miniWorkStation (Minisforum MS-01 - i9-13900H), on Proxmox VE 8.2.4 x86_64 (6.8.8-3-pve), using version 2024.07.19, I watch the Youtube 8K video for 10 minutes and the GPU never freezes, I checked dmesg without any errors, its works perfectly fine.
[image]

pasbec, 2024-07-21:
> I followed your advice with the modified minor version, dmesg says:
>
> [ 5.863414] i915 0000:00:02.3: [drm] _ERROR_ GT0: IOV: Unable to confirm version 1.9 (0000000000000000) [5.863572] i915 [...]

I'm running out of ideas but if even 6.5.13-3-pve is not working with `sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-0/' Makefile`. Below is a comparison of i915-sriov-dkms 6.1 vs. 2024.07.19 for Proxmox VE:

### i915-sriov-dkms 6.1
Available via `git reset --hard 42b49ff`
#### Minor GuC firmware check
**0** or **4**, incorrect, hardcoded
#### Kernels
- *6.5.13-**3**-pve:* **compiled**
- *6.5.13-**5**-pve:* did not compile
- *6.8.4-3-pve:* did not compile
- *6.8.8-2-pve:* did not compile

### i915-sriov-dkms 2024.07.19
Available via `git reset --hard 21a2f4a1`
#### Minor GuC firmware check
**9**, correct, can be changed via `GUCFIRMWARE_MINOR` e.g., `sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-0/' Makefile`
#### Kernels
- *6.5.13-**3**-pve:* compiles, **same binary as i915-sriov-dkms 6.1 for *any* incorrect firmware version != 9**
- *6.5.13-**5**-pve:* compiles
- *6.8.4-3-pve:* compiles
- *6.8.8-2-pve:* compiles

So either your freezes are related to the 6.8 kernel or to the corrected firmware version.

But the i915-sriov-dkms module version 2024.07.19 with `sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-0/' Makefile` will compile effectively the same binary for 6.5.13-**3**-pve which was the only working version before all the latest changes since 6.1.

If you still have freezes with this combination and it was working before with 6.1, it *must* be related to something else.

pasbec, 2024-07-23:
I have succesfully finished a 24 hour 3D + 4k video test running in 2 Windows 11 VMs in parallel without issues on my Promox VE 8.2 host with kernel 6.8.8-2-pve (I don't want to see the next electricity bill). And the host with an ASUS W680 + i9-14900K is running absolutely fine for 6 days now. Really no problems at all. It needs some new BIOS [thanks to Intel](https://www.youtube.com/watch?v=wkrOYfmXhIc) and must therefore be rebooted now.

My dmesg got filled with a lot of messages like `i915 0000:00:02.0: VF{i} FLR` in the meantime. That seems to be some Function Level Reset related to SR-IOV. Not sure if this is [OK](https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3?permalink_comment_id=4714186#gistcomment-4714186) but at least I could not see any effect.

The small N100 is also running with the loaded module in kernel 6.8.8-2 for several days now but it was just idling without any VF in use.

Not sure why this is different for my two machines. I'm running both machines with the kernel command line potion `split_lock_detect=off i915.enable_fbc=1 i915.enable_guc=3 i915.max_vfs=7 ignore_msrs=1 report_ignored_msrs=0` - not sure if some options are related? At least the i9-13900H should be very similar to mine. Don't know, sorry.

I will post an update whenever I encounter any kind of issue with the 6.8 kernel.

[image]

[image]

[image]

<details>
<summary>dmesg (shortened)</summary>

```
[    0.000000] Kernel is locked down from EFI Secure Boot mode; see man kernel_lockdown.7
[    0.222775] Booting paravirtualized kernel on bare hardware
[    0.224356] Unknown kernel command line parameters "BOOT_IMAGE=/vmlinuz-6.8.8-2-pve boot=zfs split_lock_detect=off", will be passed to user space.
[    0.706592] iommu: Default domain type: Passthrough (set via kernel command line)
               use xe.force_probe='a780' and i915.force_probe='!a780'
[  123.042157] i915 0000:00:02.0: Running in SR-IOV PF mode
[  123.043176] i915 0000:00:02.0: [drm] VT-d active for gfx access
[  123.043215] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[  123.043927] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[  123.044397] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[  123.051438] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[  123.055207] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[  123.055210] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[  123.057969] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[  123.058344] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[  123.058345] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[  123.058859] i915 0000:00:02.0: [drm] GuC RC: enabled
[  123.059378] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[  123.059452] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[  123.059967] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[  123.061022] i915 0000:00:02.0: 7 VFs could be associated with this PF
[  123.061440] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[  123.061453] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[  123.061662] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[  124.416831] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
               use xe.force_probe='a780' and i915.force_probe='!a780'
[  124.417046] i915 0000:00:02.1: enabling device (0000 -> 0002)
[  124.417099] i915 0000:00:02.1: Running in SR-IOV VF mode
[  124.418216] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[  124.420238] i915 0000:00:02.1: [drm] VT-d active for gfx access
[  124.420311] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[  124.421069] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[  124.421577] i915 0000:00:02.1: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[  124.421586] i915 0000:00:02.1: HuC firmware PRELOADED
[  124.424119] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[  124.424132] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[  124.424507] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 2
[  124.425213] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[  124.425232] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='a780' and i915.force_probe='!a780'
[  124.425416] i915 0000:00:02.2: enabling device (0000 -> 0002)
[  124.425596] i915 0000:00:02.2: Running in SR-IOV VF mode
[  124.426237] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[  124.428463] i915 0000:00:02.2: [drm] VT-d active for gfx access
[  124.428572] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[  124.429665] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[  124.430379] i915 0000:00:02.2: GuC firmware PRELOADED version 1.9 submission:SR-IOV VF
[  124.430395] i915 0000:00:02.2: HuC firmware PRELOADED
[  124.433390] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[  124.433415] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[... cut ...]
[279494.605572] i915 0000:00:02.0: VF7 FLR
[308266.377520] i915 0000:00:02.0: VF7 FLR
[317529.066836] i915 0000:00:02.0: VF3 FLR
[317530.838814] i915 0000:00:02.0: VF3 FLR
[317530.961348] i915 0000:00:02.0: VF3 FLR
[317531.307410] i915 0000:00:02.0: VF4 FLR
[317533.080184] i915 0000:00:02.0: VF4 FLR
[317533.197102] i915 0000:00:02.0: VF4 FLR
[317537.107788] i915 0000:00:02.0: VF3 FLR
[317539.345480] i915 0000:00:02.0: VF4 FLR
[317562.808160] i915 0000:00:02.0: VF4 FLR
[368277.136137] i915 0000:00:02.0: VF1 FLR
```

<details>

johntdavis84, 2024-07-24:
Thanks for taking the time to test this and post an update. I’m really confident to try making the switch to 6.8 this weekend.
(I need to make sure the latest version of kernel 6.8.x is actually installed correctly on Proxmox now. Since I’ve had 6.5.13-5 pinned, I get scary warnings every time it does a system update about a dpkg-configure failure.)

“My dmesg got filled with a lot of messages like i915 0000:00:02.0: VF{i} FLR in the meantime. That seems to be some Function Level Reset related to SR-IOV. Not sure if this is OK <https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3?permalink_comment_id=4714186#gistcomment-4714186> but at least I could not see any effect.”

I see these all the time on a working Proxmox 8.2.x (6.5.13-5)-based install on an HP Elite Mini 600 G9. It isn’t associated with any sort of performance glitches or issues on my system. I’ve just been ignoring it.

-- -- --
John T Davis
***@***.***

makoONE, 2024-07-26:
I carried out further tests with my previous setup and came to the following conclusion:
With kernel 6.2.16-20-pve it runs unconditionally stable. With kernel 6.5.13-5-pve and various 6.8.x-pve kernels, freezes occur early or late.
Through pasbec's last post (thanks for that) I became aware of the kernel command line options “split_lock_detect=off ignore_msrs=1 report_ignored_msrs=0”, which I had not used before.
I included them and after that it has been running stable with the current kernel 6.8.8-3-pve for hours.
However, I now see lots of dmesg entries like this:

[  527.228951] ------------[ cut here ]------------
[  527.228953] i915 0000:00:02.0: drm_WARN_ON(plane_state->ggtt_vma)
[  527.228989] WARNING: CPU: 2 PID: 130 at /var/lib/dkms/i915-sriov-dkms/2024.07.19/build/drivers/gpu/drm/i915/display/intel_atomic_plane.c:135 intel_plane_destroy_state+0x93/0xe0 [i915]
[  527.229097] Modules linked in: veth vfio_pci vfio_pci_core vfio_iommu_type1 vfio iommufd dm_snapshot joydev input_leds hid_apple hid_generic usbkbd usbmouse usbhid hid apple_mfi_fastcharge ebtable_filter ebtables ip_set ip6table_raw iptable_raw ip6table_filter ip6_tables iptable_filter nf_tables qrtr softdog sunrpc binfmt_misc bonding tls nfnetlink_log nfnetlink intel_rapl_msr intel_rapl_common intel_uncore_frequency intel_uncore_frequency_common x86_pkg_temp_thermal intel_powerclamp coretemp snd_sof_pci_intel_tgl snd_sof_intel_hda_common snd_hda_codec_hdmi soundwire_intel snd_sof_intel_hda_mlink kvm_intel soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp kvm snd_sof snd_sof_utils snd_soc_hdac_hda snd_hda_ext_core irqbypass mt7921e snd_soc_acpi_intel_match crct10dif_pclmul mt7921_common snd_soc_acpi polyval_clmulni soundwire_generic_allocation polyval_generic ghash_clmulni_intel mt792x_lib soundwire_bus sha256_ssse3 mt76_connac_lib sha1_ssse3 snd_soc_core aesni_intel snd_hda_codec_realtek mt76
[  527.229127]  snd_compress ac97_bus crypto_simd snd_hda_codec_generic cryptd snd_pcm_dmaengine snd_hda_intel mac80211 snd_intel_dspcfg snd_intel_sdw_acpi btusb snd_hda_codec btrtl btintel snd_hda_core btbcm snd_hwdep btmtk cmdlinepart snd_pcm cfg80211 spi_nor snd_timer rapl bluetooth pcspkr intel_cstate wmi_bmof libarc4 mei_me snd ecdh_generic mtd ecc ee1004 soundcore mei mei_vsc_hw intel_pmc_core intel_vsec pmt_telemetry pmt_class acpi_tad acpi_pad mac_hid zfs(PO) spl(O) vhost_net vhost vhost_iotlb tap parport_pc ppdev lp parport efi_pstore dmi_sysfs ip_tables x_tables autofs4 btrfs blake2b_generic xor raid6_pq i915(OE) dm_thin_pool dm_persistent_data dm_bio_prison dm_bufio libcrc32c xe drm_gpuvm drm_exec gpu_sched drm_buddy i2c_algo_bit drm_suballoc_helper drm_ttm_helper ttm drm_display_helper xhci_pci nvme xhci_pci_renesas crc32_pclmul nvme_core i2c_i801 spi_intel_pci xhci_hcd ahci cec igc spi_intel i2c_smbus libahci nvme_auth rc_core video wmi
[  527.229175] CPU: 2 PID: 130 Comm: kworker/2:1 Tainted: P     U  W  OE      6.8.8-3-pve #1
[  527.229177] Hardware name: Micro Computer(HK) Tech Limited Venus series/AHBNB, BIOS 1.0S 05/11/2024
[  527.229178] Workqueue: events intel_atomic_helper_free_state_worker [i915]
[  527.229264] RIP: 0010:intel_plane_destroy_state+0x93/0xe0 [i915]
[  527.229363] Code: 4c 8b 6f 50 4d 85 ed 75 03 4c 8b 2f e8 96 2d 63 c6 48 c7 c1 d8 9b d0 c0 4c 89 ea 48 c7 c7 ac 94 d3 c0 48 89 c6 e8 8d 54 c4 c5 <0f> 0b 48 83 bb e0 00 00 00 00 74 89 49 8b 04 24 48 8b 78 08 4c 8b
[  527.229365] RSP: 0018:ffffbd948058fd88 EFLAGS: 00010246
[  527.229366] RAX: 0000000000000000 RBX: ffff9f722a751400 RCX: 0000000000000000
[  527.229367] RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
[  527.229367] RBP: ffffbd948058fda0 R08: 0000000000000000 R09: 0000000000000000
[  527.229368] R10: 0000000000000000 R11: 0000000000000000 R12: ffff9f722911f000
[  527.229369] R13: ffff9f72022124f0 R14: 0000000000000005 R15: 0000000000000004
[  527.229369] FS:  0000000000000000(0000) GS:ffff9f816f500000(0000) knlGS:0000000000000000
[  527.229370] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  527.229371] CR2: 00007e7956b96028 CR3: 0000000116438000 CR4: 0000000000f52ef0
[  527.229372] PKRU: 55555554
[  527.229372] Call Trace:
[  527.229373]  <TASK>
[  527.229375]  ? show_regs+0x6d/0x80
[  527.229381]  ? __warn+0x89/0x160
[  527.229383]  ? intel_plane_destroy_state+0x93/0xe0 [i915]
[  527.229470]  ? report_bug+0x17e/0x1b0
[  527.229473]  ? handle_bug+0x46/0x90
[  527.229476]  ? exc_invalid_op+0x18/0x80
[  527.229478]  ? asm_exc_invalid_op+0x1b/0x20
[  527.229481]  ? intel_plane_destroy_state+0x93/0xe0 [i915]
[  527.229569]  ? intel_plane_destroy_state+0x93/0xe0 [i915]
[  527.229660]  drm_atomic_state_default_clear+0x1d7/0x310
[  527.229663]  intel_atomic_state_clear+0x12/0x40 [i915]
[  527.229755]  __drm_atomic_state_free+0x26/0xe0
[  527.229757]  intel_atomic_helper_free_state_worker+0x5b/0x90 [i915]
[  527.229858]  process_one_work+0x16a/0x350
[  527.229862]  worker_thread+0x306/0x440
[  527.229864]  ? __pfx_worker_thread+0x10/0x10
[  527.229866]  kthread+0xef/0x120
[  527.229868]  ? __pfx_kthread+0x10/0x10
[  527.229869]  ret_from_fork+0x44/0x70
[  527.229871]  ? __pfx_kthread+0x10/0x10
[  527.229872]  ret_from_fork_asm+0x1b/0x30
[  527.229875]  </TASK>
[  527.229875] ---[ end trace 0000000000000000 ]---

Is it possible to find out the cause of this or what is the best way to fix the errors?

makoONE, 2024-07-26:
Oh what a pity, a few hours later another freeze with kernel 6.8.8-3-pve.
So keep searching and hoping...

pasbec, 2024-07-26:
Sorry to hear that. You mentioned initially that kernel 6.5 has also been working before. Maybe it is worth to check if kernel 6.5.13-**3**-pve is working stable with the latest versions 2024.07.19 or 2024.07.24 (haven't tried the latter) of the dkms module. If not, you should test kernel 6.5.13-3-pve with the old version `git reset --hard 42b49ff` 6.1 **of the dkms module** again (make sure to delete /usr/src/i915-sriov-dkms before re-adding. If kernel 6.5.13-3-pve is freezing and it has not done that before, it must be something else.

mag1024, 2024-09-05:
I too am unable to get a 12th Gen CPU to work with a 6.8.12 kernel, whereas 6.5 used to work fine.
One interesting thing I noticed is that even if uninstall this module -- which seems to work because I get a message about 'max_vfs' being an unknown parameter -- I still get the NULL pointer deref.

<details><summary>dmsg snippet</summary><p>

```
[    4.314982] Setting dangerous option enable_guc - tainting kernel
[    4.314984] i915: unknown parameter 'max_vfs' ignored
[    4.316075] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.316079] BUG: kernel NULL pointer dereference, address: 0000000000000008
[    4.316082] fbcon: Taking over console
[    4.316084] #PF: supervisor read access in kernel mode
[    4.316086] #PF: error_code(0x0000) - not-present page
[    4.316087] PGD 0 P4D 0
[    4.316088] Oops: 0000 [#1] PREEMPT SMP NOPTI
[    4.316090] CPU: 12 PID: 896 Comm: (udev-worker) Tainted: P     U     O       6.8.12-1-pve #1
[    4.316092] Hardware name: ASUSTeK COMPUTER INC. System Product Name/Pro WS W680-ACE IPMI, BIOS 3802 08/08/2024
[    4.316094] RIP: 0010:kernfs_find_and_get_ns+0x16/0x80
[    4.316097] Code: ac fd ff ff 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 0f 1f 44 00 00 55 48 89 e5 41 56 49 89 d6 41 55 49 89 f5 41 54 53 <48> 8b 47 08 48 89 fb 48 85 c0 48 0f 44 c7 4c 8b 60 50 49 83 c4 60
[    4.316099] RSP: 0018:ffffb0a00090b488 EFLAGS: 00010246
[    4.316101] RAX: 0000000000000000 RBX: ffffffff830f1d20 RCX: 0000000000000000
[    4.316102] RDX: 0000000000000000 RSI: ffffffff830f1e68 RDI: 0000000000000000
[    4.316103] RBP: ffffb0a00090b4a8 R08: 0000000000000000 R09: 0000000000000000
[    4.316104] R10: 0000000000000000 R11: 0000000000000000 R12: ffff9d2dc32a2090
[    4.316105] R13: ffffffff830f1e68 R14: 0000000000000000 R15: ffffffff84062520
[    4.316106] FS:  00007664d0f6a8c0(0000) GS:ffff9d3cff600000(0000) knlGS:0000000000000000
[    4.316108] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    4.316109] CR2: 0000000000000008 CR3: 0000000105a84002 CR4: 0000000000f70ef0
[    4.316110] PKRU: 55555554
[    4.316111] Call Trace:
[    4.316113]  <TASK>
[    4.316114]  ? show_regs+0x6d/0x80
[    4.316117]  ? __die+0x24/0x80
[    4.316119]  ? page_fault_oops+0x176/0x4f0
[    4.316122]  ? do_user_addr_fault+0x2ed/0x650
[    4.316123]  ? irq_work_queue+0x2f/0x70
[    4.316126]  ? exc_page_fault+0x83/0x1b0
[    4.316129]  ? asm_exc_page_fault+0x27/0x30
[    4.316132]  ? kernfs_find_and_get_ns+0x16/0x80
[    4.316134]  sysfs_unmerge_group+0x1d/0x70
[    4.316136]  dpm_sysfs_remove+0x2b/0x70
[    4.316139]  device_del+0xa2/0x3e0
[    4.316141]  platform_device_del.part.0+0x18/0x80
[    4.316143]  platform_device_unregister+0x20/0x40
[    4.316145]  sysfb_disable+0x2f/0x90
[    4.316148]  aperture_remove_conflicting_pci_devices+0x9c/0xb0
[    4.316150]  drm_aperture_remove_conflicting_pci_framebuffers+0x15/0x20
[    4.316154]  i915_driver_probe+0x808/0xbe0 [i915]
[    4.316220]  ? drm_privacy_screen_get+0x16d/0x190
[    4.316223]  ? acpi_dev_found+0x64/0x80
[    4.316226]  i915_pci_probe+0xc1/0x150 [i915]
[    4.316290]  local_pci_probe+0x44/0xb0
[    4.316293]  pci_device_probe+0xc5/0x250
[    4.316296]  really_probe+0x1b7/0x420
[    4.316298]  __driver_probe_device+0x7d/0x180
[    4.316300]  driver_probe_device+0x24/0xa0
[    4.316301]  __driver_attach+0xf4/0x1f0
[    4.316303]  ? __pfx___driver_attach+0x10/0x10
[    4.316305]  bus_for_each_dev+0x8a/0xf0
[    4.316307]  driver_attach+0x1e/0x30
[    4.316308]  bus_add_driver+0x14e/0x280
[    4.316310]  driver_register+0x5e/0x130
[    4.316312]  __pci_register_driver+0x5e/0x70
[    4.316314]  i915_pci_register_driver+0x23/0x30 [i915]
[    4.316375]  i915_init+0x23/0xc0 [i915]
[    4.316442]  ? __pfx_i915_init+0x10/0x10 [i915]
[    4.316504]  do_one_initcall+0x5b/0x340
[    4.316508]  do_init_module+0x97/0x280
[    4.316511]  load_module+0x21b0/0x2260
[    4.316515]  init_module_from_file+0x96/0x100
[... 24 lines cut ...]
[    4.316572] RDX: 0000000000000000 RSI: 00005858dbd257f0 RDI: 0000000000000013
[    4.316573] RBP: 00005858dbd257f0 R08: 0000000000000000 R09: 00005858dbd1ae70
[    4.316575] R10: 0000000000000013 R11: 0000000000000246 R12: 0000000000020000
[    4.316576] R13: 0000000000000000 R14: 00005858dbd25690 R15: 0000000000000017
[    4.316578]  </TASK>
[    4.316579] Modules linked in: i2c_mux soundwire_bus i915(+) kvm snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine irqbypass crct10dif_pclmul polyval_clmulni polyval_generic snd_hda_intel ghash_clmulni_intel sha256_ssse3 snd_intel_dspcfg sha1_ssse3 snd_intel_sdw_acpi aesni_intel snd_hda_codec crypto_simd drm_buddy snd_hda_core cryptd ttm dvb_usb_rtl28xxu(+) snd_hwdep dvb_usb_v2 snd_pcm drm_display_helper dvb_core mei_hdcp mei_pxp snd_timer rapl cmdlinepart spi_nor eeepc_wmi intel_cstate ipmi_ssif acpi_ipmi mc ast cec snd mei_me wmi_bmof mtd cp210x pl2303 pcspkr soundcore rc_core ipmi_si mei i2c_algo_bit usbserial ipmi_devintf intel_pmc_core ipmi_msghandler parport_pc parport intel_vsec pmt_telemetry pmt_class acpi_pad acpi_tad input_leds joydev mac_hid vhost_net vhost vhost_iotlb tap efi_pstore dmi_sysfs ip_tables x_tables autofs4 usbkbd usbmouse cdc_ether usbnet mii zfs(PO) spl(O) btrfs blake2b_generic xor raid6_pq libcrc32c hid_generic usbhid hid mfd_aaeon asus_wmi ledtrig_audio sparse_keymap platform_profile
[... cut ...]
[    6.874733] RBP: ffffb0a00090b4a8 R08: 0000000000000000 R09: 0000000000000000
[    6.874735] R10: 0000000000000000 R11: 0000000000000000 R12: ffff9d2dc32a2090
[    6.874736] R13: ffffffff830f1e68 R14: 0000000000000000 R15: ffffffff84062520
[    6.874738] FS:  00007664d0f6a8c0(0000) GS:ffff9d3cff600000(0000) knlGS:0000000000000000
[    6.874740] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.874741] CR2: 0000000000000008 CR3: 0000000105a84002 CR4: 0000000000f70ef0
[    6.874743] PKRU: 55555554
```
</p></details>

DrymarchonShaun, 2024-09-30:
For what it's worth, I'm getting the same on 6.6.52 immediately after I run my window manager
```
kernel: BUG: kernel NULL pointer dereference, address: 000000000000057d
kernel: #PF: supervisor read access in kernel mode
kernel: #PF: error_code(0x0000) - not-present page
```

vit0sd, 2024-10-08:
This patch worked for me
 
diff --git a/drivers/gpu/drm/i915/display/intel_atomic_plane.c b/drivers/gpu/drm/i915/display/intel_atomic_plane.c
index 230e00c..d36a41b 100644
--- a/drivers/gpu/drm/i915/display/intel_atomic_plane.c
+++ b/drivers/gpu/drm/i915/display/intel_atomic_plane.c
@@ -1152,7 +1152,7 @@ intel_cleanup_plane_fb(struct drm_plane *plane,
 	if (!obj)
 		return;
 
-	if (state->rps_interactive) {
+	if (state && state->rps_interactive) {
 		intel_rps_mark_interactive(&to_gt(dev_priv)->rps, false);
 		state->rps_interactive = false;
 	}

ChaimHong, 2024-11-19:
The same problem occurs, pve kernel 6.5.13-5, i915-sriov-dkms 2024.7.17. `i915-sriov-dkms/2024.07.17, 6.5.13-5-pve, x86_64: installed`
I installed xfce4 and connected it to the monitor.

makoONE, 2024-12-07:
I use proxmox on my homeserver with a minimal Xfce4 desktop environment and connect from this with Remmina/RDP to a Windows 11 VM that also uses the iGPU of the server via SR-IOV. In this scenario I continued to experience freezes with the dkms module from here and all new kernel versions after 6.2 and the system had to be reset.

I have been using the dkms module from the repo listed below for a few days now and have not experienced any more freezes since then. I am using the current and official Proxmox kernel 6.8.12-4-pve but also with the new opt-in kernel 6.11.0-1-pve there are no problems so far.

https://github.com/bbaa-bbaa/i915-sriov-dkms

Only when starting the desktop environment on the server does a dmesg error entry appear in the form:
```
i915 0000:00:02.0: [drm] *ERROR* [CRTC:81:pipe A] mismatch in post_csc_lut hw_state doesn't match sw_state
``` 

However, this does not seem to have any effect or I have not noticed it yet.

ChaimHong, 2024-12-11:
> I use proxmox on my home server with a minimal Xfce4 desktop environment and connect from this with Remmina/RDP to a Windows 11 VM that also uses the iGPU of the server via SR-IOV. In this scenario [...]

so What is the reason?

kitor, 2024-12-19:
> This patch worked for me

For the record, I have the same issue on plain Debian host with `6.11.10-1` kernel. It crashes seconds after SDDM starts.
The patch got corrupted as it was not properly wrapped in code block so here it is:

```
diff --git a/drivers/gpu/drm/i915/display/intel_atomic_plane.c b/drivers/gpu/drm/i915/display/intel_atomic_plane.c
index 230e00c..d36a41b 100644
--- a/drivers/gpu/drm/i915/display/intel_atomic_plane.c
+++ b/drivers/gpu/drm/i915/display/intel_atomic_plane.c
@@ -1152,7 +1152,7 @@ intel_cleanup_plane_fb(struct drm_plane *plane,
        if (!obj)
                return;
 
-       if (state->rps_interactive) {
+       if (state && state->rps_interactive) {
                intel_rps_mark_interactive(&to_gt(dev_priv)->rps, false);
                state->rps_interactive = false;
        }
```

In my case this fixed the immediate crash, but this seems to be just hiding the problem, not fixing it.
My computer (Lenovo Yoga 6 14IAP8, i5-1240p) was unable to output anything to external displays on thunderbolt dock, with multiple things in logs:
```
i915 0000:00:02.0: [drm] Port A asks to use VBT vswing/preemph tables
WARNING: CPU: 5 PID: 274 at /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_bios.c:2708 intel_bios_init+0x1803/0x20b0 [i915]
#...
i915 0000:00:02.0: [drm] *ERROR* AUX USBC3/DDI TC3/PHY TC3: not done (status 0x2c000aff)
#...
i915 0000:00:02.0: [drm] *ERROR* CPU pipe B FIFO underrun: port,transcoder,
#...
i915 0000:00:02.0: [drm] Selective fetch area calculation failed in pipe A
i915 0000:00:02.0: [drm] *ERROR* CPU pipe B FIFO underrun: port,transcoder,
i915 0000:00:02.0: [drm] *ERROR* [CRTC:132:pipe B] mismatch in post_csc_lut hw_state doesn't match sw_state
------------[ cut here ]------------
pipe state doesn't match!
WARNING: CPU: 4 PID: 1384 at /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/display/intel_modeset_verify.c:220 intel_modeset_verify_crtc+0x3d1/0x400 [i915]
```

kitor, 2025-01-01:
I decided to build 6.5.10 kernel for this device and everything works flawlessly. I was able to create only one virtual device, works fine with Windows 10 VM and looking glass.

kitor, 2025-01-17:
Hey, with latest changes pulled (I see support for 6.12) now I'm on 6.12.9 and it seems to work stable so far :)