strongtz/i915-sriov-dkms #183: Fail to load i915 on guest OS (Ubuntu 23.10 aka 6.5.0-44-generic) "err -2"

sheetsg77, 2024-07-20:
I'm receiving an error after installing dkms per readme (no errors on compile by the way), updating grub and then initramfs.  Then adding VF to hardware and restarting.

Output from dmseg | grep i915

```
greg@ub-mantic:~$ sudo dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.5.0-44-generic root=UUID=71cdc55b-a9db-471c-8879-eddfea9a9dd3 ro console=tty1 console=ttyS0 i915.enable=guc=3
[    0.010879] Kernel command line: BOOT_IMAGE=/vmlinuz-6.5.0-44-generic root=UUID=71cdc55b-a9db-471c-8879-eddfea9a9dd3 ro console=tty1 console=ttyS0 i915.enable=guc=3
[    3.451740] i915: loading out-of-tree module taints kernel.
[    3.451767] i915: module verification failed: signature and/or required key missing - tainting kernel
[    3.451821] i915: Unknown symbol drm_dp_clock_recovery_ok (err -2)
[    3.451823] i915: Unknown symbol drm_dp_read_channel_eq_delay (err -2)
[    3.451829] i915: Unknown symbol drm_dp_cec_unset_edid (err -2)
[    3.451831] i915: Unknown symbol drm_lspcon_set_mode (err -2)
[    3.451832] i915: Unknown symbol drm_dp_downstream_444_to_420_conversion (err -2)
[    3.451835] i915: Unknown symbol drm_dp_atomic_find_time_slots (err -2)
[    3.451836] i915: Unknown symbol drm_dsc_compute_rc_parameters (err -2)
[    3.451840] i915: Unknown symbol drm_dp_get_adjust_request_voltage (err -2)
[    3.451847] i915: Unknown symbol drm_dp_pcon_frl_prepare (err -2)
[    3.451849] i915: Unknown symbol drm_dp_downstream_max_tmds_clock (err -2)
[    3.451855] i915: Unknown symbol drm_dp_read_mst_cap (err -2)
[    3.451864] i915: Unknown symbol drm_dp_channel_eq_ok (err -2)
[    3.451865] i915: Unknown symbol drm_dp_dsc_sink_line_buf_depth (err -2)
[    3.451878] i915: Unknown symbol drm_dp_downstream_max_dotclock (err -2)
[    3.451881] i915: Unknown symbol drm_dp_read_sink_count_cap (err -2)
[    3.451883] i915: Unknown symbol drm_dp_bw_code_to_link_rate (err -2)
[    3.451889] i915: Unknown symbol drm_dp_pcon_frl_enable (err -2)
[    3.451894] i915: Unknown symbol drm_dp_pcon_dsc_max_slices (err -2)
[    3.451895] i915: Unknown symbol drm_dp_lttpr_pre_emphasis_level_3_supported (err -2)
[    3.451897] i915: Unknown symbol drm_dp_mst_put_port_malloc (err -2)
[    3.451900] i915: Unknown symbol drm_dp_get_adjust_tx_ffe_preset (err -2)
[    3.451903] i915: Unknown symbol drm_atomic_get_mst_payload_state (err -2)
[    3.451907] i915: Unknown symbol drm_buddy_init (err -2)
[    3.451909] i915: Unknown symbol drm_dp_aux_unregister (err -2)
[    3.451924] i915: Unknown symbol drm_dp_mst_update_slots (err -2)
[    3.451942] i915: Unknown symbol drm_dp_pcon_hdmi_frl_link_error_count (err -2)
[    3.451955] i915: Unknown symbol drm_dp_mst_topology_mgr_suspend (err -2)
[    3.451966] i915: Unknown symbol drm_dp_pcon_dsc_max_slice_width (err -2)
[    3.451973] i915: Unknown symbol drm_dp_psr_setup_time (err -2)
[    3.451979] i915: Unknown symbol drm_dp_128b132b_link_training_failed (err -2)
[    3.451984] i915: Unknown symbol drm_dp_vsc_sdp_log (err -2)
[    3.451998] i915: Unknown symbol drm_dp_mst_connector_late_register (err -2)
[    3.452001] i915: Unknown symbol drm_dp_dsc_sink_supported_input_bpcs (err -2)
[    3.452021] i915: Unknown symbol cec_notifier_conn_register (err -2)
[    3.452022] i915: Unknown symbol drm_dp_128b132b_read_aux_rd_interval (err -2)
[    3.452025] i915: Unknown symbol drm_dp_read_lttpr_common_caps (err -2)
[    3.452034] i915: Unknown symbol drm_dp_downstream_rgb_to_ycbcr_conversion (err -2)
[    3.452036] i915: Unknown symbol drm_hdmi_avi_infoframe_content_type (err -2)
[    3.452050] i915: Unknown symbol drm_dp_dpcd_write (err -2)
[    3.452061] i915: Unknown symbol drm_dp_dpcd_read_phy_link_status (err -2)
[    3.452063] i915: Unknown symbol drm_dp_pcon_frl_configure_2 (err -2)
[    3.452064] i915: Unknown symbol drm_buddy_free_list (err -2)
[    3.452088] i915: Unknown symbol drm_dp_cec_unregister_connector (err -2)
[    3.452096] i915: Unknown symbol drm_buddy_alloc_blocks (err -2)
[    3.452098] i915: Unknown symbol drm_dp_dpcd_read (err -2)
[    3.452101] i915: Unknown symbol drm_hdmi_avi_infoframe_colorimetry (err -2)
[    3.452102] i915: Unknown symbol drm_dp_pcon_pps_override_param (err -2)
[    3.452109] i915: Unknown symbol drm_dp_128b132b_eq_interlane_align_done (err -2)
[    3.452118] i915: Unknown symbol drm_buddy_fini (err -2)
[    3.452121] i915: Unknown symbol drm_dp_mst_atomic_wait_for_dependencies (err -2)
[    3.452142] i915: Unknown symbol drm_dp_mst_topology_mgr_set_mst (err -2)
[    3.452173] i915: Unknown symbol drm_dp_atomic_release_time_slots (err -2)
[    3.452178] i915: Unknown symbol drm_dp_mst_detect_port (err -2)
[    3.452184] i915: Unknown symbol drm_dp_lttpr_max_lane_count (err -2)
[    3.452198] i915: Unknown symbol drm_dp_128b132b_lane_channel_eq_done (err -2)
[    3.452220] i915: Unknown symbol drm_edp_backlight_disable (err -2)
[    3.452233] i915: Unknown symbol drm_dp_get_phy_test_pattern (err -2)
[    3.452236] i915: Unknown symbol drm_dp_lttpr_count (err -2)
[    3.452247] i915: Unknown symbol drm_dp_read_dpcd_caps (err -2)
[    3.452252] i915: Unknown symbol drm_dp_downstream_min_tmds_clock (err -2)
[... 57 lines cut ...]
[    3.453310] i915: Unknown symbol drm_dp_mst_topology_mgr_destroy (err -2)
[    3.453315] i915: Unknown symbol drm_dp_mst_atomic_setup_commit (err -2)
[    3.453327] i915: Unknown symbol drm_dp_read_downstream_info (err -2)
[    3.453334] i915: Unknown symbol drm_dp_pcon_hdmi_link_mode (err -2)
[    3.453335] i915: Unknown symbol drm_dsc_dp_rc_buffer_size (err -2)
[    3.453342] i915: Unknown symbol drm_dp_get_dual_mode_type_name (err -2)
[    3.453361] i915: Unknown symbol drm_buddy_block_trim (err -2)
[    3.453363] i915: Unknown symbol drm_connector_attach_content_protection_property (err -2)
[    3.453373] i915: Unknown symbol drm_dp_mst_get_port_malloc (err -2)
[    3.453381] i915: Unknown symbol drm_dp_dual_mode_set_tmds_output (err -2)
[    3.453488] i915: Unknown symbol drm_dp_mst_get_edid (err -2)
[    3.453493] i915: Unknown symbol drm_dp_dsc_sink_max_slice_count (err -2)
[    3.453508] i915: Unknown symbol drm_dp_get_adjust_request_pre_emphasis (err -2)
[    3.453527] i915: Unknown symbol drm_dp_set_subconnector_property (err -2)
[    3.453575] i915: Unknown symbol drm_dp_read_sink_count (err -2)
[    3.453659] i915: Unknown symbol drm_dp_send_power_updown_phy (err -2)
[    3.453669] i915: Unknown symbol drm_dp_mst_root_conn_atomic_check (err -2)
[    3.453699] i915: Unknown symbol drm_dp_add_payload_part1 (err -2)
[    3.453722] i915: Unknown symbol drm_dp_mst_topology_mgr_resume (err -2)
[    3.453727] i915: Unknown symbol drm_dp_downstream_debug (err -2)
[    3.453757] i915: Unknown symbol drm_edp_backlight_set_level (err -2)
[    3.453758] i915: Unknown symbol drm_dp_set_phy_test_pattern (err -2)
[    3.453807] i915: Unknown symbol drm_dsc_dp_pps_header_init (err -2)
[    3.453855] i915: Unknown symbol drm_dp_read_clock_recovery_delay (err -2)
```

pasbec, 2024-07-20:
Haven't tested it in Ubuntu 23.04 but Ubuntu 22.04 with kernel 6.5.0-44-generic. Try `apt install linux-modules-extra-6.5.0-44-generic` and reboot.

sheetsg77, 2024-07-20:
Thanks - that worked.  Now onto testing Jellyfin

I'm also going to experiment with Ubuntu 24.10 (6.8 kernel), I'll let you know

sheetsg77, 2024-07-20:
resolved by pasbec suggestion to apt install linux-modules-extra-6.5.0-44-generic and then reboot.

closed 2024-07-20 by sheetsg77