strongtz/i915-sriov-dkms #186: ERROR GT0: IOV: Unable to confirm version 1.4

Aliang-code, 2024-07-28:
kernel: 6.8.8-3-pve
cpu: i5-12600T (UHD Graphics 770)

I noticed that my system log has the following output, however it looks like 3 virtualized graphics cards were created, I'm not sure if this is correct..
```
0000:00:02.0 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 770] (rev 0c)
0000:00:02.1 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 770] (rev 0c)
0000:00:02.2 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 770] (rev 0c)
0000:00:02.3 VGA compatible controller: Intel Corporation Alder Lake-S GT1 [UHD Graphics 770] (rev 0c)
```

```
Jul 28 19:50:58 pve kernel: pci 0000:00:02.1: [8086:4690] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
Jul 28 19:50:58 pve kernel: pci 0000:00:02.1: DMAR: Skip IOMMU disabling for graphics
Jul 28 19:50:58 pve kernel: pci 0000:00:02.1: Adding to iommu group 42
Jul 28 19:50:58 pve kernel: pci 0000:00:02.1: vgaarb: bridge control possible
Jul 28 19:50:58 pve kernel: pci 0000:00:02.1: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
Jul 28 19:50:58 pve kernel: i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
Jul 28 19:50:58 pve kernel: xe 0000:00:02.1: Your graphics device 4690 is not officially supported
by xe driver in this kernel version. To force Xe probe,
use xe.force_probe='4690' and i915.force_probe='!4690'
module parameters or CONFIG_DRM_XE_FORCE_PROBE='4690' and
CONFIG_DRM_I915_FORCE_PROBE='!4690' configuration options.
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: enabling device (0000 -> 0002)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: Running in SR-IOV VF mode
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] VT-d active for gfx access
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] Using Transparent Hugepages
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: HuC firmware PRELOADED
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: [drm] PMU not supported for this GPU.
Jul 28 19:50:58 pve kernel: [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
Jul 28 19:50:58 pve kernel: pci 0000:00:02.2: [8086:4690] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
Jul 28 19:50:58 pve kernel: pci 0000:00:02.2: DMAR: Skip IOMMU disabling for graphics
Jul 28 19:50:58 pve kernel: pci 0000:00:02.2: Adding to iommu group 43
Jul 28 19:50:58 pve kernel: pci 0000:00:02.2: vgaarb: bridge control possible
Jul 28 19:50:58 pve kernel: pci 0000:00:02.2: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
Jul 28 19:50:58 pve kernel: i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
Jul 28 19:50:58 pve kernel: xe 0000:00:02.2: Your graphics device 4690 is not officially supported
by xe driver in this kernel version. To force Xe probe,
use xe.force_probe='4690' and i915.force_probe='!4690'
module parameters or CONFIG_DRM_XE_FORCE_PROBE='4690' and
CONFIG_DRM_I915_FORCE_PROBE='!4690' configuration options.
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: enabling device (0000 -> 0002)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: Running in SR-IOV VF mode
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] VT-d active for gfx access
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] Using Transparent Hugepages
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: HuC firmware PRELOADED
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: [drm] PMU not supported for this GPU.
Jul 28 19:50:58 pve kernel: [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
Jul 28 19:50:58 pve kernel: pci 0000:00:02.3: [8086:4690] type 00 class 0x030000 PCIe Root Complex Integrated Endpoint
Jul 28 19:50:58 pve kernel: pci 0000:00:02.3: DMAR: Skip IOMMU disabling for graphics
Jul 28 19:50:58 pve kernel: pci 0000:00:02.3: Adding to iommu group 44
Jul 28 19:50:58 pve kernel: pci 0000:00:02.3: vgaarb: bridge control possible
Jul 28 19:50:58 pve kernel: pci 0000:00:02.3: vgaarb: VGA device added: decodes=io+mem,owns=none,locks=none
Jul 28 19:50:58 pve kernel: i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
Jul 28 19:50:58 pve kernel: i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
Jul 28 19:50:58 pve kernel: i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
Jul 28 19:50:58 pve kernel: xe 0000:00:02.3: Your graphics device 4690 is not officially supported
by xe driver in this kernel version. To force Xe probe,
use xe.force_probe='4690' and i915.force_probe='!4690'
module parameters or CONFIG_DRM_XE_FORCE_PROBE='4690' and
CONFIG_DRM_I915_FORCE_PROBE='!4690' configuration options.
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: enabling device (0000 -> 0002)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: Running in SR-IOV VF mode
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] VT-d active for gfx access
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] Using Transparent Hugepages
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: HuC firmware PRELOADED
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
Jul 28 19:50:58 pve kernel: i915 0000:00:02.3: [drm] PMU not supported for this GPU.
Jul 28 19:50:58 pve kernel: [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
Jul 28 19:50:58 pve kernel: i915 0000:00:02.0: Enabled 3 VFs
```

pasbec, 2024-07-28:
The firmware version in the code has recently been bumped to 1.9 since 1.4 is quite old.

But my experience is that the dkms module seems to "work" even if the firmware version does not match. Still I'm wondering why Proxmox VE is not providing you the latest firmware. Guess the package `pve-firmware` is a dependency of the kernel anyway. On my hosts, `dpkg-query -L pve-firmware | grep i915` gives the following output:
```
/lib/firmware/i915
/lib/firmware/i915/adlp_dmc.bin
/lib/firmware/i915/adlp_dmc_ver2_12.bin
/lib/firmware/i915/adlp_dmc_ver2_16.bin
/lib/firmware/i915/adlp_guc_62.0.3.bin
/lib/firmware/i915/adlp_guc_69.0.3.bin
/lib/firmware/i915/adlp_guc_70.1.1.bin
/lib/firmware/i915/adlp_guc_70.bin
/lib/firmware/i915/adls_dmc_ver2_01.bin
/lib/firmware/i915/bxt_dmc_ver1_07.bin
/lib/firmware/i915/bxt_guc_62.0.0.bin
/lib/firmware/i915/bxt_guc_70.1.1.bin
/lib/firmware/i915/bxt_huc_2.0.0.bin
/lib/firmware/i915/cml_guc_62.0.0.bin
/lib/firmware/i915/cml_guc_70.1.1.bin
/lib/firmware/i915/cml_huc_4.0.0.bin
/lib/firmware/i915/dg1_dmc_ver2_02.bin
/lib/firmware/i915/dg1_guc_70.bin
/lib/firmware/i915/dg1_huc.bin
/lib/firmware/i915/dg2_dmc_ver2_08.bin
/lib/firmware/i915/dg2_guc_70.bin
/lib/firmware/i915/dg2_huc_gsc.bin
/lib/firmware/i915/ehl_guc_62.0.0.bin
/lib/firmware/i915/ehl_guc_70.1.1.bin
/lib/firmware/i915/ehl_huc_9.0.0.bin
/lib/firmware/i915/glk_dmc_ver1_04.bin
/lib/firmware/i915/glk_guc_62.0.0.bin
/lib/firmware/i915/glk_guc_70.1.1.bin
/lib/firmware/i915/glk_huc_4.0.0.bin
/lib/firmware/i915/icl_dmc_ver1_09.bin
/lib/firmware/i915/icl_guc_62.0.0.bin
/lib/firmware/i915/icl_guc_70.1.1.bin
/lib/firmware/i915/icl_huc_9.0.0.bin
/lib/firmware/i915/kbl_dmc_ver1_04.bin
/lib/firmware/i915/kbl_guc_62.0.0.bin
/lib/firmware/i915/kbl_guc_70.1.1.bin
/lib/firmware/i915/kbl_huc_4.0.0.bin
/lib/firmware/i915/mtl_dmc.bin
/lib/firmware/i915/mtl_gsc_1.bin
/lib/firmware/i915/mtl_guc_70.bin
/lib/firmware/i915/mtl_huc_gsc.bin
/lib/firmware/i915/rkl_dmc_ver2_03.bin
/lib/firmware/i915/skl_dmc_ver1_27.bin
/lib/firmware/i915/skl_guc_62.0.0.bin
/lib/firmware/i915/skl_guc_70.1.1.bin
/lib/firmware/i915/skl_huc_2.0.0.bin
/lib/firmware/i915/tgl_dmc_ver2_12.bin
/lib/firmware/i915/tgl_guc_62.0.0.bin
/lib/firmware/i915/tgl_guc_69.0.3.bin
/lib/firmware/i915/tgl_guc_70.1.1.bin
/lib/firmware/i915/tgl_guc_70.bin
/lib/firmware/i915/tgl_huc.bin
/usr/share/doc/pve-firmware/licenses/LICENSE.i915
/lib/firmware/i915/tgl_huc_7.9.3.bin
```

Is this the same for you and do also have pve-firmware version 3.12-1 installed (check with `apt policy pve-firmware`)?
Btw, how did you get the kernel 6.8.8-3-pve? I can only see 6.8.8-2-pve.

If you want to test the previously allowed guc firmware minor version 4 instead of the updated minor version 9. You could just run the following in the root of your the i915-sriov-dkms repo:

```
sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-4/' Makefile
rm -rf /usr/src/i915-sriov-dkms-* /var/lib/dkms/i915-sriov-dkms
dkms add .
dkms install -m i915-sriov-dkms -v $(cat VERSION) -k $(uname -r) --force
```

This will change `GUCFIRMWARE_MINOR=9` to `GUCFIRMWARE_MINOR=4` in the Makefile.

Just for testing it would be enough to run just following:
```
GUCFIRMWARE_MINOR=4 dkms install -m i915-sriov-dkms -v $(cat VERSION) -k $(uname -r) --force
```
But this would not survive system updates in the future.

Aliang-code, 2024-07-28:
> The firmware version in the code has recently been bumped to 1.9 since 1.4 is quite old.
>
> But my experience is that the dkms module seems to "work" even if the firmware version does not match. Still [...]

I just update by `apt update`. Here is my output:
```
root@pve:~# dpkg-query -L pve-firmware | grep i915
/lib/firmware/i915
/lib/firmware/i915/adlp_dmc.bin
/lib/firmware/i915/adlp_dmc_ver2_12.bin
/lib/firmware/i915/adlp_dmc_ver2_16.bin
/lib/firmware/i915/adlp_guc_62.0.3.bin
/lib/firmware/i915/adlp_guc_69.0.3.bin
/lib/firmware/i915/adlp_guc_70.1.1.bin
/lib/firmware/i915/adlp_guc_70.bin
/lib/firmware/i915/adls_dmc_ver2_01.bin
/lib/firmware/i915/bxt_dmc_ver1_07.bin
/lib/firmware/i915/bxt_guc_62.0.0.bin
/lib/firmware/i915/bxt_guc_70.1.1.bin
/lib/firmware/i915/bxt_huc_2.0.0.bin
/lib/firmware/i915/cml_guc_62.0.0.bin
/lib/firmware/i915/cml_guc_70.1.1.bin
/lib/firmware/i915/cml_huc_4.0.0.bin
/lib/firmware/i915/dg1_dmc_ver2_02.bin
/lib/firmware/i915/dg1_guc_70.bin
/lib/firmware/i915/dg1_huc.bin
/lib/firmware/i915/dg2_dmc_ver2_08.bin
/lib/firmware/i915/dg2_guc_70.bin
/lib/firmware/i915/dg2_huc_gsc.bin
/lib/firmware/i915/ehl_guc_62.0.0.bin
/lib/firmware/i915/ehl_guc_70.1.1.bin
/lib/firmware/i915/ehl_huc_9.0.0.bin
/lib/firmware/i915/glk_dmc_ver1_04.bin
/lib/firmware/i915/glk_guc_62.0.0.bin
/lib/firmware/i915/glk_guc_70.1.1.bin
/lib/firmware/i915/glk_huc_4.0.0.bin
/lib/firmware/i915/icl_dmc_ver1_09.bin
/lib/firmware/i915/icl_guc_62.0.0.bin
/lib/firmware/i915/icl_guc_70.1.1.bin
/lib/firmware/i915/icl_huc_9.0.0.bin
/lib/firmware/i915/kbl_dmc_ver1_04.bin
/lib/firmware/i915/kbl_guc_62.0.0.bin
/lib/firmware/i915/kbl_guc_70.1.1.bin
/lib/firmware/i915/kbl_huc_4.0.0.bin
/lib/firmware/i915/mtl_dmc.bin
/lib/firmware/i915/mtl_gsc_1.bin
/lib/firmware/i915/mtl_guc_70.bin
/lib/firmware/i915/mtl_huc_gsc.bin
/lib/firmware/i915/rkl_dmc_ver2_03.bin
/lib/firmware/i915/skl_dmc_ver1_27.bin
/lib/firmware/i915/skl_guc_62.0.0.bin
/lib/firmware/i915/skl_guc_70.1.1.bin
/lib/firmware/i915/skl_huc_2.0.0.bin
/lib/firmware/i915/tgl_dmc_ver2_12.bin
/lib/firmware/i915/tgl_guc_62.0.0.bin
/lib/firmware/i915/tgl_guc_69.0.3.bin
/lib/firmware/i915/tgl_guc_70.1.1.bin
/lib/firmware/i915/tgl_guc_70.bin
/lib/firmware/i915/tgl_huc.bin
/usr/share/doc/pve-firmware/licenses/LICENSE.i915
/lib/firmware/i915/tgl_huc_7.9.3.bin
```

pasbec, 2024-07-30:
The file list seems similar but the exact version is not shown.

What is the version of your installed `pve-firmware`? You can find it with `apt policy pve-firmware`
Have you tried changing `GUCFIRMWARE_MINOR` to 4 as explained above?

scyto, 2024-09-18:
> What is the version of your installed `pve-firmware`?

```
root@pve1:~# apt policy pve-firmware
pve-firmware:
  Installed: 3.13-1
  Candidate: 3.13-2
  Version table:
     3.13-2 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
 *** 3.13-1 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
        100 /var/lib/dpkg/status
     3.12-1 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.11-1 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.10-1 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.9-2 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.9-1 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.8-5 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.8-3 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.8-2 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
     3.7-1 500
        500 http://download.proxmox.com/debian/pve bookworm/pve-no-subscription amd64 Packages
```

scyto, 2024-09-18:
umm myabe i am a numbnuts.... but is the issue the format

```
[    5.005904] i915 0000:00:02.6: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.4 (0000000000000000)
[    5.005917] i915 0000:00:02.6: [drm] *ERROR* GT0: IOV: Found interface version 0.1.4.1
``
its looking for 1.4 but it got 0.1.4.1 - i dont think thats older than 1.4 -= i think it is newer and now is prepended with a 0. ?

MacharaStormwing, 2024-09-22:
So do I understand this correctly:
The drivers have a certain interface version (mine 0.1.13.4) but the firmware has a different one and that is the issue?

I don't remember having the issue before updating just today (kernel 6.8.12-2) or I did not notice it.

Is there a solution to this?

[    5.534849] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    5.534868] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4

Thats the whole log: https://pastebin.com/Ra9C5HUD

scyto, 2024-09-22:
There is a thread on GitHub, it appears to be a non-breaking error. All it is saying it is looking for 1.13 but found 0.1.13.4 just seems a number format issue, i.e 0.1.13.4 is 1.13. Just ignore it until they fix the code error.

scyto, 2024-09-22:
I don't think that's the issue, the number mismatch issue is the same for different versions. 1.13 and 0.1.13.1 are the same, and the error likely indicates a bug in assessing the number format for some reason.

scyto, 2024-09-22:
Confirmed all that is needed is the firmware version command.

So this fixed it for me:

`GUCFIRMWARE_MINOR=4 dkms install -m i915-sriov-dkms -v 2024.08.09 --force --kernelsourcedir /usr/src/linux-headers-6.8.12-2-pve/ `

that said, no reason not to be on latest firmwwares either :-)

pasbec, 2024-09-23:
@MacharaStormwin: Have you tried to use `GUCFIRMWARE_MINOR=13` for the 1.13.x firmware?

johntdavis84, 2024-09-23:
Is this a purely cosmetic thing? I used to see a similar error in my logs, but on boot with PVE 8.2.3 and kernel 6.8.8.4, I’m not seeing it.

```
***@***.***:~# dmesg -T | grep -i GT0
[Mon Sep 23 18:29:35 2024] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.20.0
[Mon Sep 23 18:29:35 2024] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[Mon Sep 23 18:29:35 2024] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[Mon Sep 23 18:29:35 2024] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[Mon Sep 23 18:29:35 2024] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.2: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.3: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.4: [drm] GT0: GUC: interface version 0.1.9.0
[Mon Sep 23 18:29:37 2024] i915 0000:00:02.4: [drm] GT0: GUC: interface version 0.1.9.0
``` 

PVE Version:
```
pve-manager/8.2.3/b4648c690591095f (running kernel: 6.8.8-4-pve)
pve-firmware: 3.13-1
```

DKMS:
```
# dkms status
i915-sriov-dkms/2024.08.09, 6.8.8-4-pve, x86_64: installed
```

For reference, I only have 4 VFs enabled. The iGPU on an i7-12700T can probably only barely support that, to be honest.

bolzerrr, 2024-09-24:
> @MacharaStormwin: Have you tried to use `GUCFIRMWARE_MINOR=13` for the 1.13.x firmware?

I run into the same issue. Tried:

`GUCFIRMWARE_MINOR=13 dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
Module i915-sriov-dkms-2024.09.21 for kernel 6.8.12-1-pve (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

i915.ko:
 - Uninstallation
   - Deleting from: /lib/modules/6.8.12-1-pve/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....

i915.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.8.12-1-pve/updates/dkms/
depmod...`

But still:

[   10.214385] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   10.214425] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   10.215489] i915 0000:00:02.3: [drm] VT-d active for gfx access
[   10.215526] i915 0000:00:02.3: [drm] Using Transparent Hugepages
[   10.216493] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   10.216538] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   10.217056] i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   10.217065] i915 0000:00:02.3: HuC firmware PRELOADED
[   10.217516] i915 0000:00:02.3: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   10.217523] i915 0000:00:02.3: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[   10.219976] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[   10.219990] i915 0000:00:02.3: [drm] PMU not supported for this GPU.

scyto, 2024-09-24:
> Before uninstall, this module version was ACTIVE on this kernel.

Did you did both uninstall and remove  before creating the new one?
After installing the new one did you run these 3 commands:
```
update-grub 
update-initramfs -u 
proxmox-boot-tool refresh
```

pasbec, 2024-10-03:
Here are some more findings:
- [GuC Firmware v70.29.2 → VF API 1.13.4](https://gitlab.com/kernel-firmware/linux-firmware/-/commit/9889ca654ef0a251e633c43ce51e86ad911479b9): `GUCFIRMWARE_MAJOR=1`, `GUCFIRMWARE_MINOR=13`
- [GuC Firmware v70.20.0 → VF API 1.9.0](https://gitlab.com/kernel-firmware/linux-firmware/-/commit/842afb27a53ccc04f5ec2f43c9bad5a661f718d4): `GUCFIRMWARE_MAJOR=1`, `GUCFIRMWARE_MINOR=9`
- [GuC Firmware v70.13.1 → VF API 1.4.1](https://gitlab.com/kernel-firmware/linux-firmware/-/commit/44a9510c94ac0334931b6c89dd240ffe5bf1e5fa): `GUCFIRMWARE_MAJOR=1`, `GUCFIRMWARE_MINOR=4`

For Proxmox VE and [pve-firmware](https://git.lierfang.com/proxmox/pve-firmware/-/commits/master) this probably means the following from comparing the submodule update dates:
- `pve-firmware >= 3.13-2`: `GUCFIRMWARE_MAJOR=1`, `GUCFIRMWARE_MINOR=13`
- `pve-firmware >= 3.10-1`: `GUCFIRMWARE_MAJOR=1`, `GUCFIRMWARE_MINOR=9`
- `pve-firmware >= 3.8-4`: `GUCFIRMWARE_MAJOR=1`, `GUCFIRMWARE_MINOR=4`

It seems however that the source code of this **i915-sriov-dkms repo is [too old](https://github.com/strongtz/i915-sriov-dkms/pull/75) for GuC Firmware v70.29.2**. It is possible to get rid of the firmware mismatch but there still is some mmio request failure with this version - not sure if this can be safely ignored:

[   11.244021] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   11.245446] i915 0000:00:02.1: Running in SR-IOV VF mode
[   11.247615] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[   11.249818] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   11.251290] i915 0000:00:02.1: [drm] Using Transparent Hugepages
**[   11.253631] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4**
**[   11.255921] i915 0000:00:02.1: GuC firmware PRELOADED version 1.13 submission:SR-IOV VF**
**[   11.257287] i915 0000:00:02.1: HuC firmware PRELOADED**
**[   11.259004] i915 0000:00:02.1: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0**
**[   11.260350] i915 0000:00:02.1: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT**
[   11.264827] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   11.266189] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   11.267754] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[   11.275837] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   11.277031] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none

I fear to get rid of this problem the whole repo would have to be updated from a more recent version of Intels [LTS repo](https://github.com/intel/linux-intel-lts/tree/master/drivers/gpu/drm/i915).

bbaa-bbaa, 2024-10-04:
> I've modified the `Makefile` in my own fork (branch) with an attempt to autodetect the correct version. But it currently requires a reboot after each firmware upgrade to work which I don't like very much.

In fact, intel's upstream repo only compares the MAJOR version, so we can maybe safely remove the check.
https://github.com/intel/linux-intel-lts/blame/5404e6dd8524e9cc698099c8780a6889e24ecbfd/drivers/gpu/drm/i915/gt/iov/intel_iov_query.c#L97-L112

> I fear to get rid of this problem the whole repo would have to be updated from a more recent version of Intels [LTS repo](https://github.com/intel/linux-intel-lts/tree/master/drivers/gpu/drm/i915).

Porting this dkms module to the most recent 6.6 kernel branch means that we need to remove support for kernels before 6.6 and rebuild the 6.6~6.11 adaptation code. I may try that when I have time,

pasbec, 2024-10-05:
I guess there is still enough hope the new xe driver will support SR-IOV at some point.

At the moment I'm running stable on my i9-14900K (GPU id a780)
```
               use xe.force_probe='a780' and i915.force_probe='!a780'
[   32.971076] i915 0000:00:02.0: Running in SR-IOV PF mode
[   32.972270] i915 0000:00:02.0: [drm] VT-d active for gfx access
[   32.972918] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[   32.973626] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[   32.974226] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[   32.979801] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[   32.983420] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.29.2
[   32.983526] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[   32.989282] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[   32.989802] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[   32.989909] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[   32.990620] i915 0000:00:02.0: [drm] GuC RC: enabled
[   32.991451] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[   32.991782] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[   32.992672] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 2
[   32.994483] i915 0000:00:02.0: 7 VFs could be associated with this PF
[   32.994932] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   32.995972] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[   34.439249] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
               use xe.force_probe='a780' and i915.force_probe='!a780'
[   34.442422] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   34.442936] i915 0000:00:02.1: Running in SR-IOV VF mode
[   34.444407] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[   34.445733] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   34.446527] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   34.447912] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[   34.449753] i915 0000:00:02.1: GuC firmware PRELOADED version 1.13 submission:SR-IOV VF
[   34.450116] i915 0000:00:02.1: HuC firmware PRELOADED
[   34.452736] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[   34.453124] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   34.453962] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 3
[   34.457075] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[   34.457555] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
```

But I see some errors on my N100 (GPU id 46d1) the latest firmware:
```
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[    7.605500] i915 0000:00:02.0: Running in SR-IOV PF mode
[    7.608067] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    7.625413] i915 0000:00:02.0: vgaarb: deactivate vga console
[    7.625596] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    7.626127] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    7.634587] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc.bin (v2.20)
[    7.641087] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.29.2
[    7.641107] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    7.669675] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    7.671535] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    7.671539] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    7.672100] i915 0000:00:02.0: [drm] GuC RC: enabled
[    7.672200] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[    7.672206] i915 0000:00:02.0: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[    7.674541] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.697202] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[    7.699852] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    7.728000] fbcon: i915drmfb (fb0) is primary device
[    7.800611] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   11.146822] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
               use xe.force_probe='46d1' and i915.force_probe='!46d1'
[   11.155042] i915 0000:00:02.1: enabling device (0000 -> 0002)
[   11.156436] i915 0000:00:02.1: Running in SR-IOV VF mode
[   11.158592] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[   11.161534] i915 0000:00:02.1: [drm] VT-d active for gfx access
[   11.162950] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[   11.165338] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[... 5 lines cut ...]
[   11.178071] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[   11.178457] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[   11.188994] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[... cut ...]
[123672.855859]  ? intel_dp_aux_xfer+0x616/0x690 [i915]
[123672.856675]  intel_dp_aux_transfer+0x12f/0x330 [i915]
```

But still both 3D acceleration and video encoding/decoding seems to work ok-ish for the N100 via remote desktop (despite the errors above):
[image]

I agree that the removing the major version check could be an option (or test it against multiple known versions). Funny enough, even in case of the wrong major version (firmware mismatch), VA-API still seems to remain functional for the virtual functions...

johntdavis84, 2024-10-05:
> On Oct 5, 2024, at 1:57 PM, pasbec ***@***.***> wrote:
> 
> I guess there is still enough hope the new xe driver will support SR-IOV at some point.

Looks like we’re almost there. Xe is the default driver in Linux 6.12. However, the Xe driver has not been updated to know which iGPUs support SR-IOV, so right now, it’s treating none of them as having SR-IOV support.

But it looks like this is something that can eventually be toggled on by changing a build (?) flag?
See: https://github.com/intel/linux-intel-lts/issues/33#issuecomment-2389418647

bbaa-bbaa, 2024-10-08:
> I may try that when I have time.
Done. Anyone interested in it can try PR #207 .

haoyouab, 2024-10-11:
> I guess there is still enough hope the new xe driver will support SR-IOV at some point.
>
> At the moment I'm running stable on my i9-14900K (GPU id a780)
>
> ```
>                use xe.force_probe='a780' [...]

Hi @pasbec. I'm using 14700K with Fedora 40 now and I wonder if you get `GPU id a780` working or not. Which `GUCFIRMWARE_MINOR` are you using? I tried `GUCFIRMWARE_MINOR=13` but still seeing errors like:
```
[  139.235566] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
               use xe.force_probe='a780' and i915.force_probe='!a780'
[  139.235858] i915 0000:00:02.1: enabling device (0000 -> 0002)
[  139.235910] i915 0000:00:02.1: Running in SR-IOV VF mode
[  139.236177] i915 0000:00:02.1: [drm] GT0: Incompatible option enable_guc=7 - undocumented flag
[  139.236429] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[  139.236470] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[  139.236968] i915 0000:00:02.1: [drm] VT-d active for gfx access
[  139.237018] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[  139.237742] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[  139.237755] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[  139.238013] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[  139.238017] i915 0000:00:02.1: HuC firmware PRELOADED
[  139.240781] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[  139.240790] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[  139.241350] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 2
[  139.241757] i915 0000:00:02.0: Enabled 1 VFs
```

haoyouab, 2024-10-13:
Fedora 40 running on i7-14700k confirmed working with the following setup:
```bash
sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-13/' Makefile
dkms remove i915-sriov-dkms/2024.09.21 --all
dkms install -m i915-sriov-dkms -v $(cat VERSION) -k $(uname -r) --force
echo 2 > /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_numvfs
```
On the host:
[image]

`lspci` output:
[image]

Latest `intel_gpu_top` output on the host:
[image]

I got error code 43 on windows guest at first. After I did the steps below it works smoothly:
1. add `<vendor_id state='on' value='GenuineIntel'/>` using `virsh edit`
2. remove the line `<feature policy='disable' name='hypervisor'/>` from `<cpu>` section

On the windows guest:
[image]

hmoffatt, 2024-11-25:
> But still:

This is what I see on the N100 (Alder Lake) also, with the latest Promox 8.3 kernel (6.8.12-4) and pve-firmware (3.14-1);

```
[   10.121376] i915 0000:00:02.7: Running in SR-IOV VF mode
[   10.121576] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   10.121613] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   10.121999] i915 0000:00:02.7: [drm] VT-d active for gfx access
[   10.122023] i915 0000:00:02.7: [drm] Using Transparent Hugepages
[   10.122489] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[   10.122526] i915 0000:00:02.7: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[   10.122840] i915 0000:00:02.7: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[   10.122844] i915 0000:00:02.7: HuC firmware PRELOADED
[   10.123124] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   10.123129] i915 0000:00:02.7: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
[   10.124798] i915 0000:00:02.7: [drm] Protected Xe Path (PXP) protected content support initialized
[   10.124808] i915 0000:00:02.7: [drm] PMU not supported for this GPU.
[   10.124895] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.7 on minor 7
```

In a linux guest I see non-stop timeout errors

```
[    4.371366] [drm:fw_domains_get_with_fallback [i915]] *ERROR* gt: timed out waiting for forcewake ack request.
[    4.371486] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
[    4.972254] [drm:fw_domains_get_with_fallback [i915]] *ERROR* gt: timed out waiting for forcewake ack request.
[    4.972370] i915 0000:01:00.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by fw_domains_get_with_fallback+0x1d8/0x230 [i915]
```

stefyelan, 2024-11-27:
I have the same issue. My device info as blew:

pveversion:
  `pve-manager/8.2.8/a577cfa684c7476d (running kernel: 6.8.12-1-pve)`
Linux Kernel:
  `6.8.12-1-pve`
pve-firmware:
```
  Installed: 3.14-1
  Candidate: 3.14-1
```
CPU:
  `13th Gen Intel(R) Core(TM) i9-13900H`
GPU：
  `Intel® HD Graphics：Intel Corporation Raptor Lake-P [UHD Graphics] [8086:a720]`

Enabled 3 VFs:
```
  00:02.0 VGA compatible controller: Intel Corporation Raptor Lake-P [UHD Graphics] (rev 04)
  00:02.1 VGA compatible controller: Intel Corporation Raptor Lake-P [UHD Graphics] (rev 04)
  00:02.2 VGA compatible controller: Intel Corporation Raptor Lake-P [UHD Graphics] (rev 04)
  00:02.3 VGA compatible controller: Intel Corporation Raptor Lake-P [UHD Graphics] (rev 04)
```

dmesg message info:
```

[    2.578692] xe 0000:00:02.0: Your graphics device a720 is not officially supported
               by xe driver in this kernel version. To force Xe probe,
               use xe.force_probe='a720' and i915.force_probe='!a720'
               module parameters or CONFIG_DRM_XE_FORCE_PROBE='a720' and
               CONFIG_DRM_I915_FORCE_PROBE='!a720' configuration options.

[    4.286746] xe 0000:00:02.1: Your graphics device a720 is not officially supported
               by xe driver in this kernel version. To force Xe probe,
               use xe.force_probe='a720' and i915.force_probe='!a720'
               module parameters or CONFIG_DRM_XE_FORCE_PROBE='a720' and
               CONFIG_DRM_I915_FORCE_PROBE='!a720' configuration options.
[    4.286782] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    4.286806] i915 0000:00:02.1: Running in SR-IOV VF mode
[    4.287528] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    4.287562] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    4.288244] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    4.288268] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    4.288929] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    4.288963] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    4.289430] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    4.289431] i915 0000:00:02.1: HuC firmware PRELOADED
[    4.292607] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    4.292610] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
```

I'm not sure if this is correct..

stefyelan, 2024-11-28:
> Fedora 40 running on i7-14700k confirmed working with the following setup:
> 
> ```shell
> sed -i 's/GUCFIRMWARE_MINOR:-9/GUCFIRMWARE_MINOR:-13/' Makefile
> dkms remove i915-sriov-dkms/2024.09.21 --all
> dkms install -m i915-sriov-dkms -v $(cat VERSION) -k $(uname -r) --force
> ```
> 
> Add the following to kernel cmdline:
> 
> ```
> intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=7
> ```
> 
> Reboot and then:
> 
> ```
> echo 2 > /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_numvfs
> ```
> 
> On the host: ![image](https://private-user-images.githubusercontent.com/19225626/376007549-cf00d7d0-0d68-442c-80cb-bc8642bd7621.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI2OTUzMTMsIm5iZiI6MTczMjY5NTAxMywicGF0aCI6Ii8xOTIyNTYyNi8zNzYwMDc1NDktY2YwMGQ3ZDAtMGQ2OC00NDJjLTgwY2ItYmM4NjQyYmQ3NjIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA4MTAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzOWU2MzA5YmMwNzkyNGNkNWRmMzRkMmU5ZDc0YjYxODFjMmFhMTNjMmE2NzZhZjc1YjI3NDQwYTY5NTYyNzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yRqpcSOPnRJ4ZnWkVxUVtUSMvR9lYyonWw4o2sf70x8)
> 
> `lspci` output: ![image](https://private-user-images.githubusercontent.com/19225626/376007558-6225e723-e562-4df5-91df-6f8dc3db9468.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI2OTUzMTMsIm5iZiI6MTczMjY5NTAxMywicGF0aCI6Ii8xOTIyNTYyNi8zNzYwMDc1NTgtNjIyNWU3MjMtZTU2Mi00ZGY1LTkxZGYtNmY4ZGMzZGI5NDY4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA4MTAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdjOTY5YjJhZTQwOTVjZTA1MTFjOWQzMGVhMDhmZTU4YTJjZDdmM2RlYzAxYTUyOWIwNzg3NDcwMGNmZTdhZTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.kpkUGX_pzMlIj-hXlZR2fq0Kj5thJzY6oyqpbV94t1E)
> 
> Latest `intel_gpu_top` output on the host: ![image](https://private-user-images.githubusercontent.com/19225626/376007323-75bb36c0-908f-4b58-a8cd-bcc865ffa876.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI2OTUzMTMsIm5iZiI6MTczMjY5NTAxMywicGF0aCI6Ii8xOTIyNTYyNi8zNzYwMDczMjMtNzViYjM2YzAtOTA4Zi00YjU4LWE4Y2QtYmNjODY1ZmZhODc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA4MTAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3ODg3ZmVmODUzMDMyMmJiMjlmMjU0Y2U2NWFjMzFhYWYxZWE5NDcxMjRkM2NmZGU2YzlkZmFhYzA4NzBhMTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1H086Y0-KLtp6rQSTstsAJf--61gBIhnvHazp88Bx_s)
> 
> I got error code 43 on windows guest at first. After I did the steps below it works smoothly:
> 
> 1. add `<vendor_id state='on' value='GenuineIntel'/>` using `virsh edit`
> 2. remove the line `<feature policy='disable' name='hypervisor'/>` from `<cpu>` section
> 
> On the windows guest: ![image](https://private-user-images.githubusercontent.com/19225626/376007627-5b65103f-b8f6-411d-ae83-ae4d4065ed7a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI2OTUzMTMsIm5iZiI6MTczMjY5NTAxMywicGF0aCI6Ii8xOTIyNTYyNi8zNzYwMDc2MjctNWI2NTEwM2YtYjhmNi00MTFkLWFlODMtYWU0ZDQwNjVlZDdhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI3VDA4MTAxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0MmQ3ZTZmZTVhYWRlZWE1MDU5NzY0MjI0MDE1OGFjM2RkNDdjMmIwZTJiMjBjNTRjNmEwNWMzMzExMjc4MTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Wk0DZph8ql7QXe8zQLpMUt5LWXnaT7vGtMxrjFd58tc)

For which linux kernel version do you use, please.
I get the same problem in Fedora. many thanks

haoyouab, 2024-12-23:
> For which linux kernel version do you use, please. I get the same problem in Fedora. many thanks

Fedora Workstation 40. Kernel 6.10.12-200.fc40.x86_64

sacredx72, 2025-01-17:
@hmoffatt @pasbec 
Did you manage to install the driver without errors? I tried different builds and the result is the same, errors are present in the log
[   10.123124] i915 0000:00:02.7: [drm] *ERROR* GT0: GUC: mmio request 0x4100: failure 201/0
[   10.123129] i915 0000:00:02.7: [drm] *ERROR* GT0: Failed to retrieve hwconfig table: -ENOENT
In PVE, as the Intel driver is installed, and there is an output of the load command "intel_gpu_top"

But at the same time, when starting Plex or Windows VM, the load status does not change, all the time at 0, I am sure that the video core is not loading correctly. Do you have any progress in solving this problem "hwconfig table: -ENOENT", "GUC: mmio request 0x4100: failure 201/0" ?