strongtz/i915-sriov-dkms #187: docs：PVE Host Installation Steps (Tested Kernel 6.5 and 6.8) missing step

Diluka, 2024-07-30:
> Update grub and initramfs by executing update-grub and update-initramfs -u

this step cannot be omitted

Diluka, 2024-07-30:
https://github.com/strongtz/i915-sriov-dkms/issues/172#issuecomment-2146758012

pasbec, 2024-07-30:
In point 7 of the cited [installation steps](https://pve.proxmox.com/wiki/Host_Bootloader) it says: 
> Optionally pin the kernel version and update the boot config via `proxmox-boot-tool`.

The documention of `proxmox-boot-tool` can be found easily via Google or any other search engine: https://pve.proxmox.com/wiki/Host_Bootloader

Point is, the necessary steps simply depends on your system. Updating Grub does not make sense in case you are using systemd-boot, SecureBoot requires import of the dkms MOK key. Pinninf the kernel is also not *required* but *recommended*. So there is multiple options and not universal stpes to finish the install. Btw, refreshing the Proxmox boot conifg with `proxmox-boot-tool refresh` will automaticall run `update-grub` as far as I can see. An update of the initramfs is usually not necessary.

closed 2024-12-29 by strongtz