strongtz/i915-sriov-dkms #190: pve linux 6.8.12-1-pve compiling issue

ksong008, 2024-08-08:
pve linux 6.8.12-1-pve compiling failed

here is the log
[log.txt](https://github.com/user-attachments/files/16536848/log.txt)

submatrix, 2024-08-08:
Same issue and build errors here. Is it possible to force use of an older kernel in the meantime?

johntdavis84, 2024-08-08:
Yes, absolutely. I’d actually suggest that pinning the Proxmox kernel be added to the project’s instructions as a troubleshooting/stability enhancement option.

I wrote up a process tutorial for kernel pinning here, using the proxmox-boot-tool.
(There were already tutorials, but I needed more details, having never done it before, and decided to write up a more detailed one to share.)

https://forum.proxmox.com/threads/how-to-pin-a-machine-to-proxmox-8-1-or-any-previous-version.145752/￼
How to Pin a Machine to Proxmox 8.1 (or Any Previous Version)?
forum.proxmox.com

MacharaStormwing, 2024-08-08:
I have the same issue.
I manually downloaded the version proxmox-hernel 6.8.8-4-pve (included proxmox-headers) and then pinned that with 
proxmox-boot-tool kernel pin 6.8.8-4-pve
proxmox-boot-tool refresh

[johntdavis84](https://github.com/johntdavis84): the link has some letters appended that makes the link not work. I removed them and got there manually.

johntdavis84, 2024-08-08:
Thanks for letting me know. Sorry about that. Not sure what happened; the link works fine from my end. Maybe because I’m logged in?

Does this work? https://forum.proxmox.com/threads/how-to-pin-a-machine-to-proxmox-8-1-or-any-previous-version.145752/

freeDom-, 2024-08-08:
Also got an error building 6.8.12-1 - reverted back to 6.8.8-4 for now and pinned that.

pasbec, 2024-08-08:
I've created #191 with a small fix for proxmox-kernel-6.8.12-1-pve.

IanRanson, 2024-08-08:
Just tested the pull request #191 and seems to be working for me on proxmox-kernel-6.8.12-1-pve, thanks so much @pasbec!

closed 2024-08-09 by ksong008

scyto, 2024-09-18:
I compiled on 6.8.12.-2 today and it compiled just fine