strongtz/i915-sriov-dkms #194: echo 2 > sriov_numvfs: Permission denied

luyanfeng, 2024-08-16:
```

# echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
sh: 6: cannot create /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs: Permission denied
```

what`s  wrong ?

pasbec, 2024-08-16:
Are you trying this as ordinary user or superuser (as root or via su, sudo -i or sudo and tee)?

luyanfeng, 2024-08-20:
> Are you trying this as ordinary user or superuser (as root or via su, sudo -i or sudo and tee)?

user:  root
pve: 8.2.4
kernel: 6.8.12-1-pve

2017fighting, 2024-08-23:
try this

`echo 2 > sudo tee /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`

LechTheMitch, 2024-09-05:
I have the same issue and I am logged as the root user and i tried doing it using sysfsutils but still didn't work

2017fighting, 2024-09-06:
maybe you don't have `0000:00:02.0` folder

LechTheMitch, 2024-09-06:
> maybe you don't have `0000:00:02.0` folder

No it exits and the file that we need to modify is there "sriov_numvfs" it's just unwritable even with sysfsutils and using the root account

veapon, 2024-09-23:
Same issue, the 0000:00:02.0 folder does exist, and i've run chmod -R 777 inside the folder,  also no luck.
[image]

LsHallo, 2024-09-28:
System:
i5-12600k
Z690-P D4

Having the same issue: `root@px1:~/i915-sriov-dkms# echo 2 > /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_numvfs 
-bash: echo: write error: Numerical result out of range`
Kernel: `Linux px1 6.8.12-1-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-1 (2024-08-05T16:17Z) x86_64 GNU/Linux`
SRIOV works successfully for my mellanox connectx4 lx. So I'm pretty confident with my SRIOV setup.

sriov_* are present on device:
```
root@px1:~/i915-sriov-dkms# ls /sys/devices/pci0000\:00/0000\:00\:02.0/
ari_enabled           consistent_dma_mask_bits  dma_mask_bits    firmware_node  i2c-4  i2c-9        label           max_link_width  power        reset_method  resource4                sriov_offset         subsystem                                                       vendor
boot_vga              current_link_speed        driver           graphics       i2c-5  index        link            modalias        power_state  resource      revision                 sriov_stride         subsystem_device
broken_parity_status  current_link_width        driver_override  i2c-10         i2c-6  iommu        local_cpulist   msi_bus         remove       resource0     rom                      sriov_totalvfs       subsystem_vendor
class                 d3cold_allowed            drm              i2c-11         i2c-7  iommu_group  local_cpus      msi_irqs        rescan       resource2     sriov_drivers_autoprobe  sriov_vf_device      supplier:mei:0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1
config                device                    enable           i2c-12         i2c-8  irq          max_link_speed  numa_node       reset        resource2_wc  sriov_numvfs             sriov_vf_total_msix  uevent
```

dkms:
```
root@px1:~/i915-sriov-dkms# dkms status
i915-sriov-dkms/2024.09.21, 6.8.12-1-pve, x86_64: installed
i915-sriov-dkms/2024.09.21, 6.8.12-2-pve, x86_64: installed (WARNING! Diff between built and installed module!)
kernel-mft-dkms/4.29.0, 6.8.12-1-pve, amd64: installed
kernel-mft-dkms/4.29.0, 6.8.12-2-pve, x86_64: installed
```

kernel headers:
```
root@px1:~/i915-sriov-dkms# apt list --installed | grep headers

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

proxmox-default-headers/stable,now 1.1.0 all [installed]
proxmox-headers-6.8.12-1-pve/stable,now 6.8.12-1 amd64 [installed]
proxmox-headers-6.8.12-2-pve/stable,now 6.8.12-2 amd64 [installed,automatic]
proxmox-headers-6.8.4-2-pve/stable,now 6.8.4-2 amd64 [installed]
proxmox-headers-6.8/stable,now 6.8.12-2 all [installed,automatic]
pve-headers/stable,now 8.2.0 all [installed]
```

grub: `GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on intel_iommu=pt i915.enable_guc=3 i915.max_vfs=7"`

Mellanox VFS works fine:
```
01:00.0 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx]
01:00.1 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:00.2 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:00.3 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:00.4 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:00.5 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:00.6 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:00.7 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
01:01.0 Ethernet controller: Mellanox Technologies MT27710 Family [ConnectX-4 Lx Virtual Function]
```

pasbec, 2024-10-03:
I strongly believe `-bash: echo: write error: Numerical result out of range` is not the same issue as `sh: 6: cannot create /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs: Permission denied`. What is the ouput of `cat /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_totalvfs` in your case @LsHallo?

LsHallo, 2024-10-04:
```
root@px1:~# cat /sys/devices/pci0000\:00/0000\:00\:02.0/sriov_totalvfs
7
```
I had to uninstall the dkms again since my system hardlocked a lot after I installed it. 
The weird thing is.. I had it running before as I evaluated my new server CPU. And I also successfully passed through a virtual GPU to a VM and encoded a video in Jellyfin on it. So I find it weird that I'm having all these problems.
Granted it was about 3 months ago and the kernel was slightly older. Maybe that's all there is to it..

closed 2024-12-29 by strongtz

KYLN24, 2025-02-05:
Try writing to `/sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs`? I find it work for me.

```
echo 2 > /sys/module/i915/drivers/pci:i915/0000:00:02.0/sriov_numvfs
```

and then check `/sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`

```
cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
```