strongtz/i915-sriov-dkms #195: i915 support on Intel Ultra CPU

aznablegs, 2024-08-18:
Hi there, 

Thanks for the driver and guide which is very helpful. I have applied to 12th gen CPU and working like charm. Recently just bought a new mini PC with Intel Ultra CPU, seems the dkms driver doesn't work. 

After dig out some document, found Intel mentioned it is not supported by SR-IOV but ATS. Any idea if I still can use your driver to support this function? Thanks.

https://www.intel.com/content/www/us/en/support/articles/000093216/graphics/processor-graphics.html

wolfcar, 2024-09-11:
There is a post on the official Intel forum at https://www.intel.com/content/www/us/en/support/articles/000008563/ethernet-products.html. I don’t have an Ultra PC, could you try it and see if it works successfully?

aznablegs, 2024-09-11:
thanks. but seems nothing output from the scan. after some troubleshooting, I think the driver under proxmox still not fully supported under the kernel 6.8...

tristan-k, 2024-09-18:
I tried but failed. See my comment [here](https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3?permalink_comment_id=5124494#gistcomment-5124494).

tristan-k, 2024-09-20:
I gave it another try but the guests are still not able to use the VFs of the iGPU. 

Here is what I did. See [this comment](https://gist.github.com/scyto/e4e3de35ee23fdb4ae5d5a3b85c16ed3?permalink_comment_id=5196749#gistcomment-5196749) for further explanations.

On the host:

```
apt install build-essentials dkms sysfsutils
$ cd ~
$ git clone https://github.com/strongtz/i915-sriov-dkms.git
mv i915-sriov-dkms/ /usr/src/i915-sriov-dkms-2024.08.09
$ cd /usr/src/i915-sriov-dkms-2024.08.09
$ dkms add .
$ dkms install -m i915-sriov-dkms -v 2024.08.09 --force --kernelsourcedir /usr/src/linux-headers-6.8.12-2-pve/
$ echo 2 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
$ cat /etc/default/grub | grep GRUB_CMDLINE_LINUX_DEFAULT
GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on i915.force_probe=7d55 i915.enable_guc=3 i915.max_vfs=7"
$ update-grub
$ update-initramfs -u -k $(uname -r)
```

```
$ mkdir firmware
$ cd firmware
$ wget -r -nd -e robots=no -A '*.bin' --accept-regex '/plain/' https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915/
$ mv *.bin /lib/firmware/i915/
$ cp /lib/firmware/i915/mtl_guc_70.bin /lib/firmware/i915/mtl_guc_70.6.4.bin
$ reboot
```

```
$ dkms status
i915-sriov-dkms/2024.08.09, 6.8.12-2-pve, x86_64: installed
$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.1 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.2 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.3 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.4 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.5 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.6 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
00:02.7 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
$ inxi -G | grep Device-1
  Device-1: Intel Meteor Lake-P [Intel Arc Graphics] driver: i915 v: kernel
```

On the guest (Ubunut 24.04)

```
$ apt install linux-image-6.8.0-41-generic linux-headers-6.8.0-41-generic linux-modules-extra-6.5.0-41-generic
$ cat /etc/default/grub | grep GRUB_DEFAULT
GRUB_DEFAULT="Advanced options for Ubuntu>Ubuntu, with Linux 6.8.0-41-generic"
$ cd ~
$ git clone https://github.com/strongtz/i915-sriov-dkms.git
mv i915-sriov-dkms/ /usr/src/i915-sriov-dkms-2024.08.09
$ cd /usr/src/i915-sriov-dkms-2024.08.09
$ dkms add .
$ dkms install -m i915-sriov-dkms -v 2024.08.09 --force --kernelsourcedir /usr/src/linux-headers-6.8.0-41-generic/
$ cat /etc/default/grub | grep GRUB_CMDLINE_LINUX_DEFAULT
GRUB_CMDLINE_LINUX_DEFAULT="noquiet nosplash i915.force_probe=7d55 i915.enable_guc=3"
$ update-grub
$ update-initramfs -u -k $(uname -r)
$ reboot
```

```
$ dkms status
i915-sriov-dkms/2024.08.09, 6.8.0-41-generic, x86_64: installed
$ lspci | grep VGA
06:10.0 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)
$ inxi -G | grep Device-1
  Device-1: Intel Meteor Lake-P [Intel Arc Graphics] driver: N/A
```

I guess progress for Core Ultra devices can be followed [here](https://github.com/intel/linux-intel-lts/issues/33). Currently it's not possible to use SR-IOV with Meteor Lake unless I did something wrong.

aznablegs, 2024-09-22:
from Intel site, I just see the latest Ultra CPU doesn't support SR-IOV but sth called ATS. Still, can't find any relevant doc regarding this topic.

https://www.intel.com/content/www/us/en/support/articles/000093216/graphics/processor-graphics.html

mm2293, 2024-10-16:
> Meteor Lake supports GPU SR-IOV. I'm running it on Bee-Link GTi14s right now. However, i915's start is as you mentioned, not graceful.

Interesting... Could you please make a test with Windows 10 or 11?

celesrenata, 2024-10-16:
Way ahead of you, I'm still fighting to get it to work.

I've got QEMU to report the PCI card from 0000:02.00
I got all the settings that intel drivers expect and then some in there
But I'm up against error 43, you can restart the device and it claims it works, but I'm still fighting that.

As for successes, I have Ubuntu 24 successfully running LLM tasks, as well as transcoding jobs from pods.
I am working on Arch and NixOS configs to run inside the VM to give linux headless features to the project next.

My work towards making everythign work lives here: https://github.com/celesrenata/nixos-k3s-configs

You can explore the settings required from this thread I am on for windows 11, windows 10 should already work, but I haven't tested it since it is so close to EOL: https://github.com/kubevirt/kubevirt/issues/11338#issuecomment-2409818352

mm2293, 2024-10-16:
> Way ahead of you, I'm still fighting to get it to work.
> 
> I've got QEMU to report the PCI card from 0000:02.00 I got all the settings that intel drivers expect and then some in there But I'm up against error 43, you can restart the device and it claims it works, but I'm still fighting that.
> 
> As for successes, I have Ubuntu 24 successfully running LLM tasks, as well as transcoding jobs from pods. I am working on Arch and NixOS configs to run inside the VM to give linux headless features to the project next.
> 
> My work towards making everythign work lives here: https://github.com/celesrenata/nixos-k3s-configs
> 
> You can explore the settings required from this thread I am on for windows 11, windows 10 should already work, but I haven't tested it since it is so close to EOL: [kubevirt/kubevirt#11338 (comment)](https://github.com/kubevirt/kubevirt/issues/11338#issuecomment-2409818352)

So you think the problem is that the PCI card from is reported on the wrong address? Do I understand right? As I could see in your kubevirt link you overwrite the address so that it gets reported on 0000:02.00. Why do you think it‘ll work on Windows 10 but not on Windows 11?

jeeftor, 2024-10-16:
I'm beyond surprised you got Nix to support the processor.. Is your nix config available somewhere? Maybe I can give it a run...

jeeftor, 2024-10-16:
This is your intel config?

https://github.com/celesrenata/nixos-k3s-configs/blob/main/nixos-kube-configs/gremlin-1/overlays/i915-sriov-dkms.nix

Your Nix skill is impressive ... i always get my head so confused with nix :) (even though I'm using nix-darwin on most of my macs)

celesrenata, 2024-10-18:
> > Way ahead of you, I'm still fighting to get it to work.
> > I've got QEMU to report the PCI card from 0000:02.00 I got all the settings that intel drivers expect and then some in there But I'm up against error 43, you can restart the device and it claims it works, but I'm still fighting that.
> > As for successes, I have Ubuntu 24 successfully running LLM tasks, as well as transcoding jobs from pods. I am working on Arch and NixOS configs to run inside the VM to give linux headless features to the project next.
> > My work towards making everythign work lives here: https://github.com/celesrenata/nixos-k3s-configs
> > You can explore the settings required from this thread I am on for windows 11, windows 10 should already work, but I haven't tested it since it is so close to EOL: [kubevirt/kubevirt#11338 (comment)](https://github.com/kubevirt/kubevirt/issues/11338#issuecomment-2409818352)
> 
> So you think the problem is that the PCI card is reported on the wrong address? Do I understand right? As I could see in your kubevirt link you overwrite the address so that it gets reported on 0000:02.00. Why do you think it‘ll work on Windows 10 but not on Windows 11?

I'll spin up windows 10 to verify this weekend.

celesrenata, 2024-10-18:
> This is your intel config?
> 
> https://github.com/celesrenata/nixos-k3s-configs/blob/main/nixos-kube-configs/gremlin-1/overlays/i915-sriov-dkms.nix
> 
> Your Nix skill is impressive ... i always get my head so confused with nix :) (even though I'm using nix-darwin on most of my macs)

https://github.com/celesrenata/nixos-k3s-configs/blob/main/nixos-kube-configs/gremlin-1/overlays/i915-sriov-dkms.nix is the main module from this repo I import.
https://github.com/celesrenata/nixos-k3s-configs/blob/main/nixos-kube-configs/gremlin-1/overlays/intel-gfx-sriov.nix I borrow from the original intel sr-iov project to create the VFs
https://github.com/celesrenata/nixos-k3s-configs/blob/main/nixos-kube-configs/gremlin-1/overlays/intel-firmware.nix to hunt down the specific firmware files.
and to add the custom kernel config values before compiling. https://github.com/celesrenata/nixos-k3s-configs/blob/main/nixos-kube-configs/gremlin-1/overlays/kernel.nix

celesrenata, 2024-10-21:
> I'll spin up windows 10 to verify this weekend. 

Nope, win10 also fails.

mm2293, 2024-10-21:
What makes you think that the driver fully supports srvio for meteor lake?

ccoles146, 2024-10-22:
This is what Intel says: https://dgpu-docs.intel.com/driver/kernel-driver-types.html

> The following features are only available in the out-of-tree i915 kernel module:
> -GPU debugging
> -Single Root I/O Virtualization (SR-IOV)
> -Virtual Memory Binding (VM_BIND) and Ultra Low Latency Submission (ULLS)

Although I tried to get it working with the Xe drivers, it seems it is not supported. So far I haven't got the dkms driver working either.

celesrenata, 2024-10-26:
> What makes you think that the driver fully supports sriov for meteor lake?

I thought I'd give it a try as I am consistently able to get LLMs via SR-IOV running in parallel in Ubuntu 24.04 KubeVirts with this driver.

celesrenata, 2024-12-11:
https://lore.kernel.org/lkml/CAPM=9txbfH8vf-YjwTXEYL729a6r2eeLBxCJc3MSD-t5jXVA-w@mail.gmail.com/ looks like 6.13 has some sriov work, we might see it then.

mm2293, 2025-01-27:
https://www.intel.com/content/www/us/en/support/articles/000093216/graphics/processor-graphics.html

Looks like Meteor Lake and Luna Lake won‘t support SRIOV…

sylv-io, 2025-01-28:
> https://lore.kernel.org/lkml/CAPM=9txbfH8vf-YjwTXEYL729a6r2eeLBxCJc3MSD-t5jXVA-w@mail.gmail.com/ looks like 6.13 has some sriov work, we might see it then.

If you look in the xe sriov code and even in [drm-xe-next](https://gitlab.freedesktop.org/drm/xe/kernel), you will see that there is currently no SR-IOV support at all.
They have added a `has_sriov` flag to [xe_device_desc](https://github.com/torvalds/linux/blob/ffd294d346d185b70e28b1a28abe367bbfe53c04/drivers/gpu/drm/xe/xe_pci.c#L65), but no platform has it set yet. e.g.:[mtl_desc](https://github.com/torvalds/linux/blob/ffd294d346d185b70e28b1a28abe367bbfe53c04/drivers/gpu/drm/xe/xe_pci.c#L330)

jeeftor, 2025-02-03:
@celesrenata  - didn't you get this working? Or am I lagging on my understanding (https://github.com/strongtz/i915-sriov-dkms/issues/210#issuecomment-2626587269)

bbaa-bbaa, 2025-02-03:
Intel seems to have enabled the `has_sriov` flag for BMG/PTL on the [mainline-tracking](https://github.com/intel/mainline-tracking/tree/linux/v6.13) repo.

https://github.com/intel/mainline-tracking/commit/9c5c1bf6386aacefc3d869ade9417906ad5f44d8#diff-bdafa5dc7a7551347b083d58fd09f84944ea4683979fc5edf8d6c17bf652d2d6

It is worth noting that Battlemage is stated as unsupported in Intel's support documentation, but the `has_sriov` flag is enabled in the repo above.

johntdavis84, 2025-02-03:
On one hand, this is excellent news. :)

On the other, it sure would be nice if they toggled it on for 12th and 13th gen CPUs.

celesrenata, 2025-02-07:
https://github.com/user-attachments/assets/24fafe23-394f-41ee-8199-b7c0ddb93a66

I have attached a video proving the GPU acting as a normal GPU as well. I haven't successfully connected to Xorg via XRDP yet, but I have validated it is using the GPU with `glxgears`. I apologize I do not have a visual to show you of the desktop, I am still working on that. :)

This is for Intel 185H using  ubuntu 24.04, 6.8.0-41-generic, using commit `e26ce8952e465762fc0743731aa377ec0b2889ff` I have had no success using anything newer sadly.

bbaa-bbaa, 2025-02-07:
> I have had no success using anything newer sadly.

The `has_sriov` flag for Meteor Lake has not been removed. Have you set the `i915.force_probe` kernel parameter to ensure the i915 driver is loaded?
https://github.com/strongtz/i915-sriov-dkms/blame/234e8cf71741d8dd110d79fe66dd8c8444c581a3/drivers/gpu/drm/i915/i915_pci.c#L797

celesrenata, 2025-02-07:
> The `has_sriov` flag for Meteor Lake has not been removed. Have you set the `i915.force_probe` kernel parameter to ensure the i915 driver is loaded? [...]

I use the following kernel parameters:

```nix
boot.kernelParams = [
    "intel_iommu=on"
    "iommu=pt"
    "i915.enable_guc=3"
    "i915.max_vfs=7"
    "i915.force_probe=7d55"
    "module_blacklist=xe"
    "boot.shell_on_fail"
    "hugepagesz=1G"
    "hugepages=2"
    "hugepagesz=2M"
    "hugepages=512"
    "zswap.enabled=1"
    "zswap.compressor=zstd"
    "zswap.zpool=z3fold"
  ];
```

The symptom I get is either reboots or hard locks above 6.6. I have enabled logging to try to recover what errors occur, but I haven't had much luck hunting it down as it doesn't leave a trace other than an off center line at the top of the screen in the console at the time of the crash. It typically occurs at the time of creating the VFs on boot.

bbaa-bbaa, 2025-02-07:
Can you try manually activating VF in the tty to capture any kernel logs that might appear?
If a Kernel Panic occurs, use [Kdump](https://wiki.archlinux.org/title/Kdump) might be helpful.
It seems that Nix also includes a [crashdump](https://github.com/NixOS/nixpkgs/blob/master/nixos/modules/misc/crashdump.nix) module.

celesrenata, 2025-02-07:
```nix
boot.crashDump.enable = true;
```

is on, I can upgrade one of the hosts upto a newer kernel to test this weekend.

paulandrewcrouch, 2025-09-07:
Just revisiting this, is the consensus given the statement on the Intel site that despite getting dkms to show the 8 SRIOV, this will never work on Meteor Lake (I want to run Frigate and Windows in Proxmox VM's and share the GPU).   Thank you for everyone's efforts above!

mundschenk-at, 2025-11-29:
> Just revisiting this, is the consensus given the statement on the Intel site that despite getting dkms to show the 8 SRIOV, this will never work on Meteor Lake (I want to run Frigate and Windows in Proxmox VM's and share the GPU). Thank you for everyone's efforts above!

It seems to work fine on my Core Ultra 7 155H (but I've only used the virtual GPU with Jellyfin, not Frigate).