strongtz/i915-sriov-dkms #196: Proxmox Host Full System Update: Error autoinstalling 6.8.8-4-pve when kernel 6.5.13-5 is pinned. What do I need to do to make sure 6.8.8-4 is usable?

johntdavis84, 2024-08-26:
Hello,

I just updated my Proxmox system, and saw this output on the CLI.

```
Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.8.8-2-pve -C /lib/modules/6.8.8-2-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build KVER=6.8.8-2-pve...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.8-2-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.8.8-2-pve failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/proxmox-kernel-6.8.8-2-pve-signed.postinst line 20.
dpkg: error processing package proxmox-kernel-6.8.8-2-pve-signed (--configure):
 installed proxmox-kernel-6.8.8-2-pve-signed package post-installation script subprocess returned error exit status 2
Setting up libaom3:amd64 (3.6.0-1+deb12u1) ...
Setting up proxmox-kernel-6.8.4-3-pve-signed (6.8.4-3) ...
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/dkms 6.8.4-3-pve /boot/vmlinuz-6.8.4-3-pve
dkms: running auto installation service for kernel 6.8.4-3-pve.
Sign command: /lib/modules/6.8.4-3-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.8.4-3-pve -C /lib/modules/6.8.4-3-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build KVER=6.8.4-3-pve...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.4-3-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.8.4-3-pve failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/proxmox-kernel-6.8.4-3-pve-signed.postinst line 20.
dpkg: error processing package proxmox-kernel-6.8.4-3-pve-signed (--configure):
 installed proxmox-kernel-6.8.4-3-pve-signed package post-installation script subprocess returned error exit status 2
Setting up proxmox-headers-6.5.13-6-pve (6.5.13-6) ...
Setting up bind9-libs:amd64 (1:9.18.28-1~deb12u2) ...
Setting up proxmox-headers-6.8.8-4-pve (6.8.8-4) ...
Setting up pve-firmware (3.13-1) ...
Setting up ifupdown2 (3.2.0-1+pmx9) ...
Setting up pve-cluster (8.0.7) ...
Setting up proxmox-firewall (0.5.0) ...
Setting up libpve-rs-perl (0.8.9) ...
Setting up proxmox-kernel-6.8.8-4-pve-signed (6.8.8-4) ...
Examining /etc/kernel/postinst.d.
run-parts: executing /etc/kernel/postinst.d/dkms 6.8.8-4-pve /boot/vmlinuz-6.8.8-4-pve
dkms: running auto installation service for kernel 6.8.8-4-pve.
Sign command: /lib/modules/6.8.8-4-pve/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.8.8-4-pve -C /lib/modules/6.8.8-4-pve/build M=/var/lib/dkms/i915-sriov-dkms/6.5.13-5/build KVER=6.8.8-4-pve...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.8-4-pve (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.8.8-4-pve failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/postinst.d at /var/lib/dpkg/info/proxmox-kernel-6.8.8-4-pve-signed.postinst line 20.
dpkg: error processing package proxmox-kernel-6.8.8-4-pve-signed (--configure):
 installed proxmox-kernel-6.8.8-4-pve-signed package post-installation script subprocess returned error exit status 2
Setting up libpve-cluster-perl (8.0.7) ...
Setting up bind9-host (1:9.18.28-1~deb12u2) ...
dpkg: dependency problems prevent configuration of proxmox-kernel-6.8:
 proxmox-kernel-6.8 depends on proxmox-kernel-6.8.8-4-pve-signed | proxmox-kernel-6.8.8-4-pve; however:
  Package proxmox-kernel-6.8.8-4-pve-signed is not configured yet.
  Package proxmox-kernel-6.8.8-4-pve is not installed.
  Package proxmox-kernel-6.8.8-4-pve-signed which provides proxmox-kernel-6.8.8-4-pve is not configured yet.

dpkg: error processing package proxmox-kernel-6.8 (--configure):
 dependency problems - leaving unconfigured
 
 ...
 
 Errors were encountered while processing:
 proxmox-kernel-6.8.8-2-pve-signed
 proxmox-kernel-6.8.4-3-pve-signed
 proxmox-kernel-6.8.8-4-pve-signed
 proxmox-kernel-6.8
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I'm not sure why it's trying to build the dkms module against 6.8.x, as I haven't configured it to do that (I installed it against 6.5.13-5). I'm concerned, however, that it's reporting that it couldn't configure the 6.8.8 kernels, though. At some point, I'm going to want to unpin 6.5.13-5 and switch over to the 6.8 series. 

Are my 6.8.x kernels damaged at this point? Do I need to do anything to fix this?

Thanks!

pasbec, 2024-08-29:
I'd recommend to [remove the module, reinstall all your kernels via apt](https://github.com/strongtz/i915-sriov-dkms/issues/173#issuecomment-2245656209) and install the latest version of this dkms package. It should work fine with all revent kernels.

Charlesm54, 2024-08-29:
@pasbec thanks for the reply:) but this wash not about my problem but from johntdavis84

And wanted to explain what is best practice to proceed and that is reading and research the problem and make off-side backups.   

And yes Today I solved my Proxmox kernel problem after months running on a pinned Kernel.
Yeah I am so happy:)<  lossing the server wash no option.

After many reading work and research and learning a lot in that process myself.

I made sure my backups were  in place made with PBS and a bare-metal backup with Clonezilla.
 I decided to apt update && apt dist-upgade the server.
And after that I saw what happend reading the Consult /var/lib/dkms/blksnap/6.0.3.1221/build/make.log for more information.

This blksnap came from a veeam agent that I installed but the log file showed me that the blksnap didn't remove correctly.
I removed veeam many months ago.

Today after I found out what this blksnap wash I uninstalled it and Proxmox did the rest and I removed the pinned old kernel and booted to the new  one 6.8.12-1-pve.

https://pve.proxmox.com/wiki/Host_Bootloader#sysboot_proxmox_boot_tool

https://forum.proxmox.com/threads/how-to-pin-unpin-a-specific-kernel.111732/

https://clonezilla.org/clonezilla-live.php

greetings
Charlesm

johntdavis84, 2024-09-08:
> @johntdavis84 Any progress on your problem ? your problem is bad install or remove from this driver i915-sriov-dkms Maybe you can remember what you did with this driver?
> 
> Maybe this one is your error? driver i915-sriov-dkms
> 
> https://forum.proxmox.com/threads/kernel-upgrade-6-5-13-1-failed-to-install.144756/
> 
> Consult /var/lib/dkms/i915-sriov-dkms/6.5.13-5/build/make.log
> 
> I use WinSCP to login to Proxmox and search the log.
> 
> I have a similar problem from a bad remove with a snap install from veeam- dkms/blksnap And for months I use this old and pinned kernel 6.2.16-19-pve without any problems and today after a bare-metal backup I did a apt update && apt dist-upgrade Same problem the new the kernel....

Thanks for this. :) 

I completely removed the previous SR-IOV DKMS install, which was quite outdated anyway, reinstalled the kernels I wanted, and then installed the latest version of the DKMS driver. It's working great. :)

I also updated the DKMS driver inside the Lubuntu VM I had using it. No issues there.

Previously, I had the 6.5.13-5 kernel pinned in the PVE host. It's nice to be on the latest 6.8.x.
Should I pin the latest 6.8 kernel? What's the safest way to keep the SR-IOV support working and update the kernel at the same time? I think I just need to update to the latest kernel, reboot the PVE host, and then uninstall and reinstall the DKMS driver. Is there more to it than that?

closed 2025-09-05 by johntdavis84