strongtz/i915-sriov-dkms #198: Build SRIOV fails for Ubuntu 24.04 LTS guest

Donimax, 2024-09-16:
Hello everyone,
I am following the pve host installation steps for kernel 6.5 and 6.8 from the README.
When compiling the module in step 5, I get the following errors:

````
root@ws01 ~/i915-sriov-dkms # dkms install -m i915-sriov-dkms -v 2024.08.09 --force
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.8.0-44-generic -C /lib/modules/6.8.0-44-generic/build M=/var/lib/dkms/i915-sriov-dkms/2024.08.09/build......................................................................(bad exit status: 2)
ERROR (dkms apport): binary package for i915-sriov-dkms: 2024.08.09 not found
Error! Bad return status for module build on kernel: 6.8.0-44-generic (x86_64)
Consult /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/make.log for more information.
````

`/var/lib/dkms/i915-sriov-dkms/2024.08.09/build/make.log`

````
DKMS make.log for i915-sriov-dkms-2024.08.09 for kernel 6.8.0-44-generic (x86_64)
Mon Sep 16 08:15:00 PM CEST 2024
make: Entering directory '/usr/src/linux-headers-6.8.0-44-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
KERNELRELEASE=6.8.0-44-generic
KERNELVERSION=6.8.0
EXTRAVERSION_MAJOR=44
EXTRAVERSION_MINOR=0
EXTRAVERSION_NAME=generic
EXTRAVERSION_DEFINE=EXTRAVERSION_GENERIC
LSBRELEASE=24.04
LSBRELEASE_MAJOR=24
LSBRELEASE_MINOR=4
LSBRELEASE_NAME=Ubuntu
LSBRELEASE_DEFINE=RELEASE_UBUNTU
GUCFIRMWARE_MAJOR=1
GUCFIRMWARE_MINOR=9
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_driver.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_drm_client.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_config.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_getparam.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_hwmon.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_ioctl.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_irq.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_mitigations.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_module.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_params.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_pci.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_scatterlist.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_switcheroo.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_sysfs.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/i915_utils.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_device_info.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_memory_region.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_pcode.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_region_ttm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_runtime_pm.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_sbi.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_step.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_uncore.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/intel_wakeref.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/vlv_sideband.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/vlv_suspend.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/soc/intel_dram.o
  CC [M]  /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/soc/intel_gmch.o
[... 288 lines cut ...]
./include/drm/display/drm_dp_mst_helper.h:854:64: note: expected ‘struct drm_dp_mst_atomic_payload *’ but argument is of type ‘struct drm_atomic_state *’
  854 |                              struct drm_dp_mst_atomic_payload *payload);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/display/intel_dp_mst.h:72:3: error: too many arguments to function ‘drm_dp_add_payload_part2’
   72 |   drm_dp_add_payload_part2(&intel_dp->mst_mgr, &state->base, mst_payload_state);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/display/intel_dp_mst.c:751:9: note: in expansion of macro ‘DRM_DP_ADD_PAYLOAD_PART2’
  751 |         DRM_DP_ADD_PAYLOAD_PART2(intel_dp, state,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
./include/drm/display/drm_dp_mst_helper.h:853:5: note: declared here
  853 | int drm_dp_add_payload_part2(struct drm_dp_mst_topology_mgr *mgr,
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.08.09/build/drivers/gpu/drm/i915/display/intel_dp_mst.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.8.0-44-generic/Makefile:1925: /var/lib/dkms/i915-sriov-dkms/2024.08.09/build] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-44-generic'
````

**System:**
Ubuntu 24.04.1 LTS
6.8.0-44-generic

Thanks in advance.

LMaier-dev, 2024-09-17:
My build fails with the exact same log when building in the guest OS

I am also running Ubuntu 24.04.1 LTS
6.8.0-45-generic

CrAazZyMaN21, 2024-09-17:
A fallback solution would be installing 6.8.0-41-generic which is the last working kernel.

tristan-k, 2024-09-19:
I just tried Kernel `` after the same build error but I'm still having issues.

```
$  inxi -G
Graphics:
  Device-1: Intel Meteor Lake-P [Intel Arc Graphics] driver: N/A
  Display: server: No display server data found. Headless machine? tty: 214x52
  API: EGL v: 1.5 drivers: swrast platforms: surfaceless,device
  API: OpenGL v: 4.5 vendor: mesa v: 24.0.9-0ubuntu0.1 note: console (EGL sourced)
    renderer: llvmpipe (LLVM 17.0.6 256 bits)
$ dkms status
i915-sriov-dkms/2024.08.09, 6.8.0-41-generic, x86_64: installed
root@ubuntu24:~# uname -a
Linux ubuntu24 6.8.0-41-generic #41-Ubuntu SMP PREEMPT_DYNAMIC Fri Aug  2 20:41:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
root@ubuntu24:~# dmesg | grep i915
[    3.206696] i915: loading out-of-tree module taints kernel.
[    3.206757] i915: module verification failed: signature and/or required key missing - tainting kernel
[    3.206860] i915: Unknown symbol drm_dp_clock_recovery_ok (err -2)
[    3.206865] i915: Unknown symbol drm_dp_read_channel_eq_delay (err -2)
[    3.206874] i915: Unknown symbol drm_dp_cec_unset_edid (err -2)
[    3.206876] i915: Unknown symbol drm_lspcon_set_mode (err -2)
[    3.206879] i915: Unknown symbol drm_dp_downstream_444_to_420_conversion (err -2)
[    3.206885] i915: Unknown symbol drm_dp_atomic_find_time_slots (err -2)
[    3.206887] i915: Unknown symbol drm_dsc_compute_rc_parameters (err -2)
[    3.206893] i915: Unknown symbol drm_dp_get_adjust_request_voltage (err -2)
[    3.206902] i915: Unknown symbol drm_dp_pcon_frl_prepare (err -2)
[    3.206905] i915: Unknown symbol drm_dp_downstream_max_tmds_clock (err -2)
[    3.206916] i915: Unknown symbol drm_dp_read_mst_cap (err -2)
[    3.206929] i915: Unknown symbol drm_dp_channel_eq_ok (err -2)
[    3.206930] i915: Unknown symbol drm_dp_dsc_sink_line_buf_depth (err -2)
[    3.206947] i915: Unknown symbol drm_dp_downstream_max_dotclock (err -2)
[    3.206953] i915: Unknown symbol drm_dp_read_sink_count_cap (err -2)
[    3.206957] i915: Unknown symbol drm_dp_bw_code_to_link_rate (err -2)
[    3.206965] i915: Unknown symbol drm_dp_pcon_frl_enable (err -2)
[    3.206975] i915: Unknown symbol drm_dp_pcon_dsc_max_slices (err -2)
[    3.206977] i915: Unknown symbol drm_dp_lttpr_pre_emphasis_level_3_supported (err -2)
[    3.206981] i915: Unknown symbol drm_dp_mst_put_port_malloc (err -2)
[    3.206985] i915: Unknown symbol drm_dp_get_adjust_tx_ffe_preset (err -2)
[    3.206991] i915: Unknown symbol drm_atomic_get_mst_payload_state (err -2)
[    3.207000] i915: Unknown symbol drm_buddy_init (err -2)
[    3.207004] i915: Unknown symbol drm_dp_aux_unregister (err -2)
[    3.207057] i915: Unknown symbol drm_dp_mst_update_slots (err -2)
[    3.207077] i915: Unknown symbol drm_dp_pcon_hdmi_frl_link_error_count (err -2)
[    3.207090] i915: Unknown symbol drm_dp_mst_topology_mgr_suspend (err -2)
[    3.207105] i915: Unknown symbol drm_dp_pcon_dsc_max_slice_width (err -2)
[    3.207114] i915: Unknown symbol drm_dp_psr_setup_time (err -2)
[    3.207121] i915: Unknown symbol drm_dp_128b132b_link_training_failed (err -2)
[    3.207126] i915: Unknown symbol drm_dp_vsc_sdp_log (err -2)
[    3.207144] i915: Unknown symbol drm_dp_mst_connector_late_register (err -2)
[    3.207148] i915: Unknown symbol drm_dp_dsc_sink_supported_input_bpcs (err -2)
[    3.207173] i915: Unknown symbol cec_notifier_conn_register (err -2)
[    3.207174] i915: Unknown symbol drm_dp_128b132b_read_aux_rd_interval (err -2)
[    3.207177] i915: Unknown symbol drm_dp_read_lttpr_common_caps (err -2)
[    3.207188] i915: Unknown symbol drm_dp_downstream_rgb_to_ycbcr_conversion (err -2)
[    3.207191] i915: Unknown symbol drm_hdmi_avi_infoframe_content_type (err -2)
[    3.207206] i915: Unknown symbol drm_dp_dpcd_write (err -2)
[    3.207218] i915: Unknown symbol drm_dp_dpcd_read_phy_link_status (err -2)
[    3.207220] i915: Unknown symbol drm_dp_pcon_frl_configure_2 (err -2)
[    3.207221] i915: Unknown symbol drm_buddy_free_list (err -2)
[    3.207250] i915: Unknown symbol drm_dp_cec_unregister_connector (err -2)
[    3.207261] i915: Unknown symbol drm_buddy_alloc_blocks (err -2)
[    3.207263] i915: Unknown symbol drm_dp_dpcd_read (err -2)
[    3.207268] i915: Unknown symbol drm_hdmi_avi_infoframe_colorimetry (err -2)
[    3.207270] i915: Unknown symbol drm_dp_pcon_pps_override_param (err -2)
[    3.207279] i915: Unknown symbol drm_dp_128b132b_eq_interlane_align_done (err -2)
[    3.207292] i915: Unknown symbol drm_buddy_fini (err -2)
[    3.207297] i915: Unknown symbol drm_dp_mst_atomic_wait_for_dependencies (err -2)
[    3.207326] i915: Unknown symbol drm_dp_mst_topology_mgr_set_mst (err -2)
[    3.207375] i915: Unknown symbol drm_dp_atomic_release_time_slots (err -2)
[    3.207383] i915: Unknown symbol drm_dp_mst_detect_port (err -2)
[... 63 lines cut ...]
[    3.209130] i915: Unknown symbol drm_dp_mst_topology_mgr_destroy (err -2)
[    3.209138] i915: Unknown symbol drm_dp_mst_atomic_setup_commit (err -2)
[    3.209153] i915: Unknown symbol drm_dp_read_downstream_info (err -2)
[    3.209162] i915: Unknown symbol drm_dp_pcon_hdmi_link_mode (err -2)
[    3.209163] i915: Unknown symbol drm_dsc_dp_rc_buffer_size (err -2)
[    3.209172] i915: Unknown symbol drm_dp_get_dual_mode_type_name (err -2)
[    3.209193] i915: Unknown symbol drm_buddy_block_trim (err -2)
[    3.209195] i915: Unknown symbol drm_connector_attach_content_protection_property (err -2)
[    3.209206] i915: Unknown symbol drm_dp_mst_get_port_malloc (err -2)
[... cut ...]
[    3.209597] i915: Unknown symbol drm_dp_add_payload_part1 (err -2)
[    3.209624] i915: Unknown symbol drm_dp_mst_topology_mgr_resume (err -2)
[    3.209632] i915: Unknown symbol drm_dp_downstream_debug (err -2)
[    3.209728] i915: Unknown symbol drm_edp_backlight_set_level (err -2)
[    3.209730] i915: Unknown symbol drm_dp_set_phy_test_pattern (err -2)
[    3.209801] i915: Unknown symbol drm_dsc_dp_pps_header_init (err -2)
[    3.209873] i915: Unknown symbol drm_dp_read_clock_recovery_delay (err -2)
```

celesrenata, 2024-09-22:
I also have this problem with ubuntu 24.04, I step around it with the following in a script. I borrow this kernel module for NixOS for SR-IOV for Kubernetes, works great!

```bash
sudo apt install -y linux-headers-6.8.0-41-generic linux-headers-6.8.0-41 linux-image-6.8.0-41-generic linux-modules-6.8.0-41-generic linux-tools-6.8.0-41-generic linux-modules-extra-6.8.0-41-generic 
sudo apt-mark hold linux-headers-6.8.0-41-generic linux-headers-6.8.0-41 linux-image-6.8.0-41-generic linux-modules-6.8.0-41-generic linux-tools-6.8.0-41-generic linux-modules-extra-6.8.0-41-generic 
sudo mv /usr/bin/linux-check-removal /usr/bin/linux-check-removal.orig
echo -e '#!/bin/sh\necho "Overriding default linux-check-removal script!"\nexit 0' | sudo tee /usr/bin/linux-check-removal
sudo chmod +x /usr/bin/linux-check-removal
sudo apt remove -y linux-headers-\$(uname -r) linux-image-\$(uname -r) linux-modules-\$(uname -r) linux-tools-\$(uname -r) linux-modules-extra-\$(uname -r)
echo "options i915 force_probe=7d55 enable_guc=3" | sudo tee -a /etc/modprobe.d/i915.conf
sudo mkdir -p /lib/firmware/i915
sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_gsc_102.0.0.1511.bin -O /lib/firmware/i915/mtl_gsc_102.0.0.1511.bin
sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_guc_70.6.4.bin -O /lib/firmware/i915/mtl_guc_70.6.4.bin
sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_huc_8.4.3_gsc.bin -O /lib/firmware/i915/mtl_huc_8.4.3_gsc.bin
sudo shutdown -r now &
git clone https://github.com/strongtz/i915-sriov-dkms
sudo apt install build-* dkms -y
cd i915-sriov-dkms && sudo dkms add .
cd i915-sriov-dkms && sudo dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
sudo update-initramfs -u
sudo shutdown -r now &
```

bbaa-bbaa, 2024-10-08:
Try PR #205 based on the 6.6 branch.

Donimax, 2024-10-09:
> Try PR #207 based on the 6.6 branch.

Thx works 
I tried your branch on a fresh Ubuntu 24.04.1 LTS `6.8.0-45-generic`.

```
root@ubuntu-intel:~# lspci -vs 07:00.0
07:00.0 Display controller: Intel Corporation AlderLake-S GT1 (rev 0c)
        Subsystem: ASRock Incorporation AlderLake-S GT1
        Physical Slot: 0-6
        Flags: bus master, fast devsel, latency 0, IRQ 54
        Memory at 80000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 382800000000 (64-bit, prefetchable) [size=512M]
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
        Kernel driver in use: i915
        Kernel modules: xe, i915

root@ubuntu-intel:~# dmesg | grep i915
[    5.161475] i915: loading out-of-tree module taints kernel.
[    5.161504] i915: module verification failed: signature and/or required key missing - tainting kernel
[    7.216041] i915 0000:07:00.0: Running in SR-IOV VF mode
[    7.216539] i915 0000:07:00.0: [drm] GT0: GUC: interface version 0.1.4.1
[    7.216880] i915 0000:07:00.0: [drm] VT-d active for gfx access
[    7.216909] i915 0000:07:00.0: [drm] Using Transparent Hugepages
[    7.217771] i915 0000:07:00.0: [drm] GT0: GUC: interface version 0.1.4.1
[    7.218243] i915 0000:07:00.0: [drm] GT0: GUC: interface version 0.1.4.1
[    7.218568] i915 0000:07:00.0: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    7.218570] i915 0000:07:00.0: HuC firmware PRELOADED
[    7.220559] i915 0000:07:00.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    7.220565] i915 0000:07:00.0: [drm] PMU not supported for this GPU.
[    7.220667] [drm] Initialized i915 1.6.0 20201103 for 0000:07:00.0 on minor 0

root@ubuntu-intel:~# ls -lisa /dev/dri/render*
728 0 crw-rw----+ 1 root render 226, 128 Oct  9 09:21 /dev/dri/renderD128

root@ubuntu-intel:~# vainfo
libva info: VA-API version 1.20.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_20
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.18.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 24.1.0 ()
vainfo: Supported profile and entrypoints
```

yduanBioinfo, 2024-11-01:
> I also have this problem with ubuntu 24.04, I step around it with the following in a script. I borrow this kernel module for NixOS for SR-IOV for Kubernetes, works great!
> 
> ```shell
> sudo apt install -y linux-headers-6.8.0-41-generic linux-headers-6.8.0-41 linux-image-6.8.0-41-generic linux-modules-6.8.0-41-generic linux-tools-6.8.0-41-generic linux-modules-extra-6.8.0-41-generic 
> sudo apt-mark hold linux-headers-6.8.0-41-generic linux-headers-6.8.0-41 linux-image-6.8.0-41-generic linux-modules-6.8.0-41-generic linux-tools-6.8.0-41-generic linux-modules-extra-6.8.0-41-generic 
> sudo mv /usr/bin/linux-check-removal /usr/bin/linux-check-removal.orig
> echo -e '#!/bin/sh\necho "Overriding default linux-check-removal script!"\nexit 0' | sudo tee /usr/bin/linux-check-removal
> sudo chmod +x /usr/bin/linux-check-removal
> sudo apt remove -y linux-headers-\$(uname -r) linux-image-\$(uname -r) linux-modules-\$(uname -r) linux-tools-\$(uname -r) linux-modules-extra-\$(uname -r)
> echo "options i915 force_probe=7d55 enable_guc=3" | sudo tee -a /etc/modprobe.d/i915.conf
> sudo mkdir -p /lib/firmware/i915
> sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_gsc_102.0.0.1511.bin -O /lib/firmware/i915/mtl_gsc_102.0.0.1511.bin
> sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_guc_70.6.4.bin -O /lib/firmware/i915/mtl_guc_70.6.4.bin
> sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_huc_8.4.3_gsc.bin -O /lib/firmware/i915/mtl_huc_8.4.3_gsc.bin
> sudo shutdown -r now &
> git clone https://github.com/strongtz/i915-sriov-dkms
> sudo apt install build-* dkms -y
> cd i915-sriov-dkms && sudo dkms add .
> cd i915-sriov-dkms && sudo dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
> sudo update-initramfs -u
> sudo shutdown -r now &
> ```

Thanks! It works.

mackerel-12138, 2024-12-22:
> I also have this problem with ubuntu 24.04, I step around it with the following in a script. I borrow this kernel module for NixOS for SR-IOV for Kubernetes, works great!我在 ubuntu 24.04 中也遇到了这个问题，我在脚本中使用以下方法逐步解决了它。我借用了 NixOS 的这个内核模块，用于 Kubernetes 的 SR-IOV，效果很好！
> 
> ```shell
> sudo apt install -y linux-headers-6.8.0-41-generic linux-headers-6.8.0-41 linux-image-6.8.0-41-generic linux-modules-6.8.0-41-generic linux-tools-6.8.0-41-generic linux-modules-extra-6.8.0-41-generic 
> sudo apt-mark hold linux-headers-6.8.0-41-generic linux-headers-6.8.0-41 linux-image-6.8.0-41-generic linux-modules-6.8.0-41-generic linux-tools-6.8.0-41-generic linux-modules-extra-6.8.0-41-generic 
> sudo mv /usr/bin/linux-check-removal /usr/bin/linux-check-removal.orig
> echo -e '#!/bin/sh\necho "Overriding default linux-check-removal script!"\nexit 0' | sudo tee /usr/bin/linux-check-removal
> sudo chmod +x /usr/bin/linux-check-removal
> sudo apt remove -y linux-headers-\$(uname -r) linux-image-\$(uname -r) linux-modules-\$(uname -r) linux-tools-\$(uname -r) linux-modules-extra-\$(uname -r)
> echo "options i915 force_probe=7d55 enable_guc=3" | sudo tee -a /etc/modprobe.d/i915.conf
> sudo mkdir -p /lib/firmware/i915
> sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_gsc_102.0.0.1511.bin -O /lib/firmware/i915/mtl_gsc_102.0.0.1511.bin
> sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_guc_70.6.4.bin -O /lib/firmware/i915/mtl_guc_70.6.4.bin
> sudo wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_huc_8.4.3_gsc.bin -O /lib/firmware/i915/mtl_huc_8.4.3_gsc.bin
> sudo shutdown -r now &
> git clone https://github.com/strongtz/i915-sriov-dkms
> sudo apt install build-* dkms -y
> cd i915-sriov-dkms && sudo dkms add .
> cd i915-sriov-dkms && sudo dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
> sudo update-initramfs -u
> sudo shutdown -r now &
> ```

It works! thank you

closed 2024-12-29 by strongtz