strongtz/i915-sriov-dkms #199: Can't get this fully working on guest debian 12 VM

ratbich, 2024-09-18:
Hello! I have been struggling for days trying different things to try to get my guest debian 12 VM to pass through properly.

My CPU is i9-12900HK (Minisforum NAB9)

Everything seems to be working fine on the proxmox host, kernel 6.8.12-1-pve

dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-1-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
[    0.058843] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-1-pve root=/dev/mapper/pve-root ro intel_iommu=on i915.enable_guc=3 i915.max_vfs=7
............................
[    3.161893] i915 0000:00:02.0: Enabled 7 VFs

I ran a Windows 11 Guest VM, and the drivers installed and showed up properly. Intel Iris XE. Running
intel_gpu_top -d drm:/dev/dri/renderD128 on the proxmox host, I see the GPU being utilized.

Now comes to Debian 12. I have tried multiple kernels... compiling my own versions....multiple times... fresh VMs...
Every kernel version ends up with same problem, 

dmesg | grep i915 , i915: Unknown symbol intel_xxxxxx(err -2)

```
lspci | grep -i vga
06:10.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
```

this will show here, Kernel modules: i915, but never IN USE like it should

```
lspci -vs 06:10.0
06:10.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c) (prog-if 00 [VGA controller])
        Subsystem: Intel Corporation Alder Lake-P GT2 [Iris Xe Graphics]
        Physical Slot: 16-2
        Flags: bus master, fast devsel, latency 0
        Memory at 80000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 800000000 (64-bit, prefetchable) [size=512M]
        Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit-
        Kernel modules: i915
```

```
dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.6-custom root=UUID=38946e7e-e85c-443a-9202-ce57f3238fdb ro quiet i915.enable_guc=3
[    0.024014] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.6-custom root=UUID=38946e7e-e85c-443a-9202-ce57f3238fdb ro quiet i915.enable_guc=3
[    2.238263] i915: loading out-of-tree module taints kernel.
[    2.238682] i915: Unknown symbol intel_pxp_irq_enable (err -2)
[    2.239170] i915: Unknown symbol intel_pxp_debugfs_register (err -2)
[    2.239737] i915: Unknown symbol intel_pxp_resume_complete (err -2)
[    2.240218] i915: Unknown symbol intel_pxp_terminate (err -2)
[    2.240461] i915: Unknown symbol intel_pxp_runtime_suspend (err -2)
[    2.240747] i915: Unknown symbol intel_pxp_session_management_init (err -2)
[    2.240909] i915: Unknown symbol intel_soc_pmic_exec_mipi_pmic_seq_element (err -2)
[    2.241601] i915: Unknown symbol intel_pxp_irq_handler (err -2)
[    2.242686] i915: Unknown symbol intel_pxp_suspend (err -2)
[    2.242772] i915: Unknown symbol intel_pxp_gsccs_init (err -2)
[    2.243369] i915: Unknown symbol intel_pxp_gsccs_fini (err -2)
[    2.243414] i915: Unknown symbol intel_pxp_suspend_prepare (err -2)
[    2.243606] i915: Unknown symbol intel_pxp_irq_disable (err -2)
```

I just don't know what else to try besides the instruction on the github page. Is there a certain kernel that works properly with the i9-12900HK and the i915-sriov-dkms? Does anyone have it working on that CPU and Debian 12?

ratbich, 2024-09-18:
RESOLVED!
Kernel 6.9 from backports worked with the same procedure

closed 2024-09-18 by ratbich

WPF0414, 2024-10-02:
> 已解决！ 来自反向移植的内核 6.9 可以使用相同的程序
我也碰到这个问题了，请问能不能说一下详细操作？非常感谢！

WPF0414, 2024-10-02:
> RESOLVED! Kernel 6.9 from backports worked with the same procedure

I have also encountered this problem. Could you please explain the detailed operation? Thank you very much!