strongtz/i915-sriov-dkms #205: IGPU Code 43

Nightmare41241, 2024-10-05:
i am using a laptop with i7-1260P intel iris xe. i follow the instruction of sriov. when i install intel driver from intel web and restart, windows cannot use gpu and show code43. I am using pop os with 6.6 xanmod v3.

Nightmare41241, 2024-10-05:
ive tried the old ver of gpu driver and still code43

pasbec, 2024-10-05:
I have no idea about your virtualization setup / hypervisor but with kvm, there are a few things required. Make sure you have configured the following:
- not using any other display `-vga none -nographic`
- host CPU type `-cpu host` (other cpu flags might be necessary to get good performance, e.g. `hv_passthough`)
- use q35 machine type `-machine type=pc-q35-8.1` (or similar)
- VF device like this `-device vfio-pci,host=0000:00:02.1,id=hostpci0,bus=ich9-pcie-port-1,addr=0x0` (or similar)

In Proxmox make sure to use pcie and primary gpu options for the passthrough.

closed 2024-12-29 by strongtz