strongtz/i915-sriov-dkms #206: Host hangs on suspend

lucker999, 2024-10-06:
On 6.6.53-1-arch-lts with `intel_iommu=on i915.enable_guc=3 i915.max_vfs=7` and `cat /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs` 2 the host system hangs out once a suspend command issued (either with button, lid closed or as `systemctl suspend`. The stranger thing is there is nothing in the logs after that, not even a suspend command. 
I tried the module after the issues #175  and #204 have been fixed. The memory leak stopped, but even if no VM is running (even virt-manager has not been launched) the system hangs with black screen and only way to revive the box is to hard reset.

bbaa-bbaa, 2024-10-06:
Can you try to suspend normally with all VF's turned off? (with `echo 0 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs` )

lucker999, 2024-10-07:
Yes, with VF turned off PC suspends and resumes normally as expected

bbaa-bbaa, 2024-10-08:
I have no ideas about that. Can you try PR #205 based on 6.6 branch?

lucker999, 2024-10-08:
Just tried with arch-6.11.2-5 on two boxes (i7-1360P Iris Xe and i5-12450H UHD Graphics (48EU)) with the same outcome: once VF is enabled any suspend attempts resulting in black screen and unresponsive box. The only way is to force reboot.
I believe this is a new issue because a month or two ago I gave the module a try and the only problem I noticed was memory leak. But I am not sure, I just might not suspend any machine while I tried.

bbaa-bbaa, 2024-10-09:
Did you notice the kernel panic when the system hung? If you have a laptop with a PS/2 keyboard the cap locks indicator will blink.

lucker999, 2024-10-09:
I don't think so, because both boxes went rather quiet after black screen, fans almost stopped, no CPU load. But I'm not sure if this is an evidence, never witnessed a kernel panic close, so I only assuming. 
Alas, no PS/2 port, the both boxes quite new, no blinking noticed on laptop as well, that's I'm sure of.

sheaahhoi1, 2024-10-26:
@lucker999 1. I encountered a similar situation on PVE, but I know where the problem is, the machine is not connected to the display device where it was originally stored, and I can't access the PVE site, but the PING passes.
2. When I moved to another place to connect the display device and restarted the machine, everything was fine, PING passed and at the same time I visited the PVE website, and the display device (PVE image appeared) did not have any error messages.
3.Maybe i915-sriov-dkms preset this point in the preset modification.
4. Solution (1) must be connected to the display device / HDMI cheat
[image]
 (2) ask for guidance to modify the default connection to the display device problem.

lucker999, 2024-10-28:
@sheaahhoi1 thank you for the suggestion. Alas, I doubt that the problem I experience related to a display device (at least as I understand the suggestion). It happens even without any VM running, just when a VF created with a rebooted box where no applications running at all. Th PC just hangs on a sleep command when a VF is created, that's it. I haven't even got to a VM settings.

bbaa-bbaa, 2024-11-17:
It seems that the problem was fixed in https://github.com/intel/linux-intel-lts/commit/6cb32a4a630c4f474a280c0d9a2236c1e10e03a8 but then revert in https://github.com/intel/linux-intel-lts/commit/e46efe82261f5f422d1cba2d494d8eddcf5f4a32 .

There are some previous cases where this patch may help, can you try testing it? https://github.com/strongtz/i915-sriov-dkms/issues/78#issuecomment-2019463505

bbaa-bbaa, 2024-12-28:
try #236

michael-pptf, 2025-01-26:
@sheaahhoi1 When you can't access PVE web console, chances are the NIC naming scheme changed and you don't even know it. Worse, when you virtualize the iGPU, you lose display output as well.

Once and for all solution: Bind the NIC with MAC address and assign it a static name.

Steps:
`$ ip a` and look for the NIC that is being used to access the web UI. Write down the MAC addess.

Create a file with the following contents:
```
[Match]
MACAddress=aa:bb:cc:dd:ee:ff #your NIC MAC address
Type=ether

[Link]
Name=lan0 #or whatever the name you like, just don't use ethxxx
```

Edit `/etc/network/interfaces` and change the NIC name that is bind to Proxmox web UI, usually in the form of enpsxxxx or ethxxx

Refresh `initramfs` by running `initramf-update -u -k all`

trablin45, 2025-01-28:
Im having the same issue with windows 11 vm, when windows enters suspend then when to trying to restart the vm crashes.