strongtz/i915-sriov-dkms #208: Invalid when the PVE file system is ZFS

zz541843, 2024-10-06:
I tried for a long time and could only come to this conclusion : (

When installing PVE with two SSDs and choosing ZFS RAID1, I could enter the system normally and follow the documentation to successfully install the drivers. However, after rebooting, the virtual GPU was not generated. I repeated this process many times but still failed.

Then I tried installing with a single SSD (after rebooting, the GNU/Linux option appeared, which I suspect is related to this issue since ZFS can only boot with UEFI), and I completed the process smoothly.

I tried many kernel versions but they didn't work, so I upgraded from the default 6.8.4-2 to the latest 6.8.12-2 for testing.

My technical skills are limited, and after consulting with ChatGPT, I could only provide the following information for reference (I swear my steps were correct, and I wasted 10 hours on this)

```
root@junsl:~# dmesg | grep i915
               use xe.force_probe='4690' and i915.force_probe='!4690'
[    6.021221] i915: module verification failed: signature and/or required key missing - tainting kernel
[    6.215890] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    6.236886] i915 0000:00:02.0: vgaarb: deactivate vga console
[    6.236932] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    6.237549] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    6.238060] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[    6.238226] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    6.252276] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.29.2
[    6.252281] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    6.265687] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    6.265704] i915 0000:00:02.0: [drm] GT0: GUC: submission disabled
[    6.265712] i915 0000:00:02.0: [drm] GT0: GUC: SLPC disabled
[    6.266448] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    6.266537] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    6.323457] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[    6.365506] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    6.369746] fbcon: i915drmfb (fb0) is primary device
[    6.477963] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```

pasbec, 2024-10-07:
I'm running fine with encrypted zfs root, grub for boot and even secure boot enabled. It has nothing to do with the module - although the boot process might be different.

Your log shows that you seem to have secure boot enabled but did not import the machine owner key from dkms. Tty to use mokutil to import it or disable secure boot in your BIOS.

zz541843, 2024-10-07:
> I'm running fine with encrypted zfs root, grub for boot and even secure boot enabled. It has nothing to do with the module I believe - although the boot process might be different.我在启用加密的 zfs root、grub for boot 甚至安全引导的情况下运行良好。我相信它与模块无关 - 尽管启动过程可能不同。
> 
> Your log shows (`module verification failed: signature and/or required key missing - tainting kernel`) that you seem to have secure boot enabled but did not import the machine owner key from dkms. Tty to use mokutil to import it or disable secure boot in your BIOS.您的日志显示 （） `module verification failed: signature and/or required key missing - tainting kernel` ） 您似乎启用了安全启动，但未从 dkms 导入计算机所有者密钥。使用 mokutil 导入它或在 BIOS 中禁用安全启动。

My BIOS has turned off secure boot, and I have not changed the BIOS settings for either a single hard drive or ZFS RAID 1

zz541843, 2024-10-07:
`module verification failed: signature and/or required key missing - tainting kernel`
This sentence also exists when I succeed

```
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-2-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt pcie_acs_override=downstream i915.enable_guc=3 i915.max_vfs=3
[    0.076310] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-2-pve root=/dev/mapper/pve-root ro quiet intel_iommu=on iommu=pt pcie_acs_override=downstream i915.enable_guc=3 i915.max_vfs=3
               use xe.force_probe='4690' and i915.force_probe='!4690'
[    4.782131] i915: module verification failed: signature and/or required key missing - tainting kernel
[    5.028871] i915 0000:00:02.0: Running in SR-IOV PF mode
[    5.029148] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    5.130580] i915 0000:00:02.0: vgaarb: deactivate vga console
[    5.130623] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    5.131192] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    5.132138] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    5.132159] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[    5.147423] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/tgl_guc_70.bin version 70.29.2
[    5.147427] i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
[    5.149580] i915 0000:00:02.0: [drm] GT0: HuC: authenticated for all workloads!
[    5.149870] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    5.149871] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    5.150175] i915 0000:00:02.0: [drm] GuC RC: enabled
[    5.150655] mei_pxp 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: bound 0000:00:02.0 (ops i915_pxp_tee_component_ops [i915])
[    5.150741] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.234002] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 1
[    5.270575] i915 0000:00:02.0: 3 VFs could be associated with this PF
[    5.308445] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.311255] fbcon: i915drmfb (fb0) is primary device
[    5.420717] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    5.806648] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
               use xe.force_probe='4690' and i915.force_probe='!4690'
[    5.806700] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    5.806715] i915 0000:00:02.1: Running in SR-IOV VF mode
[    5.807461] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    5.807616] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    5.808612] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    5.808626] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    5.809117] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    5.809128] i915 0000:00:02.1: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    5.809533] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    5.809534] i915 0000:00:02.1: HuC firmware PRELOADED
[    5.812535] i915 0000:00:02.1: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.812538] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    5.812732] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.1 on minor 0
[    5.812904] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[    5.812906] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='4690' and i915.force_probe='!4690'
[    5.812954] i915 0000:00:02.2: enabling device (0000 -> 0002)
[    5.812969] i915 0000:00:02.2: Running in SR-IOV VF mode
[    5.813502] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    5.813577] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    5.813837] i915 0000:00:02.2: [drm] VT-d active for gfx access
[    5.813849] i915 0000:00:02.2: [drm] Using Transparent Hugepages
[    5.814243] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Unable to confirm version 1.13 (0000000000000000)
[    5.814252] i915 0000:00:02.2: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    5.814394] i915 0000:00:02.2: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    5.814396] i915 0000:00:02.2: HuC firmware PRELOADED
[    5.815725] i915 0000:00:02.2: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.815727] i915 0000:00:02.2: [drm] PMU not supported for this GPU.
[    5.815848] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.2 on minor 2
[    5.815998] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[    5.816000] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=none
[    5.816002] i915 0000:00:02.2: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
               use xe.force_probe='4690' and i915.force_probe='!4690'
[    5.816034] i915 0000:00:02.3: enabling device (0000 -> 0002)
[    5.816045] i915 0000:00:02.3: Running in SR-IOV VF mode
[... cut ...]
[    5.817040] i915 0000:00:02.3: [drm] *ERROR* GT0: IOV: Found interface version 0.1.13.4
[    5.817182] i915 0000:00:02.3: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    5.817183] i915 0000:00:02.3: HuC firmware PRELOADED
[    5.818513] i915 0000:00:02.3: [drm] Protected Xe Path (PXP) protected content support initialized
[    5.818515] i915 0000:00:02.3: [drm] PMU not supported for this GPU.
[    5.818543] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.3 on minor 3
[    5.818634] i915 0000:00:02.0: Enabled 3 VFs
```

pasbec, 2024-10-07:
Strange - just to be sure regarding secure boot you could check the output of `mokutil --sb-state` after boot.

And what is the kernel comamnd line with zfs? In the failed example with zfs above, your output of `dmesg | grep i915` shows no kernel command line. Are you sure `i915.enable_guc=3 i915.max_vfs=3` is enabled when you use zfs root?

To find out which bootloader Proxmox uses, please refer to [the official documentation](https://pve.proxmox.com/wiki/Host_Bootloader#sysboot_determine_bootloader_used). A description on how to change the kernel command line depending on the bootloader (systemd-boot or grub), is also [available](https://pve.proxmox.com/pve-docs/pve-admin-guide.html#sysboot_edit_kernel_cmdline).

In case of systemdboot, make sure `cat /etc/kernel/cmdline` returns something *like*:
```
root=ZFS=rpool/ROOT/pve-1 boot=zfs intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=3
```

With grub, make sure `cat /etc/default/grub` contains somthing *like*:
```
GRUB_CMDLINE_LINUX_DEFAULT="intel_iommu=on iommu=pt i915.enable_guc=3 i915.max_vfs=3"
```

Whenever you apply changes to one of these files, you need to run `proxmox-boot-tool refresh` afterwards, otherwise your changes won't have any effect.

Also check that SR-IOV is enabled in the BIOS.

zz541843, 2024-10-07:
First of all, thank you very much! 
Secondly, as a programmer with limited knowledge of Linux, I apologize to you and pay my respects. 
Thank you for your follow-up response. 
After checking, I found that the system uses system boot for startup, whereas I had been modifying the GRUB boot parameters, because in my country, community resources mainly cover GRUB booting, and information is limited. After I modified the parameters in /etc/kernel/cmdline and refreshed it, and rebooted, it was successful. 
God bless you !

closed 2024-10-07 by zz541843

pasbec, 2024-10-07:
No worries, you are very welcome. Glad your problem is solved!