strongtz/i915-sriov-dkms #209: Just FIY everyone who wants up to day kernel and hardware accelerated OpenGL in Windows VM

pg2web, 2024-10-09:
I finally tired waiting when Intel will provide official SRIOV support for mainline kernels and found the alternative. I suggest to try Mvisor lightweight QEMU alternative and their Windows virtio-opengl driver that uses /dev/dri/ render. It's very close to hardware and will fit most of your needs. I even run it in podman container for better isolation.
It even allows to run very picky Virtualbox based LDplayer android emulator with good FPS as nested->nested guest.

###

$ cat Dockerfile 
```
FROM fedora:40

RUN dnf -y update && dnf install -y meson ninja-build libdrm-devel libpciaccess-devel mesa-libgbm-devel libva-devel libffi-devel alsa-lib-devel gcc-c++ libstdc++-devel protobuf-compiler acpica-tools yaml-cpp-devel glib2-devel libblkid-devel pixman-devel protobuf-devel libepoxy-devel cmake openssl libzstd-devel SDL2-devel gdb git && dnf clean all

RUN cd /opt && git clone https://github.com/tenclass/mvisor && git clone https://gitlab.freedesktop.org/virgl/virglrenderer

RUN cd /opt/virglrenderer && git reset --hard 8df4cba170940dad9350a99900293adbcef39b6c && meson build --prefix=/usr && ninja install -C build

RUN cd /opt/mvisor && meson setup build -Dsdl=true -Dvgpu=true && meson compile -C build/
```

$ cat mvisor.sh 
```
#!/bin/bash

podman run -d --rm --name mvisor \
--device /dev/kvm \
--device /dev/dri \
-v /data/mvisor:/data:z \
-v /run/user/$(id -u)/:/run/user/$(id -u) \
-v /data/vm/ISO:/ISO:z \
-e WAYLAND_DISPLAY=$WAYLAND_DISPLAY \
-e XDG_RUNTIME_DIR=/run/user/$(id -u) \
-e DISPLAY \
mvisor:latest /opt/mvisor/build/mvisor -c /data/win10.yaml
```
[image]

jpleite, 2024-10-28:
Hi @pg2web I have a fundamental doubt. I'm running Proxmox on a Meteor Lake 125H. GPU to windows vm is not working properly with dkms (error 43). Your suggestion is to install Fedora as main OS, run MVisor and install windows?

madalinignisca, 2025-05-26:
I can't understand how is this issue related to the project. Why are you pointing people to another project?