strongtz/i915-sriov-dkms #210: Support for Core Ultra 7 155H - Meteor Lake

jeeftor, 2024-10-16:
For the life of me I can't seem to get this thing to install on Kernel 6.8.12 with an 155H processor. Can somebody confirm whether its supposed to be supported at this juncture. 

I haven't been able to get past this error

```bash
root@pve4:~# modprobe -v i915
install /bin/false enable_guc=3 max_vfs=7
modprobe: ERROR: ../libkmod/libkmod-module.c:1047 command_do() Error running install command '/bin/false' for module i915: retcode 1
modprobe: ERROR: could not insert 'i915': Invalid argument
root@pve4:~#
```

EPiSKiNG, 2024-10-20:
Following and current have vf showing up in proxmox, but haven't verified functionality.  Will update in the next day or two.  I have the same CPU (beelink GTi14)

EPiSKiNG, 2024-10-20:
Here are my current steps that are allowing VFs to show up in the host from a fresh 8.2.2 install:

```
bash -c "$(wget -qLO - https://github.com/tteck/Proxmox/raw/main/misc/post-pve-install.sh)"

nano /etc/apt/sources.list
#ADD THE FOLLOWING:
	#non-free firmwares
	deb http://deb.debian.org/debian bookworm non-free-firmware

	#non-free drivers and components
	deb http://deb.debian.org/debian bookworm non-free
	
#Install Intel GPU Drivers
apt update && apt install intel-media-va-driver-non-free intel-gpu-tools vainfo -y
apt update && apt install pve-headers-$(uname -r)
apt install git sysfsutils dkms build-* unzip -y
echo "options i915 force_probe=7d55 enable_guc=3" | tee -a /etc/modprobe.d/i915.conf
mkdir -p /lib/firmware/i915
wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_gsc_102.0.0.1511.bin -O /lib/firmware/i915/mtl_gsc_102.0.0.1511.bin
wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_guc_70.6.4.bin -O /lib/firmware/i915/mtl_guc_70.6.4.bin 
wget https://github.com/intel-gpu/intel-gpu-firmware/raw/main/firmware/mtl_huc_8.4.3_gsc.bin -O /lib/firmware/i915/mtl_huc_8.4.3_gsc.bin
reboot

git clone https://github.com/strongtz/i915-sriov-dkms.git
cd ~/i915-sriov-dkms
dkms add .
dkms install -m i915-sriov-dkms -v $(cat VERSION) --force
nano /etc/default/grub
# change GRUB_CMDLINE_LINUX_DEFAULT to intel_iommu=on i915.enable_guc=3 i915.max_vfs=7, or add to it if you have other arguments there already.
update-grub
echo "devices/pci0000:00/0000:00:02.0/sriov_numvfs = 7" > /etc/sysfs.conf
reboot

```

Will continue on my progress after dinner/tomorrow.... Hopefully that gets you pointed in the right direction :)

ccoles146, 2024-10-21:
Did you make any progress with this. I'm currently trying to get an Ultra 5 working? I thought I was making progress using Ubuntu 24.10 with kernel 6.11 but it seems not.

EPiSKiNG, 2024-10-22:
I wasn't able to get the VFs to work properly for the nodes. I gave up and went the LXC route for my needs.  I'm sure it will get ironed out in the next couple of months, but nothing I did got me to the point where the vGPU would function properly in the VM.  I'm also contemplating just doing a full passthrough to an Ubuntu 24.04 VM and running my containers inside there.  Fingers crossed for full 14th gen iGPU support in the future. :)

mm2293, 2024-10-22:
You think full gpu passthrough would work? I've tried to get it working with Windows 11, but I couldn't get rid of error 43

mm2293, 2024-10-22:
> I got the same error 43.

With full passthough?

ccoles146, 2024-10-22:
@mm2293  yes that's full passthrough.
I've not got a VM to boot yet using SRIOV.
I just installed the bbaa-bbaa branch to see if it helped with kernel 6.11 but I get errors still.

```sh
    1.841400] i915: loading out-of-tree module taints kernel.
[    1.841476] i915: module verification failed: signature and/or required key missing - tainting kernel
[    2.114336] i915 0000:00:02.0: Force probing unsupported Device ID 7dd5, tainting kernel
[    2.114390] i915 0000:00:02.0: Running in SR-IOV PF mode
[    2.114556] i915 0000:00:02.0: [drm] GT0: Incompatible option enable_guc=3 - HuC is not supported!
[    2.115596] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.115607] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.115644] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    2.115962] i915 0000:00:02.0: [drm] *ERROR* Unexpected child device config size 40 (expected 39 for VBT version 256)
[    2.128460] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.138339] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.23)
[    2.157480] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.29.2
[    2.170018] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    2.170023] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    2.170256] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    2.176915] mei_gsc_proxy 0000:00:16.0-0f73db04-97ab-4125-b893-e904ad0d5464: bound 0000:00:02.0 (ops i915_gsc_proxy_component_ops [i915])
[    2.177251] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.29.2
[    2.177253] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[    2.200368] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for clear media
[    2.200688] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[    2.200689] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[    2.200767] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
[    2.203403] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.237552] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 1
[    2.251708] i915 0000:00:02.0: 7 VFs could be associated with this PF
[    2.285025] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.286645] fbcon: i915drmfb (fb1) is primary device
[    2.348121] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.0.10.1878, svn 1)
[    2.368152] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
[    3.436565] i915 0000:00:02.0: [drm] fb1: i915drmfb frame buffer device
[    9.409793] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
[    9.409875] i915 0000:00:02.1: Force probing unsupported Device ID 7dd5, tainting kernel
[    9.409885] i915 0000:00:02.1: enabling device (0000 -> 0002)
[    9.409948] i915 0000:00:02.1: [drm] Cannot read valid display IP version; disabling display.
[    9.409957] i915 0000:00:02.1: Running in SR-IOV VF mode
[    9.409958] i915 0000:00:02.1: [drm] preliminary graphics version 12.70 media version 13.00
[    9.410151] i915 0000:00:02.1: [drm] GT0: Incompatible option enable_guc=3 - HuC is not supported!
[    9.410295] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[    9.410333] i915 0000:00:02.1: [drm] GT0: GMD_ID 0x311c004 version 12.71 step B0
[    9.411368] i915 0000:00:02.1: [drm] GT1: GUC: interface version 0.1.13.4
[    9.411537] i915 0000:00:02.1: [drm] GT1: GMD_ID 0x3400008 version 13.0 step C0
[    9.412282] i915 0000:00:02.1: [drm] VT-d active for gfx access
[    9.412332] i915 0000:00:02.1: [drm] Using Transparent Hugepages
[    9.412940] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[    9.416700] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
[    9.417249] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    9.417252] i915 0000:00:02.1: HuC firmware N/A
[    9.421001] i915 0000:00:02.1: [drm] GT1: GUC: interface version 0.1.13.4
[    9.423490] i915 0000:00:02.1: [drm] GT1: GUC: interface version 0.1.13.4
[    9.423679] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
[    9.423681] i915 0000:00:02.1: HuC firmware PRELOADED
[    9.426713] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
[    9.426834] [drm] Initialized i915 1.6.0 for 0000:00:02.1 on minor 2
[    9.427321] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
[    9.427325] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
```

EPiSKiNG, 2024-10-22:
For full GPU passthrough I imagine you just need to clean install latest proxmox, spin up a VM and pass through the GPU.  I don't think there's anything else to it than that?

mm2293, 2024-10-22:
> For full GPU passthrough I imagine you just need to clean install latest proxmox, spin up a VM and pass through the GPU. I don't think there's anything else to it than that? I don't even think you need to install the drivers on the host or do any other kernel/boot edits.

I had the same thought until I got the error 43 in Windows 11…

mm2293, 2024-10-22:
> @mm2293 yes that's full passthrough. I've not got a VM to boot yet using SRIOV. I just installed the bbaa-bbaa branch to see if it helped with kernel 6.11 but I get errors still.
> 
> ```shell
>     1.841400] i915: loading out-of-tree module taints kernel.
> [    1.841476] i915: module verification failed: signature and/or required key missing - tainting kernel
> [    2.114336] i915 0000:00:02.0: Force probing unsupported Device ID 7dd5, tainting kernel
> [    2.114390] i915 0000:00:02.0: Running in SR-IOV PF mode
> [    2.114556] i915 0000:00:02.0: [drm] GT0: Incompatible option enable_guc=3 - HuC is not supported!
> [    2.115596] i915 0000:00:02.0: [drm] VT-d active for gfx access
> [    2.115607] i915 0000:00:02.0: vgaarb: deactivate vga console
> [    2.115644] i915 0000:00:02.0: [drm] Using Transparent Hugepages
> [    2.115962] i915 0000:00:02.0: [drm] *ERROR* Unexpected child device config size 40 (expected 39 for VBT version 256)
> [    2.128460] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
> [    2.138339] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.23)
> [    2.157480] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.29.2
> [    2.170018] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
> [    2.170023] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
> [    2.170256] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
> [    2.176915] mei_gsc_proxy 0000:00:16.0-0f73db04-97ab-4125-b893-e904ad0d5464: bound 0000:00:02.0 (ops i915_gsc_proxy_component_ops [i915])
> [    2.177251] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.29.2
> [    2.177253] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
> [    2.200368] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for clear media
> [    2.200688] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
> [    2.200689] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
> [    2.200767] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
> [    2.203403] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
> [    2.237552] [drm] Initialized i915 1.6.0 for 0000:00:02.0 on minor 1
> [    2.251708] i915 0000:00:02.0: 7 VFs could be associated with this PF
> [    2.285025] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
> [    2.286645] fbcon: i915drmfb (fb1) is primary device
> [    2.348121] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.0.10.1878, svn 1)
> [    2.368152] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
> [    3.436565] i915 0000:00:02.0: [drm] fb1: i915drmfb frame buffer device
> [    9.409793] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=io+mem
> [    9.409875] i915 0000:00:02.1: Force probing unsupported Device ID 7dd5, tainting kernel
> [    9.409885] i915 0000:00:02.1: enabling device (0000 -> 0002)
> [    9.409948] i915 0000:00:02.1: [drm] Cannot read valid display IP version; disabling display.
> [    9.409957] i915 0000:00:02.1: Running in SR-IOV VF mode
> [    9.409958] i915 0000:00:02.1: [drm] preliminary graphics version 12.70 media version 13.00
> [    9.410151] i915 0000:00:02.1: [drm] GT0: Incompatible option enable_guc=3 - HuC is not supported!
> [    9.410295] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
> [    9.410333] i915 0000:00:02.1: [drm] GT0: GMD_ID 0x311c004 version 12.71 step B0
> [    9.411368] i915 0000:00:02.1: [drm] GT1: GUC: interface version 0.1.13.4
> [    9.411537] i915 0000:00:02.1: [drm] GT1: GMD_ID 0x3400008 version 13.0 step C0
> [    9.412282] i915 0000:00:02.1: [drm] VT-d active for gfx access
> [    9.412332] i915 0000:00:02.1: [drm] Using Transparent Hugepages
> [    9.412940] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
> [    9.416700] i915 0000:00:02.1: [drm] GT0: GUC: interface version 0.1.13.4
> [    9.417249] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
> [    9.417252] i915 0000:00:02.1: HuC firmware N/A
> [    9.421001] i915 0000:00:02.1: [drm] GT1: GUC: interface version 0.1.13.4
> [    9.423490] i915 0000:00:02.1: [drm] GT1: GUC: interface version 0.1.13.4
> [    9.423679] i915 0000:00:02.1: GuC firmware PRELOADED version 0.0 submission:SR-IOV VF
> [    9.423681] i915 0000:00:02.1: HuC firmware PRELOADED
> [    9.426713] i915 0000:00:02.1: [drm] PMU not supported for this GPU.
> [    9.426834] [drm] Initialized i915 1.6.0 for 0000:00:02.1 on minor 2
> [    9.427321] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=none,decodes=none:owns=io+mem
> [    9.427325] i915 0000:00:02.1: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=none:owns=none
> ```

I think we have to wait for the xe driver to Support SRIOV :/

johntdavis84, 2024-10-23:
> Are the xe drivers something attached to a kernel release - or we have to just wait on them to be written and they can be back-fit into an existing kernel?

The Xe driver is the current generation of Intel iGPU driver. It’s been in development for quite a while now, and as of Kernel 6.12 became the default in-kernel driver, replacing the i915 kernel driver.

iGPU (and A-series dGPU?) SR-IOV support was kind of shoe-horned into i915 based on the work done to get SR-IOV working in Xe, from my understanding. However, while the Xe driver should have more stable and featureful SR-IOV support than i915 going forward, the version of the Xe driver in the current 6.12 kernel does not enable SR-IOV support for iGPUs.

Given the latest dev notes, a lot of us are guessing we might not see SR-IOV enabled in Xe for iGPUs until at least 6.13.

-- -- --
John T Davis

mm2293, 2024-10-27:
> For full GPU passthrough I imagine you just need to clean install latest proxmox, spin up a VM and pass through the GPU. I don't think there's anything else to it than that? I don't even think you need to install the drivers on the host or do any other kernel/boot edits.

Have you tried the full passthrough?

EPiSKiNG, 2024-10-27:
Indeed, and I managed to get it to work properly to passthrough to Ubuntu
server 24.04.  If I remember correctly, I installed drivers in both host
and node.

mm2293, 2024-10-27:
> Indeed, and I managed to get it to work properly to passthrough to Ubuntu server 24.04. If I remember correctly, I installed drivers in both host and node.
> […](#)

Okay, hmm, what the hell is the problem with Windows then 😫

mm2293, 2024-10-28:
> Indeed, and I managed to get it to work properly to passthrough to Ubuntu server 24.04. If I remember correctly, I installed drivers in both host and node.
> […](#)

Could you please share your configs?

bartje, 2024-11-13:
> > Indeed, and I managed to get it to work properly to passthrough to Ubuntu server 24.04. If I remember correctly, I installed drivers in both host and node.
> 
> > […](#)
> 
> > On Sun, Oct 27, 2024, 12:33 AM mm2293 ***@***.***> wrote: For full GPU passthrough I imagine you just need to clean install latest proxmox, spin up a VM and pass through the GPU. I don't think there's anything else to it than that? I don't even think you need to install the drivers on the host or do any other kernel/boot edits. Have you tried the full passthrough?
> 
> 
> Could you please share your configs?

I am also interested how to do it exactely, which drivers to install?

bartje, 2024-11-14:
EPisKiNG,
thanks for the response, but I'am rather a beginner in this stuff.

Add the following lines to your `/etc/default/grub` file:
     GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=o
update-grub and reboot

then adding a line to de vfio.conf file (xxx:xxx is the ID that can be found with lspci -nn | grep -i vga)
echo "options vfio-pci ids=xxxx:xxxx" > /etc/modprobe.d/vfio.conf
update-initramfs -u
reboot.

I do not need to install the intel drivers on the host?

EPiSKiNG, 2024-11-16:
I believe the drivers should be installed, but if you're using modern PVE, you should already have them in the system.  Technically, it shouldn't matter I don't think?  Since we're passing through the GPU device in a "raw" manner rather than utilizing the vGPU functions.  

Good luck!

EPiSKiNG, 2024-11-22:
Proxmox 8.3 released with option for 6.11 kernel.   Wonder if that makes things any easier?

thk-socal, 2024-11-28:
I have 8.3 running. For an LLM VM on Ubuntu 24.04 LTS, I passed through the RAW Devices to the VM.
0000:00:2 for the GPU
0000:00:0b for the NPU

On the VM, I installed both the GPU and NPU drivers and both devices are available and working. They even pass through to the docker container where I am running Intel Analytics accelerated ollama.

I installed nothing extra on the host and just pass through the raw device to the VM.

bartje, 2024-11-28:
@thk-socal,
what steps did you follow, I achieved a passthrough (see a few posts above) of the GPU, but never succeeded in installing the drivers on the guest (debian or ubunto)?

thk-socal, 2024-11-28:
@bartje I used the 24.04 section from Intel to install the drivers on my Ubuntu 24.04 LTS server. I am using the GPU as an AI processor, so it is NOT marked as Primary GPU in the hardware device passthrough on Proxmox:

https://dgpu-docs.intel.com/driver/client/overview.html

I then confirmed that it was indeed recognized and running via: cliinfo

bartje, 2024-11-28:
That worked! thanks.  Now trying to let it work in Debian...

mm2293, 2025-01-28:
Bad news:
https://www.intel.com/content/www/us/en/support/articles/000093216/graphics/processor-graphics.html

EPiSKiNG, 2025-01-30:
RIP.

jeeftor, 2025-01-31:
[image]

This one? 

That makes no sense to me...

tristan-k, 2025-01-31:
Didn't @Celesrenata already proove that it's possible to use SR-IOV with Meteor Lake on her [NixOS configs](https://github.com/celesrenata/nixos-k3s-configs)?

celesrenata, 2025-01-31:
> Didn't @Celesrenata already proove that it's possible to use SR-IOV with Meteor Lake on her [NixOS configs](https://github.com/celesrenata/nixos-k3s-configs)? 

Yes

celesrenata, 2025-02-03:
[image]

I have played around with a few options.

The old base using revision: `e26ce8952e465762fc0743731aa377ec0b2889ff` works great for me in 6.6

I've had nothing but stability issues when moving to 6.9+ and newer revisions. I've tried digging into crashlogs, but it dies before writing the logs.

I may be missing something in my configs for higher level compatibility.

But they indeed run. I had success with AI on them, no success on accelerated video in NixOS or Windows yet.

chenxiex, 2025-02-08:
I have tried with Ultra 5 125U. The module can be installed and the VFs show on the host. But when I passthrough the VF to a Windows guest, it doesn't work. The "pro" Windows driver recommanded in README doesn't support meteor lake and won't install at all, and the common driver reports error 43.

celesrenata, 2025-02-10:
I was able to identify the source of my crashes using coredump. It appears to be coming from `ClusterPlex` which is are Kubernetes pods utilizing the `vaapi` driver.
Which confirms one of my hunches. It did appear much more stable than previous 6.11 builds which would crash during boot.

```bash
[root@gremlin-3:/var/lib/systemd/coredump]# ls -l
total 30672
-rw-r-----+ 1 root root 5188128 Feb  9 02:16 'core.Plex\x20Media\x20Serv.1000.081070912f1541a8a6aa986e6cbffb0d.1270037.1739096181000000.zst'
-rw-r-----+ 1 root root 4948134 Feb  9 02:26 'core.Plex\x20Media\x20Serv.1000.081070912f1541a8a6aa986e6cbffb0d.1284558.1739096769000000.zst'
-rw-r-----+ 1 root root 5021939 Feb  9 02:36 'core.Plex\x20Media\x20Serv.1000.081070912f1541a8a6aa986e6cbffb0d.1299055.1739097360000000.zst'
-rw-r-----+ 1 root root 4469084 Feb  9 02:45 'core.Plex\x20Media\x20Serv.1000.081070912f1541a8a6aa986e6cbffb0d.1313381.1739097948000000.zst'
-rw-r-----+ 1 root root 7440421 Feb  9 02:06 'core.Plex\x20Media\x20Serv.1000.081070912f1541a8a6aa986e6cbffb0d.52116.1739095587000000.zst'
-rw-r-----+ 1 root root 4322103 Feb  6 19:55 'core.Plex\x20Transcoder.1000.e958575896b44a9994e63d78f17c810b.1108347.1738900542000000.zst'
```

I also attempted to update just the guest systems to newer drivers, I didn't expect it to work, it didn't due to a compatibility between graphics IP versions `12.70` and `13.00`

RomRider, 2025-02-14:
I can't even get full passthrough to work... Ubuntu 24.04 and 24.10... but if I plug a screen through HDMI, I get the console from the guest
```
ansible@swarm01:~$ vainfo
Trying display: wayland
Trying display: x11
error: can't connect to X server!
error: failed to initialize display
```

* Running pve 8.3.3
* Blacklisted i915 and xe on the host
* `GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on iommu=pt"` on the host
* installed the linux drivers from this link on the guest (shared above): https://dgpu-docs.intel.com/driver/client/overview.html

EPiSKiNG, 2025-03-05:
Nice!  Gonna give this a shot!  

Thanks!!

EPiSKiNG, 2025-03-05:
Are you running this on the GTi-14?  If so, did you change any BIOS settings away from default?

Thanks!

jpleite, 2025-03-05:
> GTi14

Running on minisforum 125H but is the same GPU family. Did not change anything on BIOS. BTW. I was able also to run Openweb UI on a Ubuntu 22.04 LXC. Used this docker https://github.com/mattcurf/ollama-intel-gpu. I confirm it is working like a charm using Arc on chip GPUs

EPiSKiNG, 2025-03-05:
Nice, so LXCs are also able to use the GPU.  Fantastic!

jpleite, 2025-03-05:
@mm2293 your output matches mine. Did you select device 0000:00:02.7 for example in PCI device like my screenshot? Also please check Arc driver in windows is version 32.0.101.6129

mm2293, 2025-03-05:
> [@mm2293](https://github.com/mm2293) your output matches mine. Did you select device 0000:00:02.7 for example in PCI device like my screenshot? Also please check Arc driver in windows is version 32.0.101.6129

Yeah, I've selected 0000:00:02.7 with all the settings on your screenshots. I also installed driver 32.0.101.6129 on the second try - but unfortunately no luck.. Did you try a newer driver too or did you install the old driver intentionally?

Thanks!

jpleite, 2025-03-06:
> Yeah, I've selected 0000:00:02.7 with all the settings on your screenshots. I also installed driver 32.0.101.6129 on the second try - but unfortunately no luck.. Did you try a newer driver too or [...]

Tried many drivers until this one worked out. Thought it would work for all but it seems black magic still. I kept trying installing the drivers since November last year

mm2293, 2025-03-06:
> Tried many drivers until this one worked out. Thought it would work for all but it seems black magic still. I kept trying installing the drivers since November last year

Hmm, okay, strange.. What Windows 11 Version are you on right now?

Thanks!

jpleite, 2025-03-06:
> Hmm, okay, strange.. What Windows 11 Version are you on right now?
>
> Thanks!

Using Win 11 Pro 23H2. Also set display to none and I'm using Windows App Beta to remotely access from Mac

[image]

[image]

mm2293, 2025-03-06:
> Using Win 11 Pro 23H2. Also set display to none and I'm using Windows App Beta to remotely access from Mac
>
> [image]
>
> [image]

Ok, thanks for your support. Unfortunately I'm not getting it to work.
btw, do you have similiar log entries in journalctl?

```
Mar 06 17:18:31 pve1 QEMU[59055]: kvm: VFIO_MAP_DMA failed: Invalid argument
Mar 06 17:18:31 pve1 QEMU[59055]: kvm: vfio_container_dma_map(0x59ff7d813e60, 0x383800000000, 0x1000000, 0x738d09400000) = -22 (Invalid argument)
```

Thanks!

jpleite, 2025-03-08:
```
root@milknucproxmox:~# journalctl
Oct 17 08:23:29 milknucproxmox kernel: Linux version 6.8.4-2-pve (build@proxmox) (gcc (Debian 12.2.0-14) 12.2.0,>
Oct 17 08:23:29 milknucproxmox kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.4-2-pve root=/dev/mapper/pve-r>
Oct 17 08:23:29 milknucproxmox kernel: KERNEL supported cpus:
Oct 17 08:23:29 milknucproxmox kernel:   Intel GenuineIntel
Oct 17 08:23:29 milknucproxmox kernel:   AMD AuthenticAMD
Oct 17 08:23:29 milknucproxmox kernel:   Hygon HygonGenuine
Oct 17 08:23:29 milknucproxmox kernel:   Centaur CentaurHauls
Oct 17 08:23:29 milknucproxmox kernel:   zhaoxin   Shanghai  
Oct 17 08:23:29 milknucproxmox kernel: x86/split lock detection: #AC: crashing the kernel on kernel split_locks >
Oct 17 08:23:29 milknucproxmox kernel: BIOS-provided physical RAM map:
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000000100000-0x000000005ea29fff] usable
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x000000005ea2a000-0x000000005ea2afff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x000000005ea2b000-0x0000000063263fff] usable
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000063264000-0x0000000066f06fff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000066f07000-0x0000000066fecfff] ACPI data
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000066fed000-0x0000000067106fff] ACPI NVS
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000067107000-0x0000000069f58fff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000069f59000-0x0000000069ffefff] type 20
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x0000000069fff000-0x0000000069ffffff] usable
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x000000006a000000-0x00000000787fffff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
Oct 17 08:23:29 milknucproxmox kernel: BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
```

mm2293, 2025-03-09:
> ```
> root@milknucproxmox:~# journalctl
> Oct 17 08:23:29 milknucproxmox kernel: Linux version 6.8.4-2-pve (build@proxmox) (gcc (Debian 12.2.0-14) 12.2.0,>
> Oct 17 08:23:29 milknucproxmox kernel: Command [...]

So I guess there is no „VFIO_MAP_DMA failed“ in your case?

tristan-k, 2025-03-11:
@mm2293 @jpleite 

I'm [still unable](https://github.com/strongtz/i915-sriov-dkms/issues/195#issuecomment-2363549281) to use SR-IOV with a Intel NUC14 `Intel Core Ultra 5 125H` on Proxmox `8.4.3` and Kernel `6.8.12-2-pve`. I followed the [instructions](https://github.com/strongtz/i915-sriov-dkms?tab=readme-ov-file#pve-host-installation-steps-tested-kernel-68) but I'm getting `no driver bound to device; cannot configure SR-IOV` errors.

```
$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)

$ dmesg | grep SR-IOV
[    4.065205] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV
[    4.065232] pci 0000:00:02.0: no driver bound to device; cannot configure SR-IOV

$ dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-8-pve root=/dev/mapper/pve-root ro noquiet nosplash intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
[    0.081555] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-8-pve root=/dev/mapper/pve-root ro noquiet nosplash intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe
```

I cant even write to `/sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs`

```
$ echo 1 > /sys/devices/pci0000:00/0000:00:02.0/sriov_numvfs
-bash: echo: write error: No such file or directory
```

mm2293, 2025-03-11:
> modprobe: ERROR: could not insert 'i915': Key was rejected by service

Looks like you have enabled secure boot. 
Try the following:

[image]
Reboot an enter password.
Source: https://www.derekseaman.com/2024/07/proxmox-ve-8-2-windows-11-vgpu-vt-d-passthrough-with-intel-alder-lake.html

tristan-k, 2025-03-11:
@mm2293 

That was it. Now the SR-IOV devices are loaded but I'm getting a lot of errors:

```
$ dmesg | grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-8-pve root=/dev/mapper/pve-root ro noquiet nosplash intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe i915.force_probe=7d55
[    0.081289] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-6.8.12-8-pve root=/dev/mapper/pve-root ro noquiet nosplash intel_iommu=on i915.enable_guc=3 i915.max_vfs=7 module_blacklist=xe i915.force_probe=7d55
[    3.679243] i915 0000:00:02.0: Force probing unsupported Device ID 7d55, tainting kernel
[    3.679574] i915 0000:00:02.0: [drm] Found METEORLAKE (device ID 7d55) display version 14.00 stepping C0
[    3.679967] i915 0000:00:02.0: Running in SR-IOV PF mode
[    3.681828] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    3.748385] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.748408] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    3.761281] i915 0000:00:02.0: vgaarb: VGA decodes changed: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.769636] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/mtl_dmc.bin (v2.23)
[    3.791901] i915 0000:00:02.0: [drm] GT0: GuC firmware i915/mtl_guc_70.bin version 70.36.0
[    3.803854] i915 0000:00:02.0: [drm] GT0: GUC: submission enabled
[    3.803860] i915 0000:00:02.0: [drm] GT0: GUC: SLPC enabled
[    3.804105] i915 0000:00:02.0: [drm] GT0: GUC: RC enabled
[    3.808377] mei_gsc_proxy 0000:00:16.0-0f73db04-97ab-4125-b893-e904ad0d5464: bound 0000:00:02.0 (ops i915_gsc_proxy_component_ops [i915])
[    3.808733] i915 0000:00:02.0: [drm] GT1: GuC firmware i915/mtl_guc_70.bin version 70.36.0
[    3.808736] i915 0000:00:02.0: [drm] GT1: HuC firmware i915/mtl_huc_gsc.bin version 8.5.4
[    3.834204] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for clear media
[    3.834594] i915 0000:00:02.0: [drm] GT1: GUC: submission enabled
[    3.834596] i915 0000:00:02.0: [drm] GT1: GUC: SLPC enabled
[    3.834676] i915 0000:00:02.0: [drm] GT1: GUC: RC enabled
[    3.836976] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    3.910162] [drm] Initialized i915 1.6.0 20230929 for 0000:00:02.0 on minor 1
[    3.911880] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.974720] i915 0000:00:02.0: [drm] GT1: Loaded GSC firmware i915/mtl_gsc_1.bin (cv1.0, r102.1.15.1926, svn 1)
[    3.994974] i915 0000:00:02.0: [drm] GT1: HuC: authenticated for all workloads
[    4.023459] fbcon: i915drmfb (fb0) is primary device
[    4.023526] i915 [CRTC:80:pipe A] fastset requirement not met in dpll_hw_state.cx0pll C10
[    4.023527] i915 expected:
[    4.023528] i915 found:
[    4.023528] i915 [CRTC:80:pipe A] fastset requirement not met in infoframes.enable (expected 0x00000010, found 0x00000031)
[    4.023530] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_hdisplay (expected 640, found 1280)
[    4.023532] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_htotal (expected 784, found 1688)
[    4.023532] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_hblank_start (expected 640, found 1280)
[    4.023533] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_hblank_end (expected 784, found 1688)
[    4.023534] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_hsync_start (expected 648, found 1328)
[    4.023534] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_hsync_end (expected 744, found 1440)
[    4.023535] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_vdisplay (expected 480, found 1024)
[    4.023535] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_vblank_start (expected 480, found 1024)
[    4.023536] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_vsync_start (expected 482, found 1025)
[    4.023537] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_vsync_end (expected 484, found 1028)
[    4.023537] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_vtotal (expected 509, found 1066)
[    4.023538] i915 [CRTC:131:pipe B] fastset requirement not met in hw.pipe_mode.crtc_vblank_end (expected 509, found 1066)
[    4.023539] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_hdisplay (expected 640, found 1280)
[    4.023539] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_htotal (expected 784, found 1688)
[    4.023540] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_hblank_start (expected 640, found 1280)
[    4.023540] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_hblank_end (expected 784, found 1688)
[    4.023541] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_hsync_start (expected 648, found 1328)
[    4.023542] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_hsync_end (expected 744, found 1440)
[    4.023542] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_vdisplay (expected 480, found 1024)
[    4.023543] i915 [CRTC:131:pipe B] fastset requirement not met in hw.adjusted_mode.crtc_vblank_start (expected 480, found 1024)
[... cut ...]
[   59.388308] i915 0000:00:02.0: [drm] *ERROR* GT1: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[   59.388318] i915 0000:00:02.0: [drm] GT1: IOV: Failed to save VF6 state (-EPROTO)
[   59.388526] i915 0000:00:02.0: [drm] *ERROR* GT0: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[   59.388535] i915 0000:00:02.0: [drm] GT0: IOV: Failed to save VF7 state (-EPROTO)
[   59.388715] i915 0000:00:02.0: [drm] *ERROR* GT1: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
[   59.388724] i915 0000:00:02.0: [drm] GT1: IOV: Failed to save VF7 state (-EPROTO)
```

tristan-k, 2025-03-11:
I installed a Ubuntu 24.04.2 VM and followed the [Linux Guest Installation Steps](https://github.com/strongtz/i915-sriov-dkms?tab=readme-ov-file#linux-guest-installation-steps-tested-ubuntu-2404kernel-68) but I'm hitting another roadblock. The SR-IOV device is recognized in the VM but is seems no driver is loaded. Secure Boot is definitley disabled.

On the guest:

```
$ cat /etc/default/grub | grep GRUB_CMDLINE_LINUX_DEFAULT
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash i915.enable_guc=3 module_blacklist=xe i915.force_probe=7d55"

$ dmesg | grep i915
               use xe.force_probe='7d55' and i915.force_probe='!7d55'
[    3.522689] i915: loading out-of-tree module taints kernel.
[    3.522736] i915: module verification failed: signature and/or required key missing - tainting kernel
[    3.834825] i915 0000:00:10.0: Your graphics device 7d55 is not properly supported by i915 in this
               kernel version. To force driver probe anyway, use i915.force_probe=7d55

$ inxi -G | grep Device-1
  Device-1: Intel Meteor Lake-P [Intel Arc Graphics] driver: N/A

$ lspci | grep VGA
00:10.0 VGA compatible controller: Intel Corporation Meteor Lake-P [Intel Arc Graphics] (rev 08)

$ lspci -n -s 00:10 -v
00:10.0 0300: 8086:7d55 (rev 08) (prog-if 00 [VGA controller])
	Subsystem: 1043:88c8
	Physical Slot: 16
	Flags: bus master, fast devsel, latency 0
	Memory at 800000000 (64-bit, prefetchable) [size=16M]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Kernel modules: xe, i915
```

tristan-k, 2025-03-11:
I also installed Windows 11 24H2 VM with Intel Arc Drivers `32.0.101.6129`. At first I was getting `Error Code 43` in the device manager. After adding `args: -cpu 'host,vendor=GenuineIntel'` to `/etc/pve/qemu-server/VM-ID.conf` the error went away. If I disable the `Default: Display` in Proxmox and set the SR-IOV device to `Primary GPU` and connect to the VM via RDP or VNC I only get a black screen.

Here is a screenshot with `Default: Display` enabled and `Primary GPU` disabled for the SR-IOV `Raw device` `0000:00:02.2` while running Unigine Heaven benchmark.

[image]

My guess is that the SR-IOV device isn't able to create a virtual display.

mm2293, 2025-03-12:
> args: -cpu 'host,vendor=GenuineIntel'

Thank you so much for this. This was the last piece of the puzzle. I finally got it working.
Right now, I'm using the newest Intel drivers and W11 23H2. I have no issues with Display set to none and Raw device 0000:00:02.7.

tristan-k, 2025-03-12:
Glad I could help you. Are you using a Metero Lake or Alder Lake System?

mm2293, 2025-03-12:
> Glad I could help you. Are you using a Metero Lake or Alder Lake System?

Meteor Lake: Intel Ultra 155H.

You may want to try with Windows 11 23H2

ValeZAA, 2025-03-28:
You need kernel 6.13.7 in experimental Debian

https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg2019570.html

paulandrewcrouch, 2025-04-06:
I hope this isn't too far off the mark (if it is, I'll delete it).   I've just purchased the Geekom GT1 with Meteor Lake with the hope of using it to play games with my daughter.   Alas am getting the 43 issue in Windows (using Proxmox 8.3.5).   I've tried the xe driver, i915 driver and vfio-pci for full pass through.   I can use it on full pass through in Ubuntu Frigate VM but nothign on Windows.   What I really want to do is share this across a W11 Pro install (for gaming) and Ubuntu for Frigate.   I thought this thread concluded it wasn't possible and the Intel site says SRIOV isn't available on Meteor Lake, but then these last posts suggests that perhaps it is.   Before I spend another day messing about, is it possible?   Thank you so much.

cristovamperes, 2025-07-06:
I'm having the same problems using an Intel 265K

mhemantha, 2025-07-12:
> I'm having the same problems using an Intel 265K

With proxmox 8.4 and the [opt-in 6.14.5 kernel](https://forum.proxmox.com/threads/opt-in-linux-6-14-kernel-for-proxmox-ve-8-available-on-test-no-subscription.164497/), sriov driver here worked without a hitch for win10 passthrough for me on a 265K thanks to the excellent instructions by @jpleite and hints by @tristan-k about i915.force_probe and qemu -cpu args

EPiSKiNG, 2025-07-12:
> With proxmox 8.4 and the [opt-in 6.14.5 kernel](https://forum.proxmox.com/threads/opt-in-linux-6-14-kernel-for-proxmox-ve-8-available-on-test-no-subscription.164497/), sriov driver here worked [...]

Is that for vGPU or straight passthrough?

jeeftor, 2025-07-14:
`apt update`
`apt install proxmox-kernel-6.14`
`reboot`

casperfrx, 2025-07-15:
Thanks but the problem for me the problem was never the latest kernel. I always ran Proxmox on 6.14 because I run fairly new hardware. But the many possible configurations on modprobe and the like are all over the place.

mhemantha, 2025-07-17:
> Is that for vGPU or straight passthrough?

with the vGPU. Note: The 265k (which is _Arrow Lake_) host is running headless, I haven't tested graphics output to a display. But I was able to utilise the virtual iGPU unit in Win10 for AI inference with openvino.

On ubuntu 24.04 LTS _privileged_ lxc, errors same as what @duggasco wrote about came up when intel_gpu_top was invoked. Those stopped when I added the `-d sriov` filter to that command .    
>[115685.095671] i915 0000:00:02.0: [drm] *ERROR* GT1: GUC: CT: Sending action 0x550b failed (-EIO) status=0XE0000102
>[115685.095672] i915 0000:00:02.0: [drm] GT1: IOV: Failed to save VF2 state (-EPROTO

I saw vGPU utilisation (without the above or any other errors) in both ffmpeg  (forcing it with `-hwaccel vaapi  -init_hw_device vaapi=va:/dev/dri/renderD130   -init_hw_device opencl@va`)  and in ai inference with ipex ollama, but again since the machine is remote, wasn't able to test display output. In this lxc ubuntu guest, I've installed [nonfree drivers from intel](https://dgpu-docs.intel.com/driver/client/overview.html) I've passed only one of the vGPUs to this lxc.

[5 more replies]