strongtz/i915-sriov-dkms #212: TrueNAS Scale build

blaz0r, 2024-10-20:
Hi,

I have a TrueNAS Scale install which runs 6.6.44  -  unfortunately the build errors out with ISO99 macro expansion errors, I believe because of the "+" in the uname -a: Linux truenas 6.6.44-production+truenas

Updating the Makefile as below to convert the + to _ seemed like a quick win, but unfortunately results in even more, different, macro errors.

EXTRAVERSION_DEFINE := $(shell var=$(EXTRAVERSION_NAME); var=$$(echo $$var | sed 's/[-+]/_/g' | awk '{print toupper($$0)}'); echo EXTRAVERSION_$${var:-EMPTY})

Logs attached from dkms build.

I'm really at my limit of c/build tool chain knowledge here, any ideas how to advance?  

Thank you, and kind regards,

[make.log.underscore.txt](https://github.com/user-attachments/files/17450669/make.log.underscore.txt)
[make.log.plus.txt](https://github.com/user-attachments/files/17450670/make.log.plus.txt)

scyto, 2024-11-18:
did you ever solve this?

blaz0r, 2024-11-18:
> did you ever solve this?
> 
> my log looks like this, not sure i am getting even as far as you did..
> 
> ```
> root@truenas[/home/truenas_admin/i915-sriov-dkms]# cat /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/make.log
> DKMS make.log for i915-sriov-dkms-2024.09.21 for kernel 6.6.44-production+truenas (x86_64)
> Sun Nov 17 19:21:13 PST 2024
> make: Entering directory '/usr/src/linux-headers-6.6.44-production+truenas'
> KERNELRELEASE=6.6.44-production+truenas
> KERNELVERSION=6.6.44
> EXTRAVERSION_MAJOR=0
> EXTRAVERSION_MINOR=0
> EXTRAVERSION_NAME=production+truenas
> EXTRAVERSION_DEFINE=EXTRAVERSION_PRODUCTION+TRUENAS
> LSBRELEASE=12
> LSBRELEASE_MAJOR=12
> LSBRELEASE_MINOR=0
> LSBRELEASE_NAME=Debian
> LSBRELEASE_DEFINE=RELEASE_DEBIAN
> GUCFIRMWARE_MAJOR=1
> GUCFIRMWARE_MINOR=9
> make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/i915.mod] Killed
> make[2]: *** Waiting for unfinished jobs....
> make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build/drivers/gpu/drm/i915/i915_driver.o] Killed
> make[1]: *** [/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: /var/lib/dkms/i915-sriov-dkms/2024.09.21/build] Error 2
> make: *** [Makefile:237: __sub-make] Error 2
> make: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
> ```

Sadly not.  I’ve basically resigned myself to waiting for the next truenas scale release to hopefully have the presumed next lts kernel 6.12 that should have all these sr-iov changes merged… so i guess a few months.  If not, I’ll consider a switch back to proxmox with truenas as a guest.

szlcug, 2025-02-28:
did you ever solve this?

christaikobo, 2025-05-20:
@blaz0r hi, do you mind sharing some experience as to how to run truenas scale as a VM guest under PVE?

I am running the 25.04 version of truenas scale in a VM.

Haven't been able to make it work. `/dev/dri` is empty, and `dmesg | grep i915` shows

```
[   18.980961] i915 0000:06:10.0: [drm] Found ALDERLAKE_S (device ID 4692) display version 12.00 stepping C0
[   20.982802] i915 0000:06:10.0: [drm] *ERROR* Device is non-operational; MMIO access returns 0xFFFFFFFF!
[   20.983174] i915 0000:06:10.0: [drm] *ERROR* Device initialization failed (-5)
[   20.983178] i915 0000:06:10.0: probe with driver i915 failed with error -5
```

christaikobo, 2025-06-11:
I figured it out. Install headers on a normal system, build it, and copy and replace the i915.ko and kvmgt.ko to truenas. It works.

GavinIves, 2025-07-11:
> I figured it out. Install headers on a normal system, build it, and copy and replace the i915.ko and kvmgt.ko to truenas. It works.
> 
> If you don't want to go through the hassle, here are the files for 25.04. [prebuilt.zip](https://github.com/user-attachments/files/20696724/prebuilt.zip)

What is your kernel version? I have 25.04.1. Your module does not work.

BeingNoAbility, 2026-11-07:
Hi, I figured out how to properly build i915-sriov-dkms on TrueNAS 25.10.8.
Here is the instructions.

To avoid modifying your production bare-metal system, perform this build inside a clean TrueNAS 25.10.8 VM and copy the resulting .ko kernel module over once compiled.

**Why this works:** Standard sudo shells in TrueNAS enforce sudo auditing, which gets overloaded during heavy compilation and causes the build to fail. Spawning a clean root session via systemd-run bypasses this auditing overhead.

### 1.Access the root shell; Log in to the TrueNAS shell and elevate to root:
`sudo -i
`
### 2.Start a clean root session (bypasses sudo auditing)
`systemd-run --pty --uid=0 --gid=0 /bin/bash
`
### 3.Enable TrueNAS developer tools
`install-dev-tools
`
### 4.Install build dependencies and kernel headers
`apt install build-essential dkms git linux-headers-truenas-production-amd64
`
### 5.Clone the repository
`git clone -b kernel-v6.18 https://github.com/strongtz/i915-sriov-dkms.git
`
### 6.Build the DKMS module
`dkms build ./i915-sriov-dkms
`
### 7.Retrieve the compiled .ko file; Once the build finishes, locate your compiled kernel module in the DKMS output directory:
`/var/lib/dkms/i915-sriov-dkms/2025.12.10.2/6.12.113-production+truenas/x86_64/module/
`
### Copy this .ko file over to your bare-metal TrueNAS installation.

BeingNoAbility, 2026-11-07:
> Hi, I figured out how to properly build i915-sriov-dkms on TrueNAS 25.10.8. Here is the instructions.
>
> To avoid modifying your production bare-metal system, perform this build inside a clean TrueNAS [...]

[module.zip](https://github.com/user-attachments/files/34514183/module.zip) If you need, for TrueNAS Scale 25.10.8 only. You need to build a new one for newer system version.

christaikobo, 2026-11-07:
Honestly I don't bother with truenas anymore. Happy for you and thanks for sharing.

But truenas is not trying to appeal to community users for sure.