strongtz/i915-sriov-dkms #213: adding a systemd service for AUR pkg

user7z, 2024-10-24:
needing to echo number of VFs every single boot is not practical , plz provide the aur PKGBUILD with a service file that do it automatically, one would just change a variable that defines number of VFs needed

bbaa-bbaa, 2024-10-27:
I created an AUR package that includes a tmpfiles.d configuration: https://aur.archlinux.org/packages/i915-sriov-dkms
#207 is also available in aur: https://aur.archlinux.org/packages/i915-sriov-dkms-bbaa-git

user7z, 2024-10-27:
@bbaa-bbaa thank you , dont forget to submit a request to take of the existing orphaned package [i915-sriov-dkms-git](https://aur.archlinux.org/packages/i915-sriov-dkms-git)
Then delete the git one named by you , i well give it try thank you again

bbaa-bbaa, 2024-10-28:
> Then delete the git one named by you

The AUR package named by me is based on the Linux 6.6 module branch PR #207 . I think this fork still requires further testing to merge.

You can use [i915-sriov-dkms](https://aur.archlinux.org/packages/i915-sriov-dkms) package for current branch.

strongtz, 2024-10-28:
To be honest since I can't test this on my hardware anymore, I would be happy to add someone as a co-maintainer of the AUR package :)

PR #239 (2025-01-03): ci: auto build package and create release
May simplify the installation and update process.
GitHub Action will automatically create a Release when the version is bumped.

Closes #173 
Closes #200 
Closes #213 
Closes #222
closed 2025-01-03 by strongtz